fwupd/fwupd #6565: OverrideESPMountPoint does not take effect

Piroro-hs, 2023-12-21:
**Describe the bug**
ESP does not detected even if OverrideESPMountPoint configured
Does Udisks always required?

**fwupd version information**
Please provide the version of the daemon and client.

```
# fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.14
compile   org.freedesktop.Passim        0.1.4
compile   com.hughsie.libjcat           0.1.14
compile   org.freedesktop.fwupd         1.9.10
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    6.6.7-arch1-1.1
runtime   org.freedesktop.fwupd         1.9.10
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): pacman

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```
# fwupdmgr get-devices --show-all-devices
WARNING: UEFI ESP partition not detected or configured
See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
Gigabyte Technology Co., Ltd. Z690I A ULTRA PLUS D4
│
├─Unknown Device:
│     Device ID:          9c55e7677e0a67ce1d50c15c05efe7864e6016ff
│     Serial Number:      290461295
│     GUID:               6a07b70c-8f04-5b03-a4c8-0a606d2a44be ← DRM\VEN_ACR&DEV_0583
│
├─Unknown Device:
│     Device ID:          f911a17457132c3d9c6daed53a308bfddd2dd78b
│     Serial Number:      0000000000001
│     GUID:               a65cbb64-b03b-58b6-9e87-31df36e40f65 ← DRM\VEN_IOC&DEV_FFFF
│
├─Unknown Device:
│     Device ID:          0ca7d9ea83b2d18d8523bb2eed1753317fe1a969
│     GUID:               06d8ee08-fa01-5ed7-be1c-b77936b6f43a ← GPIO\ID_INTC1056:00
│
├─12th Gen Intel Core™ i7-12700K:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000032
│     Vendor:             Intel
│     GUIDs:              615995d1-bdd0-5048-a291-83eaef0de820 ← CPUID\PRO_0&FAM_06&MOD_97
│                         c9697a05-dbd8-58ac-833c-5a4fa8fdbcd7 ← CPUID\PRO_0&FAM_06&MOD_97&STP_2
│     Device Flags:       • Internal device
│     Device Requests:    • Message
│
├─AlderLake-S GT1:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    0c
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              0116f464-64b8-531a-a445-591713da1abc ← PCI\VEN_8086&DEV_4680
│                         31966e36-1fab-52a2-a1a1-81de2070a82d ← PCI\VEN_8086&DEV_4680&SUBSYS_1458D000
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│     Device Requests:    • Message
│
├─Arc A770:
│ │   Device ID:          3743975ad7f64f8d6575a9ae49fb3a8856fe186f
[... 170 lines cut ...]
```

```
# tree /boot
/boot
|  |--EFI
|  |  |--BOOT
|  |  |  |--BOOTx64.efi
|  |--booster-linux.img
|  |--cmdline
|  |--intel-ucode.img
|  |--loader
|  |  |--random-seed
|  |--splash.bmp
|  |--vmlinuz-linux
```

**Additional questions**

- Operating system and version: Arch Linux, 6.6.7-arch1-1.1 x86_64 GNU/Linux
- Have you tried rebooting?: y
- Is this a regression?: n
- Are you using an NVMe disk?: y
- Is secure boot enabled?: n
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?: n

hughsie, 2023-12-21:
>  Does Udisks always required?

We've done basically no testing without UDisks.

superm1, 2023-12-21:
I think we need to see output for `sudo  fwupdtool esp-list -vv` to see what's going wrong in your case.

Piroro-hs, 2023-12-25:
```
10:49:02.966 FuDebug              verbose to debug (on console 1)
10:49:02.967 FuEngine             starting fwupd 1.9.10…
10:49:02.969 FuContext            failed to call org.freedesktop.UDisks2.Manager.GetBlockDevices(): The name is not activatable
10:49:02.970 FuContext            failed to call org.freedesktop.UDisks2.Manager.GetBlockDevices(): The name is not activatable
No ESP or BDP found
```

superm1, 2023-12-26:
Please post your modified conffile

Piroro-hs, 2023-12-26:
/etc/fwupd.conf
```ini
[fwupd]
DisabledPlugins=test;test_ble
OnlyTrusted=true
AllowEmulation=false
```

/etc/uefi.conf
```ini
EspLocation=/boot
```

superm1, 2023-12-26:
Your conffiles seem wrong to me.
1. You sholud have `/etc/fwupd/` directory for storing stuff.
2. The `EspLocation` key goes into `/etc/fwupd/fwupd.conf`.  See `man fwupd.conf` for more details.

Piroro-hs, 2023-12-26:
Oops, wrong config directory is just mistypes. The `fwupd.conf` is actually sits inside `/etc/fwupd`.
However, move `EspLocation` key to `/etc/fwupd/fwupd.conf` does not change anything...

superm1, 2023-12-26:
Can you share the output for that tool I asked after you've moved it?  Also is it already mounted before you started?

Piroro-hs, 2023-12-26:
```
16:54:08.445 FuDebug              verbose to debug (on console 1)
16:54:08.445 FuEngine             starting fwupd 1.9.10…
16:54:08.446 FuContext            failed to call org.freedesktop.UDisks2.Manager.GetBlockDevices(): The name is not activatable
16:54:08.447 FuContext            failed to call org.freedesktop.UDisks2.Manager.GetBlockDevices(): The name is not activatable
No ESP or BDP found
```

`/boot` is mounted as vfat.

superm1, 2023-12-26:
Oh I think that tool might only work with Udisks.  Fixing the conffile - did it fix the actual issue?

Piroro-hs, 2023-12-28:
I confirmed it works. Thanks!

closed 2023-12-28 by superm1