fwupd/fwupd #6574: TPM PCR0 recontruction fails on Pluton fTPM

JohnRTitor, 2023-12-26:
**Describe the bug**
My Gigabyte UEFI BIOS has an option to select which TPM chip to use. By default it uses AMD fTPM. After manually enabling Pluton fTPM via Gigabyte UEFI, TPM PCR0 reconstruction status changed to Invalid.

**Steps to Reproduce**

- Enable Pluton fTPM on a supported CPU (Ryzen 6000, Ryzen 7000) via UEFI interface
- Boot into a linux system
- Run ` fwupdmgr security --force`

**Expected behavior**
```
TPM PCR0 reconstruction:       Valid
```

**fwupd version information**
Please provide the version of the daemon and client.

```
compile   org.freedesktop.fwupd         1.9.5
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.kernel                    6.6.8-x64v4-xanmod1
runtime   org.freedesktop.fwupd         1.9.5
```

**Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):**

Preloaded with Kubuntu, package manager is apt

**fwupd device information**

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```
[Output.txt](https://github.com/fwupd/fwupd/files/13771159/Output.txt)

</details>

**Additional questions**

- Operating system and version: Kubuntu 23.10
- Kernel: Generic 6.5.0-14 and Xanmod 6.6.8
- Have you tried rebooting? Yes
- Is this a regression? Likely not

hughsie, 2023-12-26:
Could you build fwupd locally for us and grab the output of `sudo ./plugins/tpm/fwupdtpmevlog -p 0` please?

JohnRTitor, 2023-12-26:
Yeah sure, one question before I do, is there a simple way to revert to the original fwupd provided by Ubuntu later, if something goes wrong?

superm1, 2023-12-26:
You don't have to install, just build in local directory and run binary from local directory.

JohnRTitor, 2023-12-26:
Ahh some tech support here

```
$ fwupdmgr security
WARNING: This package has not been validated, it may not work properly.
Unsupported daemon version 1.9.5, client version is 1.9.10-98-gcb7060fcd

```
Installed it in /usr/local

superm1, 2023-12-26:
You don't need to install it to `/usr/local` to run what is needed.  The binary Richard asked for the output doesn't install (it's a noinst binary), it's in the build directory only.  But since you're installed you can also try `fwupdtool security` and it will run the version you just built so you can confirm the bug still exists with the most recent version.

When you want to clean everything up you can just purge everything in `/usr/local/*`.

JohnRTitor, 2023-12-26:
Here's the output:
And looks like my CPU status got changed to Invalid/Unsupported. (it was Valid in original distro build)
```
$ sudo fwupdtool security        
[sudo] password for masum: 
Loading…                 [*****                                  ]16:48:58.387 FuPluginUefiCapsule  skipping device that failed coldplug: ESRT GUID '00000000-0000-0000-0000-000000000000' was not valid
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
Host Security ID: HSI:0 (v2.0.0)

HSI-1
✔ BIOS firmware updates:         Enabled
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI bootservice variables:    Locked
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled
✘ Supported CPU:                 Invalid

HSI-2
✔ IOMMU:                         Enabled
✘ TPM PCR0 reconstruction:       Invalid

HSI-3
✔ Pre-boot DMA protection:       Enabled
✘ CET Platform:                  Not supported
✘ Suspend-to-idle:               Disabled
✘ Suspend-to-ram:                Enabled

HSI-4
✔ SMAP:                          Enabled
✘ Encrypted RAM:                 Not supported

Runtime Suffix -!
✔ fwupd plugins:                 Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Encrypted
✔ Linux kernel:                  Untainted

This system has a low HSI security level.
 » https://fwupd.github.io/hsi.html#low-security-level

The TPM PCR0 differs from reconstruction.
 » https://fwupd.github.io/hsi.html#pcr0-tpm-event-log-reconstruction

```

superm1, 2023-12-26:
That CPU thing regressing again appears to be a different bug.  I can reproduce it too with main.

Can you please share the output for the tool Richard asked for?

JohnRTitor, 2023-12-26:
The thing is I can't seem to find the tool script/executable?
```shell
$ pwd
/home/masum/Desktop/Dev-Enviornment/fwupd
$ ls -a plugins/tpm 
.                  fu-tpm-eventlog-common.c  fu-tpm.rs           meson.build
..                 fu-tpm-eventlog-common.h  fu-tpm-v1-device.c  README.md
fu-self-test.c     fu-tpm-eventlog-parser.c  fu-tpm-v1-device.h  tests
fu-tpm-device.c    fu-tpm-eventlog-parser.h  fu-tpm-v2-device.c
fu-tpm-device.h    fu-tpm-plugin.c           fu-tpm-v2-device.h
fu-tpm-eventlog.c  fu-tpm-plugin.h           fuzzing

```

superm1, 2023-12-26:
It should be in your "build" directory, not the source directory.

superm1, 2023-12-26:
@hughsie the unsupported CPU issue is caused by 19aa82453a3663cd2f84ee18b392a61ea61216d4....

```
git bisect start '--term-new=broken' '--term-old=fixed'
# status: waiting for both good and bad commits
# fixed: [640f82210537454db15f3e9b495829a78590f6f6] trivial: Show the NO_PROBE_COMPLETE message in non-supported builds
git bisect fixed 640f82210537454db15f3e9b495829a78590f6f6
# status: waiting for bad commit, 1 good commit known
# broken: [cb7060fcd1353c858f16c46b43ecfb280bb4da7f] trivial: fix a spurious warning argument in fu_udev_device_get_dev
git bisect broken cb7060fcd1353c858f16c46b43ecfb280bb4da7f
# fixed: [8afede84fcfb789ee36c2292139ce042ed193140] Release fwupd 1.9.10
git bisect fixed 8afede84fcfb789ee36c2292139ce042ed193140
# fixed: [ce389a87964d15c372ab6fd02e8bfaea38959a41] trivial: Add fu_input_stream_read_uXX helpers
git bisect fixed ce389a87964d15c372ab6fd02e8bfaea38959a41
# fixed: [e91c8810b8373bafb5cc53e68a4ec2cac97a1221] trivial: Fix parsing EFI volumes with offset supplied
git bisect fixed e91c8810b8373bafb5cc53e68a4ec2cac97a1221
# fixed: [1b069bfbdd6db35fbd0ef20496d91a35eedf1c99] synaptics-mst: Speed up device enumeration
git bisect fixed 1b069bfbdd6db35fbd0ef20496d91a35eedf1c99
# fixed: [cc1222ac42d9437c639f2031058deb7fa1ac61fe] synaptics-mst: Add support for the Luxshare Quad USB4 Dock
git bisect fixed cc1222ac42d9437c639f2031058deb7fa1ac61fe
# fixed: [b69b45555a10228c42b69f8d0481fffc054f0abe] Include the offline report in an archive
git bisect fixed b69b45555a10228c42b69f8d0481fffc054f0abe
# broken: [19aa82453a3663cd2f84ee18b392a61ea61216d4] Deallocate the GUdevDevice after the probe has completed
git bisect broken 19aa82453a3663cd2f84ee18b392a61ea61216d4
# fixed: [286858196e5bdacda40abfaa2956c4cfa41c1c56] trivial: Show the NO_PROBE_COMPLETE message in non-supported builds
git bisect fixed 286858196e5bdacda40abfaa2956c4cfa41c1c56
# first broken commit: [19aa82453a3663cd2f84ee18b392a61ea61216d4] Deallocate the GUdevDevice after the probe has completed
```

JohnRTitor, 2023-12-26:
Got it, here's the output: 
```
$ sudo ./plugins/tpm/fwupdtpmevlog -p 0 

PCR:                    BIOS (0)
Type:                   0x8
Description:            ev-s-crtm-version
ChecksumSha256:         d681837b3b8f5fd685bfad042ca78e6884e10e1c8f272dbccd2fd10be738a7fa
BlobStr:                BSsQp8fZZUGBQCrd6Ur2PA==

PCR:                    BIOS (0)
Type:                   0x8000000a
ChecksumSha256:         d992c9ba3d94ee8378de03db7580013dda6bb878083b34a708d8f6c785fa5016
BlobStr:                D0ZWX01BSU5fV1JBUFBFUgAAAAEAAAAAAPC6AAAAAAA=

PCR:                    BIOS (0)
Type:                   0x4
Description:            ev-separator
ChecksumSha256:         df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119
BlobStr:                AAAAAA==

Reconstructed PCRs:
  PCR 0:                SHA256(f31d0c2019152f20970aa1f38fc8fcf4f57dfbc15bc3bdd5625a15335d762590)
```

superm1, 2023-12-26:
Can you please post the same output while you're in fTPM mode?

JohnRTitor, 2023-12-26:
I ran the command with Pluton fTPM mode enabled.

superm1, 2023-12-26:
Can you please switch to AMD fTPM mode and run it? We need to compare.

JohnRTitor, 2023-12-26:
Switched to AMD fTPM mode.
```
├─TPM:
│     Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
│     Summary:            TPM 2.0 Device
│     Current version:    6.31.0.6
│     Vendor:             Advanced Micro Devices, Inc. (TPM:AMD)
│     GUIDs:              9305de1c-1e12-5665-81c4-37f8e51219b8 ← TPM\VEN_AMD&DEV_0001
│                         78a291ae-b499-5b0f-8f1d-74e1fefd0b1c ← TPM\VEN_AMD&MOD_AMD
│                         65a3fced-b423-563f-8098-bf5c329fc063 ← TPM\VEN_AMD&DEV_0001&VER_2.0
│                         5e704f0d-83cb-5364-8384-f46d725a23b8 ← TPM\VEN_AMD&MOD_AMD&VER_2.0
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device can recover flash failures
│                         • Full disk encryption secrets may be invalidated when updating
│                         • Signed Payload
│   

```

```
$ sudo fwupdtool security
HSI-2
✔ IOMMU:                         Enabled
✔ TPM PCR0 reconstruction:       Valid
```

The output you asked.
```
$ sudo ./plugins/tpm/fwupdtpmevlog -p 0 
[sudo] password for masum: 
PCR:                    BIOS (0)
Type:                   0x8
Description:            ev-s-crtm-version
ChecksumSha256:         d681837b3b8f5fd685bfad042ca78e6884e10e1c8f272dbccd2fd10be738a7fa
BlobStr:                BSsQp8fZZUGBQCrd6Ur2PA==

PCR:                    BIOS (0)
Type:                   0x8000000a
ChecksumSha256:         d992c9ba3d94ee8378de03db7580013dda6bb878083b34a708d8f6c785fa5016
BlobStr:                D0ZWX01BSU5fV1JBUFBFUgAAAAEAAAAAAPC6AAAAAAA=

PCR:                    BIOS (0)
Type:                   0x4
Description:            ev-separator
ChecksumSha256:         df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119
BlobStr:                AAAAAA==

Reconstructed PCRs:
  PCR 0:                SHA256(f31d0c2019152f20970aa1f38fc8fcf4f57dfbc15bc3bdd5625a15335d762590)
```

superm1, 2023-12-26:
One more thing to contrast; can you look at the output for `sudo tpm2_pcrread` in both cases for PCR 0?

From what you've posted so far the event log and reconstructed value is the same in both cases actually.  Assuming the TPM is actually "working" in the case of Pluton enabled this looks like a BIOS bug to me that the event log isn't being updated for what is being measured in the Pluton case.

JohnRTitor, 2023-12-27:
> One more thing to contrast; can you look at the output for `sudo tpm2_pcrread` in both cases for PCR 0?

```
$ sudo tpm2_pcrread 
sudo: tpm2_pcrread: command not found
```
And I dont see a similar binary in the plugins/tpm folder.

> From what you've posted so far the event log and reconstructed value is the same in both cases actually. 

Perhaps I need to build the binary again while in AMD fTPM mode?

superm1, 2023-12-27:
It's part of the tpm2 tools package, comes separate from fwupd.

JohnRTitor, 2023-12-27:
While in AMD fTPM mode:
```
$ sudo tpm2_pcrread 
  sha1:
  sha256:
    0 : 0xF31D0C2019152F20970AA1F38FC8FCF4F57DFBC15BC3BDD5625A15335D762590
    1 : 0xB390100E6DDCC39E006A8B6566B5A65F4461B25FE552E1FED625634E411E61D0
    2 : 0x3D458CFE55CC03EA1F443F1562BEEC8DF51C75E14A9FCF9A7234A13F198E7969
    3 : 0x3D458CFE55CC03EA1F443F1562BEEC8DF51C75E14A9FCF9A7234A13F198E7969
    4 : 0xDE64867C86EEBED7C47EC40B71B3A7A04D383F7BE063A8B440BF87C00A05841D
    5 : 0x18872CC6B150F0380F526ABD8165B7795FE98F18D19F594F7C0E06F5948F4BF4
    6 : 0x3D458CFE55CC03EA1F443F1562BEEC8DF51C75E14A9FCF9A7234A13F198E7969
    7 : 0x6144CAEB0EE71439D0488B2744236BE01A6558F781C5E7EE6BCF42D845A8A7C8
    8 : 0x77FDC003BB9740E1EC77CAE6A384369FFAE78FA7F027ACB53A687FD3C849F8CE
    9 : 0x6FB9B53DAB3648B5A59FBE5DD7DA395A54C0DDBB98F221550401F8770D4680D3
    10: 0xF91C800989E3919DEA96B58458E757A0AF592B16971FD7B78C4356FA5F206F31
    11: 0x0000000000000000000000000000000000000000000000000000000000000000
    12: 0x0000000000000000000000000000000000000000000000000000000000000000
    13: 0x0000000000000000000000000000000000000000000000000000000000000000
    14: 0x6AC305367F9587C017EDB26A93553A3585718E0F75F6D6A37E10493978FCC15D
    15: 0x0000000000000000000000000000000000000000000000000000000000000000
    16: 0x0000000000000000000000000000000000000000000000000000000000000000
    17: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    18: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    19: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    20: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    21: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    22: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    23: 0x0000000000000000000000000000000000000000000000000000000000000000
  sha384:

```

In Pluton fTPM mode
```
$ sudo tpm2_pcrread 
  sha1:
  sha256:
    0 : 0x0189A8726CB1234DA4D0A49947E10E1081E3A2874FA4AC499F864D0C662DDBE7
    1 : 0x24FC50EC773E901C205A19E33D0399072B166C2DA84A80818E31A5F9362F39D5
    2 : 0x3D458CFE55CC03EA1F443F1562BEEC8DF51C75E14A9FCF9A7234A13F198E7969
    3 : 0x3D458CFE55CC03EA1F443F1562BEEC8DF51C75E14A9FCF9A7234A13F198E7969
    4 : 0xDE64867C86EEBED7C47EC40B71B3A7A04D383F7BE063A8B440BF87C00A05841D
    5 : 0x18872CC6B150F0380F526ABD8165B7795FE98F18D19F594F7C0E06F5948F4BF4
    6 : 0x3D458CFE55CC03EA1F443F1562BEEC8DF51C75E14A9FCF9A7234A13F198E7969
    7 : 0x6144CAEB0EE71439D0488B2744236BE01A6558F781C5E7EE6BCF42D845A8A7C8
    8 : 0x77FDC003BB9740E1EC77CAE6A384369FFAE78FA7F027ACB53A687FD3C849F8CE
    9 : 0x6FB9B53DAB3648B5A59FBE5DD7DA395A54C0DDBB98F221550401F8770D4680D3
    10: 0x3E4D2A3AF80ABFE8B3074F5858DDE1FB26536DAC217800606DB3DD82E6BEE257
    11: 0x0000000000000000000000000000000000000000000000000000000000000000
    12: 0x0000000000000000000000000000000000000000000000000000000000000000
    13: 0x0000000000000000000000000000000000000000000000000000000000000000
    14: 0x6AC305367F9587C017EDB26A93553A3585718E0F75F6D6A37E10493978FCC15D
    15: 0x0000000000000000000000000000000000000000000000000000000000000000
    16: 0x0000000000000000000000000000000000000000000000000000000000000000
    17: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    18: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    19: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    20: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    21: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    22: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
    23: 0x0000000000000000000000000000000000000000000000000000000000000000
  sha384:

```

superm1, 2023-12-27:
FWIW I checked this on the machines I have.

* I can reproduce this on a Lenovo Z13 (Containing `AMD Ryzen 5 PRO 6650U`) in Pluton TPM mode as well.
* I can not reproduce this on a Framework 13 (Containing `AMD Ryzen 7640U`) in Pluton TPM mode.

I believe this is a BIOS bug; perhaps in the implementation used by certain vendors.

I suggest you raise this with your board vendor.

closed 2023-12-27 by superm1

JohnRTitor, 2023-12-28:
Guess it's time to contact Gigabyte support, given that they recently enabled Pluton with the latest stable BIOS update.
TPM is working fine on Windows. Bitlocker and other TPM related services are working as well with the Pluton fTPM.