fwupd/fwupd #6578: Failed to get devices when Dell HD22Q Docking was plugged

KaiChuan-Hsieh, 2023-12-28:
**Describe the bug**
Failed to get devices when Dell HD22Q was plugged after upgrade fwupd to 1.9.10.

Dec 28 09:42:59 202002-27723 systemd[1]: Starting Firmware update daemon...
Dec 28 09:43:00 202002-27723 fwupd[2984]: 01:43:00.980 FuEngine             failed to add device usb:04:00:01:04:03: No device GType set
Dec 28 09:43:17 202002-27723 fwupd[2984]: 01:43:17.586 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-4/drm_dp_aux4: failed to add device using on synaptics_mst: failed to enable remote control: remote command failed: invalid [1]
Dec 28 09:43:19 202002-27723 fwupd[2984]: 01:43:19.712 FuMain               Daemon ready for requests (locale en_US.UTF-8)
Dec 28 09:43:22 202002-27723 systemd[1]: Started Firmware update daemon.

**Steps to Reproduce**
Run command
$ sudo fwupdtool get-devices --verbose

**Expected behavior**
Can get devices successfully without error.

**fwupd version information**
compile   com.hughsie.libxmlb           0.3.6
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.9.10
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.9.10
runtime   org.kernel                    6.2.0-39-generic

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
I install it on Ubuntu 22.04, and utilize ppa to create 1.9.10 fwupd packages.
$ sudo add-apt-repository ppa:kchsieh/fwupd
$ sudo apt install fwupd

<details>

<summary>**fwupd device information**</summary>
Dell Inc. Precision 5750
│
├─DELL0990:00 04F3:311C:
│     Device ID:          b8cfad8dcc60878190ea039a952b3d7cd97cad91
│     Summary:            Touchpad
│     Current version:    0x000a
│     Bootloader Version: 0x0001
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              46598f8d-5777-50c4-8da9-cda0cc8bbc26 ← HIDRAW\VEN_04F3&DEV_311C
│                         d00322f0-ff1f-5458-9c64-8a771fb0b72e ← HIDRAW\VEN_04F3&DEV_311C&MOD_0000
│                         6cf7ae6a-8b98-5b63-9af8-b23b932f65de ← ELANTP\ICTYPE_13&MOD_0000
│                         ac3627a9-8463-5e63-b6b1-be9e0bcf27d2 ← ELANTP\ICTYPE_13&MOD_0000&DRIVER_HID
│     Device Flags:       • Internal device
│                         • Updatable
│     Device Requests:    • Message
│                         • Image
│   
├─HD22Q:
│ │   Device ID:          495a64cd518871c4511a85022df727c538ef67ba
│ │   Summary:            High performance dock
│ │   Current version:    01.00.00.03
│ │   Vendor:             Dell Inc. (USB:0x413C)
│ │   Install Duration:   1 minute
│ │   Serial Number:      8L281L3/3110746456619804
│ │   GUID:               67de0057-2f75-5bea-a388-ced11251dd78 ← USB\VID_413C&PID_B06E&hub&atomic_embedded
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │                       • Device can recover flash failures
│ │                       • Device is usable for the duration of the update
│ │                       • Signed Payload
│ │   Device Requests:    • Image
│ │ 
│ ├─Package level of Dell dock:
│ │     Device ID:        2a1e7ab23626ef9e1e5407b45605680dbc982939
│ │     Summary:          A representation of dock update status
[... 285 lines cut ...]
│     GUID:               5f7cc4ea-9f7a-5c5f-b535-dfbcc73f5ba6 ← UEFI\CRT_2493B118EABD632FD3BA49C333A8329B1982E872
│   
└─Xeon W-10855M CPU @ 2.80GHz:
      Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
      Current version:    0x000000f8
      Vendor:             Intel
      GUIDs:              a45b0522-5722-54bd-b802-86cd044262df ← CPUID\PRO_0&FAM_06&MOD_A5
                          7b9b6e8c-226c-5db6-86cb-ea3187578013 ← CPUID\PRO_0&FAM_06&MOD_A5&STP_2
      Device Flags:       • Internal device
      Device Requests:    • Message
    

</details>

**Additional questions**

- Operating system and version:
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.3 LTS
Release:        22.04
Codename:       jammy
- Have you tried rebooting?
coldboot warmboot can both reproduce the issue
- Is this a regression?
yes

superm1, 2023-12-30:
Can you please provide output for

`sudo fwupdtool get-devices -vv`

@CragW ∆

superm1, 2023-12-30:
Can  you please reproduce on 1_9_X branch? We've had some changes to synaptics mst plugin for Cayenne support which is listed here in your output which isn't yet tagged into a new release.

CragW, 2023-12-31:
> Can you please provide output for
> 
> `sudo fwupdtool get-devices -vv`
> 
> @CragW ∆

Sure, will check on Tue. Was out for a family event.

superm1, 2024-01-01:
I was thinking more about this issue; and _perhaps_ the actual issue is that the `dell_dock` plugin and the `synaptics_mst` plugin are both touching the same hardware.  We used to have a way to prevent Synaptics MST from probing stuff in a Dell Dock since the `dell_dock` plugin was the preferred update method. 

To prove if this is the case; you can try to blacklist the `dell_dock` plugin in the `DisabledPlugins` key in `/etc/fwupd/fwupd.conf`.  
* If that helps then I think we need a way to block the MST hub in this dock from probing in the synaptics MST plugin. 
* If this doesn't help; it's a different root cause.  It could be something to do with the host hardware containing a Synaptics MST hub or the monitor being connected containing one.  It could even be exposing a kernel driver MST bug.  You might look at kernel log with drm.debug output turned up while running `fwupd` or `fwupdtool` enumerate for hints.

CragW, 2024-01-02:
> Dec 28 09:43:00 202002-27723 fwupd[2984]: 01:43:00.980 FuEngine failed to add device usb:04:00:01:04:03: No device GType set

@superm1  We can remove `Realtek NIC` device from dell.quirk to address this error.

> Dec 28 09:43:17 202002-27723 fwupd[2984]: 01:43:17.586 FuEngine failed to add device /sys/devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-4/drm_dp_aux4: failed to add device using on synaptics_mst: failed to enable remote control: remote command failed: invalid [1]

`dell_dock` plugin doesn't use `drm_dp_aux`;  This error is introduced by `synaptics_mst` itself trying to enable remote control, maybe handle the error in a nicer way.

superm1, 2024-01-02:
> @superm1  We can remove `Realtek NIC` device from dell.quirk to address this error.
>
>
> `dell_dock` plugin doesn't use `drm_dp_aux`;  This error is introduced by `synaptics_mst` itself trying to [...]

But if it's the same physical device and Dell dock enabled remote control already it might cause this kind of failure is my theory.

The verbose log and the experiment I suggested should confirm it.

CragW, 2024-01-02:
I've disabled `dell_dock` plugin, the error generated by `synaptcis_mst` happens anyway regardless dock, see:
[no_dell_dock_plugin.txt](https://github.com/fwupd/fwupd/files/13807035/no_dell_dock_plugin.txt)

```
03:53:17.858 FuEngine             plugins disabled: dell_dock, test, test_ble
...

03:53:18.405 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-6/drm_dp_aux6: failed to add device using on synaptics_mst: failed to enable remote control: remote command failed: invalid [1]
```

superm1, 2024-01-02:
> I've disabled `dell_dock` plugin, the error generated by `synaptcis_mst` happens anyway regardless dock, see: [...]

Is that even with a power cycled dock?

superm1, 2024-01-02:
@CloudExt can you please look at this?

superm1, 2024-01-02:
Something funky to me is that multiple dp aux devices have the exact same DPCD:
```
03:53:18.045 FuDevice             using d762543f8c20f636e6fff031a000078d3e10c600 for PCI_SLOT_NAME=0000:00:02.0:drm_dp_aux1
03:53:18.046 FuDpauxDevice        DPAUX read @0x500:90 cc 24 53 59 4e 41 62 11 10 06 05 00 
03:53:18.046 FuStruct             DpauxDpcd:
  ieee_oui: 0x90cc24
  dev_id: SYNAb.
  hw_rev: 0x10
  fw_ver: 0x60500
03:53:18.046 FuDevice             using 585c346c009da3cc19a400fb59a4ed8d0cbd0151 for PCI_SLOT_NAME=0000:00:02.0:drm_dp_aux2
03:53:18.067 FuBackend            failed to open device /sys/devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-2/drm_dp_aux2: DPCD read failed: failed to read 14: No such device or address
03:53:18.068 FuDevice             using 2d9bd19a73af0e1160733d42c55f1d9de90ed451 for PCI_SLOT_NAME=0000:00:02.0:drm_dp_aux3
03:53:18.089 FuBackend            failed to open device /sys/devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-3/drm_dp_aux3: DPCD read failed: failed to read 14: No such device or address
03:53:18.089 FuDevice             using cd3818e2d3a168ec7e137e21fdfb21df5ad2e175 for PCI_SLOT_NAME=0000:00:02.0:drm_dp_aux4
03:53:18.110 FuBackend            failed to open device /sys/devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-4/drm_dp_aux4: DPCD read failed: failed to read 14: No such device or address
03:53:18.111 FuDevice             using 47a0754f6a2c9a7eb5f2883e4b78834a64451577 for PCI_SLOT_NAME=0000:00:02.0:drm_dp_aux5
03:53:18.148 FuBackend            failed to open device /sys/devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-5/drm_dp_aux5: DPCD read failed: failed to read 14: Input/output error
03:53:18.148 FuDevice             using d7bfd19a68de57bb7de8a0cc8f7b97680677abc4 for PCI_SLOT_NAME=0000:00:02.0:drm_dp_aux6
03:53:18.185 FuDpauxDevice        DPAUX read @0x500:90 cc 24 53 59 4e 41 62 11 10 06 05 00 
03:53:18.185 FuStruct             DpauxDpcd:
  ieee_oui: 0x90cc24
  dev_id: SYNAb.
  hw_rev: 0x10
  fw_ver: 0x60500
03:53:18.186 FuDevice             using a4d90c62862392b0f200f174dfb874c27bbbe86c for PCI_SLOT_NAME=0000:00:02.0:drm_dp_aux7
03:53:18.224 FuDpauxDevice        DPAUX read @0x500:00 00 00 00 00 00 00 00 00 00 00 00 00 
03:53:18.224 FuStruct             DpauxDpcd:
  ieee_oui: 0x0
  hw_rev: 0x0
  fw_ver: 0x0
```

I expected only one aux end point to have that DPCD.  It showing up on multiple is odd to me.  In the topology connected how many Synaptics MST hubs are really there?

Can this reproduce on a system that uses `amdgpu` instead of `i915`?

superm1, 2024-01-02:
> @superm1 We can remove `Realtek NIC` device from dell.quirk to address this error.
> 
> >

go ahead and follow up with a PR for this

CragW, 2024-01-02:
> Is that even with a power cycled dock?

Yup, even with a new powered dock.

CloudExt, 2024-01-02:
How about just remove DELL docks in `synaptics-mst.quirk`?

CloudExt, 2024-01-02:
I think `synaptics_mst` uses DP aux and `dell_dock` on HID, I have no idea why the error occurred even two plugins touched the same device, the RC commands should work even you touched with different protocol at the same time.
Does any modification in `dp_aux` causes this issue?

@CragW Would it work when you disable `synaptics_mst` plugin?

superm1, 2024-01-02:
I notice in the [kernel source](https://github.com/torvalds/linux/blob/610a9b8f49fbcf1100716370d3b5f6f884a2835a/drivers/gpu/drm/display/drm_dp_mst_topology.c#L2315) that when a port is _added_ for MST it's given the name `DPMST`.

Considering this I think it's not really a new hub, but it's a given port that happens to be created downstream which happens to have the exact same DPCD.

A simple solution to this issue would be for the daemon to skip any aux devices with the name `DPMST`, but I suspect this also breaks anything actually cascaded.

hughsie, 2024-01-02:
> for the daemon to skip any aux devices with the name DPMST

I think that scuppers some of the plans we have for the future...

superm1, 2024-01-02:
> > for the daemon to skip any aux devices with the name DPMST
> 
> I think that scuppers some of the plans we have for the future...

I _think_ this should help: https://github.com/fwupd/fwupd/pull/6593.  Can still do downstream devices, but it won't fail if they're not really there.

superm1, 2024-01-02:
Hopefully this is fixed in both `main` and `1_9_X` branches now.  Please recheck.

CragW, 2024-01-03:
Issue still there by the build from source at 36ae23df8.

[synaptics_mst_enable_rc_invalid.log](https://github.com/fwupd/fwupd/files/13821330/synaptics_mst_enable_rc_invalid.log)

superm1, 2024-01-03:
That's unfortunate. Perhaps setting of the name by fu-udev-device isn't actually working. Can you dig into it?

superm1, 2024-01-03:
How about if you blacklist mediatek scalar plugin - does it keep happening?

superm1, 2024-01-03:
Maybe https://github.com/fwupd/fwupd/pull/6602 helps?

CragW, 2024-01-04:
> How about if you blacklist mediatek scalar plugin - does it keep happening?

The error persists; Also `$fwupdtool --plugins synaptics_mst get-devices` alone does see the error.

superm1, 2024-01-04:
> fwupdtool --plugins synaptics_mst get-devices

Can we see updated log for this for any other ideas?

CragW, 2024-01-04:
> Maybe #6602 helps?

Unfortunatelly no. 

> Perhaps setting of the name by fu-udev-device isn't actually working. Can you dig into it?

The device name is yet settled so `DPMST` didn't hit.

Just out of curiousity since the plugin attempts to try dpaux anyway why a test on `enable_rc` command introduces a critical error? I would expect to downgrade the error level as unsupported.

[fwupdtool-synaptics-mst-plugin-only.log](https://github.com/fwupd/fwupd/files/13830130/fwupdtool-synaptics-mst-plugin-only.log)

closed 2024-01-05 by hughsie