fwupd/fwupd #6584: Flatpak wiki page seems out of date

nfrizzell, 2023-12-30:
The Github wiki page for the Flatpak package says that network access is not supported, and that it's necessary to manually download CAB files to be installed:

https://github.com/fwupd/fwupd/wiki/fwupd-flatpak#installation

However, it seems like the Flatpak does have network access after all:

```
$ flatpak info --show-permissions org.freedesktop.fwupd
[Context]
shared=network;
devices=all;
filesystems=/sys;xdg-download;/boot;

[System Bus Policy]
org.freedesktop.fwupd=talk
org.freedesktop.login1=talk
org.freedesktop.UPower=talk
org.freedesktop.UDisks2=talk
```

I was also able to successfully update the firmware on my laptop running Debian 12 using the Flatpak without having to manually download any firmware. Am I misunderstanding something simple here, or should the wiki page be updated to reflect this?

Version: Flatpak v1.9.10

superm1, 2023-12-30:
The original version uploaded to flathub had that too:
https://github.com/flathub/org.freedesktop.fwupd/commit/43ed08e12cbb4bda41bb3e9b74a8d798b5425cfc

I'd say it's likely a wiki page out of date issue. @hughsie ?

hughsie, 2023-12-31:
I've updated the wiki, thanks!

closed 2023-12-31 by hughsie