fwupd/fwupd #6585: Confusing suggested configuration change

SoMuchForSubtlety, 2023-12-31:
**Describe the bug**
`fwupdmgr security` shows the following message
```
╔══════════════════════════════════════════════════════════════════════════════╗
║ Configuration Change Suggested: Encrypted RAM                                ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Encrypted RAM makes it impossible for information that is stored in device   ║
║ memory to be read if the memory chip is removed and accessed.                ║
║                                                                              ║
║ This tool can change the BIOS setting 'com.thinklmi.TSME' from 'Enable' to   ║
║ 'Enable' automatically, but it will only be active after restarting the      ║
║ computer.                                                                    ║
║                                                                              ║
║ You should ensure you are comfortable restoring the setting from the system  ║
║ firmware setup, as this change may cause the system to not boot into Linux   ║
║ or cause other system instability.                                           ║
╚══════════════════════════════════════════════════════════════════════════════╝
```

`from 'Enable' to  'Enable' ` does not make sense. The output also reports `✘ Encrypted RAM: Not supported`, maybe that's related?

**Steps to Reproduce**
run `fwupdmgr security`

**fwupd version information**

```shell
compile   com.hughsie.libxmlb           0.3.14
compile   com.hughsie.libjcat           0.1.14
compile   org.freedesktop.fwupd         1.9.10
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.10
runtime   org.kernel                    6.6.8-200.fc39.x86_64
```

installed via `dnf`

<details>

<summary><b>fwupd device information</b></summary>

Please provide the output of the fwupd devices recognized in your system.

```
LENOVO 20UD0013GE
│
├─AMD Ryzen 7 PRO 4750U with Radeon Graphics:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x08600109
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              1f2e0c2f-e078-5fcb-a8de-f78eff9a470a ← CPUID\PRO_0&FAM_17&MOD_60
│ │                       5fd5eb41-dd4a-5742-8448-861eaaa61cf8 ← CPUID\PRO_0&FAM_17&MOD_60&STP_1
│ │   Device Flags:       • Internal device
│ │   Device Requests:    • Message
│ │ 
│ ├─Graphics Processing Unit (GPU):
│ │ │   Device ID:        e32966e41f54b47823eef28bd8b4bd9e1018b923
│ │ │   Summary:          Renoir Generic VBIOS
│ │ │   Current version:  017.010.000.031.000000
│ │ │   Vendor:           Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│ │ │   GUID:             ffd1eae2-b5dd-58c4-b217-5ff61b0543b8 ← AMD\113-RENOIR-037
│ │ │   Device Flags:     • Internal device
│ │ │   Device Requests:  • Message
│ │ │ 
│ │ └─LP140WF9-SPF1:
│ │       Device ID:      aec1a869eb0df71b7cea6b3ac71d39b830faf164
│ │       GUID:           269df938-248e-5e93-8de5-c7046e71fa11 ← DRM\VEN_LGD&DEV_0608
[... 211 lines cut ...]
│     Update State:       Success
│     Problems:           [31m[1m• Device requires AC power to be connected[0m
│     GUID:               58ac1529-d706-459c-9f98-c711bb777410
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│     Device Requests:    • Message
│                         • Image (custom)
│   
└─UEFI Platform Key:
      Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
      Summary:            Lenovo Ltd. PK CA 2012
      Vendor:             Lenovo Ltd.
      GUID:               71599d14-9b31-5270-b3bd-74c494585820 ← UEFI\CRT_9AEF2123F4DE7C19AFABD909BB2C8CAC4411E07E
```

</details>

**Additional questions**

- Operating system and version: Fedora 39
- Have you tried rebooting? yes
- Is this a regression? ¯\\\_(ツ)\_/¯

superm1, 2023-12-31:
Can you please provide output from

`sudo fwupdtool security -vv` to better understand this issue?

SoMuchForSubtlety, 2023-12-31:
[security vv.log](https://github.com/fwupd/fwupd/files/13802233/security.vv.log)

superm1, 2024-01-02:
A few things:
1. Can you check what `cat /sys/bus/pci/drivers/ccp/*/tsme_status` returns?  This will help clarify if the problem is the obsoleting of MSR and PCI_PSP plugins on your system.
2 Can you please share `/proc/cmdline`?
3. Can you please try https://github.com/fwupd/fwupd/pull/6587?  I think this will stop it offering to change the setting at least.

SoMuchForSubtlety, 2024-01-02:
> 1. Can you check what `cat /sys/bus/pci/drivers/ccp/*/tsme_status` returns? 

```
0
```

> 2. Can you please share `/proc/cmdline`?

```
BOOT_IMAGE=(hd0,gpt2)/vmlinuz-6.6.8-200.fc39.x86_64 root=UUID=497dde77-e636-410c-a2f6-3e7cda1a6dca ro rootflags=subvol=root rhgb quiet rhgb quiet amd_pstate=passive preempt=full
```

>  3. Can you please try #6587

I followed the instructions from the wiki, but get the following error when running the newly installed version
```
fwupdmgr: error while loading shared libraries: libfwupd.so.3: cannot open shared object file: No such file or directory
```

superm1, 2024-01-02:
Fedora doesn't set /usr/local/lib64 by default in the library search path.

Try running with 

`LD_LIBRARY_PATH=/usr/local/lib64` in your environment

SoMuchForSubtlety, 2024-01-02:
Thanks, I can confirm that the change suggestion no longer appears.

superm1, 2024-01-02:
Okay. Then I'll backport that to 1_9_X.

Regarding the reason it suggested it though in the first place; that's pretty odd.

Can you please reset BIOS default settings and then try to turn it back on again?

SoMuchForSubtlety, 2024-01-02:
After resetting the BIOS, fwupd reported `✘ Encrypted RAM: Not supported` again, but still offered to enable it. I applied the setting and after a reboot, fwupd now reports `✔ Encrypted RAM: Encrypted`.

superm1, 2024-01-02:
Okay then there is definitely some bug in the BIOS. I'll fix the fwupd side of it, but you should report the BIOS side of it to Lenovo.

superm1, 2024-01-02:
Fixed fwupd side of the issue in d74375a19c4edfa35792bb88159a754351841741 on 1_9_X and e8c45d7e52945917601e9260bc5e5986bab6abb9 in main.

closed 2024-01-02 by superm1