fwupd/fwupd #6589: dell_dock: device does not set FWUPD_REQUEST_FLAG_NON_GENERIC_MESSAGE

CragW, 2024-01-02:
**Describe the bug**
After the firmware is installed, a human unfriendly message was prompted to user while some critical errors are seen on the daemon side.

```
$ sudo ./dist/fwupd.sh
09:35:59.416 FuMain               Daemon ready for requests (locale en_US.UTF-8)
09:36:04.726 GLib-GObject         invalid cast from 'GUnixInputStream' to 'GSeekable'
09:36:04.726 GLib-GIO             g_seekable_can_seek: assertion 'G_IS_SEEKABLE (seekable)' failed
```

```
$ ./dist/fwupdmgr.sh install ~/Downloads/cab/b1275193349c3ebd1db3821462143ac859302afe3572060fab7ee4793b4892bd-DellDockFirmwareUpdateLinux_01.00.16.cab --allow-reinstall
WARNING: This package has not been validated, it may not work properly.
Decompressing…           [                                       ]
request org.freedesktop.fwupd.request.remove-replug is not a GENERIC_MESSAGE and the device does not set FWUPD_REQUEST_FLAG_NON_GENERIC_MESSAGE
```

**Steps to Reproduce**
1. build fwupd from source
2. launch the daemon
3. install firmware for dell dock

**Expected behavior**
No errors.

**fwupd version information**
```shell
$ ./dist/fwupdmgr.sh --version
compile   com.hughsie.libxmlb           0.3.6
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         2.0.0
runtime   org.freedesktop.fwupd-efi     2.0.0
compile   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         2.0.0
runtime   org.kernel                    6.5.0-1007-oem
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
source

<details>

<summary>**fwupd device information**</summary>

```shell
Idle…: 0%
[31m[1mWARNING[0m: This package has not been validated, it may not work properly.
Dell Inc. XPS 13 9310
│
├─WD22TB4:
│ │   Device ID:          26140adcae5d9ca71008259caf4bb8639005cfde
│ │   Summary:            High performance dock
│ │   Current version:    01.01.00.05
│ │   Vendor:             Dell Inc. (USB:0x413C)
│ │   Install Duration:   1 minute
│ │   Serial Number:      6R2X0L3/3107576664195875
│ │   GUID:               cd357cf1-40b2-5d87-b8df-bb2dd82774aa ← USB\VID_413C&PID_B06E&hub&embedded
│ │   Device Flags:       • Updatable
│ │                       • Device stages updates
│ │                       • Device can recover flash failures
│ │                       • Device is usable for the duration of the update
│ │                       • Signed Payload
│ │ 
│ ├─Package level of Dell dock:
│ │     Device ID:        cc4b0e729e344517c4690908fe1b4aaadea014ac
│ │     Summary:          A representation of dock update status
│ │     Current version:  01.00.32.01
│ │     Vendor:           Dell Inc. (USB:0x413C)
│ │     Install Duration: 5 seconds
[... 194 lines cut ...]
│     Summary:            Dell Inc. Platform Key
│     Vendor:             Dell Inc.
│     GUID:               5f7cc4ea-9f7a-5c5f-b535-dfbcc73f5ba6 ← UEFI\CRT_2493B118EABD632FD3BA49C333A8329B1982E872
│   
└─Unifying Receiver:
      Device ID:          1bf177ee2510c9dcbe9fa4cb630af962633789e2
      Summary:            Miniaturised USB wireless receiver
      Current version:    RQR12.11_B0032
      Bootloader Version: BOT01.04_B0016
      Vendor:             HIDRAW:0x046D|USB:0x046D
      Install Duration:   30 seconds
      GUIDs:              9d131a0c-a606-580f-8eda-80587250b8d6
                          279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
      Device Flags:       • Updatable
                          • Signed Payload
    
```

</details>

**Additional questions**

- Operating system and version: Ubuntu 22.04
- Have you tried rebooting? Yes.
- Is this a regression? Yes.

hughsie, 2024-01-02:
> request org.freedesktop.fwupd.request.remove-replug is not a GENERIC_MESSAGE and the device does not set FWUPD_REQUEST_FLAG_NON_GENERIC_MESSAGE

For this, I've pushed 08264f58f5dfe2178cc0968da197fb24197a397a -- can you re-pull and rebuild and get the new warning please.

> invalid cast from 'GUnixInputStream' to 'GSeekable'

For this, can you set `G_DEBUG=fatal-criticals` and then get me a backtrace of where it crashes please.

CragW, 2024-01-02:
> For this, I've pushed [08264f5](https://github.com/fwupd/fwupd/commit/08264f58f5dfe2178cc0968da197fb24197a397a) -- can you re-pull and rebuild and get the new warning please.

New screen output:
```
$ ./dist/fwupdmgr.sh install ~/Downloads/cab/b1275193349c3ebd1db3821462143ac859302afe3572060fab7ee4793b4892bd-DellDockFirmwareUpdateLinux_01.00.16.cab --allow-reinstall
WARNING: This package has not been validated, it may not work properly.
Decompressing…           [                                       ]
request org.freedesktop.fwupd.request.remove-replug is not a GENERIC_MESSAGE and device cc4b0e729e344517c4690908fe1b4aaadea014ac [dell_dock] does not set FWUPD_REQUEST_FLAG_NON_GENERIC_MESSAGE
```

> For this, can you set `G_DEBUG=fatal-criticals` and then get me a backtrace of where it crashes please.

[fatal-criticals.txt](https://github.com/fwupd/fwupd/files/13809075/fatal-criticals.txt)

It failed to cast `stream` as is a `GInputStream`. 
https://github.com/fwupd/fwupd/blob/main/libfwupdplugin/fu-input-stream.c#L300

hughsie, 2024-01-02:
> It failed to cast stream as is a GInputStream.
> https://github.com/fwupd/fwupd/blob/main/libfwupdplugin/fu-input-stream.c#L300

Can you paste the backtrace please, you can use `bt<enter>` in the gdb prompt to get it. Thanks.

CragW, 2024-01-02:
> Can you paste the backtrace please, you can use `bt<enter>` in the gdb prompt to get it. Thanks.

Does this help?

libfwupdplugin.so!fu_input_stream_read_byte_array(GInputStream * stream, gsize offset, gsize count, GError ** error) (/home/crabx/mywork/fwupd/libfwupdplugin/fu-input-stream.c:310)
libfwupdplugin.so!fu_input_stream_read_bytes(GInputStream * stream, gsize offset, gsize count, GError ** error) (/home/crabx/mywork/fwupd/libfwupdplugin/fu-input-stream.c:350)
libfwupdplugin.so!fu_bytes_get_contents_fd(gint fd, gsize count, GError ** error) (/home/crabx/mywork/fwupd/libfwupdplugin/fu-bytes.c:123)
fu_daemon_daemon_method_call(GDBusConnection * connection, const gchar * sender, const gchar * object_path, const gchar * interface_name, const gchar * method_name, GVariant * parameters, GDBusMethodInvocation * invocation, gpointer user_data) (/home/crabx/mywork/fwupd/src/fu-daemon.c:1892)
libgio-2.0.so.0![Unknown/Just-In-Time compiled code] (Unknown Source:0)
libglib-2.0.so.0!g_main_context_dispatch (Unknown Source:0)
libglib-2.0.so.0![Unknown/Just-In-Time compiled code] (Unknown Source:0)
libglib-2.0.so.0!g_main_loop_run (Unknown Source:0)
fu_daemon_start(FuDaemon * self) (/home/crabx/mywork/fwupd/src/fu-daemon.c:66)
main(int argc, char ** argv) (/home/crabx/mywork/fwupd/src/fu-main.c:218)

hughsie, 2024-01-02:
> Does this help?

It does! Can you try main now please? I merged `wip/hughsie/parse-stream` which replaces that code completely.

CragW, 2024-01-02:
> It does! Can you try main now please? I merged `wip/hughsie/parse-stream` which replaces that code completely.

The critical errors in daemon are gone from `main` branch, cheers.

> request org.freedesktop.fwupd.request.remove-replug is not a GENERIC_MESSAGE and the device does not set FWUPD_REQUEST_FLAG_NON_GENERIC_MESSAGE

Non-generic-message is the last issue, #6591 doesn't fix it yet, BTW this error happens to developer build only.

If it is a supported build, then the update messages are prompted okay:

```
$ ./dist/fwupdmgr.sh install ~/Downloads/cab/b1275193349c3ebd1db3821462143ac859302afe3572060fab7ee4793b4892bd-DellDockFirmwareUpdateLinux_01.00.16.cab --allow-reinstall
Waiting…                 [***************************************] Less than one minute remaining…
Successfully installed firmware
The update will continue when the device USB cable has been unplugged.
The update will continue when the device USB cable has been unplugged.
The update will continue when the device USB cable has been unplugged.
The update will continue when the device USB cable has been unplugged.
The update will continue when the device USB cable has been unplugged.
```

hughsie, 2024-01-02:
@CragW can you edit `libfwupdplugin/fu-device.c` and add something like:

	@@ -5921,6 +5921,7 @@ fu_device_emit_request(FuDevice *self, FwupdRequest *request, FuProgress *progre
		}
		if (!fwupd_request_has_flag(request, FWUPD_REQUEST_FLAG_ALLOW_GENERIC_MESSAGE) &&
		    !fu_device_has_request_flag(self, FWUPD_REQUEST_FLAG_NON_GENERIC_MESSAGE)) {
	+               g_error("HERE");
			g_set_error(error,
				    G_IO_ERROR,
				    G_IO_ERROR_NOT_SUPPORTED,

and then get the backtrace from gdb please.

CragW, 2024-01-02:
> g_error("HERE");

[update-message.txt](https://github.com/fwupd/fwupd/files/13810132/update-message.txt)

libglib-2.0.so.0!g_logv (Unknown Source:0)
libglib-2.0.so.0!g_log (Unknown Source:0)
libfwupdplugin.so!fu_device_emit_request(FuDevice * self, FwupdRequest * request, FuProgress * progress, GError ** error) (/home/crabx/mywork/fwupd/libfwupdplugin/fu-device.c:5924)
libfwupdplugin.so!fu_device_write_firmware(FuDevice * self, GInputStream * stream, FuProgress * progress, FwupdInstallFlags flags, GError ** error) (/home/crabx/mywork/fwupd/libfwupdplugin/fu-device.c:4466)
libfwupdplugin.so!fu_plugin_device_write_firmware(FuPlugin * self, FuDevice * device, GInputStream * stream, FuProgress * progress, FwupdInstallFlags flags, GError ** error) (/home/crabx/mywork/fwupd/libfwupdplugin/fu-plugin.c:786)
libfwupdplugin.so!fu_plugin_runner_write_firmware(FuPlugin * self, FuDevice * device, GInputStream * stream, FuProgress * progress, FwupdInstallFlags flags, GError ** error) (/home/crabx/mywork/fwupd/libfwupdplugin/fu-plugin.c:2151)
libfwupdengine.so!fu_engine_write_firmware(FuEngine * self, const gchar * device_id, GInputStream * stream_fw, FuProgress * progress, FwupdInstallFlags flags, GError ** error) (/home/crabx/mywork/fwupd/src/fu-engine.c:3202)
libfwupdengine.so!fu_engine_install_blob(FuEngine * self, FuDevice * device, GInputStream * stream_fw, FuProgress * progress, FwupdInstallFlags flags, FwupdFeatureFlags feature_flags, GError ** error) (/home/crabx/mywork/fwupd/src/fu-engine.c:3383)
libfwupdengine.so!fu_engine_install_release(FuEngine * self, FuRelease * release, GInputStream * stream, FuProgress * progress, FwupdInstallFlags flags, GError ** error) (/home/crabx/mywork/fwupd/src/fu-engine.c:2403)
libfwupdengine.so!fu_engine_install_releases(FuEngine * self, FuEngineRequest * request, GPtrArray * releases, FuCabinet * cabinet, FuProgress * progress, FwupdInstallFlags flags, GError ** error) (/home/crabx/mywork/fwupd/src/fu-engine.c:1885)
fu_daemon_authorize_install_queue(FuMainAuthHelper * helper_ref) (/home/crabx/mywork/fwupd/src/fu-daemon.c:794)
fu_daemon_authorize_install_cb(GObject * source, GAsyncResult * res, gpointer user_data) (/home/crabx/mywork/fwupd/src/fu-daemon.c:751)
libgio-2.0.so.0![Unknown/Just-In-Time compiled code] (Unknown Source:0)
libfwupdengine.so!fu_polkit_authority_check_cb(GObject * source_object, GAsyncResult * res, gpointer user_data) (/home/crabx/mywork/fwupd/src/fu-polkit-authority.c:75)
libgio-2.0.so.0!g_simple_async_result_complete (Unknown Source:0)
libpolkit-gobject-1.so.0![Unknown/Just-In-Time compiled code] (Unknown Source:0)
libglib-2.0.so.0!g_main_context_dispatch (Unknown Source:0)
libglib-2.0.so.0![Unknown/Just-In-Time compiled code] (Unknown Source:0)
libglib-2.0.so.0!g_main_loop_run (Unknown Source:0)
fu_daemon_start(FuDaemon * self) (/home/crabx/mywork/fwupd/src/fu-daemon.c:70)

hughsie, 2024-01-02:
Ahh of course! Can you re-pull wip/hughsie/legacy-generic-message and try again please.

CragW, 2024-01-02:
> Ahh of course! Can you re-pull wip/hughsie/legacy-generic-message and try again please.

It works! We have a clean table!

superm1, 2024-01-02:
should have closed with e035a7a

closed 2024-01-02 by superm1