fwupd/fwupd #6590: 504 error when trying to upgrade firmware

aragilar, 2024-01-02:
**Describe the bug**
I get the following when running `sudo fwupdmgr upgrade`:
```
Perform operation? [Y|n]: Y
Downloading…             [        |                              ]
Failed to download, server response was 504: <html>
<head><title>504 Gateway Time-out</title></head>
<body>
<center><h1>504 Gateway Time-out</h1></center>
</body>
</html>
```

**Steps to Reproduce**
1. Run `sudo fwupdmgr upgrade`

**Expected behavior**
Firmware to be downloaded and installed.

**fwupd version information**
```
$ fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.14
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.9.10
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.9.10
runtime   org.kernel                    6.6.8-amd64
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

apt (Debian testing/unstable)

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```
$ fwupdmgr get-devices --show-all-devices
LENOVO 20FNCTO1WW
│
├─B140XTN02.E:
│     Device ID:          10feb8f8641be9b4f44d52078baa6338209a53ef
│     GUID:               d83861aa-1b88-55ed-b844-e259c7f2c9d8 ← DRM\VEN_AUO&DEV_2E3C
│     Device Flags:       • Internal device
│     Device Requests:    • Message
│   
├─Core™ i3-6100U CPU @ 2.30GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f0
│     Vendor:             Intel
│     GUIDs:              28d7f31c-1663-5b15-a900-94a71cbcd53f ← CPUID\PRO_0&FAM_06&MOD_4E
│                         e4bc7423-9fec-55a6-8f6c-827c40992491 ← CPUID\PRO_0&FAM_06&MOD_4E&STP_3
│     Device Flags:       • Internal device
│     Device Requests:    • Message
│   
├─MZ7LF128HCHP-000L1:
│     Device ID:          e421b2fc248391f6fe3e55ddbb3c9043be068bd0
│     Summary:            ATA drive
│     Current version:    FXT03L1Q
│     Vendor:             Samsung (ATA:0x144D, OUI:002538)
│     Serial Number:      S2G8NX0H411661
│     GUIDs:              4eda6340-82d2-5e5c-a96d-66a84f196628 ← IDE\SAMSUNG_MZ7LF128HCHP-000L1______________FXT03L1Q
│                         b88485fe-dec6-5fed-ac34-f26e125a3743 ← IDE\0SAMSUNG_MZ7LF128HCHP-000L1______________
│                         6895fb8e-42a0-5ab4-beb3-76133abeb9b6 ← SAMSUNG MZ7LF128HCHP-000L1
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│                         • Image
│                         • Image (custom)
│   
├─Skylake GT2 [HD Graphics 520]:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    07
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              bdcb90a0-e7cd-5d77-995d-d79b7183aa29 ← PCI\VEN_8086&DEV_1916
│                         4b4fbe62-e31c-56fd-86cc-7b71c53890b2 ← PCI\VEN_8086&DEV_1916&SUBSYS_17AA5053
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│     Device Requests:    • Message
│   
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.10
│ │   Minimum Version:    0.1.10
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUID:               c9a0acef-aa63-4b85-a52d-a9384163c0e8
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Device is usable for the duration of the update
│ │   Device Requests:    • Message
│ │                       • Image
│ │                       • Image (custom)
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  13
│       Minimum Version:  13
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            d4b03e62-4f5c-5e54-abec-7b7a4c90a58c ← UEFI\CRT_255F29D0DCC95E7F77B06ECCDF60EE9F567CBCE03FB753B09BA5BBA7F04C1F58&ARCH_X64
│                         f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Only version upgrades are allowed
│                         • Signed Payload
│       Device Requests:  • Message
│                         • Image
│     
├─UEFI Device Firmware:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    2952791221
│     Minimum Version:    2952791221
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               304a041a-be63-4f29-b429-168def233a23
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│                         • Image
│                         • Image (custom)
│   
└─UEFI Platform Key:
      Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
      Summary:            TPCDL-KEK
      Vendor:             Lenovo(Beijing) Ltd.
      GUID:               78139237-4daf-51d0-a807-e874a6127e65 ← UEFI\CRT_0E9CF5A2CB58F6EFDD6565A7D8DBD0163894C367
```

</details>

**Additional questions**

- Operating system and version: Debian testing/unstable
- Have you tried rebooting? No, but I have tried this a few times (I presume some backend service is down).
- Is this a regression? I guess so?

hughsie, 2024-01-02:
What does `fwupdmgr upgrade -vv` say please`

aragilar, 2024-01-02:
It looks like whatever system that was down is now up, the download worked and the upgrade was installed. Thanks (I presume you fixed the thing that was down)!

closed 2024-01-02 by aragilar

hughsie, 2024-01-02:
I didn't touch anything :) I assume the local CDN node you were using magically started working.