fwupd/fwupd #6601: fwupdmgr security reports same info twice

mcatanzaro, 2024-01-03:
**Describe the bug**
Look at the following output of `fwupdmgr security`:

```
$ fwupdmgr security
Host Security ID: HSI:0 (v1.9.10)

HSI-1
✔ BIOS firmware updates:         Enabled
✔ UEFI bootservice variables:    Locked
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled
✘ Fused platform:                Unknown
✘ Fused platform:                Unknown
✘ Supported CPU:                 Invalid
✘ TPM v2.0:                      Not found

HSI-2
✔ IOMMU:                         Enabled
✘ SPI write protection:          Unknown
✘ SPI write protection:          Unknown
✘ Platform debugging:            Unknown
✘ Platform debugging:            Unknown

HSI-3
✘ SPI replay protection:         Unknown
✘ SPI replay protection:         Unknown
✘ Pre-boot DMA protection:       Disabled
✘ Suspend-to-idle:               Disabled
✘ Suspend-to-ram:                Enabled

HSI-4
✘ Processor rollback protection: Unknown
✘ Processor rollback protection: Unknown
✘ Encrypted RAM:                 Unknown
✘ Encrypted RAM:                 Unknown

Runtime Suffix -!
✔ fwupd plugins:                 Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Encrypted
✔ Linux kernel:                  Untainted

This system has a low HSI security level.
 » https://fwupd.github.io/hsi.html#low-security-level

Host Security Events
  2023-10-10 14:21:29:  ✔ IOMMU device protection enabled
  2023-10-10 14:21:29:  ✘ Pre-boot DMA protection changed: Invalid → Disabled
  2023-07-05 11:49:50:  ✘ IOMMU device protection disabled
  2023-07-05 11:49:50:  ✘ Pre-boot DMA protection changed: Disabled → Invalid
```

Notice that several items are reported twice:

 * Fused platform
 * SPI write protection
 * Platform debugging
 * SPI replay protection
 * Processor rollback protection
 * Encrypted RAM

They should surely only be reported once.

**Steps to Reproduce**

1. Run `fwupdmgr security`

**Expected behavior**

Each check should be displayed only once, not twice

**fwupd version information**

```
$ fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.14
compile   com.hughsie.libjcat           0.1.14
compile   org.freedesktop.fwupd         1.9.10
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.10
runtime   org.kernel                    6.6.7-200.fc39.x86_64
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

This is fwupd-1.9.10-1.fc39

<summary>**fwupd device information**</summary>

```
$ fwupdmgr get-devices --show-all-devices
To Be Filled By O.E.M. To Be Filled By O.E.M.
│
├─Unknown Device:
│     Device ID:          a29de8124928f7dc31db026069ac1d601a56056f
│     GUID:               2d9ac22e-0ffe-5691-9e21-f9fecce8761c ← GPIO\ID_AMDIF030:00
│   
├─AMD Ryzen Threadripper 1950X 16-Core Processor:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x08001137
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              5fa201c0-e152-5003-a56b-93a9dd683fc7 ← CPUID\PRO_0&FAM_17&MOD_01
│ │                       0eab26c9-612c-5655-957c-6a4bec51e214 ← CPUID\PRO_0&FAM_17&MOD_01&STP_1
│ │   Device Flags:       • Internal device
│ │   Device Requests:    • Message
│ │ 
│ └─Secure Processor:
│       Device ID:        c54ab0237d7a8db8c717b68e0be78e4374a2a079
│       Vendor:           Advanced Micro Devices, Inc. (PCI:0x1022)
│       GUIDs:            aa0a9592-5ebf-51ed-b868-10b866630b66 ← PCI\VEN_1022&DEV_1456
│                         38f5fb89-5a5f-5bdc-90d3-e94363e1c9be ← PCI\VEN_1022&DEV_1456&SUBSYS_10221456
│       Device Flags:     • Internal device
│       Device Requests:  • Message
│     
├─Ellesmere [Radeon RX 470/480/570/570X/580/580X/590] (Radeon RX 580 Pulse 4GB):
│ │   Device ID:          1e8438ff6674a9df9ab19014dd190d4d8f50ea7a
│ │   Summary:            E387 Polaris20 XL A1 GDDR5 128Mx32 4GB
│ │   Current version:    015.050.002.001.000000
│ │   Vendor:             Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002, PCI:0x1022)
│ │   GUID:               884c0933-2918-5ff3-89a8-426c6686f9c7 ← AMD\113-2E3871U-O4A
│ │   Device Flags:       • Cryptographic hash verification is available
│ │ 
│ └─Unknown Device:
│       Device ID:        d324166a5e48e27ba0360ac27eeaa89559930203
│       Serial Number:    6CM7321MGS
│       GUID:             d0adeec9-136e-5b2a-b47b-c2ef5a9653c2 ← DRM\VEN_HWP&DEV_3183
│     
├─GPIO controller:
│     Device ID:          f685512aa07369c9e77742acef941d779d31e766
│     GUID:               37b440a9-2473-5087-a39b-db84f32a8ed8 ← GPIO\ID_AMDI0030:00
│   
├─SSD 970 EVO 1TB:
│     Device ID:          08719eac4fd20b7fb04497ff483fe71d94a92780
│     Summary:            NVM Express solid state drive
│     Current version:    2B2QEXE7
│     Vendor:             Samsung (NVME:0x144D)
│     Serial Number:      S467NF0K607390K
│     GUIDs:              47335265-a509-51f7-841e-1c94911af66b ← NVME\VEN_144D&DEV_A808
│                         c9d531ea-ee7d-5562-8def-c64d0d144813 ← NVME\VEN_144D&DEV_A808&SUBSYS_144DA801
│                         70520c98-99f3-5955-a1be-33b0a4cdce23 ← Samsung SSD 970 EVO 1TB
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│     Device Requests:    • Message
│                         • Image
│                         • Image (custom)
│   
├─UEFI Platform Key:
│     Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
│     Summary:            Root Agency
│     GUID:               3e989e5c-2bc2-5087-8216-43b67436b3b4 ← UEFI\CRT_CB5B70FA28620B74EF4943C7F12EE9265087E201
│   
├─UEFI dbx:
│     Device ID:          362301da643102b9f38477387e2193e57abaa590
│     Summary:            UEFI revocation database
│     Current version:    371
│     Minimum Version:    371
│     Vendor:             UEFI:Linux Foundation
│     Install Duration:   1 second
│     GUID:               f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Only version upgrades are allowed
│                         • Signed Payload
│     Device Requests:    • Message
│                         • Image
│   
└─X399 Series Chipset SATA Controller:
      Device ID:          7d29f2075dcafb4488b40c73f199cf46bb76bddb
      Current version:    02
      Vendor:             Advanced Micro Devices, Inc. [AMD] (PCI:0x1022)
      GUIDs:              37076a7c-c106-5f7d-b9ad-098dcd405a55 ← PCI\VEN_1022&DEV_43B6
                          e511e400-c9e1-523b-a3ab-326677431701 ← PCI\VEN_1022&DEV_43B6&SUBSYS_1B211062
                          1d3e715a-7904-528d-8c2f-ae4648e476e7 ← PCI\VEN_1022&DEV_1453
                          1b20334c-5fb2-5388-8b96-b8f7672d2cd8 ← PCI\VEN_1022&DEV_1453&SUBSYS_10221453
      Device Flags:       • Internal device
                          • Cryptographic hash verification is available
      Device Requests:    • Message
```

**Additional questions**

- Operating system and version: Fedora 39
- Have you tried rebooting? yes
- Is this a regression? yes

superm1, 2024-01-03:
Can you repro on fwupd 1.9.11?

hughsie, 2024-01-03:
Is it the same with https://bodhi.fedoraproject.org/updates/FEDORA-2024-237c69f13e ?

mcatanzaro, 2024-01-03:
I installed fwupd 1.9.11 and ran `systemctl restart fwupd` but did not reboot my computer. The output is identical except for the fwupd version number. Not fixed.

mcatanzaro, 2024-01-03:
> The output is identical except for the fwupd version number. 

(Well, actually I see there are a couple new checks, CET Platform and SMAP, which are displayed only once and not twice.)

superm1, 2024-01-03:
Can you please provide `sudo fwupdtool security --verbose --verbose` output for us to understand why this is happening?

superm1, 2024-01-03:
Also how many CCP devices do you have in `/sys/bus/pci/drivers/ccp`?

Maybe there is more than one for your machine.

mcatanzaro, 2024-01-03:
> Can you please provide sudo fwupdtool security --verbose --verbose output for us to understand why this is happening?

Here: [log.txt](https://github.com/fwupd/fwupd/files/13821162/log.txt)

> Also how many CCP devices do you have in /sys/bus/pci/drivers/ccp?
>
> Maybe there is more than one for your machine.

Aha, yes, I have two. I think this is expected, right? One CCP per CCD, perhaps? (I wonder what would happen in the future when AMD starts releasing non-heterogenous CPUs where each CCD is not equivalent....)

superm1, 2024-01-03:
Yeah if you have two CCP devices this is expected behavior.

mcatanzaro, 2024-01-03:
I guess you mean "expected" as in "I understand why this is happening" and not "it's OK to show this output to the user," right?

Possible solution, show something like:

```
✘ Fused platform (CCP 0):                Unknown
✘ Fused platform (CCP 1):                Unknown
```

instead of:

```
✘ Fused platform:                Unknown
✘ Fused platform:                Unknown
```

mcatanzaro, 2024-01-03:
(I also wonder whether the requirement should be to have one CCP that supports each required feature, or whether every CCP must support each required feature, or whether the requirement should be different based on each feature. No clue.)

superm1, 2024-01-03:
In general the multiple CCP endpoints is more relevant for multi-socket systems than it is for single socket but multiple CCDs.  I guess a superset of all CCP devices makes most sense like @hughsie suggested.

superm1, 2024-01-03:
Actually I don't know that a combination of all devices actually makes proper sense.

These are **system** attributes that are exported by the PSP _through_ the CCP PCI device.

I am actually not sure that a system that supports them will actually export through multiple CCP devices. I'm pretty sure that only the first gets exported.

So in the case of multiple devices and not supported it's probably better to just duplicate or obsolete the attributes instead.

superm1, 2024-01-08:
I've merged the PR that I think should fix this, but can you please report back with your test results?

closed 2024-01-08 by superm1

mcatanzaro, 2024-01-08:
Yes, that's on my to-do list.

hughsie, 2024-01-09:
@mcatanzaro I can do you a copr build if that would help.

mcatanzaro, 2024-01-09:
Yes, that would speed things up, thanks!

hughsie, 2024-01-09:
> Yes, that would speed things up, thanks!

https://copr.fedorainfracloud.org/coprs/rhughes/fwupd/build/6876534/

mcatanzaro, 2024-01-09:
Confirmed: it's fixed