fwupd/fwupd #6603: Swap appearing as encrypted even the machine doesn't have swap enabled

mhalano, 2024-01-03:
**Describe the bug**
When I execute `fwupdmgr security` it shows my swap as encrypted. The problem is I don't have swap enabled on my machine.

**Steps to Reproduce**
Disable swap and execute `fwupdmgr security`
**Expected behavior**
A clear and concise description of what you expected to happen.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
compile   com.hughsie.libxmlb           0.3.14
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.9.11
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.9.11
runtime   org.kernel                    6.5.0-14-generic

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
apt. Package from Debian on Ubuntu.

<details>
Please provide the output of the fwupd devices recognized in your system.

```shell
Dell Inc. Dell G15 5511
│
├─DELL0A70:00 04F3:317E:
│     Device ID:          f0b5af888f84b1518085848a548471c338c31187
│     Summary:            Touchpad
│     Current version:    0x0005
│     Bootloader Version: 0x0001
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              db972db2-d0dd-557e-9ddd-ea0a77323371 ← HIDRAW\VEN_04F3&DEV_317E
│                         38d75275-10b3-5f2f-8c9f-a8d60111c768 ← HIDRAW\VEN_04F3&DEV_317E&MOD_0089
│                         4eed77b6-0567-5942-a842-2413a8622e3a ← ELANTP\ICTYPE_10&MOD_0089
│                         44426a60-0574-56d9-889f-48746ea06ba8 ← ELANTP\ICTYPE_10&MOD_0089&DRIVER_HID
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─Unknown Device:
│     Device ID:          d324166a5e48e27ba0360ac27eeaa89559930203
│     Serial Number:      H4ZN900158
│     GUID:               e43a20c5-53bc-57ca-8b70-d9a4e9d5c6e7 ← DRM\VEN_SAM&DEV_0F52
│   
├─Unknown Device:
│     Device ID:          bbc98c073ac3282642278424d95dc59fcd055153
│     GUID:               c9ddd069-c7f2-58c5-b382-0487e683463a ← GPIO\ID_INT34C6:00
│   
├─11th Gen Intel Core™ i7-11800H @ 2.30GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x0000004e
│     Vendor:             Intel
│     GUIDs:              a6bd4ca5-75a6-5796-b564-66b5cab1b11b ← CPUID\PRO_0&FAM_06&MOD_8D
│                         d9dd5e77-df17-5bab-b5ec-22827598bfed ← CPUID\PRO_0&FAM_06&MOD_8D&STP_1
│     Device Flags:       • Internal device
│   
├─184X2:
│     Device ID:          aec1a869eb0df71b7cea6b3ac71d39b830faf164
│     GUID:               eca528bb-e474-56d5-ac6e-2ffc768b2a1f ← DRM\VEN_AUO&DEV_AE8B
│     Device Flags:       • Internal device
│   
├─GA106M [GeForce RTX 3060 Mobile / Max-Q]:
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Current version:    a1
[... 120 lines cut ...]
│                         • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
      Summary:            UEFI ESRT device
      Current version:    2451608321
      Minimum Version:    2451608321
      Vendor:             Dell (DMI:Dell Inc.)
      Update State:       Success
      GUID:               096da7a8-2137-4343-8d99-bdc6f640016d
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Supported on remote server
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```

</details>

**Additional questions**

- Operating system and version: Ubuntu 23.10
- Have you tried rebooting? No
- Is this a regression? Yes

mhalano, 2024-01-03:
```
➜ fwupdmgr security                      
Host Security ID: HSI:3! (v1.9.11)

HSI-1
✔ BIOS firmware updates:         Enabled
✔ MEI key manifest:              Valid
✔ csme manufacturing mode:       Locked
✔ csme override:                 Locked
✔ csme v0:15.0.45.2411:          Valid
✔ Platform debugging:            Disabled
✔ SPI write:                     Disabled
✔ SPI lock:                      Enabled
✔ SPI BIOS region:               Locked
✔ Supported CPU:                 Valid
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI bootservice variables:    Locked

HSI-2
✔ BIOS rollback protection:      Enabled
✔ Intel BootGuard ACM protected: Valid
✔ Intel BootGuard:               Enabled
✔ Intel BootGuard OTP fuse:      Valid
✔ Intel BootGuard verified boot: Valid
✔ Intel GDS mitigation:          Enabled
✔ IOMMU:                         Enabled
✔ Platform debugging:            Locked
✔ TPM PCR0 reconstruction:       Valid

HSI-3
✔ Intel BootGuard error policy:  Valid
✔ CET Platform:                  Supported
✔ Pre-boot DMA protection:       Enabled
✔ Suspend-to-idle:               Enabled
✔ Suspend-to-ram:                Disabled

HSI-4
✔ SMAP:                          Enabled
✘ Encrypted RAM:                 Not supported

Runtime Suffix -!
✔ fwupd plugins:                 Untainted
✔ Linux swap:                    Encrypted
✘ CET OS Support:                Not supported
✘ Linux kernel lockdown:         Disabled
✘ Linux kernel:                  Tainted
✘ UEFI secure boot:              Disabled

This system has HSI runtime issues.
 » https://fwupd.github.io/hsi.html#hsi-runtime-suffix

Host Security Events
  2023-12-05 02:04:17:  ✔ Intel CET Enabled changed: Enabled → Supported
  2023-09-16 22:51:13:  ✔ Pre-boot DMA protection is enabled
  2023-09-13 18:20:16:  ✘ Secure Boot disabled
  2023-09-13 18:20:16:  ✘ Pre-boot DMA protection is disabled
  2023-09-13 12:27:38:  ✔ TPM v2.0 changed: Not found → Found
  2023-09-13 12:11:31:  ✘ TPM v2.0 changed: Found → Not found
  2023-09-11 13:40:01:  ✘ Kernel lockdown disabled
  2023-09-11 13:40:01:  ✔ TPM v2.0 changed: Not found → Found
  2023-09-11 13:31:29:  ✘ Kernel is tainted

Host Security ID attributes uploaded successfully, thanks!
```

mhalano, 2024-01-03:
```
➜ free -h 
               total        used        free      shared  buff/cache   available
Mem:            62Gi        15Gi        30Gi       559Mi        17Gi        47Gi
Swap:             0B          0B          0B
```

superm1, 2024-01-04:
What was the last version that worked properly and didn't show this?

mhalano, 2024-01-04:
@superm1 version 1.9.10 (I just downgraded to test) shows Linux swap as disabled in green.

mhalano, 2024-01-04:
```
➜ fwupdmgr security                                                                                                                     
Host Security ID: HSI:3! (v1.9.10)

HSI-1
✔ BIOS firmware updates:         Enabled
✔ MEI key manifest:              Valid
✔ csme manufacturing mode:       Locked
✔ csme override:                 Locked
✔ csme v0:15.0.45.2411:          Valid
✔ Platform debugging:            Disabled
✔ SPI write:                     Disabled
✔ SPI lock:                      Enabled
✔ SPI BIOS region:               Locked
✔ Supported CPU:                 Invalid
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI bootservice variables:    Locked

HSI-2
✔ BIOS rollback protection:      Enabled
✔ Intel BootGuard ACM protected: Valid
✔ Intel BootGuard:               Enabled
✔ Intel BootGuard OTP fuse:      Valid
✔ Intel BootGuard verified boot: Valid
✔ Intel GDS mitigation:          Enabled
✔ IOMMU:                         Enabled
✔ Platform debugging:            Locked
✔ TPM PCR0 reconstruction:       Valid

HSI-3
✔ Intel BootGuard error policy:  Valid
✔ Intel CET Enabled:             Enabled
✔ Pre-boot DMA protection:       Enabled
✔ Suspend-to-idle:               Enabled
✔ Suspend-to-ram:                Disabled

HSI-4
✔ Intel SMAP:                    Enabled
✘ Encrypted RAM:                 Not supported

Runtime Suffix -!
✔ fwupd plugins:                 Untainted
✔ Linux swap:                    Disabled
✘ Intel CET Active:              Not supported
✘ Linux kernel lockdown:         Disabled
✘ Linux kernel:                  Tainted
✘ UEFI secure boot:              Disabled

This system has HSI runtime issues.
 » https://fwupd.github.io/hsi.html#hsi-runtime-suffix

Host Security Events
  2024-01-03 21:03:24:  ✔ Supported CPU changed: Valid → Invalid
  2024-01-03 21:03:24:  ✔ CET Platform changed: Supported → Enabled
  2023-12-05 02:04:17:  ✔ Supported CPU changed: Invalid → Valid
  2023-12-05 02:04:17:  ✔ Intel CET Enabled changed: Enabled → Supported
  2023-11-20 21:00:19:  ✔ Supported CPU changed: Valid → Invalid
  2023-09-16 22:51:13:  ✔ Pre-boot DMA protection is enabled
  2023-09-13 18:20:16:  ✘ Secure Boot disabled
  2023-09-13 18:20:16:  ✘ Pre-boot DMA protection is disabled
  2023-09-13 12:27:38:  ✔ TPM v2.0 changed: Not found → Found
  2023-09-13 12:11:31:  ✘ TPM v2.0 changed: Found → Not found
  2023-09-11 13:40:01:  ✘ Kernel lockdown disabled
  2023-09-11 13:40:01:  ✔ TPM v2.0 changed: Not found → Found

Host Security ID attributes uploaded successfully, thanks!
```

superm1, 2024-01-04:
Can you please double check `/proc/swaps` after it fails?  Maybe you've got a `/dev/zram0`

superm1, 2024-01-04:
Maybe caused by https://github.com/systemd/zram-generator

mhalano, 2024-01-04:
Nothing,
[image]

superm1, 2024-01-04:
Bizarre. I can't reproduce it.

superm1, 2024-01-04:
Let's see `sudo fwupdtool security --verbose --verbose` when it fails.

mhalano, 2024-01-04:
@superm1 Here is the log. [log.txt](https://github.com/fwupd/fwupd/files/13830072/log.txt)

superm1, 2024-01-04:
Nothing extra is in that log. @hughsie any ideas?

hughsie, 2024-01-04:
> 12:59:55.200 Fwupd                auto-setting org.fwupd.hsi.Kernel.Swap result encrypted

which I assume is being set by:

	/* none configured */
	if (!fu_linux_swap_get_enabled(swap)) {
		fwupd_security_attr_add_flag(attr, FWUPD_SECURITY_ATTR_FLAG_SUCCESS);
		fwupd_security_attr_set_result(attr, FWUPD_SECURITY_ATTR_RESULT_NOT_ENABLED);
		return;
	}

So I think this regressed in fa97f360e0a478344d5acf1728d2338b6f420cfa

closed 2024-01-04 by hughsie

mhalano, 2024-01-04:
Thank you. I will test as soon as possible, that would be when Debian package 1.9.12.

mhalano, 2024-01-05:
I compiled .deb packages from source and got this:
```➜ fwupdmgr security   
Host Security ID: HSI:3! (v2.0.0)

HSI-1
✔ BIOS firmware updates:         Enabled
✔ MEI key manifest:              Valid
✔ csme manufacturing mode:       Locked
✔ csme override:                 Locked
✔ csme v0:15.0.45.2411:          Valid
✔ Platform debugging:            Disabled
✔ SPI write:                     Disabled
✔ SPI lock:                      Enabled
✔ SPI BIOS region:               Locked
✔ Supported CPU:                 Valid
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI bootservice variables:    Locked

HSI-2
✔ BIOS rollback protection:      Enabled
✔ Intel BootGuard ACM protected: Valid
✔ Intel BootGuard:               Enabled
✔ Intel BootGuard OTP fuse:      Valid
✔ Intel BootGuard verified boot: Valid
✔ Intel GDS mitigation:          Enabled
✔ IOMMU:                         Enabled
✔ Platform debugging:            Locked
✔ TPM PCR0 reconstruction:       Valid

HSI-3
✔ CET Platform:                  Supported
✔ Intel BootGuard error policy:  Valid
✔ Pre-boot DMA protection:       Enabled
✔ Suspend-to-idle:               Enabled
✔ Suspend-to-ram:                Disabled

HSI-4
✔ SMAP:                          Enabled
✘ Encrypted RAM:                 Not supported

Runtime Suffix -!
✔ fwupd plugins:                 Untainted
✔ Linux swap:                    Disabled
✘ CET OS Support:                Not supported
✘ Linux kernel lockdown:         Disabled
✘ Linux kernel:                  Tainted
✘ UEFI secure boot:              Disabled

This system has HSI runtime issues.
 » https://fwupd.github.io/hsi.html#hsi-runtime-suffix
```
It seems working now.

mhalano, 2024-01-05:
I tried to use `sudo fwupdtool security` and with version 1.9.11 showed Linux swap as disabled, even the command `fwupdmgr security` showed as encrypted.
Yesterday I built .deb packages and executed `fwupdmgr security` and showed the correct value, but this shows that can be discrepancies between the client and the debug tool.

superm1, 2024-01-05:
The fix is backported to 1_9_X branch as well.  It will be part of 1.9.12 or 2.0.0 release.

mhalano, 2024-01-05:
@superm1 Good idea, but I would like to show the difference between `fwupdmgr` and `fwupdtool` when executing the `security` command. I can't think why the output would be different since the command isn't seem related to the running daemon. Can you think about a reason to the commands have different outputs?

superm1, 2024-01-05:
It should be on both main and 1_9_X now.  You should be able to use either for validating the fix on either `fwupdmgr` or `fwupdtool`.

When using `fwupdmgr` it uses the daemon (`fwupd`) running as a service triggered via dbus activation.  When you run `fwupdtool` the service runs directly as root in the environment provided by how it was launched.  If there are differences between the two that are consistently reproducible it's possible that a systemd bug is a play or something else significantly different about how they launch.  
Here's how you debug:
1. Stop the service from systemd (`sudo systemctl stop fwupd.service`).
2. Manually run the daemon with `--verbose --verbose`: `sudo /usr/libexec/fwupd/fwupd --verbose --verbose 2>&1 | tee fwupd.txt`
3. Reproduce the issue from the `fwupdmgr` client.

If the issue reproduces check fwupd.txt for the debug log.

If that doesn't reproduce then modify the systemd unit to add that `--verbose --verbose` to the `ExecStart` line.  Don't forget to do `sudo systemctl daemon-reload`.  After you've reproduced it you should be able to get the journal from fwupd.service to get the debug log.

mhalano, 2024-01-06:
I got the logs. I had to put `--verbose --verbose` directly in the systemd unit because if I ran the daemon standalone and then tried to use `fwupdmgr` to make the problem happen it gave me a version mismatch error.

[fwupd-journalctl.txt](https://github.com/fwupd/fwupd/files/13847932/fwupd-journalctl.txt)

mhalano, 2024-01-06:
I got this line repeated some times:
```
16195]: 01:38:20.021 Fwupd                auto-setting org.fwupd.hsi.Kernel.Swap result encrypted
```

mhalano, 2024-01-06:
Here is the log for the `fwupdtool` with the same `--verbose --verbose` options passed.
[fwupdtool-log.txt](https://github.com/fwupd/fwupd/files/13847957/fwupdtool-log.txt)

superm1, 2024-01-06:
AFAICT it's correctly marked as not-enabled now.  I think the problem is just a spurious debug line now, I'll open a PR for that.

mhalano, 2024-01-06:
@superm1 Maybe your fix could be a better approach and we could revert [this](https://github.com/fwupd/fwupd/pull/6609) PR? What you think, @hughsie?

superm1, 2024-01-06:
I think both can be there now together.