fwupd/fwupd #6610: `fwupdmgr security` output does not consistently decide whether a CPU is supported

amaxine, 2024-01-04:
**Describe the bug**
When running `fwupdmgr security`, the output doesn't remain consistent between runs, eventually determining that my CPU is unsupported by the tool. The event log suggests that the CPU should always be reported as unsupported, though I'm not sure why it was and then wasn't unsupported. 

<details>
<summary>Verbose output when supported</summary>
(fwupdmgr:3672): GLib-DEBUG: 17:45:20.437: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:3672): GLib-GIO-DEBUG: 17:45:20.440: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:3672): dconf-DEBUG: 17:45:20.440: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:3672): dconf-DEBUG: 17:45:20.440: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:3672): dconf-DEBUG: 17:45:20.440: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:3672): dconf-DEBUG: 17:45:20.440: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:3672): dconf-DEBUG: 17:45:20.440: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:3672): GLib-GIO-DEBUG: 17:45:20.440: _g_io_module_get_default: Found default implementation gnome (GProxyResolverGnome) for ‘gio-proxy-resolver’
(fwupdmgr:3672): GLib-GIO-DEBUG: 17:45:20.441: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:3672): GLib-GIO-DEBUG: 17:45:20.441: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:3672): dconf-DEBUG: 17:45:20.442: watch_established: "/system/proxy/" (establishing: 1)
(fwupdmgr:3672): dconf-DEBUG: 17:45:20.442: watch_established: "/system/proxy/http/" (establishing: 1)
(fwupdmgr:3672): dconf-DEBUG: 17:45:20.442: watch_established: "/system/proxy/https/" (establishing: 1)
(fwupdmgr:3672): dconf-DEBUG: 17:45:20.442: watch_established: "/system/proxy/ftp/" (establishing: 1)
(fwupdmgr:3672): dconf-DEBUG: 17:45:20.442: watch_established: "/system/proxy/socks/" (establishing: 1)
(fwupdmgr:3672): Fwupd-DEBUG: 17:45:21.275: Emitting ::status-changed() [idle]
Host Security ID: [1mHSI:3! (v1.9.10)[0m

[1mHSI-1[0m
✔ BIOS firmware updates:         [32m[1mEnabled[0m
✔ Fused platform:                [32m[1mLocked[0m
✔ Supported CPU:                 [32m[1mInvalid[0m
✔ TPM empty PCRs:                [32m[1mValid[0m
✔ TPM v2.0:                      [32m[1mFound[0m
✔ UEFI bootservice variables:    [32m[1mLocked[0m
✔ UEFI platform key:             [32m[1mValid[0m
✔ UEFI secure boot:              [32m[1mEnabled[0m

[1mHSI-2[0m
✔ SPI write protection:          [32m[1mEnabled[0m
✔ IOMMU:                         [32m[1mEnabled[0m
✔ Platform debugging:            [32m[1mLocked[0m
✔ TPM PCR0 reconstruction:       [32m[1mValid[0m

[1mHSI-3[0m
✔ SPI replay protection:         [32m[1mEnabled[0m
✔ Pre-boot DMA protection:       [32m[1mEnabled[0m
✔ Suspend-to-idle:               [32m[1mEnabled[0m
✔ Suspend-to-ram:                [32m[1mDisabled[0m

[1mHSI-4[0m
✘ Processor rollback protection: [31m[1mDisabled[0m
✘ Encrypted RAM:                 [31m[1mNot supported[0m

[1mRuntime Suffix -![0m
✔ fwupd plugins:                 [32m[1mUntainted[0m
✔ Linux kernel:                  [32m[1mUntainted[0m
✘ Linux kernel lockdown:         [31m[1mUnknown[0m
✘ Linux swap:                    [31m[1mUnencrypted[0m

This system has HSI runtime issues.
 » https://fwupd.github.io/hsi.html#hsi-runtime-suffix

FuMain-INFO: 17:45:21.280:   AppstreamId:          org.fwupd.hsi.Kernel.Swap
  Created:              2024-01-03
  HsiResult:            not-encrypted
  HsiResultFallback:    not-enabled
  Flags:                runtime-issue|action-config-os
  Name:                 Linux swap
  Summary:              Linux Swap
  Description:          Linux Kernel Swap temporarily saves information to disk as you work. If the information is not protected, it could be accessed by someone if they obtained the disk.
  Plugin:               linux_swap
  Uri:                  https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Kernel.Swap

FuMain-INFO: 17:45:21.280:   AppstreamId:          org.fwupd.hsi.Uefi.SecureBoot
[... 338 lines cut ...]
      Device ID:          03281da317dccd2b18de2bd1cc70a782df40ed7e
      Summary:            NVM Express solid state drive
      Current version:    624331WD
      Vendor:             Sandisk Corp (NVME:0x15B7)
      Serial Number:      23402U803056
      GUIDs:              69749ddc-cd7a-542e-981f-620656c59918 ← NVME\VEN_15B7&DEV_5030
                          53d3c0f5-42b2-5cc9-8a52-fa350528dc02 ← NVME\VEN_15B7&DEV_5030&SUBSYS_15B75030
                          4520f03f-c729-51ab-b4f4-d92eb4023470 ← WD_BLACK SN850X 4000GB
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
      Device Requests:    • Message
                          • Image
                          • Image (custom)
```

</details>

**Additional questions**

- Operating system and version: NixOS 24.05 (nixpkgs@9be0cfc0c7617723382a5330adb1ab0d6967c33d)
- Have you tried rebooting? Yes, reboots result in stating the CPU is supported until it eventually becomes unsupported
- Is this a regression? N/A - new hardware

superm1, 2024-01-04:
It should be fixed in 1.9.11, but we found a side effect that will be fixed in 1.9.12

superm1, 2024-01-04:
If you can please test the 1_9_X branch to confirm everything is fixed.

amaxine, 2024-01-05:
I updated to 1.9.11 last night and still experienced the issue, but I'll switch to head of 1_9_X later and report back.

amaxine, 2024-01-05:
Running fwupd@c4ea9fc3264c0976684d2399664c435141da6d4a I've had the issue reoccur again. Here's the output, but please let me know if I can provide more.

```
(fwupdmgr:77593): GLib-DEBUG: 23:23:07.785: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:77593): GLib-GIO-DEBUG: 23:23:07.790: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:77593): dconf-DEBUG: 23:23:07.790: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:77593): dconf-DEBUG: 23:23:07.790: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:77593): dconf-DEBUG: 23:23:07.790: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:77593): dconf-DEBUG: 23:23:07.790: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:77593): dconf-DEBUG: 23:23:07.790: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:77593): GLib-GIO-DEBUG: 23:23:07.790: _g_io_module_get_default: Found default implementation gnome (GProxyResolverGnome) for ‘gio-proxy-resolver’
(fwupdmgr:77593): GLib-GIO-DEBUG: 23:23:07.790: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:77593): GLib-GIO-DEBUG: 23:23:07.790: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:77593): dconf-DEBUG: 23:23:07.791: watch_established: "/system/proxy/" (establishing: 1)
(fwupdmgr:77593): dconf-DEBUG: 23:23:07.791: watch_established: "/system/proxy/http/" (establishing: 1)
(fwupdmgr:77593): dconf-DEBUG: 23:23:07.791: watch_established: "/system/proxy/https/" (establishing: 1)
(fwupdmgr:77593): dconf-DEBUG: 23:23:07.791: watch_established: "/system/proxy/ftp/" (establishing: 1)
(fwupdmgr:77593): dconf-DEBUG: 23:23:07.791: watch_established: "/system/proxy/socks/" (establishing: 1)
(fwupdmgr:77593): Fwupd-DEBUG: 23:23:07.792: Emitting ::status-changed() [idle]
Host Security ID: [1mHSI:0! (v1.9.12)[0m

[1mHSI-1[0m
✔ BIOS firmware updates:         [32m[1mEnabled[0m
✔ Fused platform:                [32m[1mLocked[0m
✔ TPM empty PCRs:                [32m[1mValid[0m
✔ TPM v2.0:                      [32m[1mFound[0m
✔ UEFI bootservice variables:    [32m[1mLocked[0m
✔ UEFI platform key:             [32m[1mValid[0m
✔ UEFI secure boot:              [32m[1mEnabled[0m
✘ Supported CPU:                 [31m[1mUnknown[0m

[1mHSI-2[0m
✔ SPI write protection:          [32m[1mEnabled[0m
✔ IOMMU:                         [32m[1mEnabled[0m
✔ Platform debugging:            [32m[1mLocked[0m
✔ TPM PCR0 reconstruction:       [32m[1mValid[0m

[1mHSI-3[0m
✔ SPI replay protection:         [32m[1mEnabled[0m
✔ Pre-boot DMA protection:       [32m[1mEnabled[0m
✔ Suspend-to-idle:               [32m[1mEnabled[0m
✔ Suspend-to-ram:                [32m[1mDisabled[0m
✘ CET Platform:                  [31m[1mNot supported[0m

[1mHSI-4[0m
✔ SMAP:                          [32m[1mEnabled[0m
✘ Processor rollback protection: [31m[1mDisabled[0m
✘ Encrypted RAM:                 [31m[1mNot supported[0m

[1mRuntime Suffix -![0m
✔ fwupd plugins:                 [32m[1mUntainted[0m
✔ Linux kernel:                  [32m[1mUntainted[0m
✘ Linux kernel lockdown:         [31m[1mUnknown[0m
✘ Linux swap:                    [31m[1mUnencrypted[0m

This system has a low HSI security level.
 » https://fwupd.github.io/hsi.html#low-security-level

This system has HSI runtime issues.
 » https://fwupd.github.io/hsi.html#hsi-runtime-suffix

FuMain-INFO: 23:23:07.800:   AppstreamId:          org.fwupd.hsi.SupportedCpu
  Created:              2024-01-05
  HsiLevel:             1
  HsiResultFallback:    valid
  Flags:                missing-data|action-contact-oem
  Name:                 Supported CPU
  Summary:              Processor Security Checks
  Description:          Each system should have tests to ensure firmware security.
  Plugin:               core
[... 83 lines cut ...]
  Plugin:               core
  Uri:                  https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.SupportedCpu

FuMain-INFO: 23:23:07.800:   AppstreamId:          org.fwupd.hsi.SupportedCpu
  Created:              2024-01-04
  HsiLevel:             1
  HsiResult:            not-valid
  HsiResultFallback:    valid
  Flags:                missing-data|action-contact-oem
  Name:                 Supported CPU
  Summary:              Processor Security Checks
  Description:          Each system should have tests to ensure firmware security.
  Plugin:               core
  Uri:                  https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.SupportedCpu

Host Security Events
  2024-01-05 18:45:47:  [32m[1m✔[0m Supported CPU changed: Invalid → Valid
  2024-01-05 18:45:43:  [31m[1m✘[0m Supported CPU changed: Valid → Invalid
  2024-01-05 13:41:31:  [32m[1m✔[0m Supported CPU changed: Invalid → Valid
  2024-01-05 11:38:43:  [31m[1m✘[0m Supported CPU changed: Valid → Invalid
  2024-01-05 11:13:53:  [32m[1m✔[0m Supported CPU changed: Invalid → Valid
  2024-01-04 19:01:05:  [31m[1m✘[0m Supported CPU changed: Valid → Invalid
  2024-01-04 18:10:06:  [32m[1m✔[0m Supported CPU changed: Invalid → Valid
  2024-01-04 17:25:58:  [31m[1m✘[0m Supported CPU changed: Valid → Invalid
```

superm1, 2024-01-05:
`sudo fwupdtool security --verbose --verbose` output please

amaxine, 2024-01-05:
Here's the same output when running as root:
```shell
(fwupdmgr:105493): GLib-DEBUG: 23:43:07.727: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:105493): GLib-GIO-DEBUG: 23:43:07.730: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:105493): dconf-DEBUG: 23:43:07.730: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:105493): dconf-DEBUG: 23:43:07.730: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:105493): dconf-DEBUG: 23:43:07.730: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:105493): dconf-DEBUG: 23:43:07.730: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:105493): dconf-DEBUG: 23:43:07.730: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:105493): GLib-GIO-DEBUG: 23:43:07.730: _g_io_module_get_default: Found default implementation gnome (GProxyResolverGnome) for ‘gio-proxy-resolver’
(fwupdmgr:105493): GLib-GIO-DEBUG: 23:43:07.735: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:105493): GLib-GIO-DEBUG: 23:43:07.735: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:105493): dconf-DEBUG: 23:43:07.735: watch_established: "/system/proxy/" (establishing: 1)
(fwupdmgr:105493): dconf-DEBUG: 23:43:07.735: watch_established: "/system/proxy/http/" (establishing: 1)
(fwupdmgr:105493): dconf-DEBUG: 23:43:07.735: watch_established: "/system/proxy/https/" (establishing: 1)
(fwupdmgr:105493): dconf-DEBUG: 23:43:07.736: watch_established: "/system/proxy/ftp/" (establishing: 1)
(fwupdmgr:105493): dconf-DEBUG: 23:43:07.736: watch_established: "/system/proxy/socks/" (establishing: 1)
(fwupdmgr:105493): Fwupd-DEBUG: 23:43:07.737: Emitting ::status-changed() [idle]
Host Security ID: [1mHSI:0! (v1.9.12)[0m

[1mHSI-1[0m
✔ BIOS firmware updates:         [32m[1mEnabled[0m
✔ Fused platform:                [32m[1mLocked[0m
✔ TPM empty PCRs:                [32m[1mValid[0m
✔ TPM v2.0:                      [32m[1mFound[0m
✔ UEFI bootservice variables:    [32m[1mLocked[0m
✔ UEFI platform key:             [32m[1mValid[0m
✔ UEFI secure boot:              [32m[1mEnabled[0m
✘ Supported CPU:                 [31m[1mUnknown[0m

[1mHSI-2[0m
✔ SPI write protection:          [32m[1mEnabled[0m
✔ IOMMU:                         [32m[1mEnabled[0m
✔ Platform debugging:            [32m[1mLocked[0m
✔ TPM PCR0 reconstruction:       [32m[1mValid[0m

[1mHSI-3[0m
✔ SPI replay protection:         [32m[1mEnabled[0m
✔ Pre-boot DMA protection:       [32m[1mEnabled[0m
✔ Suspend-to-idle:               [32m[1mEnabled[0m
✔ Suspend-to-ram:                [32m[1mDisabled[0m
✘ CET Platform:                  [31m[1mNot supported[0m

[1mHSI-4[0m
✔ SMAP:                          [32m[1mEnabled[0m
✘ Processor rollback protection: [31m[1mDisabled[0m
✘ Encrypted RAM:                 [31m[1mNot supported[0m

[1mRuntime Suffix -![0m
✔ fwupd plugins:                 [32m[1mUntainted[0m
✔ Linux kernel:                  [32m[1mUntainted[0m
✘ Linux kernel lockdown:         [31m[1mUnknown[0m
✘ Linux swap:                    [31m[1mUnencrypted[0m

This system has a low HSI security level.
 » https://fwupd.github.io/hsi.html#low-security-level

This system has HSI runtime issues.
 » https://fwupd.github.io/hsi.html#hsi-runtime-suffix

FuMain-INFO: 23:43:07.745:   AppstreamId:          org.fwupd.hsi.SupportedCpu
  Created:              2024-01-05
  HsiLevel:             1
  HsiResultFallback:    valid
  Flags:                missing-data|action-contact-oem
  Name:                 Supported CPU
  Summary:              Processor Security Checks
  Description:          Each system should have tests to ensure firmware security.
  Plugin:               core
  Uri:                  https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.SupportedCpu
[... 78 lines cut ...]
  Summary:              Processor Security Checks
  Description:          Each system should have tests to ensure firmware security.
  Plugin:               core
  Uri:                  https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.SupportedCpu

FuMain-INFO: 23:43:07.745:   AppstreamId:          org.fwupd.hsi.SupportedCpu
  Created:              2024-01-04
  HsiLevel:             1
  HsiResult:            not-valid
  HsiResultFallback:    valid
  Flags:                missing-data|action-contact-oem
  Name:                 Supported CPU
  Summary:              Processor Security Checks
  Description:          Each system should have tests to ensure firmware security.
  Plugin:               core
  Uri:                  https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.SupportedCpu

Host Security Events
  2024-01-05 18:45:47:  [32m[1m✔[0m Supported CPU changed: Invalid → Valid
  2024-01-05 18:45:43:  [31m[1m✘[0m Supported CPU changed: Valid → Invalid
  2024-01-05 13:41:31:  [32m[1m✔[0m Supported CPU changed: Invalid → Valid
  2024-01-05 11:38:43:  [31m[1m✘[0m Supported CPU changed: Valid → Invalid
  2024-01-05 11:13:53:  [32m[1m✔[0m Supported CPU changed: Invalid → Valid
  2024-01-04 19:01:05:  [31m[1m✘[0m Supported CPU changed: Valid → Invalid
```

superm1, 2024-01-05:
I specifically want fwupdtool and the double verbose. You can attach the file.

amaxine, 2024-01-05:
Sorry, i completely misread that, late night brain.
[out.txt](https://github.com/fwupd/fwupd/files/13847321/out.txt)

superm1, 2024-01-06:
Can you please see if this PR improves things?

amaxine, 2024-01-06:
Running 89aff691112d1f644feee1549e6a555e08d97706 I've not seen problems yet, but I'll switch to the fixed up commits and retest again.

amaxine, 2024-01-06:
I've not had the issue reoccur on 152d0ef5e5d4047b69f4405f1aa98cb777048957 either.

closed 2024-01-06 by superm1

amaxine, 2024-01-24:
Not sure if this is directly related, but on 1.9.12 there is a direct inconsistency between events and reported state - according to events both the CPU and CET should be unsupported, yet the output says otherwise. Prior the events consistently lined up.

```
Host Security ID: HSI:3! (v1.9.12)

HSI-1
✔ BIOS firmware updates:         Enabled
✔ Fused platform:                Locked
✔ Supported CPU:                 Valid
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI bootservice variables:    Locked
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled

HSI-2
✔ SPI write protection:          Enabled
✔ IOMMU:                         Enabled
✔ Platform debugging:            Locked
✔ TPM PCR0 reconstruction:       Valid

HSI-3
✔ SPI replay protection:         Enabled
✔ CET Platform:                  Supported
✔ Pre-boot DMA protection:       Enabled
✔ Suspend-to-idle:               Enabled
✔ Suspend-to-ram:                Disabled

HSI-4
✔ SMAP:                          Enabled
✘ Processor rollback protection: Disabled
✘ Encrypted RAM:                 Not supported

Runtime Suffix -!
✔ fwupd plugins:                 Untainted
✔ Linux kernel:                  Untainted
✘ CET OS Support:                Not supported
✘ Linux kernel lockdown:         Unknown
✘ Linux swap:                    Unencrypted

This system has HSI runtime issues.
 » https://fwupd.github.io/hsi.html#hsi-runtime-suffix

Host Security Events
  2024-01-24 20:28:39:  ✘ Supported CPU changed: Valid → Invalid
  2024-01-24 20:28:39:  ✘ CET Platform changed: Supported → Not supported
  2024-01-24 20:16:35:  ✔ CET Platform changed: Not supported → Supported
  2024-01-12 15:37:02:  ✔ Supported CPU changed: Invalid → Valid
  2024-01-12 13:21:13:  ✘ Supported CPU changed: Valid → Invalid
  2024-01-12 00:54:11:  ✔ Supported CPU changed: Invalid → Valid
  2024-01-11 23:10:24:  ✘ Supported CPU changed: Valid → Invalid
  2024-01-11 20:02:58:  ✔ Supported CPU changed: Invalid → Valid
  2024-01-11 17:18:47:  ✘ Supported CPU changed: Valid → Invalid
  2024-01-11 11:38:37:  ✔ Supported CPU changed: Invalid → Valid
```

superm1, 2024-01-24:
Can you share output for `fwupdmgr security --verbose` to check more about it?

amaxine, 2024-01-24:
```
(fwupdmgr:20356): GLib-DEBUG: 23:05:04.708: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:20356): GLib-GIO-DEBUG: 23:05:04.711: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:20356): dconf-DEBUG: 23:05:04.711: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:20356): dconf-DEBUG: 23:05:04.711: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:20356): dconf-DEBUG: 23:05:04.711: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:20356): dconf-DEBUG: 23:05:04.711: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:20356): dconf-DEBUG: 23:05:04.711: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:20356): GLib-GIO-DEBUG: 23:05:04.711: _g_io_module_get_default: Found default implementation gnome (GProxyResolverGnome) for ‘gio-proxy-resolver’
(fwupdmgr:20356): GLib-GIO-DEBUG: 23:05:04.712: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:20356): GLib-GIO-DEBUG: 23:05:04.712: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:20356): dconf-DEBUG: 23:05:04.713: watch_established: "/system/proxy/" (establishing: 1)
(fwupdmgr:20356): dconf-DEBUG: 23:05:04.713: watch_established: "/system/proxy/http/" (establishing: 1)
(fwupdmgr:20356): dconf-DEBUG: 23:05:04.713: watch_established: "/system/proxy/https/" (establishing: 1)
(fwupdmgr:20356): dconf-DEBUG: 23:05:04.713: watch_established: "/system/proxy/ftp/" (establishing: 1)
(fwupdmgr:20356): dconf-DEBUG: 23:05:04.713: watch_established: "/system/proxy/socks/" (establishing: 1)
(fwupdmgr:20356): Fwupd-DEBUG: 23:05:04.713: Emitting ::status-changed() [idle]
Host Security ID: [1mHSI:3! (v1.9.12)[0m

[1mHSI-1[0m
✔ BIOS firmware updates:         [32m[1mEnabled[0m
✔ Fused platform:                [32m[1mLocked[0m
✔ Supported CPU:                 [32m[1mValid[0m
✔ TPM empty PCRs:                [32m[1mValid[0m
✔ TPM v2.0:                      [32m[1mFound[0m
✔ UEFI bootservice variables:    [32m[1mLocked[0m
✔ UEFI platform key:             [32m[1mValid[0m
✔ UEFI secure boot:              [32m[1mEnabled[0m

[1mHSI-2[0m
✔ SPI write protection:          [32m[1mEnabled[0m
✔ IOMMU:                         [32m[1mEnabled[0m
✔ Platform debugging:            [32m[1mLocked[0m
✔ TPM PCR0 reconstruction:       [32m[1mValid[0m

[1mHSI-3[0m
✔ SPI replay protection:         [32m[1mEnabled[0m
✔ CET Platform:                  [32m[1mSupported[0m
✔ Pre-boot DMA protection:       [32m[1mEnabled[0m
✔ Suspend-to-idle:               [32m[1mEnabled[0m
✔ Suspend-to-ram:                [32m[1mDisabled[0m

[1mHSI-4[0m
✔ SMAP:                          [32m[1mEnabled[0m
✘ Processor rollback protection: [31m[1mDisabled[0m
✘ Encrypted RAM:                 [31m[1mNot supported[0m

[1mRuntime Suffix -![0m
✔ fwupd plugins:                 [32m[1mUntainted[0m
✔ Linux kernel:                  [32m[1mUntainted[0m
✘ CET OS Support:                [31m[1mNot supported[0m
✘ Linux kernel lockdown:         [31m[1mUnknown[0m
✘ Linux swap:                    [31m[1mUnencrypted[0m

This system has HSI runtime issues.
 » https://fwupd.github.io/hsi.html#hsi-runtime-suffix

FuMain-INFO: 23:05:04.719:   AppstreamId:          org.fwupd.hsi.IntelCet.Active
  Created:              2024-01-24
  HsiResultFallback:    not-supported
  Flags:                runtime-issue
  Name:                 CET OS Support
  Summary:              Control-flow Enforcement Technology
  Description:          Control-Flow Enforcement Technology detects and prevents certain methods for running malicious software on the device.
  Plugin:               cpu
  Uri:                  https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.IntelCet.Active
  Guid:                 2a1df9de-fe02-59c8-8af1-32c8bb011f94
  Guid:                 5a4bca32-6547-584f-bb36-920d9adede25

FuMain-INFO: 23:05:04.719:   AppstreamId:          org.fwupd.hsi.SupportedCpu
[... 121 lines cut ...]

FuMain-INFO: 23:05:04.720:   AppstreamId:          org.fwupd.hsi.SupportedCpu
  Created:              2024-01-11
  HsiLevel:             1
  HsiResult:            valid
  HsiResultFallback:    not-valid
  Flags:                success|action-contact-oem
  Name:                 Supported CPU
  Summary:              Processor Security Checks
  Description:          Each system should have tests to ensure firmware security.
  Plugin:               core
  Uri:                  https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.SupportedCpu

Host Security Events
  2024-01-24 21:40:22:  [31m[1m✘[0m Supported CPU changed: Valid → Invalid
  2024-01-24 21:40:22:  [31m[1m✘[0m CET Platform changed: Supported → Not supported
  2024-01-24 20:16:35:  [32m[1m✔[0m CET Platform changed: Not supported → Supported
  2024-01-12 15:37:02:  [32m[1m✔[0m Supported CPU changed: Invalid → Valid
  2024-01-12 13:21:13:  [31m[1m✘[0m Supported CPU changed: Valid → Invalid
  2024-01-12 00:54:11:  [32m[1m✔[0m Supported CPU changed: Invalid → Valid
  2024-01-11 23:10:24:  [31m[1m✘[0m Supported CPU changed: Valid → Invalid
  2024-01-11 20:02:58:  [32m[1m✔[0m Supported CPU changed: Invalid → Valid
  2024-01-11 17:18:47:  [31m[1m✘[0m Supported CPU changed: Valid → Invalid
  2024-01-11 11:38:37:  [32m[1m✔[0m Supported CPU changed: Invalid → Valid
```