fwupd/fwupd #6613: Incapable to build Ubuntu packages

mhalano, 2024-01-05:
**Describe the bug**
Even following the instructions to generate a .deb package for Ubuntu, it didn't work, I think because Ubuntu script `contrib/ci/ubuntu.sh` is incomplete if compared with Debian, that generates the packages correctly because it doesn't have a `debuild` call to create the package. It just put the binaries under `dist/` directory.

**Steps to Reproduce**
Execute:
OS=ubuntu-x86_64 ./contrib/ci//generate_docker.py build
 docker run --privileged -t -v `pwd`:/github/workspace fwupd-ubuntu-x86_64

**Expected behavior**
I was expecting .deb packages under git root, my `pwd`.

**fwupd version information**
Commit c4f8ef408b6cee2cfd4b90929b99e3677dc0f8a3

**Additional questions**

- Operating system and version: Ubuntu 23.10
- Have you tried rebooting? No
- Is this a regression? Maybe. I don't know the Ubuntu packages worked someday.

superm1, 2024-01-05:
Ubuntu CI job isn't meant to make packages, if tests different things.

You can run the debian script on Ubuntu though if you want to.

mhalano, 2024-01-05:
Yeah, using the Debian commands the packages were generated. Let's document how to do it because some people, like myself, could prefer to compile a package to test some new feature or, as I did, check the fix of some bug.

mhalano, 2024-01-05:
As you can see, the snap generation in the pipeline isn't related to the Docker containers. The base Docker container is Ubuntu Xenial. I will try to upgrade to a more recent version.

mhalano, 2024-01-05:
As you can see, the snap generation in the pipeline isn't related to the Docker containers. The Dockerfile used to build the snap is using Ubuntu Xenial. Maybe we could upgrade to a newer version?

superm1, 2024-01-05:
> 
As you can see, the snap generation in the pipeline isn't related to the Docker containers. The Dockerfile used to build the snap is using Ubuntu Xenial. Maybe we could upgrade to a newer version?

Yeah we stopped using docker for snap build because of various problems with it a while back.  I guess you're finding some remenants of it in the source tree we should have ripped out too.

Something I'm working on doing is shifting over to a development environment (see https://github.com/fwupd/fwupd/pull/6604).  Hopefully by doing this then people won't need to use packages as frequently.

mhalano, 2024-01-05:
I still in doubt about this issue. Snap issue, similar to this one, was solved by using native command on the host instead of using a Docker container. A question I would like to arise is how package building (deb, rpm, arch, etc.) fits with this new proposed workflow using venv? Maybe since we have a `build-fwupd` command we could have a `build-deb-fwupd`, `build-rpm-fwupd`, etc.? The idea here is to create an easy way to generate packages with this new proposed workflow.

superm1, 2024-01-05:
I found the old snap stuff in contrib/ci and tore it out, so that should hopefully be less confusing.

As for package building; the new workflow with venv would mean you don't use packages typically.  If you want to use packages, then they would be triggered ideally with the docker containers.

We could try to make some wrappers that build the deb/rpm packages too from the venv, but this is in general a lot more fragile and defeats the purpose of the venv in the first place.  By doing anything in the venv you wouldn't be installing a non-distro package ever.

superm1, 2024-01-13:
I thought about this more and I really want to discourage the use of rpm/deb packages outside of the distro packaging intention.  The reason for this is that what happens if there is an incompatible ABI change (think like a database schema bump)?  By building your own deb/rpm packages and replacing the distro you are suddenly on a different path.  Trying to switch to distro packages might bring weird incompatibilities we didn't for see needing to deal with.

We keep separate tracks described in https://github.com/fwupd/fwupd/blob/main/SECURITY.md for this reason.  You shouldn't go from 1.9.x -> 2.0.x -> 1.8.x for example.

I've put a bunch of effort into making the fwupd development environment work better now and I think anyone who needs to validate or use a newer version can use any of these 3 options which are safe and will never conflict with distro packages as they use their own storage mechanisms for database and cache.
1. Snap package.
2. Flatpak package
3. Fwupd development environment.

closed 2024-01-13 by superm1