fwupd/fwupd #6615: Dell WD22: failed to update USB4 controller

CragW, 2024-01-05:
**Describe the bug**
Failed to update firmware for USB4 controller, error messages is seen on the client side:
> Could not determine sysfs path for device

Also in the daemon side:
> 03:39:38.312 FuEngine             failed to attach device after failed update: failed to attach using thunderbolt: not yet initialized

Device status after install attempt is `Pending activation`.
```
│ └─USB4 controller in Dell dock:
│       Device ID:        cf294bf55b333004beb7c41f952c1838c23e1f4a
│       Summary:          USB4 controller
│       Current version:  38.0
│       Vendor:           Dell Inc. (USB:0x8087, TBT:0x00D4)
│       Install Duration: 46 seconds
│       Update Error:     Pending activation
│       GUID:             bd79ce60-525b-5f39-a3f6-c98c495039ff ← TBT-00d4b071
│       Device Flags:     • Supported on remote server
│                         • Device update needs activation
│                         • Device stages updates
│                         • Device is usable for the duration of the update
│                         • Updatable
│                         • Signed Payload
```
Full log:
https://drive.google.com/file/d/1cvB6mL5vagvEyU2EwgGDT4VajN9rWjwx/view?usp=sharing

**Steps to Reproduce**
```
$ ./dist/fwupdmgr.sh install cf294bf55b333004beb7c41f952c1838c23e1f4a
WARNING: This package has not been validated, it may not work properly.
0.	Cancel
1.	41.81
2.	41.81
3.	38.00
4.	38.00
5.	38.00
Choose release [0-5]: 1
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade USB4 controller in Dell dock from 38.0 to 41.81?                     ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ The vendor did not supply any release notes.                                 ║
║                                                                              ║
║ USB4 controller in Dell dock must remain connected for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Package level of Dell dock from 01.00.36.00 to 01.00.36.01?          ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ The vendor did not supply any release notes.                                 ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
╔══════════════════════════════════════════════════════════════════════════════╗
║ Reinstall RTS5413 in Dell dock to 01.22?                                     ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ The vendor did not supply any release notes.                                 ║
║                                                                              ║
║ RTS5413 in Dell dock must remain connected for the duration of the update    ║
║ to avoid damage.                                                             ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
╔══════════════════════════════════════════════════════════════════════════════╗
║ Reinstall RTS5487 in Dell dock to 01.57?                                     ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ The vendor did not supply any release notes.                                 ║
║                                                                              ║
║ RTS5487 in Dell dock must remain connected for the duration of the update    ║
║ to avoid damage.                                                             ║
╚══════════════════════════════════════════════════════════════════════════════╝
[... 249 lines cut ...]
      Device ID:          1bf177ee2510c9dcbe9fa4cb630af962633789e2
      Summary:            Miniaturised USB wireless receiver
      Current version:    RQR12.11_B0032
      Bootloader Version: BOT01.04_B0016
      Vendor:             HIDRAW:0x046D|USB:0x046D
      Install Duration:   30 seconds
      GUIDs:              9d131a0c-a606-580f-8eda-80587250b8d6
                          279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
      Device Flags:       • Updatable
                          • Supported on remote server
                          • Signed Payload

```

</details>

**Dock SKU**
WD22TB4

**Additional questions**

- Operating system and version: Ubutnu 22.04
- Have you tried unplugging the dock or any peripherals from your machine? Yes
- Have you tried to power cycle the dock from the AC adapter? Yes
- Is this a regression? Yes

superm1, 2024-01-05:
Any idea when last version without this behavior? Does 1.9.x work?

CragW, 2024-01-05:
Updated successfully with tag 1.9.10.  Maybe caused by recent changes.
```
│ └─Embedded Controller:
│       Device ID:        cf294bf55b333004beb7c41f952c1838c23e1f4a
│       Summary:          USB4 controller
│       Current version:  41.81
│       Vendor:           Dell Inc. (USB:0x8087, TBT:0x00D4)
│       Install Duration: 46 seconds
│       Update State:     Success
│       Last modified:    2024-01-05 04:27
│       GUID:             bd79ce60-525b-5f39-a3f6-c98c495039ff ← TBT-00d4b071
│       Device Flags:     • Supported on remote server
│                         • Device update needs activation
│                         • Device stages updates
│                         • Device is usable for the duration of the update
│                         • Updatable
│                         • Signed Payload

```

superm1, 2024-01-05:
Try porting this change into thunderbolt
https://github.com/fwupd/fwupd/commit/94e688f95cc22541ae5d6e90e0244bcb2c8aaa3e

CragW, 2024-01-05:
> Try porting this change into thunderbolt [94e688f](https://github.com/fwupd/fwupd/commit/94e688f95cc22541ae5d6e90e0244bcb2c8aaa3e)

That works, thanks @superm1.  PR coming.

closed 2024-01-05 by hughsie

mcallaghan-geotab, 2024-02-14:
which version is this fix released in?

Ubuntu 22.04 runs an older version, and even tried snap `edge` and `candidate` ; still getting:
```
Dell Inc. Precision 5570
│
├─Package level of Dell dock:
│ │   Device ID:          c7935ae89685c7ae72560c6ed0f93c0e1eefdfd0
│ │   Summary:            A representation of dock update status
│ │   Current version:    01.00.36.00
│ │   Vendor:             Dell Inc. (USB:0x413C)
│ │   Install Duration:   5 seconds
│ │   Problems:           • Device is waiting for the update to be applied
│ │   GUIDs:              af2a3b39-98c4-59a9-9cbe-a6909a7d14a2
│ │                       7fb217aa-2312-5c99-a853-56d865c76815 ← USB\VID_413C&PID_B06E&hub&salomon_mlk_status
│ │   Device Flags:       • Supported on remote server
│ │                       • Device can recover flash failures
│ │                       • Device is usable for the duration of the update
│ │                       • Updatable
│ │                       • Unsigned Payload
│ │ 
│ └─WD22 Dock Update:
│       New version:      01.00.36.01
│       Remote ID:        lvfs
│       Release ID:       75527
│       Summary:          Firmware update for WD22
│       License:          Proprietary
│       Size:             24 bytes
│       Created:          2023-12-14
│       Urgency:          Critical
│       Vendor:           Dell
│       Duration:         5 seconds
│       Release Flags:    • Trusted metadata
│                         • Is upgrade
│       Description:      
│       -Update the firmware package version of WD19/WD22 Dell docking station's composite devices to 01.00.36
│       Checksum:         0572352b20a0b071331971f0698625afdd140596f2e38a24359aaf25c8ab4cf8
│     
└─USB4 controller in Dell dock:
  │   Device ID:          54f0d9041b6c5438c7ff825f5139559c5ca1b222
  │   Summary:            USB4 controller
  │   Current version:    38.0
  │   Vendor:             Dell Inc. (USB:0x8087, TBT:0x00D4)
  │   Install Duration:   46 seconds
  │   Update Error:       Pending activation
  │   GUID:               bd79ce60-525b-5f39-a3f6-c98c495039ff ← TBT-00d4b071
  │   Device Flags:       • Supported on remote server
  │                       • Device update needs activation
  │                       • Device stages updates
  │                       • Device is usable for the duration of the update
  │                       • Updatable
  │                       • Signed Payload
  │ 
  └─USB4 Controller Update:
        New version:      41.81
        Remote ID:        lvfs
        Release ID:       75538
        Summary:          Firmware update for USB4
        License:          Proprietary
        Size:             294.9 kB
        Created:          2023-12-14
        Urgency:          Critical
        Vendor:           Dell
        Duration:         46 seconds
        Release Flags:    • Trusted metadata
                          • Is upgrade
        Description:      
        -This release contains bug fixes to improve the quality of USB4 controller in Dell dock.
        Checksum:         0572352b20a0b071331971f0698625afdd140596f2e38a24359aaf25c8ab4cf8

```

smagellan, 2025-02-27:
@CragW @superm1 still does not work:
```
fwupdmgr refresh --force
Updating lvfs
Downloading…             [   -                                   ]
Successfully downloaded new metadata: Updates have been published for 5 of 11 local devices
Embedded Controller is pending activation; use fwupdmgr activate to complete the update.
WD22TB4 is pending activation; use fwupdmgr activate to complete the update.

fwupdmgr --version
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.6
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.7
runtime   com.hughsie.libjcat           0.2.0
runtime   com.hughsie.libxmlb           0.3.21
runtime   org.kernel                    6.13.4
runtime   org.freedesktop.fwupd         2.0.6
```

Any ideas on how one can trigger activation?