fwupd/fwupd #6626: `dbx` updates not offered after enrolling own signing keys

amaxine, 2024-01-05:
**Describe the bug**
`fwupdmgr` does not offer to update dbx after enrolling own keys. Microsoft vendor keys are still enrolled in UEFI alongside them.

**Steps to Reproduce**
Enroll own keys into UEFI (example reasoning: no pre-signed shim availability) and note that `fwupdmgr` no longer offers to update dbx.

**Expected behavior**
`fwupdmgr` continues to offer to update dbx as long as vendor keys are installed, regardless of presence of other keys.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
compile   com.hughsie.libxmlb           0.3.14
compile   com.hughsie.libjcat           0.1.14
compile   org.freedesktop.fwupd         1.9.11
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    6.6.8
runtime   org.freedesktop.fwupd         1.9.11
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): fwupd from nixpkgs

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
Framework Laptop 13 (AMD Ryzen 7040Series)
│
├─AMD Ryzen 7 7840U w/ Radeon 780M Graphics:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              2a1df9de-fe02-59c8-8af1-32c8bb011f94 ← CPUID\PRO_0&FAM_19&MOD_74
│ │                       5a4bca32-6547-584f-bb36-920d9adede25 ← CPUID\PRO_0&FAM_19&MOD_74&STP_1
│ │   Device Flags:       • Internal device
│ │ 
│ ├─Graphics Processing Unit (GPU):
│ │ │   Device ID:        83eba1fa14bdefee1f6b4bcf06069b298e1c338f
│ │ │   Summary:          AMD AMD_PHOENIX_GENERIC
│ │ │   Current version:  022.012.000.024.000001
│ │ │   Vendor:           Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│ │ │   GUID:             6c23af4f-d6cd-5ffc-a502-0b85e472e7cb ← AMD\113-PHXGENERIC-001
│ │ │   Device Flags:     • Internal device
│ │ │ 
│ │ └─NE135FBM-N41:
│ │       Device ID:      10feb8f8641be9b4f44d52078baa6338209a53ef
│ │       GUID:           ace3bb34-1bbb-5a5f-9694-73c4cede9692 ← DRM\VEN_BOE&DEV_0BCA
│ │       Device Flags:   • Internal device
│ │     
│ ├─Secure Processor:
│ │     Device ID:        c54ab0237d7a8db8c717b68e0be78e4374a2a079
│ │     Current version:  00.2d.00.74
│ │     Bootloader Version:00.2d.00.74
│ │     Vendor:           Advanced Micro Devices, Inc. (PCI:0x1022)
│ │     GUIDs:            9eb6a793-7f97-5fb5-b49c-139d2ce3ee46 ← PCI\VEN_1022&DEV_15C7
│ │                       2e4eb727-a7c8-5eba-80cd-553b8d796d80 ← PCI\VEN_1022&DEV_15C7&SUBSYS_F1110006
│ │     Device Flags:     • Internal device
│ │   
│ └─System Management Unit (SMU):
│       Device ID:        db0330716216c629bb2c07256e5d018f499eb6ce
│       Summary:          Microcontroller used within CPU/APU program 0
│       Current version:  76.70.0
│       Vendor:           Advanced Micro Devices, Inc.
[... 123 lines cut ...]
│   │   ├── nixos-generation-44.efi
│   │   ├── nixos-generation-45.efi
│   │   └── nixos-generation-46.efi
│   ├── nixos
│   │   ├── 1pkczr3phhycyxbfbi81y7ivkw4dhqr2-initrd-linux-6.6.8-initrd.efi
│   │   └── z287b4560l2gx5zfn2p1zqdra5nzaga8-linux-6.6.8-bzImage.efi
│   └── systemd
│       └── systemd-bootx64.efi
└── loader
    ├── entries
    ├── entries.srel
    ├── loader.conf
    └── random-seed

8 directories, 15 files
```

**Additional questions**

- Operating system and version: NixOS 24.05 (nixpkgs@ac8f30c00827fe0fc03b763f4b060e758c38fce8)
- Have you tried rebooting? Yes
- Is this a regression? Not suer
- Are you using an NVMe disk? No
- Is secure boot enabled? Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No

hughsie, 2024-01-07:
I'm confused why you don't see any dbx device at all -- can you get us the output of `sudo fwupdtool get-devices --plugins uefi_dbx -vv --show-all` please.

amaxine, 2024-01-07:
Of course:
```
06:22:07.092 FuDebug              verbose to debug (on console 0)
06:22:07.093 FuEngine             starting fwupd 1.9.11...
06:22:07.093 FuMain               locked /run/lock/fwupdtool
06:22:07.095 GLib-GIO             Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
06:22:07.095 GLib-GIO             Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
06:22:07.097 FuMain               failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
Loading…: 0%
06:22:07.097 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for 'gio-vfs'
06:22:07.098 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
06:22:07.098 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
06:22:07.098 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x77
06:22:07.098 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
06:22:07.098 FuConfig             Failed to open file "/var/etc/fwupd/fwupd.conf": No such file or directory
06:22:07.098 FuConfig             ::configuration loaded
06:22:07.098 GLib-GIO             Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
06:22:07.108 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: crypto_LUKS
06:22:07.111 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
06:22:07.117 FuRemoteList         loading /nix/store/kr4nl9cnaagq3bw9xnlpai0vqld8sqj8-fwupd-1.9.11/share/fwupd/metainfo
06:22:07.117 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
06:22:07.117 XbSilo               file: 342f96af-67b1-060c-7962-bb07fcff230d, current:342f96af-67b1-060c-7962-bb07fcff230d, cached: (null)
06:22:07.117 XbSilo               loading silo with file contents
06:22:07.117 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
06:22:07.117 FuRemoteList         ::remote_list changed
06:22:07.117 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
06:22:07.117 FuRemoteList         ::remote_list changed
06:22:07.117 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
06:22:07.117 FuRemoteList         ::remote_list changed
06:22:07.117 FuEngine             priority local and vendor-directory is not download remote, so bumping
06:22:07.117 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
06:22:07.117 FuRemoteList         ::remote_list changed
06:22:07.117 FuEngine             priority local and vendor is not download remote, so bumping
06:22:07.117 FuRemoteList         ordering lvfs-testing=lvfs+1
06:22:07.117 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs[0]
Loading…: 1%
06:22:07.117 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
06:22:07.117 Jcat                 loaded 1 certificates
06:22:07.117 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
06:22:07.117 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
06:22:07.117 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
06:22:07.117 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
06:22:07.118 Jcat                 loaded 1 certificates
06:22:07.118 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
06:22:07.118 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
06:22:07.118 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
06:22:07.118 FuEngine             client certificate now exists: nothing to do
Loading…: 2%
06:22:07.118 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
06:22:07.118 FuHistory            got schema version of 9
Loading…: 3%
Loading…: 4%
06:22:07.120 FuPlugin             load(/nix/store/kr4nl9cnaagq3bw9xnlpai0vqld8sqj8-fwupd-1.9.11/lib/fwupd-1.9.11/libfu_plugin_modem_manager.so)
06:22:07.121 FuPluginTest         init
06:22:07.121 FuQuirks             loading quirks from /nix/store/kr4nl9cnaagq3bw9xnlpai0vqld8sqj8-fwupd-1.9.11/share/fwupd/quirks.d
06:22:07.121 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
06:22:07.121 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
06:22:07.121 XbSilo               file: 6ed05411-b282-84c7-49b3-71df2f5c6d12, current:6ed05411-b282-84c7-49b3-71df2f5c6d12, cached: (null)
06:22:07.121 XbSilo               loading silo with file contents
Loading…: 5%
06:22:07.121 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
06:22:07.121 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
06:22:07.121 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x77
06:22:07.121 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
06:22:07.121 FuConfig             Failed to open file "/var/etc/fwupd/fwupd.conf": No such file or directory
06:22:07.121 FuConfig             ::configuration loaded
[... cut ...]
  DoneSetup:            true
  CanInvalidate:        false
FuUdevBackend:
  Name:                 udev
  Enabled:              true
  DoneSetup:            true
  CanInvalidate:        false
FuBluezBackend:
  Name:                 bluez
  Enabled:              true
  DoneSetup:            true
  CanInvalidate:        false

Idle…: 0%
06:22:07.151 FuEngine             resetting update motd timeout
No detected devices
06:22:07.153 FuMain               No detected devices

06:22:07.153 FuPluginTest         destroy
```

hughsie, 2024-01-07:
> 06:22:07.135 FuEngine             disabling plugin because: failed to coldplug using uefi_dbx: Error opening file /sys/firmware/efi/efivars/dbx-d719b2cb-3d3a-4596-a3bc-dad00e67656f: No such file or directory

@vathpela we're still expecting that for custom certs, right?

brookelew, 2024-07-01:
For what its worth I managed to very badly quickly work around this by recreating `/sys/firmware/efi/efivars/dbx-d719b2cb-3d3a-4596-a3bc-dad00e67656f` with an initial forbidden hash using `efi-updatevar -b <some-efi-image-that-you-would-never-use>  -k <your-KEK> dbx`, relying on efi-updatevar to handle a properly formatted write, but I imagine the better way would be to efi-updatevar with the actual sig list.

JustinLex, 2024-07-29:
Hi @brookelew, thank you so much for the workaround. I ran this:

```
sudo efi-updatevar -b /boot/EFI/nixos/kernel-6.7-tdgxbefcxsjugr2gd3tcfathjwhaplpppfwlrt53dot3fipixbiq.efi  -k /etc/secureboot/keys/KEK/KEK.key dbx
```

And dbx updates started working again for me on NixOS on my Framework 13 AMD. Just had to restart fwupd and the update showed up.

vathpela, 2024-11-11:
So I think:
1) if `KEK` still has the `MS` entries, but you've added your own keys to `db` (or even `PK`), `dbx` updates should still work and `dbx` should probably not be empty.
2) if you've done your own `db` keys and also but replaced `KEK` and `PK` (which on some systems is known to also function as `KEK`), it's okay for `dbx` to be empty, which means missing from the filesystem is also expected.  Personally I'd probably put some kind of a sentinel entry in `dbx` but it isn't strictly required.  For instance, if I've removed MS's `db` entries, I might explicitly put them in `dbx` just to make that more legible.
3) if you've replaced `KEK`, I don't know how fwupd can do the updates for you without making some way for admins to add their own `dbx` updates into fwupd's update mechanism, so it probably shouldn't try unless we make such a mechanism.

hughsie, 2025-05-16:
> Error opening file /sys/firmware/efi/efivars/dbx-d719b2cb-3d3a-4596-a3bc-dad00e67656f

If the file doesn't work, then we have nothing to append to and I assume we'd have to drop `FU_EFIVARS_ATTR_APPEND_WRITE`. I'd be happy to accept a pull request to fix this, but it's not something I'm planning to work on myself I'm afraid.

If you want to test with a "raw write" you can do something like `sudo fwupdtool install-blob --plugins uefi-dbx dbxupdate_x64.bin -vv`

hughsie, 2025-07-02:
> probably because the variable doesn't exist

Have you got a way to generate and enroll signing keys onto something like qemu? I'm a bit confused how fwupd is supposed to do an authenticated append to something that doesn't exist, but maybe the firmware just lets us do that. Either way, it needs tests -- and I think qemu is probably the right way to do it.

brookelew, 2025-08-26:
> If the file doesn't work, then we have nothing to append to and I assume we'd have to drop `FU_EFIVARS_ATTR_APPEND_WRITE`. I'd be happy to accept a pull request to fix this, but it's not something [...]

If no one else wants to take this, I'm happy to open a PR and slowly chip away at this over the course of maybe a year

brookelew, 2025-09-04:
Ah, and the CI would run it with QEMU? Sorry, I assumed `fu-self-test.c` was more for mocked unit tests.

superm1, 2025-09-04:
If you need to mock it up with qemu I think we need a new target. I'm not against this. Get it working locally and we can help you integrate it if you have problems.

brookelew, 2025-11-18:
Yeah that's fair, the only reason why I can think of not wanting a dbx would be due to distrust of the vendors or running out of NVRAM. The first case they would (hopefully!) know what the dbx is and won't install the update, and the second won't be allowed to anyways. Exporting a fake device to the user also means downstream won't need to handle this as a special case.

hughsie, 2026-03-02:
@brookelew how does this look now? Does bfe857e1e4801337ef499cf97268735a152f620d mean we can close this?