fwupd/fwupd #6630: systemd “Failed to start fwupd-refresh.service”

equaeghe, 2024-01-06:
**Describe the bug**
systemd fails to start the fwupd-refresh.service.

**Steps to Reproduce**
`systemctl start fwupd-refresh.service`

**Expected behavior**
A successful run of fwupd-refresh.service (`fwupd --refresh`).

**fwupd version information**
```shell
# fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.14
compile   com.hughsie.libjcat           0.1.14
compile   org.freedesktop.fwupd         1.9.6
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.6
runtime   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.gusb          0.4.6
runtime   org.kernel                    6.6.7-gentoo-a
runtime   org.freedesktop.fwupd         1.9.6
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): Installed using gentoo ebuild.

**Additional questions**

- Operating system and version: `Linux TUE023621 6.6.7-gentoo-a #1 SMP Wed Dec 13 22:01:06 CET 2023 x86_64 AMD Ryzen 7 PRO 4750U with Radeon Graphics AuthenticAMD GNU/Linux`
- Have you tried rebooting? _No._
- Is this a regression? _I do not know._

**Log output** (`journalctl -xeu fwupd-refresh.service`)

```
Jan 06 14:00:12 TUE023621 systemd[1]: Starting fwupd-refresh.service...
░░ Subject: A start job for unit fwupd-refresh.service has begun execution
░░ Defined-By: systemd
░░ Support: https://gentoo.org/support/
░░ 
░░ A start job for unit fwupd-refresh.service has begun execution.
░░ 
░░ The job identifier is 183732.
Jan 06 14:00:12 TUE023621 strace[2848255]: (/usr/bin/fwupdmgr:2848255): GLib-DEBUG: 14:00:12.724: setenv()/putenv() are not thread-safe and should not be used after threads are created
Jan 06 14:00:12 TUE023621 strace[2848255]: (/usr/bin/fwupdmgr:2848255): GLib-GIO-DEBUG: 14:00:12.732: _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
Jan 06 14:00:12 TUE023621 strace[2848255]: (/usr/bin/fwupdmgr:2848255): pxbackend-DEBUG: 14:00:12.732: px_config_sysconfig_set_config_file: Could not read file /etc/sysconfig/proxy
Jan 06 14:00:12 TUE023621 strace[2848255]: (/usr/bin/fwupdmgr:2848255): pxbackend-DEBUG: 14:00:12.732: Active config plugins:
Jan 06 14:00:12 TUE023621 strace[2848255]: (/usr/bin/fwupdmgr:2848255): pxbackend-DEBUG: 14:00:12.733:  - config-env
Jan 06 14:00:12 TUE023621 strace[2848255]: (/usr/bin/fwupdmgr:2848255): pxbackend-DEBUG: 14:00:12.733:  - config-kde
Jan 06 14:00:12 TUE023621 strace[2848255]: (/usr/bin/fwupdmgr:2848255): pxbackend-DEBUG: 14:00:12.733:  - config-sysconfig
Jan 06 14:00:12 TUE023621 strace[2848255]: (/usr/bin/fwupdmgr:2848255): GLib-GIO-DEBUG: 14:00:12.733: Failed to initialize portal (GNetworkMonitorPortal) for gio-network-monitor: Not using portals
Jan 06 14:00:12 TUE023621 strace[2848255]: (/usr/bin/fwupdmgr:2848255): GLib-GIO-DEBUG: 14:00:12.739: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
Jan 06 14:00:12 TUE023621 strace[2848255]: (/usr/bin/fwupdmgr:2848255): GLib-GIO-DEBUG: 14:00:12.742: Failed to initialize networkmanager (GNetworkMonitorNM) for gio-network-monitor: The connection is closed
Jan 06 14:00:12 TUE023621 strace[2848255]: (/usr/bin/fwupdmgr:2848255): GLib-GIO-DEBUG: 14:00:12.745: _g_io_module_get_default: Found default implementation netlink (GNetworkMonitorNetlink) for ‘gio-network-monitor’
Jan 06 14:00:12 TUE023621 strace[2848255]: (/usr/bin/fwupdmgr:2848255): pxbackend-DEBUG: 14:00:12.745: px_manager_on_network_changed: Network connection changed, clearing pac data
Jan 06 14:00:12 TUE023621 strace[2848255]: (/usr/bin/fwupdmgr:2848255): pxbackend-DEBUG: 14:00:12.745: px_manager_constructed: Up and running
Jan 06 14:00:12 TUE023621 strace[2848255]: (/usr/bin/fwupdmgr:2848255): GLib-GIO-DEBUG: 14:00:12.745: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
Jan 06 14:00:12 TUE023621 systemd[1]: fwupd-refresh.service: Main process exited, code=exited, status=1/FAILURE
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://gentoo.org/support/
░░ 
░░ An ExecStart= process belonging to unit fwupd-refresh.service has exited.
░░ 
░░ The process' exit code is 'exited' and its exit status is 1.
Jan 06 14:00:12 TUE023621 systemd[1]: fwupd-refresh.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://gentoo.org/support/
░░ 
░░ The unit fwupd-refresh.service has entered the 'failed' state with result 'exit-code'.
Jan 06 14:00:12 TUE023621 systemd[1]: Failed to start fwupd-refresh.service.
░░ Subject: A start job for unit fwupd-refresh.service has failed
░░ Defined-By: systemd
░░ Support: https://gentoo.org/support/
░░ 
░░ A start job for unit fwupd-refresh.service has finished with a failure.
░░ 
░░ The job identifier is 183732 and the job result is failed.
```

equaeghe, 2024-01-06:
> What are the compile flags for fwupd you used? Specifically did you make a user for fwupd refresh?

I'm attaching the build log to give all info about compile flags. (Best viewed with console pager due to embedded color codes?) I hope it will provide the information needed to determine whether a user for fwupd refresh was made or not, because I do not know.

[fwupd-1.9.6:20231227-210818.log](https://github.com/fwupd/fwupd/files/13850012/fwupd-1.9.6.20231227-210818.log)

superm1, 2024-01-06:
I don't see anything in your log configuring `systemd_unit_user`  which is an option since fwupd 1.8.10 (5d22db02ec).  So this means it will use DynamicUser by default and you are hitting https://github.com/systemd/systemd/issues/22737.  There is nothing we can do about this in fwupd upstream.

You have 3 options:
1. Get the systemd bug I linked fixed.
2. Build fwupd with a user to run `fwupd-refresh.service`.
3. Avoid using `fwupd-refresh.service`.

closed 2024-01-06 by superm1