fwupd/fwupd #6633: Suddenly IntelCet.Enabled switches to not-supported

kernie, 2024-01-07:
**Describe the bug**
I updated my laptop (Thinkpad T16 Gen2 AMD) yesterday and checked `fwupdmgr security` afterwards, which gave me a level 4 before, now 2.

See the output of that command for CET:
[image]

**Steps to Reproduce**
Not clear to me.

**Expected behavior**
Cet should be available and supported for current software?

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.11
compile   com.hughsie.libxmlb           0.3.14
compile   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    6.6.9-1-default
runtime   org.freedesktop.fwupd         1.9.11
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): zypper

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
LENOVO 21K7CTO1WW
│
├─ELAN0688:00 04F3:320B:
│     Device ID:          80018e671bb4675a883c75ae9687a1ef8c22db13
│     Summary:            Touchpad
│     Current version:    0x0004
│     Bootloader Version: 0x0003
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              7fd09f45-2b14-5100-9334-b061ff620984 ← HIDRAW\VEN_04F3&DEV_320B
│                         6392df21-879e-5db8-8910-d23e0fab4f7c ← HIDRAW\VEN_04F3&DEV_320B&MOD_003C
│                         922006d1-d20b-50f1-90e1-3a32f9fb70ac ← ELANTP\ICTYPE_14&MOD_003C
│                         524197f9-106d-53f5-bb98-10882bb7d123 ← ELANTP\ICTYPE_14&MOD_003C&DRIVER_HID
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─AMD Ryzen 7 PRO 7840U w/ Radeon 780M Graphics:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x0a704103
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              2a1df9de-fe02-59c8-8af1-32c8bb011f94 ← CPUID\PRO_0&FAM_19&MOD_74
│ │                       5a4bca32-6547-584f-bb36-920d9adede25 ← CPUID\PRO_0&FAM_19&MOD_74&STP_1
│ │   Device Flags:       • Internal device
│ │ 
│ ├─Secure Processor:
│ │     Device ID:        c54ab0237d7a8db8c717b68e0be78e4374a2a079
│ │     Current version:  00.2d.16.64
│ │     Bootloader Version:00.2d.16.64
│ │     Vendor:           Advanced Micro Devices, Inc. (PCI:0x1022)
│ │     GUIDs:            9eb6a793-7f97-5fb5-b49c-139d2ce3ee46 ← PCI\VEN_1022&DEV_15C7
│ │                       ccdf3a32-442f-5c0b-ad12-2e7709eb8444 ← PCI\VEN_1022&DEV_15C7&SUBSYS_17AA50D9
│ │     Device Flags:     • Internal device
│ │   
[... 205 lines cut ...]
└─UEFI Platform Key:
      Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
      Summary:            Lenovo Ltd. PK CA 2012
      Vendor:             Lenovo Ltd.
      GUID:               71599d14-9b31-5270-b3bd-74c494585820 ← UEFI\CRT_9AEF2123F4DE7C19AFABD909BB2C8CAC4411E07E
```

</details>

**Additional questions**

- Operating system and version: OpenSuSE Tumbleweed
```
cat /usr/lib/os-release
NAME="openSUSE Tumbleweed"
# VERSION="20240105"
ID="opensuse-tumbleweed"
ID_LIKE="opensuse suse"
VERSION_ID="20240105"
PRETTY_NAME="openSUSE Tumbleweed"
ANSI_COLOR="0;32"
CPE_NAME="cpe:/o:opensuse:tumbleweed:20240105"
```
- Have you tried rebooting? Yes
- Is this a regression? Don't know

superm1, 2024-01-07:
Cet was just recently expanded to AMD platforms but not all of them support it.

https://fwupd.github.io/libfwupdplugin/hsi.html#cet-available

Can you share `sudo fwupdtool security --verbose --verbose` output so we can better understand why it decided not supported?

superm1, 2024-01-07:
FWIW, I have a Ryzen 6650U (Rembrandt) that shows this result (which is similar).

```
16:09:22.113 FuMain               ADDED:
FuCpuDevice:
  DeviceId:             4bde70ba4e39b28f9eab1628f9dd6e6244c03027
  Name:                 AMD Ryzen 5 PRO 6650U with Radeon Graphics
  Guid:                 52f8f9af-1ca9-5352-bef4-ceb232c888a5 <- CPUID\PRO_0&FAM_19&MOD_44
  Guid:                 e94372a3-3ffb-5d1c-a579-c415b7313e52 <- CPUID\PRO_0&FAM_19&MOD_44&STP_1
  Plugin:               cpu
  Flags:                internal|registered
  Vendor:               Advanced Micro Devices, Inc.
  VersionFormat:        hex
  Icon:                 computer
  Created:              2024-01-07
  Guid[quirk]:          22f9ecf4-588d-5c0a-8326-6ebff3655c6d <- CPUID\PRO_0&FAM_19
  PhysicalId:           cpu:0
  AcquiesceDelay:       50
  InternalFlags:        host-cpu
  HasSHSTK:             true
  HasIBT:               false
  HasTME:               false
  HasSMAP:              true
```

We need both SHSTK and IBT to mark the platform as enabled.  Here's how those are checked:
https://github.com/fwupd/fwupd/blob/fde3fbfd9e2c8b4bd8b0b5ec5b78dcdba012bcb4/plugins/cpu/fu-cpu-device.c#L264
Even if it's not marked enabled by the platform then it won't be used until the distro enables it too (which is a separate check).

kernie, 2024-01-07:
Output of `fwupdtool security --verbose --verbose`:

[fwupdtool_security__verbose__verbose.txt](https://github.com/fwupd/fwupd/files/13854271/fwupdtool_security__verbose__verbose.txt)

superm1, 2024-01-07:
```
bde70ba4e39b28f9eab1628f9dd6e6244c03027 [AMD Ryzen 7 PRO 7840U w/ Radeon 780M Graphics]
17:06:27.968 FuMain               ADDED:
FuCpuDevice:
  DeviceId:             4bde70ba4e39b28f9eab1628f9dd6e6244c03027
  Name:                 AMD Ryzen 7 PRO 7840U w/ Radeon 780M Graphics
  Guid:                 2a1df9de-fe02-59c8-8af1-32c8bb011f94 <- CPUID\PRO_0&FAM_19&MOD_74
  Guid:                 5a4bca32-6547-584f-bb36-920d9adede25 <- CPUID\PRO_0&FAM_19&MOD_74&STP_1
  Plugin:               cpu
  Flags:                internal|registered
  Vendor:               Advanced Micro Devices, Inc.
  VersionFormat:        hex
  Icon:                 computer
  Created:              2024-01-07
  Guid[quirk]:          22f9ecf4-588d-5c0a-8326-6ebff3655c6d <- CPUID\PRO_0&FAM_19
  PhysicalId:           cpu:0
  AcquiesceDelay:       50
  InternalFlags:        host-cpu
  HasSHSTK:             true
  HasIBT:               false
  HasTME:               false
  HasSMAP:              true
```

Looks very similar to the result I had. `SHSTK` but no `IBT`.

superm1, 2024-01-07:
IMO we need to double check the AMD architectural programmer manual (APM) to make sure that these leafs have the same meaning as they do for Intel.

They're available at the AMD developer documentation on AMD's website.

superm1, 2024-01-07:
I don't see anything in the [APM](https://www.amd.com/content/dam/amd/en/documents/processor-tech-docs/programmer-references/40332.pdf) about using the IBT bits.

p 671 (18.1.1):
>  Detecting and Enabling Shadow Stack Support
Support for the shadow stack feature is indicated by CPUID Fn0000_0007_ECX_x0[CET_SS](bit 
7)=1. This bit also indicates the shadow stack MSRs are present.
The shadow stack feature is enabled by setting CR4.CET (bit 23) = 1 (see “CR4 Register” on page 46). 
The shadow stack feature is operational when in protected mode (CR0.PE=1) with paging enabled 
(CR0.PG=1). Shadow stacks are disabled in virtual x86 mode (rFLAGS.VM=1).
Once the feature is enabled, shadow stack operation can be separately enabled in user mode and in 
supervisor mode using control bits in the S_CET and U_CET MSRs (see “Shadow Stack MSRs” on 
page 683)

@hughsie I think we should only look for IBT bits on Intel this means, right?

superm1, 2024-01-07:
Maybe we should add a check for CR4 register bit 23 on AMD instead.

superm1, 2024-01-07:
Oh; but CR4 can only be accessed from the kernel...

closed 2024-01-07 by superm1

Jackaed, 2024-03-09:
I am on a thinkpad t14s gen 3, and this is still marked as "Not supported". Should this be the case or should it be fixed? SHSTK is reported as being enabled, when I grep the output of the file for "SHSTK". Here's the pastebin of my output for `sudo fwupdtool security -vv` https://pastebin.com/Yme6fDsq

Jackaed, 2024-03-09:
Note that I'm on fwupd version 1.9.14 (lastest) which I imagine does contain the commits that were supposed to fix this issue.

superm1, 2024-03-09:
> FuPluginCpu          CET does not function, not supported: Child process exited with code 1

Jackaed, 2024-03-09:
My system has a Ryzen 7 6850U, shouldn't that CPU support CET? Quite hard to find anything about this online.

superm1, 2024-03-09:
> My system has a Ryzen 7 6850U, shouldn't that CPU support CET? Quite hard to find anything about this online.

You need kernel and software support to use CET too.

Jackaed, 2024-03-09:
My kernel is compiled with the `X86_USER_SHADOW_STACK` compile flag enabled, so afaik it should support CET (correct me if there are other build flags I need to enable). When you say "software support", what does that entail? Which particular pieces of software need to have CET support in order for this to work? For context, I am on NixOS.

superm1, 2024-03-09:
@hughsie iirc toolchain and then all binaries need recompile for support too, right?

Jackaed, 2024-03-09:
Having looked into this more, does any major distribution pass this test? Seems like there isn't support for this in any major language except C and rust, and even with rust it's only supported in unstable.

Does this test only check if the fwupd binary has been compiled with CET support? 

Apologies if I'm misunderstanding anything.

superm1, 2024-03-09:
> Having looked into this more, does any major distribution pass this test? Seems like there isn't support for this in any major language except C and rust, and even with rust it's only supported in [...]

I don't think I've seen anyone pass yet. If you think you have the kernel setup properly and the toolchain set up properly and fwupd built with that toolchain and still seeing issues it could be you're the first one fully exercising it and finding the problems. Open your own issue to debug it and share more details on toolchain versions and such.

Jackaed, 2024-03-09:
I think in order to do that I'd have to recompile all of nixpkgs, which I don't think I'm going to be able to achieve lmao. My initial mistake was misreading this issue's error as being the same as my error (CET Platform vs CET OS). I'll talk to some nixos maintainers and discuss the feasibility of getting this enabled, and see if I can get an issue made for tracking it (that doesn't get denied instantly lmao).

Thank you for being very helpful as always.

9SMTM6, 2024-03-11:
FYI, on arch I also don't see a way to getting this enabled without at the very least manually recompiling the kernel, which I'm not gonna do.

`linux-hardend` kernel either doesn't have it enabled or something else is missing. A search on the wiki only turns up some Arch Linux Developer wiki entry that also seems to have more questions than answers.

That the [wiki entries](https://fwupd.github.io/libfwupdplugin/hsi.html#cet-utilized-by-os) are obscured in relation to the command output, that they are basically just a copy of the hardware support and their further information link is dead also doesn't help.

Jackaed, 2024-03-11:
It actually runs far deeper than that. CET is essentially just a flag that had to be given to glibc when compiling every piece of software on the distribution, in order for that piece of software to run with CET enabled. You would have to recompile the kernel, all of your tooling and the fwupd binary itself in order for this check to pass.

This is not something you can fix as a user, this will hopefully slowly begin to be rolled out in the distributions themselves. 

It might be worth discussing whether this check is even worth having at all inside of fwupd, given that the only people who can really achieve this is people running LFS and maybe Gentoo if they're willing to spend enough time on it.

superm1, 2024-03-11:
> It might be worth discussing whether this check is even worth having at all inside of fwupd, given that the only people who can really achieve this is people running LFS and maybe Gentoo if they're willing to spend enough time on it.

My take on this is that the best way you push security forward is awareness.

JohnRTitor, 2024-03-20:
I am using NixOS on AMD Ryzen 7600 and running `sudo fwupdmgr security --force ` shows that CET Platform is supported but `CET OS is unsupported`.
Reading through the above discussion I am presuming this is something NixOS has to fix, and not a fwupd problem. Anyway, here's the verbose log if anyone is interested. 
[results.txt](https://github.com/fwupd/fwupd/files/14666121/results.txt)