fwupd/fwupd #6638: UEFI Update if UDisks is not available

ldevlzero, 2024-01-08:
**Describe the bug**
An UEFI update can probably not be performed if udisk2 is not present. Fwupd crashes in this case.

The programme crashes at this point:
https://github.com/fwupd/fwupd/blob/0f19050ededc3eff9a21042e3568198784771f73/libfwupdplugin/fu-efi-hard-drive-device-path.c#L218

The variable error_local is probably not set. But I think the original problem is that the data structures related to udisk2, such as proxy_blk, are not set.

**Steps to Reproduce**
Use fwupd if udisk2 is not available and do a UEFI update e.g. on Lenovo Thinkpad L14. Set the variable 'EspLocation' in fwupd.conf

**Expected behavior**
Successful UEFI update

**fwupd version information**
1.9.11

```shell
fwupdmgr --version
```
compile   com.hughsie.libxmlb           0.3.14
compile   com.hughsie.libjcat           0.1.14
compile   org.freedesktop.fwupd         1.9.11
compile   org.freedesktop.gusb          0.4.7
runtime   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.gusb          0.4.7
runtime   org.kernel                    6.1.43+
runtime   org.freedesktop.fwupd         1.9.11

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): source

hughsie, 2024-01-08:
> The variable error_local is probably not set

You're probably hitting the critical warning here:

	g_return_val_if_fail(G_IS_DBUS_PROXY(self->proxy_blk), 0);

The easiest thing is to just remove that check, but then how do we get the device for the mountpoint so we can use BLKSSZGET -- which we need to get the UEFI capsule data set up correctly.

I'm wondering why we try to support the non-udisks usecase at all -- @ldevlzero can you tell us why you're *not* using udisks?

ldevlzero, 2024-01-08:
I have a minimalist system and don't need udisks. There is also no polkit in the system for example, which is required for udisks. Up to version fwupd-1.9.2 it worked fine.

hughsie, 2024-01-08:
> I have a minimalist system and don't need udisks

Okay, looks like it's on you to work out how to call BLKSSZGET when using EspLocation then...

closed 2024-01-08 by superm1

superm1, 2024-01-08:
> > I have a minimalist system and don't need udisks
> 
> Okay, looks like it's on you to work out how to call BLKSSZGET when using EspLocation then...

As a poor man's solution we could just shell out and use `findmnt --target $ESPLOCATION` and scrape the results.

@ldevlzero can you do a PR for this?

superm1, 2024-01-08:
Or maybe this suggestion:

https://unix.stackexchange.com/a/128484