fwupd/fwupd #6643: fwupdtool attemps update, and looses the device during the update

peterablehmann, 2024-01-09:
**Describe the bug**
When running `fwupdtool update` fwupd shows an update for the Thunderbolt Host Controller of my Lenovo T580 laptop.
After downloading the update it prints that it has not found any supported devices. The update was not installed.

End of the output of `sudo fwupdtool update --verbose --verbose`
```bash
╔══════════════════════════════════════════════════════════════════════════════╗
║ Thunderbolt host controller von 14.00 auf 20.00 aktualisieren?               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ N27TF20W update PD                                                           ║
║                                                                              ║
║ • Patch bundle upgrade to F407.06.29                                         ║
║                                                                              ║
║ Bug Fixes Key         Summary                                                       ║
║                                                                              ║
║ TPS65982-14138        988CE/SNK was occasionally entering an incorrect state when  ║
║ the far-end/SRC prevented the SNK from doing a transmission by holding the   ║
║ CC line at SinkTxNG state, thus preventing the device from resending         ║
║ pending PD commands on the CC line.                                          ║
║                                                                              ║
║ TPS65982-14485        988CE/SNK was occasionally setting the Overcurrent flag in   ║
║ register 0x5F (Data Status Register) when FRS was enabled and the far end    ║
║ was a source capable dongle/dock, due to a spurious interrupt upon           ║
║ disconnect.                                                                  ║
║                                                                              ║
║ TPS65982-4314         VCONN acceptable range for cables is 4.75V-5.5V               ║
║                                                                              ║
║ Thunderbolt host controller und alle angeschlossenen Geräte sind während     ║
║ der Aktualisierung möglicherweise nicht nutzbar.                             ║
╚══════════════════════════════════════════════════════════════════════════════╝
Operation durchführen? [Y|n]: y
09:22:53.377 FuProgress           raw timing data was { 0,000, 0,001, 0,020, 0,001, 0,022, 0,002 } -- steps were set as [ 63 1 8 1 26 1 ] but should have been [ 0 3 44 2 46 5 ] at ../plugins/uefi-capsule/fu-uefi-capsule-plugin.c:1081
Laden …                  [ -                                     ]09:22:53.382 FuMain               locked /run/lock/fwupdtool
09:22:53.393 FuMain               failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
Schreiben …              [*******************                    ]09:22:53.393 Fwupd                downloading https://fwupd.org/downloads/49d43a0b6da0905014b62523bad96844da42865dc82ff2049f775c7b2931617c-Lenovo-ThinkPad-P52s-Thunderbolt-Firmware-N27TF20W-Secured.cab
09:22:53.393 Fwupd                Emitting ::status-changed() [downloading]
Herunterladen …          [                           |           ]09:22:54.493 Fwupd                download progress: 100%
Herunterladen …          [***************************************]09:22:54.658 Fwupd                download progress: 0%
Herunterladen …          [ -                                     ]09:22:54.660 Fwupd                download progress: 4%
Herunterladen …          [*                                      ]09:22:54.661 Fwupd                download progress: 6%
Herunterladen …          [**                                     ]09:22:54.670 Fwupd                download progress: 12%
Herunterladen …          [****                                   ]09:22:54.675 Fwupd                download progress: 18%
Herunterladen …          [*******                                ]09:22:54.676 Fwupd                download progress: 24%
Herunterladen …          [*********                              ]09:22:54.682 Fwupd                download progress: 31%
Herunterladen …          [************                           ]09:22:54.683 Fwupd                download progress: 37%
Herunterladen …          [**************                         ]09:22:54.686 Fwupd                download progress: 43%
Herunterladen …          [****************                       ]09:22:54.690 Fwupd                download progress: 56%
Herunterladen …          [*********************                  ]09:22:54.692 Fwupd                download progress: 62%
Herunterladen …          [************************               ]09:22:54.696 Fwupd                download progress: 75%
Herunterladen …          [*****************************          ]09:22:54.698 Fwupd                download progress: 87%
Herunterladen …          [*********************************      ]09:22:54.702 Fwupd                download progress: 100%
09:22:54.702 Fwupd                Emitting ::status-changed() [idle]
09:22:54.702 Fwupd                status-code was 200
09:22:54.721 FuCommon             mapped file /root/.cache/fwupd/49d43a0b6da0905014b62523bad96844da42865dc82ff2049f775c7b2931617c-Lenovo-ThinkPad-P52s-Thunderbolt-Firmware-N27TF20W-Secured.cab of size 0x3fa2a
Entpacken …              [ -                                     ]09:22:54.721 FuStruct             CabHeader:
  size: 0x3fa2a
  off_cffile: 0x2c
  nr_folders: 0x1
  nr_files: 0x5
  flags: 0x0
  set_id: 0x0
  idx_cabinet: 0x0
09:22:54.721 FuStruct             CabFolder:
  offset: 0xcd
  ndatab: 0x8
  compression: 0x0 [none]
09:22:54.722 FuStruct             CabData:
  checksum: 0xc57a6e05
  comp: 0x8000
  uncomp: 0x8000
[... 357 lines cut ...]
│   
├─UEFI Platform Key:
│     Gerätekennung:      6924110cde4fa051bfdc600a60620dc7aa9d3c6a
│     Zusammenfassung:    Lenovo Ltd. PK CA 2012
│     Anbieter:           Lenovo Ltd.
│     GUID:               71599d14-9b31-5270-b3bd-74c494585820 ← UEFI\CRT_9AEF2123F4DE7C19AFABD909BB2C8CAC4411E07E
│   
└─UHD Graphics 620:
      Gerätekennung:      5792b48846ce271fab11c4a545f7a3df0d36e00a
      Aktuelle Version:   07
      Anbieter:           Intel Corporation (PCI:0x8086)
      GUIDs:              8b72a10c-1279-5f8e-a28a-34fb11a58240 ← PCI\VEN_8086&DEV_5917
                          6abd5dd1-9f3b-5d12-971c-01eeedc01e1a ← PCI\VEN_8086&DEV_5917&SUBSYS_17AA225A
      Geräte-Bitschalter: • Internes Gerät
                          • Kryptografische Hash-Verifizierung ist verfügbar
      Device Requests:    • Message
```

</details>

**Additional questions**

- Operating system and version: NixOS 23.11
- Have you tried rebooting? Yes
- Is this a regression? I don't know.

peterablehmann, 2024-01-09:
Related to #5794

superm1, 2024-01-09:
Can you please reproduce on 1_9_X branch?

peterablehmann, 2024-01-10:
I thought I was on 1.9.x? Also, I'm not sure how to do that.

hughsie, 2024-01-10:
I think you probably need to use fwupdmgr (as in the daemon process) rather than fwupdtool for thunderbolt.

superm1, 2024-01-10:
Yeah you're reporting on 1.9.10, but I'm looking for testing on the branch specifically because we did have a thunderbolt related fix just land last week that isn't in a tagged release.

To build 1_9_X:
1. Clone tree
2. Switch to 1_9_X branch
3. Follow https://fwupd.github.io/libfwupdplugin/building.html
4. From within the venv reproduce the issue.

closed 2024-01-16 by peterablehmann