fwupd/fwupd #6646: Firmware signature missing or not trusted (but it should be signed correctly)

xsuchy09, 2024-01-10:
**Describe the bug**
Trying to update Lenovo ThinkPad T14 Gen 3 system firmware to 0.1.49 version: **firmware signature missing or not trusted**. [Lenovo said that it is signed correctly.](https://forums.lenovo.com/t5/Ubuntu/Ubuntu-23-10-21CH004XGE-upgrade-to-r23ul73w-1-49-firmware-signature-missing-or-not-trusted/m-p/5281458?page=1#6219862)

**Steps to Reproduce**
```fwupdmgr update```
(classic update procedure)

**Expected behavior**
Update firmware.

**fwupd version information**

```shell
fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.2.0-39-generic
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

apt

<details>
<summary>Console output - update</summary>

```shell
fwupdmgr update
Devices with no available firmware updates: 
 • ELAN0678:00 04F3:3195
 • Integrated RGB Camera
 • Prometheus IOTA Config
 • SKHynix HFS001TEJ9X102N
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Battery
 • Embedded Controller
 • Prometheus
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 0.1.47 to 0.1.49?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This BIOS includes the following updates:                                    ║
║                                                                              ║
║ • Notice that BIOS can't be downgraded to older BIOS version after           ║
║ upgrade to r23ul73w(1.49).                                                   ║
║ • Enhancement to address security vulnerability.                             ║
║ • Change to permit fan rotation after fan error happen.                      ║
║ • Fixed non-bootable issue when Absolute Persistence Module Activation is    ║
║ disabled permanently.                                                        ║
║ • Fixed issue that boot time is long when system is attached with            ║
║ DisplayLink Dock.                                                            ║
║ • Fixed platform profile stuck on power-saver under linux modern standby     ║
║ resume.                                                                      ║
║                                                                              ║
║ 21CGS1ES00 must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: Y
Downloading…             [***************************************]
Downloading…             [***************************************] Less than one minute remaining…
[... 229 lines cut ...]
│                         • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:          f37fb01122dd62c773f4e84ec89737e059712d59
      Summary:            UEFI ESRT device
      Current version:    1
      Minimum Version:    1
      Vendor:             DMI:LENOVO
      Update State:       Success
      GUID:               75d78aef-5e7f-0afe-b68a-aef4c52ccddd
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update

```

</details>

**Additional questions**

- Operating system and version: Ubuntu 24.04.3 LTS
- Have you tried rebooting? Yes
- Is this a regression? No

hughsie, 2024-01-10:
Another report (i've deduped above) has:

    compile   org.freedesktop.fwupd         1.9.5
    compile   com.hughsie.libxmlb           0.3.10
    compile   com.hughsie.libjcat           0.1.9

hughsie, 2024-01-10:
Interestingly, on a hunch:

jcat 0.1.14:

	[hughsie@hughsie-work Downloads]$ LANG=C jcat-tool verify firmware.jcat --public-keys /etc/pki/fwupd-metadata/ --public-keys /etc/pki/fwupd
	firmware.bin:
	    PASSED sha1: OK
	    PASSED sha256: OK
	    FAILED pkcs7: failed to verify data for O=Linux Vendor Firmware Project,CN=LVFS CA: Public key signature verification has failed. [-89]
	    FAILED gpg: '48A6D80E4538BAC2' is not a valid signature
	    FAILED: Validation failed
	firmware.metainfo.xml:
	    PASSED sha1: OK
	    PASSED sha256: OK
	    PASSED pkcs7: O=Linux Vendor Firmware Project,CN=LVFS CA
	    PASSED gpg: 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
	Validation failed

jcat 0.2.0:

	[hughsie@hughsie-work Downloads]$ LANG=C jcat-tool verify firmware.jcat --public-keys /etc/pki/fwupd-metadata/ --public-keys /etc/pki/fwupd
	firmware.bin:
	    PASSED sha1: OK
	    PASSED sha256: OK
	    PASSED pkcs7: O=Linux Vendor Firmware Project,CN=LVFS CA
	    PASSED gpg: 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
	firmware.metainfo.xml:
	    PASSED sha1: OK
	    PASSED sha256: OK
	    PASSED pkcs7: O=Linux Vendor Firmware Project,CN=LVFS CA
	    PASSED gpg: 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2

debugging now.

EvoXCX, 2024-01-10:
Same here, force updated by setting OnlyTrusted=false (Work on my P14s Gen 3 try at your own risk)

hughsie, 2024-01-10:
> debugging now.

Okay, I've worked out the root cause (wrong order for the jcat fallback on the LVFS) which is only triggering as the firmware just crept past 32MiB in size. I'm working on a LVFS fix (and libjcat enhancement...) and also a repair action to auto-resign any affected firmware.

hughsie, 2024-01-10:
LVFS fix is https://gitlab.com/fwupd/lvfs-website/-/commit/013fd574235d2864e7f56a5c88df1d2d13257fca and libjcat (just cosmetic and documentation, no need to release/update) is here: https://github.com/hughsie/libjcat/pull/87

It'll take a few mins to deploy the LVFS, and quite a few minutes to resign the affected firmware, but I'll ping this issue when it's all ready for testing.

hughsie, 2024-01-10:
Okay, initial small scale testing worked, e.g. the file signed a few days ago:

fwupdmgr get-details 3dbe8cd466d3e8d9b5b3246ec4d5ac03147659369b5ce29952e9c6487302f5a9-Lenovo-ThinkPad-P16v-Gen-1-AMD-SystemFirmware-1.24_.cab
    ....
        Release Flags:    • Trusted metadata
    ....

..and the file I've just resigned manually:

    fwupdmgr get-details 10e03df7f53c827bf449efb958f46cc7809139de4a4744b422eca030241c692c-Lenovo-ThinkPad-P16v-Gen-1-AMD-SystemFirmware-1.24_.cab

    ...
        Release Flags:    • Trusted payload
                          • Trusted metadata
    ...

(both with libjcat-0.1.14-2.fc39.x86_64) -- mass resigning of anything signed in the last  days is progressing now:

[image]

hughsie, 2024-01-10:
Okay, all firmwares have been resigned, and metadata regenerated, and so if you're affected by this can you please try `fwupdmgr refresh --force` and then `fwupdmgr update` please. Feedback either way appreciated.

xsuchy09, 2024-01-10:
Firmware is not found now - so no update available.
```shell
fwupdmgr refresh --force
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: 5 local devices supported

fwupdmgr update 
Devices with no available firmware updates: 
 • ELAN0678:00 04F3:3195
 • Integrated RGB Camera
 • Prometheus IOTA Config
 • SKHynix HFS001TEJ9X102N
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Battery
 • Embedded Controller
 • Prometheus
 • System Firmware
 • TPM
```

hughsie, 2024-01-10:
Ahh, the ODM moved it out of stable this morning -- probably to stop reports like this -- I'll email the ODM and move it back to stable in the interim. It'll take ~30 mins to rebuild the stable metadata, then please try again.

hughsie, 2024-01-10:
@xsuchy09 give it a go now perhaps?

frederic-klein, 2024-01-10:
can confirm: fixed it for me, 0.1.49 got installed. Thank you @hughsie

closed 2024-01-10 by hughsie

xsuchy09, 2024-01-10:
> @xsuchy09 give it a go now perhaps?

Thanks, it's ok now and looks like updated even it ends with https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Failed-to-run-update-on-reboot

vpavic, 2024-01-14:
I had the same _firmware signature missing or not trusted_ error attempting to update to 0.1.49, however after `fwupdmgr refresh --force` the update is no longer available.

This is T14 Gen 3 (AMD) on Ubuntu 23.10 with fwupd 1.9.5:

```
$ fwupdmgr --version 
compile   org.freedesktop.fwupd         1.9.5
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.kernel                    6.5.0-14-generic
runtime   org.freedesktop.fwupd         1.9.5
```

<details>
<summary><code>fwupdmgr get-devices</code> output</summary>

```
$ fwupdmgr get-devices
LENOVO 21CF004PGE
│
├─ELAN0678:00 04F3:3195:
│     Device ID:          f8bebdac7d9634f3890f6f6cf3eec0908c2c700b
│     Summary:            Touchpad
│     Current version:    0x0003
│     Bootloader Version: 0x0002
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              4e8cda57-37ba-5dc6-b114-7610e38714e2 ← HIDRAW\VEN_04F3&DEV_3195
│                         cc456594-6f7e-5d1b-82a6-b4558a813152 ← HIDRAW\VEN_04F3&DEV_3195&MOD_0006
│                         4b808d27-c833-50cc-8177-9a401f5c08fe ← ELANTP\ICTYPE_14
│                         d15e804c-570f-511b-803a-b38a4eddae29 ← ELANTP\ICTYPE_14&MOD_0006
│                         296e64f7-9e08-5717-bff2-651e41b0477e ← ELANTP\ICTYPE_14&MOD_0006&DRIVER_HID
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─AMD Ryzen 7 PRO 6850U with Radeon Graphics:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x0a404102
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│ │                       22f9ecf4-588d-5c0a-8326-6ebff3655c6d ← CPUID\PRO_0&FAM_19
│ │                       52f8f9af-1ca9-5352-bef4-ceb232c888a5 ← CPUID\PRO_0&FAM_19&MOD_44
│ │                       e94372a3-3ffb-5d1c-a579-c415b7313e52 ← CPUID\PRO_0&FAM_19&MOD_44&STP_1
│ │   Device Flags:       • Internal device
│ │ 
│ ├─Graphics Processing Unit (GPU):
│ │     Device ID:        04e17fcf7d3de91da49a163ffe4907855c3648be
│ │     Current version:  113-REMBRANDT-X37
│ │     Vendor:           Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│ │     GUIDs:            817e27b3-d1bb-591a-a1cc-e039d0a60be6 ← PCI\VEN_1002&DEV_1681
│ │                       96ccc33d-d14d-5331-89f9-5f03011dba49 ← PCI\VEN_1002&DEV_1681&SUBSYS_17AA50B6
│ │     Device Flags:     • Internal device
│ │   
│ └─System Management Unit (SMU):
│       Device ID:        db0330716216c629bb2c07256e5d018f499eb6ce
│       Summary:          Microcontroller used within CPU/APU program 4
│       Current version:  69.55.0
│       Vendor:           Advanced Micro Devices, Inc.
│       GUID:             79307ae6-a2ea-52e1-bf56-6abbaf3547ad ← /sys/devices/platform/AMDI0007:00
│       Device Flags:     • Internal device
│     
├─Battery:
│     Device ID:          11a7fbdd09ce583e58b899660fe10f147fd155a2
│     Summary:            UEFI ESRT device
│     Current version:    1.3.7
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
[... 196 lines cut ...]
│     Vendor:             DMI:LENOVO
│     Update State:       Success
│     GUID:               e87c4ff6-69e0-4647-a4c0-aa1f736511f6
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:          f37fb01122dd62c773f4e84ec89737e059712d59
      Summary:            UEFI ESRT device
      Current version:    1
      Minimum Version:    1
      Vendor:             DMI:LENOVO
      Update State:       Success
      GUID:               75d78aef-5e7f-0afe-b68a-aef4c52ccddd
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```

</details>

xsuchy09, 2024-01-15:
It's again missing - as you can see: https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadR23ET.firmware

hughsie, 2024-01-15:
@xsuchy09 the ODM pulled back that version due to another (not us!) firmware problem -- I think 0.1.50 is in progress now.

xsuchy09, 2024-01-15:
I know - just for info for @vpavic ;)