fwupd/fwupd #6649: fwupd fails to upgrade to 1.9.11 on ubuntu 20

mailsanchu, 2024-01-10:
```
svarkey@svarkey-Precision-7760:/tmp/fwupd-1.9.11$ sudo snap refresh fwupd --channel=latest/stable
error: cannot perform the following tasks:
- Stop snap "fwupd" services (systemctl command [stop snap.fwupd.fwupd.service] failed with exit status 5: Failed to stop snap.fwupd.fwupd.service: Unit snap.fwupd.fwupd.service not loaded.
)

```

superm1, 2024-01-10:
What version are you currently on?

mailsanchu, 2024-01-10:
`fwupd  1.9.10   5438  latest/stable  richard-hughes  -

`

superm1, 2024-01-10:
Can you check the journal and status for the service it's failing on?

By chance are you not running an Ubuntu kernel?

superm1, 2024-01-10:
Cc @valentindavid for any ideas

mailsanchu, 2024-01-10:
```
Jan 10 15:58:28 svarkey-Precision-7760 systemd[1]: Failed to put bus name to hashmap: File exists
Jan 10 15:58:28 svarkey-Precision-7760 systemd[1]: snap.fwupd.fwupd.service: Two services allocated for the same bus name org.freedesktop.fwupd, refusing operation.

```

superm1, 2024-01-10:
Sounds like you might have the distro package installed too. You need one or the other.

mailsanchu, 2024-01-10:
OK How do i upgrade the distro package?

mailsanchu, 2024-01-10:
I removed the distro and installed from snap and it works 
```
svarkey@svarkey-Precision-7760:/tmp/fwupd-1.9.11$ sudo apt-get remove fwupd
Reading package lists... Done

```

mailsanchu, 2024-01-10:
```
svarkey@svarkey-Precision-7760:/tmp/fwupd-1.9.11$ sudo snap refresh fwupd --channel=latest/stable
fwupd 1.9.11 from Richard Hughes refreshed
svarkey@svarkey-Precision-7760:/tmp/fwupd-1.9.11$ 

```

closed 2024-01-10 by superm1