fwupd/fwupd #6658: Dell Precision 7560,Precision 7760 - Version 1.27.0 not showing

skipperTux, 2024-01-12:
**Describe the bug**
Fedora Linux 39 (Workstation Edition) on a Dell Inc. Precision 7560 the latest [Version 1.27.0](https://fwupd.org/lvfs/devices/com.dell.uefib9870160.firmware) is not showing up. System Firmware Current version: 1.26.0.

**Steps to Reproduce**
Allow Firmware upgrades in System BIOS. Then do
```shell
$ fwupdmgr --force refresh
Updating lvfs
Downloading…             [  -                                    ]
Successfully downloaded new metadata: 8 local devices supported

$ fwupdmgr get-updates
Devices with no available firmware updates: 
 • System Firmware
 • TPM
 • UEFI Device Firmware
 • Unifying Receiver
Devices with the latest available firmware version:
 • Package level of Dell dock
 • RTS5413 in Dell dock
 • RTS5487 in Dell dock
 • VMM5331 in Dell dock
 • WD19DCS
 • PM9A1 NVMe Samsung 2048GB
 • UEFI Device Firmware
 • UEFI dbx
No updates available
```

**Expected behavior**
Download of the latest firmware [Version 1.27.0](https://fwupd.org/lvfs/devices/com.dell.uefib9870160.firmware), system reboot and system firmware upgrade.

**fwupd version information**
```shell
$ fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.14
compile   com.hughsie.libjcat           0.1.14
compile   org.freedesktop.fwupd         1.9.11
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.11
runtime   org.kernel                    6.6.9-200.fc39.x86_64
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): `dnf`

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
Dell Inc. Precision 7560
│
├─WD19DCS:
│ │   Device ID:          9acd1a7bce5d458fef14ce8d457f90f39818b10a
│ │   Summary:            High performance dock
│ │   Current version:    01.01.00.07
│ │   Vendor:             Dell Inc. (USB:0x413C)
│ │   Install Duration:   1 minute
│ │   Serial Number:      18FMYT3/3193360135041767
│ │   GUID:               cd357cf1-40b2-5d87-b8df-bb2dd82774aa ← USB\VID_413C&PID_B06E&hub&embedded
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │                       • Device can recover flash failures
[... 221 lines cut ...]
│     Bootloader Version: BOT03.02_B0009
│     Vendor:             Logitech (HIDRAW:0x046D, USB:0x046D)
│     Install Duration:   30 seconds
│     Serial Number:      Logitech_USB_Receiver
│     GUIDs:              cc4cbfa9-bf9d-540b-b92b-172ce31013c1
│                         279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
│     Device Flags:       • Updatable
│                         • Signed Payload
│   
└─Xeon W-11955M CPU @ 2.60GHz:
      Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
      Current version:    0x0000004e
      Vendor:             Intel
      GUIDs:              a6bd4ca5-75a6-5796-b564-66b5cab1b11b ← CPUID\PRO_0&FAM_06&MOD_8D
                          d9dd5e77-df17-5bab-b5ec-22827598bfed ← CPUID\PRO_0&FAM_06&MOD_8D&STP_1
      Device Flags:       • Internal device
```

</details>

**Additional questions**

- Operating system and version: Fedora Linux 39 (Workstation Edition)
- Have you tried rebooting? Yes
- Is this a regression? No

hughsie, 2024-01-12:
What's the output of `sudo fwupdtool hwids` please.

skipperTux, 2024-01-12:
<details>
<summary><b>Output</b> fwupdtool hwids</summary>

```shell
$ sudo fwupdtool hwids
Loading…                 [*                                      ]
Computer Information
--------------------
BiosVendor: Dell Inc.
BiosVersion: 1.26.0
BiosMajorRelease: 1
BiosMinorRelease: 26
FirmwareMajorRelease: ff
FirmwareMinorRelease: ff
Manufacturer: Dell Inc.
Family: Precision
ProductName: Precision 7560
ProductSku: A69
EnclosureKind: a
BaseboardManufacturer: Dell Inc.
BaseboardProduct: Y1R4H
Hardware IDs
------------
{9f3fd9c2-381e-57d9-a529-1a33075a8913}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{21c14527-559d-563c-8d91-e873e4320719}   <- Manufacturer + Family + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{2d27911d-97e8-501c-b99a-d4b2cb1edc80}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{007b8ebb-f6b4-52c2-ae73-c4d40b38d9a5}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
{3a27f14f-88bd-5b79-b2cb-ef6ae876c608}   <- Manufacturer + Family + ProductName + ProductSku
{a9ffd22d-8339-5094-8f38-eb2d045b5174}   <- Manufacturer + Family + ProductName
{b45152f8-4766-51e2-9aee-c42537d48d25}   <- Manufacturer + ProductSku + BaseboardManufacturer + BaseboardProduct
{6ddbc49a-7362-5a25-a5e2-dcb24bce0252}   <- Manufacturer + ProductSku
{eebda2e7-bd3a-5ce6-afb9-12ace4bd069b}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{22b9d6b6-403f-5b38-922c-0e3c0ac00cfe}   <- Manufacturer + ProductName
{691d29fe-b326-51e8-9f74-99c4c4d97c56}   <- Manufacturer + Family + BaseboardManufacturer + BaseboardProduct
{9bf5227f-7ae3-5421-9a57-a440b2cc99ed}   <- Manufacturer + Family
{5c163113-9296-5c8d-a92d-ae04eb59a0f7}   <- Manufacturer + EnclosureKind
{1694d7c8-6703-5a1b-8d5f-cad9d4b3fb31}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
{85d38fda-fc0e-5c6f-808f-076984ae7978}   <- Manufacturer
```

</details>

hughsie, 2024-01-12:
The HWID `22b9d6b6-403f-5b38-922c-0e3c0ac00cfe` matches and `b9870160-dd98-4ab8-a855-a3e8e5e30978` is the right device; this should work. What's the output of `sudo fwupdtool get-updates -vv` please.

skipperTux, 2024-01-12:
Output `sudo fwupdtool get-updates -vv` -> See attached file
[fwupdtool_get-updates_verbose.txt](https://github.com/fwupd/fwupd/files/13917984/fwupdtool_get-updates_verbose.txt)

hughsie, 2024-01-12:
Ohh hmm, interesting. The firmware has a requirement on fwupd >= 1.0.1 but the fwupd is checking that and preventing the update:

> 10:40:23.417 FuEngine             failed to set release for component: requirement 'hardware' also needs org.freedesktop.fwupd >= 1.0.8

Sure enough, this is the LVFS website code: https://gitlab.com/fwupd/lvfs-website/-/commit/565af0bf48161a1606700daecfcd39ab4472bdf9

So either fwupd or the LVFS is wrong :) Debugging now.

hughsie, 2024-01-12:
Support for HWID requires was added in 2ec78d68db5790547e3a58b0dfb3ae359892040e -- which was first in the 1.0.1 release. So, fwupd is wrong!

hughsie, 2024-01-12:
No, now *I'm* wrong. 1.0.1 added support for HWID requires, 1.0.8 added support for `or` clauses in the HWIDs requires, which is what the Dell firmware is using. I'll add a LVFS repair action now.

skipperTux, 2024-01-12:
Hm, I did a `--force refresh` and a `get-updates`, but I still get `no available firmware updates`. Anything else I need to do?

hughsie, 2024-01-12:
> Anything else I need to do?

Wait a few hours! 🙂

hughsie, 2024-01-12:
@skipperTux can you try now please?

skipperTux, 2024-01-12:
There we go...

```shell
$ fwupdmgr --force refresh
Updating lvfs
Downloading…             [   -                                   ]
Successfully downloaded new metadata: 9 local devices supported
```

<details>
<summary><b>Output</b> fwupdmgr get-updates</summary>

```shell
$ fwupdmgr get-updates
Devices with no available firmware updates: 
 • TPM
 • UEFI Device Firmware
 • Unifying Receiver
Devices with the latest available firmware version:
 • Package level of Dell dock
 • RTS5413 in Dell dock
 • RTS5487 in Dell dock
 • VMM5331 in Dell dock
 • WD19DCS
 • PM9A1 NVMe Samsung 2048GB
 • UEFI Device Firmware
 • UEFI dbx
Dell Inc. Precision 7560
│
└─System Firmware:
  │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
  │   Summary:            UEFI ESRT device
  │   Current version:    1.26.0
  │   Minimum Version:    1.26.0
  │   Vendor:             Dell (DMI:Dell Inc.)
  │   Update State:       Success
  │   GUID:               b9870160-dd98-4ab8-a855-a3e8e5e30978
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─Precision 7560,Precision 7760 System Update:
        New version:      1.27.0
        Remote ID:        lvfs
        Release ID:       75543
        Summary:          Firmware for the Dell Precision 7560,Precision 7760
        License:          Proprietary
        Size:             24,1 MB
        Created:          2023-11-03
        Urgency:          Critical
        Vendor:           Dell
        Release Flags:    • Trusted metadata
                          • Is upgrade
        Description:      
        This stable release fixes the following issues:
        
        • This release contains security updates as disclosed in the Dell Security Advisories .
        • Fixed the issue where the Admin Unlock window in the BIOS setup stops responding or the system reboots automatically. This issue occurs when you enter the wrong Master Password in the system password prompt by pressing CTRL + Enter.
        • Fixed the issue where the system stops responding during Power-On Self-Test (POST). This issue occurs when the NVMe or Trusted Platform Module firmware is interrupted while updating the firmware.
        
        Some new functionality has also been added:
        Issues:           DSA-2023-362
                          DSA-2023-359
                          DSA-2023-342
        Checksum:         d06ec78b0ee8e8bf9438c6de9e6650b9c5273258fa603199ba46448aa427f258
```

</details>

Performing the firmware upgrade in a minute and coming back.

skipperTux, 2024-01-12:
```
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    1.27.0
│ │   Minimum Version:    1.27.0
│ │   Vendor:             Dell (DMI:Dell Inc.)
│ │   Update State:       Success
```
:+1: Thanks again. Case closed.

closed 2024-01-12 by skipperTux