fwupd/fwupd #6683: confined snap errors when enumerating TPM

superm1, 2024-01-16:
**Describe the bug**
Running `fwupdtool get-devices --plugins tpm` has errors.  The device enumerates but errors are displayed.

**Steps to Reproduce**
`sudo fwupd.fwupdtool get-devices --plugins tpm` on a machine with a TPM.

**Expected behavior**
No errors

**fwupd version information**
35187d94c5707081cdfb6ccd27658c894ec275b3

Here is the enumeration for the TPM that you can see the error.
```shell
$ sudo fwupd.fwupdtool get-devices --plugins tpm
Loading…                 [*********************                  ]ERROR:tcti:src/tss2-tcti/tcti-device.c:452:Tss2_Tcti_Device_Init() Failed to open specified TCTI device file /dev/tpmrm0: Operation not permitted 
ERROR:tcti:src/tss2-tcti/tctildr-dl.c:154:tcti_from_file() Could not initialize TCTI file: libtss2-tcti-device.so.0 
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
Framework Laptop 13 (AMD Ryzen 7040Series)
│
└─TPM:
      Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
      Summary:            TPM 2.0 Device
      Current version:    6.3.1.610
      Vendor:             Microsoft (TPM:MSFT)
      GUIDs:              2a65d4b6-60a0-5e8c-acfa-d86cc3cbe4c3 ← TPM\VEN_MSFT&DEV_0001
                          31a7bb1b-f5e1-53ff-b0e5-8692a21caed6 ← TPM\VEN_MSFT&MOD_Pluton.TPM.A
                          5bea8d58-c840-513e-a5ad-0184cbaa9821 ← TPM\VEN_MSFT&DEV_0001&VER_2.0
                          e02519c6-9139-5de8-89d4-9dca2f5c3b66 ← TPM\VEN_MSFT&MOD_Pluton.TPM.A&VER_2.0
      Device Flags:       • Internal device
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device can recover flash failures
                          • Full disk encryption secrets may be invalidated when updating
                          • Signed Payload
```

Here is the messages in dmesg when running.
```
[124873.044720] audit: type=1400 audit(1705374805.596:792): apparmor="DENIED" operation="open" class="file" profile="snap.fwupd.fwupdtool" name="/etc/gnutls/config" pid=42417 comm="fwupdtool" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
[124873.100280] audit: type=1326 audit(1705374805.652:793): auid=1000 uid=0 gid=0 ses=3 subj=snap.fwupd.fwupdtool pid=42417 comm="fwupdtool" exe="/snap/fwupd/x1/bin/fwupdtool" sig=0 arch=c000003e syscall=314 compat=0 ip=0x715dc703aa7d code=0x50000
```

superm1, 2024-01-16:
This was reproduced using snapd 2.61.1.

hughsie, 2024-11-11:
@superm1 still happening? I think maybe we should open an issue against snap somehow.

valentindavid, 2024-11-11:
It seems I did not see this bug. This is confusing, `/dev/tpmrm0` should be rw. And dmesg does not seem to have any related messages.

closed 2025-05-03 by hughsie