fwupd/fwupd #6688: ThinkPad X280 can't downgrade Thunderbolt firmware version & get report "Unmatched performance for  high speed I/O"

Anifyuli, 2024-01-17:
**Describe the bug**
My system failed to downgrade it Thunderbolt firmware version from 23.00 to 20.00 and get report from firmware updater "Unmatched performance for high speed I/O". I always get this error if I try to downgrade it.
```shell
could not start thunderbolt device upgrade: could not write to /sys/devices/pci0000:00/0000:00:1c.0/0000:02:00.0/0000:03:00.0/0000:04:00.0/domain0/0-0/nvm_authenticate: Invalid argument
```

**Steps to Reproduce**
1. Downgrade it version with command `sudo fwupdmgr downgrade f52c4809a9e30e4a911351f5d9daf063e0a92515`
2. Get that error

**Expected behavior**
It can downgraded flawlessly and "Unmatched performance for high speed I/O" disappear from firmware updater

**fwupd version information**
```shell
$ fwupdmgr --version   
compile   org.freedesktop.fwupd         1.9.5
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.kernel                    6.5.0-14-generic
runtime   org.freedesktop.fwupd         1.9.5
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): `apt`

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
LENOVO 20KEA062UK
│
├─Core™ i5-8350U CPU @ 1.70GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f4
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         561403e8-143a-5071-ab09-bf5e4c146983 ← CPUID\PRO_0&FAM_06&MOD_8E
│                         54bdc429-0179-5019-9cb6-8a343324c97c ← CPUID\PRO_0&FAM_06&MOD_8E&STP_A
│     Device Flags:       • Internal device
│   
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    0.1.15
│     Minimum Version:    0.1.15
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               3f4a527b-6588-45b8-b2d3-dc61189b63cb
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    184.94.4494
│     Minimum Version:    0.0.1
│     Vendor:             Lenovo (DMI:LENOVO)
[... 456 lines cut ...]
├── initrd.img.old -> initrd.img-6.5.0-13-generic
├── memtest86+ia32.bin
├── memtest86+ia32.efi
├── memtest86+x64.bin
├── memtest86+x64.efi
├── System.map-6.2.0-33-generic
├── System.map-6.5.0-13-generic
├── System.map-6.5.0-14-generic
├── vmlinuz -> vmlinuz-6.5.0-14-generic
├── vmlinuz-6.2.0-33-generic
├── vmlinuz-6.5.0-13-generic
├── vmlinuz-6.5.0-14-generic
└── vmlinuz.old -> vmlinuz-6.5.0-13-generic

6 directories, 312 files
```

**Additional questions**

- Operating system and version: Ubuntu 23.10
- Have you tried rebooting? Yes
- Is this a regression? IDK with it
- Are you using an NVMe disk? No, I'm use M.2 SATA SSD
- Is secure boot enabled? No, it disabled by me
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No, it disabled

superm1, 2024-01-17:
Downgrades are blocked by hardware sometimes if there was a security update. This sounds like one of those cases.

Anifyuli, 2024-01-18:
So, I can't do that? Oh my gosh

superm1, 2024-01-18:
Yup.  Unfortunately we don't have any interface from the kernel about downgrade info or how to introspect the CAB to tell us the minimum after a downgrade.

closed 2024-01-18 by superm1