fwupd/fwupd #6689: Can't update firmware on Aerox 3 Wireless Mouse (`firmware is 4000 bytes smaller than the allowed minimum size of 27000 bytes`)

yukijoou, 2024-01-17:
**Describe the bug**
Trying to update from firmware version 1.9.3 to 1.11.4 using `sudo fwupdmgr update` fails with the following log:

```
Devices with no available firmware updates: 
 • SUV500M8240G
 • System Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Aerox 3 Wireless Mouse from 1.9.3 to 1.11.4?                         ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issue:                               ║
║                                                                              ║
║ • Fix issue with wireless connection dropping during use                     ║
║                                                                              ║
║ Aerox 3 Wireless Mouse and all connected devices may not be usable while     ║
║ updating.                                                                    ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y
Decompressing…           [***************************************]
firmware is 4000 bytes smaller than the allowed minimum size of 27000 bytes
```

**Steps to Reproduce**

1. Have a Steelseries Aerox 3 Wireless Mouse on firmware version 1.9.3
2. Run `sudo fwupdmgr update`
3. The firmware gets downloaded and unpacked
4. The mouse starts blinking, indicating it's no longer in normal operating mode
5. The firmware doesn't get updated
6. The mouse goes back to regular operation

**Expected behavior**
The firmware is updated.

**fwupd version information**
Please provide the version of the daemon and client.

```
compile   org.freedesktop.fwupd         1.9.11
compile   com.hughsie.libxmlb           0.3.14
compile   org.freedesktop.Passim        0.1.5
compile   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.fwupd-efi     1.4
runtime   org.freedesktop.Passim        0.1.5
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    6.7.0-zen3-1-zen
runtime   org.freedesktop.fwupd         1.9.11
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): Pacman, using Arch Linux's repositories.

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```
Intel Corporation NUC7i7BNH
│
├─Unknown Device:
│     Device ID:          57e35fbedc32c4e5e69c55e3d7810f7ae6f1427a
│     Numéro de série:    402022
│     GUID:               24cfaaa6-7ac5-5781-9692-27b71b70b250 ← DRM\VEN_GSM&DEV_59F1
│   
├─Aerox 3 Wireless Mouse:
│     Device ID:          1f914e619b232dc4da68ec3389b98b20163003a2
│     Version actuelle:   1.9.3
│     Fournisseur:        SteelSeries (USB:0x1038)
│     Durée d'installation:13 secondes
│     GUIDs:              df0cb7f0-be88-532f-87f9-65662da376b9
│                         14b18a05-6162-58f6-bae4-a6ff3c8b66c1
│                         4f3fac7c-981b-58cc-bb58-8a3b90f31c9c ← USB\VID_1038&PID_183A
│     Drapeaux de périphérique:• Mise à jour possible
│                         • Supported on remote server
│                         • Cryptographic hash verification is available
│   
├─Core™ i7-7567U CPU @ 3.50GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Version actuelle:   0x000000f4
│     Fournisseur:        Intel
│     GUIDs:              561403e8-143a-5071-ab09-bf5e4c146983 ← CPUID\PRO_0&FAM_06&MOD_8E
│                         9ca69899-3716-5857-9fd3-882a5c73236f ← CPUID\PRO_0&FAM_06&MOD_8E&STP_9
│     Drapeaux de périphérique:• Périphérique interne
│   
├─Iris Plus Graphics 650:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Version actuelle:   06
│     Fournisseur:        Intel Corporation (PCI:0x8086)
│     GUIDs:              64a4a919-4d3e-50d4-97b8-ccd95ee1c2f5 ← PCI\VEN_8086&DEV_5927
│                         0d57f774-a74c-5d8f-a988-035fc8d0cccd ← PCI\VEN_8086&DEV_5927&SUBSYS_80862068
│     Drapeaux de périphérique:• Périphérique interne
│                         • Cryptographic hash verification is available
│   
├─SUV500M8240G:
│     Device ID:          64ec4cd1e1c9565e79b00f4e7221b1c689b33e96
│     Résumé:             ATA drive
│     Version actuelle:   003056RA
│     Fournisseur:        Kingston (ATA:0x2646, OUI:0026b7)
│     Numéro de série:    50026B72820CDFED
│     GUIDs:              0da57aba-db63-5511-9b55-6129198d77d5 ← IDE\KINGSTON_SUV500M8240G___________________003056RA
│                         7b5c40b8-4fe3-51e2-9567-0e177924120c ← IDE\0KINGSTON_SUV500M8240G___________________
│                         999f3140-0df4-5045-ba94-8b24e4258a88 ← KINGSTON SUV500M8240G
│     Drapeaux de périphérique:• Périphérique interne
│                         • Mise à jour possible
│                         • Le système nécessite une source d'alimentation externe
│                         • Needs a reboot after installation
│                         • Needs shutdown after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Résumé:             UEFI ESRT device
│ │   Version actuelle:   69
│ │   Version minimum:    69
│ │   Fournisseur:        Intel Corporation (DMI:Intel Corp.)
│ │   État de mise à jour:Success
│ │   GUID:               fb9cea17-bf84-4c53-a573-625caa1a62a2
│ │   Drapeaux de périphérique:• Périphérique interne
│ │                       • Mise à jour possible
│ │                       • Le système nécessite une source d'alimentation externe
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Résumé:           UEFI revocation database
│       Version actuelle: 371
│       Version minimum:  371
│       Fournisseur:      UEFI:Linux Foundation
│       Durée d'installation:1 seconde
│       GUID:             f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│       Drapeaux de périphérique:• Périphérique interne
│                         • Mise à jour possible
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Only version upgrades are allowed
│                         • Signed Payload
│     
└─TPM:
      Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
      Résumé:             TPM 2.0 Device
      Version actuelle:   11.8.50.3425
      Fournisseur:        Intel (TPM:INTC)
      GUIDs:              34801700-3a50-5b05-820c-fe14580e4c2d ← TPM\VEN_INTC&DEV_0000
                          8e1cbc5d-5a11-5149-bfea-b6065d5296ba ← TPM\VEN_INTC&MOD_Intel
                          03f304f4-223e-54f4-b2c1-c3cf3b5817c6 ← TPM\VEN_INTC&DEV_0000&VER_2.0
                          52d7b679-db28-5bf7-bd87-41d77aeec600 ← TPM\VEN_INTC&MOD_Intel&VER_2.0
      Drapeaux de périphérique:• Périphérique interne
                          • Le système nécessite une source d'alimentation externe
                          • Needs a reboot after installation
                          • Device can recover flash failures
                          • Full disk encryption secrets may be invalidated when updating
                          • Signed Payload
```

</details>

**Additional questions**

- Operating system and version: ArchLinux, latest. Issue has been happening for at least two weeks. Before then, no firmware updates were shown to be available
- Have you tried rebooting? Yes
- Is this a regression? No idea.

superm1, 2024-01-17:
Can you try changing the quirk file like this?  You can change it on your filesystem without recompiling if you just restart fwupd.service.

```
diff --git a/plugins/steelseries/steelseries.quirk b/plugins/steelseries/steelseries.quirk
index de9399281..fc2dd90f2 100644
--- a/plugins/steelseries/steelseries.quirk
+++ b/plugins/steelseries/steelseries.quirk
@@ -48,7 +48,7 @@ GType = FuSteelseriesFizz
 Name = Aerox 3 Wireless Mouse
 Icon = input-mouse
 CounterpartGuid = USB\VID_1038&PID_1839,HIDRAW\VEN_0111&DEV_183A
-FirmwareSize = 0x27000
+FirmwareSize = 0x23000

 [USB\VID_1038&PID_183B]
 Plugin = steelseries
```

superm1, 2024-01-17:
CC @gportay.  This is from 714983d82d

yukijoou, 2024-01-17:
thanks for your swift response :)

where would i find `steelseries.quirk`? `sudo find / -iname 'steelseries.quirk'` found nothing…

superm1, 2024-01-17:
Oh; did we start compiling it in?  In that case let me just spin up a PR for you to try.  You can test it in the fwupd development environment without replacing your host fwupd.

https://github.com/fwupd/fwupd/pull/6690

gportay, 2024-01-17:
I have a very bad memory. I have no more access to the device and documentation. I will try to have a look and keep you informed. But I cannot guarantee it, sorry.

gportay, 2024-01-17:
Note: there is one firmware for the dongle, one for the mouse.

yukijoou, 2024-01-17:
umm… only the mouse is plugged in with its bundled usb-c-to-a cable right now… so, i don't see how it could be interfering…

yukijoou, 2024-01-17:
i have tried building & running the patched version, the build went fine, but i can't run any fwupd binaries. they all error out like this:

```
/tmp/tmp.6VSh8vY7Q6/fwupd/venv/bin/../dist/bin/fwupdmgr: error while loading shared libraries: libfwupd.so.3: cannot open shared object file: No such file or directory
```

the tests do pass fine though. any idea what could be going

superm1, 2024-01-17:
`source venv bin/activate`

yukijoou, 2024-01-17:
yep, i am in the venv!

```
(fwupd) [yukijoou@smolputer fwupd]$ fwupd
/tmp/tmp.6VSh8vY7Q6/fwupd/venv/bin/../dist/libexec/fwupd/fwupd: error while loading shared libraries: libfwupd.so.3: cannot open shared object file: No such file or directory
(fwupd) [yukijoou@smolputer fwupd]$ 
```

superm1, 2024-01-17:
huh....  It's supposed to be setting the `LD_LIBRARY_PATH` so it can load that properly.  Arch might be doing it a little weird.  I've tested it both on Ubuntu and Fedora and it works.  
Can you throw a `set -x` in the wrapper for fwupd?

superm1, 2024-01-17:
Here's a fix for that.  I'll send a PR separately.

```
--- a/contrib/launch-venv.sh
+++ b/contrib/launch-venv.sh
@@ -4,7 +4,7 @@ DIST="$(dirname $0)/../dist"
 BIN="$(basename $0)"
 export FWUPD_LOCALSTATEDIR=${DIST}
 export FWUPD_SYSCONFDIR=${DIST}/etc
-export LD_LIBRARY_PATH=${DIST}/lib/${gcc}:${DIST}/lib64
+export LD_LIBRARY_PATH=${DIST}/lib/${gcc}:${DIST}/lib64:${DIST}/lib
 if [ -n "${DEBUG}" ]; then
         if ! which gdbserver 1>/dev/null 2>&1; then
                 echo "install gdbserver to enable debugging"
```

yukijoou, 2024-01-17:
indeed, from my experience, arch's ld does weird things with the library path… hardcoding it to the correct location seems to fix the issue :) 

i can now run the daemon & connect to it. using `fwupdtool get-devices`, i get the following error messages before my usual device tree, could that be related to the issue?

```
18:43:57.592 FuDevice             FuSteelseriesFizz flag 0x2000 is unknown -- use fu_device_register_private_flag()
18:43:57.592 FuDevice             FuSteelseriesFizz flag 0x2000 is unknown -- use fu_device_register_private_flag()
18:43:57.756 FuEngine             steelseries [1f914e619b232dc4da68ec3389b98b20163003a2] does not declare signed/unsigned payload -- perhaps add fu_device_add_flag(device, FWUPD_DEVICE_FLAG_UNSIGNED_PAYLOAD);
18:43:58.226 FuPluginIntelMe      failed to get public key using /fpf/OemCred: generic failure [0xb]

yukijoou, 2024-01-17:
also, i can't update using this version, it reports:

```
Devices with no available firmware updates: 
 • Aerox 3 Wireless Mouse
 • SUV500M8240G
 • System Firmware
 • UEFI dbx
No updatable devices
```

superm1, 2024-01-17:
Did you run refresh yet?

yukijoou, 2024-01-17:
whoops!

just did, and it was able to download the update. the update is still failing though

```
Devices with no available firmware updates: 
 • SUV500M8240G
 • System Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Aerox 3 Wireless Mouse from 1.9.3 to 1.11.4?                         ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issue:                               ║
║                                                                              ║
║ • Fix issue with wireless connection dropping during use                     ║
║                                                                              ║
║ Aerox 3 Wireless Mouse and all connected devices may not be usable while     ║
║ updating.                                                                    ║
╚══════════════════════════════════════════════════════════════════════════════╝
Effectuer l'opération ? [Y|n]: y
Décompression…           [                                       ]  ]
checksum mismatch, got 0x2144df1c, expected 0xdb5500f5
```

gportay, 2024-01-17:
@superm1, the firmware size has not changed with 1.11.4 (since 1.9.3).

For the mouse:

```
gportay@archlinux ~/src/steelseries-firmware/aerox-3-wireless/mouse $ ls -l 1.9.3/mouse.bin 1.11.4/mouse.bin
-rw-r--r-- 1 gportay gportay 159744 Jun  1  2023 1.11.4/mouse.bin
-rw-r--r-- 1 gportay gportay 159744 Jan 17 19:32 1.9.3/mouse.bin
```

gportay, 2024-01-17:
For the dongle:

```
gportay@archlinux ~/src/steelseries-firmware/aerox-3-wireless/dongle :( $ ls -l 1.3.1/dongle.bin 1.11.4/dongle.bin
-rw-r--r-- 1 gportay gportay 143360 Jun  1  2023 1.11.4/dongle.bin
-rw-r--r-- 1 gportay gportay 143360 Jan 17 19:32 1.3.1/dongle.bin
```

Note: the version of dongle is now matching the version of mouse.

gportay, 2024-01-17:
> just did, and it was able to download the update. the update is still failing though

Hopefully.

My two cents here, fwupd is downloading the wrong firmware: the dongle to flash the mouse.

gportay, 2024-01-17:
@yukijoou, how are you flashing the mouse, using USB or via the dongle?

yukijoou, 2024-01-17:
> fwupd is downloading the wrong firmware: the dongle to flash the mouse.

can i download the firmware files manually to test out this theory?

> how are you flashing the mouse, using USB or via the dongle?

i have the mouse directly plugged into my computer. i tried using a different cable (c-to-c, rather than the provided a-to-c), and that didn't help. i tried unplugging the mouse and plugging in only the dongle, and that didn't work either, fwupd didn't seem to recognise it.

i should have probably noted beforehand that the mouse has been refusing to connect to the dongle for a while, which is what i was trying to fix with the firmware update… maybe there's something wrong hardware-wise? in which case, i am very sorry to have wasted your time

if needs be, i can try updating the firmware using steelseries' software, through a VM

gportay, 2024-01-17:
> can i download the firmware files manually to test out this theory?

I am sorry, I do not remember how to (or if it is possible even). I cannot help about that, @hughsie, @superm1 maybe?

So IIUC, your mouse is directly connected to your machine without using the original cable from Steelseries but with an a cable C-to-C of your own?

What does `fwupdmgr get-updates` says?

Note: I remember I had to use USB-A cable to test it. but I may be wrong.

gportay, 2024-01-17:
BTW, I strongly think you cannot upgrade the mouse firmware through the dongle if the mouse is not connected (paired) to it.

Could you connect the dongle and the mouse via USB and send the output of `lsusb` please?

superm1, 2024-01-17:
> I am sorry, I do not remember how to (or if it is possible even). I cannot help about that, @hughsie, @superm1 maybe?

If you take a CAB and extract it it's possible to use `fwupdtool install-blob` with the `firmware.bin` within it, but I don't think that's a generally good idea unless you know what you're doing and can recover.

yukijoou, 2024-01-17:
@gportay 

> So IIUC, your mouse is directly connected to your machine without using the original cable from Steelseries but with an a cable C-to-C of your own?

i've tried both that and using the original cable. didn't see any difference, except the c-to-c connection being more flakey, but that cable seems to generally have issues.

> What does `fwupdmgr get-updates` says?

```
Devices with no available firmware updates: 
 • SUV500M8240G
 • System Firmware
Devices with the latest available firmware version:
 • UEFI dbx
────────────────────────────────────────────────
Devices that were not updated correctly:
 • Aerox 3 Wireless Mouse (1.9.3 → 1.11.4)
Uploading firmware reports helps hardware vendors to quickly identify failing and successful updates on real devices.
[-- cut firmware report upload --]
Intel Corporation NUC7i7BNH
│
└─Aerox 3 Wireless Mouse:
  │   Device ID:          1f914e619b232dc4da68ec3389b98b20163003a2
  │   Version actuelle:   1.9.3
  │   Fournisseur:        SteelSeries (USB:0x1038)
  │   Durée d'installation:13 secondes
  │   GUIDs:              df0cb7f0-be88-532f-87f9-65662da376b9
  │                       14b18a05-6162-58f6-bae4-a6ff3c8b66c1
  │                       4f3fac7c-981b-58cc-bb58-8a3b90f31c9c ← USB\VID_1038&PID_183A
  │   Drapeaux de périphérique:• Mise à jour possible
  │                       • Supported on remote server
  │                       • Cryptographic hash verification is available
  │ 
  ├─Aerox 3 Wireless USB Receiver USB Receiver Update:
  │     Nouvelle version: 1.11.4
  │     Remote ID:        lvfs
  │     Release ID:       40205
  │     Résumé:           Firmware for the SteelSeries Rival 3 Wireless USB Receiver
  │     Licence:          Propriétaire
  │     Taille:           143,4 Ko
  │     Created:          2023-05-31
  │     Urgence:          Haute
  │     Tested by Allion:
  │       Tested:         2023-11-30
  │       Distribution:   chromeos 121
  │       Old version:    1.4.2
  │       Version[fwupd]: 1.9.7
  │     Tested by Collabora:
  │       Tested:         2023-08-09
  │       Distribution:   chromeos 117
  │       Old version:    1.11.4
  │       Version[fwupd]: 1.9.3
  │     Tested by Collabora:
  │       Tested:         2023-08-09
  │       Distribution:   chromeos 117
  │       Old version:    1.11.4
  │       Version[fwupd]: 1.9.3
  │     Tested by Collabora:
  │       Tested:         2023-08-03
  │       Distribution:   chromeos 117
  │       Old version:    1.4.2
  │       Version[fwupd]: 1.9.3
  │     Fournisseur:      SteelSeries
  │     Durée:            13 secondes
  │     Release Flags:    • Trusted metadata
  │                       • Is upgrade
  │     Description:      
  │     This stable release fixes the following issue:
  │     
  │     • Fix issue with wireless connection dropping during use
[... 61 lines cut ...]
        Created:          2022-09-09
        Urgence:          Haute
        Fournisseur:      SteelSeries
        Durée:            13 secondes
        Release Flags:    • Trusted metadata
                          • Is upgrade
        Description:      
        This stable release fixes the following issue:
        
        • Don't enter sleep in wireless mode when we are charging
        • Fix idle brightness not to be brighter than regular brightness
        • Wake up sleeping mouse if plugged into usb
        • Fix issue with modifiers being processed after other keys on windows
        • Fix BLE wake from sleep issue
        • Incorporated improvements from latest SteelSeries libraries
        • Updated lib-steel to address pixart jitter bug, smoothing bug, and 0 sample debounce bug
        • Faster wake from sleep (2s to 200ms) with slightly increased power consumption
        • Improved connection process to dongle
        • BLE connection improvements - improved pairing and Chromebook support
        • More consistent battery reporting
        • Decreased timing to prevent unintentionally entering bootloader mode
        Somme de contrôle:e73273acf749edef62cb648e947ff935d26f46dbbbec3718ea5a674f266b09fc
```

i did go ahead and submit the report, in case it's of any use

yukijoou, 2024-01-17:
i'll likely not go for the manual install for now, as it's currently the only mouse i have…

gportay, 2024-01-18:
@hughsie, @superm1, I am no more familiar with the output, but does the `Release ID:       40205` (aka `Aerox 3 Wireless USB Receiver USB Receiver Update`) is a candidate firmware for the device `1f914e619b232dc4da68ec3389b98b20163003a2`/`4f3fac7c-981b-58cc-bb58-8a3b90f31c9c ← USB\VID_1038&PID_183A` (aka `Aerox 3 Wireless Mouse`)?

If so, it should not it is a firmware for the dongle (receiver) not for the mouse.

gportay, 2024-01-18:
> Could you connect the dongle and the mouse via USB and send the output of lsusb please?

@yukijoou, could you send the output of lsusb with both USB dongle and USB mouse connected?

And the output of `fwupdmgr get-updates` with the USB dongle connected?

gportay, 2024-01-18:
@yukijoou, I you need your mouse to get fixed, you can go through the expert way with `fwupdtool install FILE [DEVICE-ID|GUID]` or `fwupdtool install-blob FILENAME DEVICE-ID`. All you need is to download the [cabinet](https://fwupd.org/downloads/f1453976fe7ea27522524872327ad5aab6fba0c962df719ac05253439d7d72a9-aerox-3-wireless-mouse.cab) file and install it to device guid `4f3fac7c-981b-58cc-bb58-8a3b90f31c9c` or id `1f914e619b232dc4da68ec3389b98b20163003a2`

But I would prefer to understand what is wrong first, if you do not mind.

yukijoou, 2024-01-18:
@gportay 

> could you send the output of lsusb with both USB dongle and USB mouse connected?

sure :)

```
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 05d5:0689 Super Gate Technology Co., Ltd KEYBOARD
Bus 001 Device 003: ID 0d8c:0134 C-Media Electronics, Inc. BIRD UM1
Bus 001 Device 004: ID 1038:183a SteelSeries ApS SteelSeries Aerox 3 Wireless
Bus 001 Device 005: ID 2516:0089 Cooler Master Co., Ltd. SK630
Bus 001 Device 006: ID 8087:0a2b Intel Corp. Bluetooth wireless interface
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 1038:1838 SteelSeries ApS SteelSeries Aerox 3 Wireless
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
```

> And the output of `fwupdmgr get-updates` with the USB dongle connected?

```
Devices with no available firmware updates: 
 • Aerox 3 Wireless Mouse via USB Receiver
 • SUV500M8240G
 • System Firmware
Devices with the latest available firmware version:
 • UEFI dbx
Intel Corporation NUC7i7BNH
│
├─Aerox 3 Wireless Mouse:
│ │   Device ID:          1f914e619b232dc4da68ec3389b98b20163003a2
│ │   Current version:    1.9.3
│ │   Vendor:             SteelSeries (USB:0x1038)
│ │   Install Duration:   13 seconds
│ │   GUIDs:              df0cb7f0-be88-532f-87f9-65662da376b9
│ │                       14b18a05-6162-58f6-bae4-a6ff3c8b66c1
│ │                       4f3fac7c-981b-58cc-bb58-8a3b90f31c9c ← USB\VID_1038&PID_183A
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Cryptographic hash verification is available
│ │ 
│ ├─Aerox 3 Wireless USB Receiver USB Receiver Update:
│ │     New version:      1.11.4
│ │     Remote ID:        lvfs
│ │     Release ID:       40205
│ │     Summary:          Firmware for the SteelSeries Rival 3 Wireless USB Receiver
│ │     License:          Proprietary
│ │     Size:             143.4 kB
│ │     Created:          2023-05-31
│ │     Urgency:          High
│ │     Tested by Allion:
│ │       Tested:         2023-11-30
│ │       Distribution:   chromeos 121
│ │       Old version:    1.4.2
│ │       Version[fwupd]: 1.9.7
│ │     Tested by Collabora:
│ │       Tested:         2023-08-09
│ │       Distribution:   chromeos 117
│ │       Old version:    1.11.4
│ │       Version[fwupd]: 1.9.3
│ │     Tested by Collabora:
│ │       Tested:         2023-08-09
│ │       Distribution:   chromeos 117
│ │       Old version:    1.11.4
│ │       Version[fwupd]: 1.9.3
│ │     Tested by Collabora:
│ │       Tested:         2023-08-03
│ │       Distribution:   chromeos 117
│ │       Old version:    1.4.2
[... 209 lines cut ...]
        Release ID:       27270
        Summary:          Firmware for the SteelSeries Rival 3 Wireless USB Receiver
        License:          Proprietary
        Size:             143.4 kB
        Created:          2023-02-16
        Urgency:          Low
        Tested by Collabora:
          Tested:         2023-08-09
          Distribution:   chromeos 117
          Old version:    1.11.4
          Version[fwupd]: 1.9.3
        Vendor:           SteelSeries
        Duration:         13 seconds
        Release Flags:    • Trusted metadata
                          • Is upgrade
        Description:      
        This stable release fixes the following issue:
        
        • Fix issue with modifiers being processed after other keys on windows.
        Checksum:         41e25df1ee09f3c05a4799e4fc325115bd5ba9490309c76c545f3e3a88c1c9b9
```

> But I would prefer to understand what is wrong first, if you do not mind.

no problems with that – i'd rather the issue gets fixed too, so that future updates go smoothly!

gportay, 2024-01-25:
> @hughsie, @superm1, I am no more familiar with the output, but does the `Release ID: 40205` (aka `Aerox 3 Wireless USB Receiver USB Receiver Update`) is a candidate firmware for the device [...]

@superm1, @hughsie, please could you confirm or not this?

hughsie, 2024-01-25:
> please could you confirm or not this?

According to the LVFS, `4f3fac7c-981b-58cc-bb58-8a3b90f31c9c` is a  SteelSeries Aerox 3 Wireless Mouse Mouse Update

gportay, 2024-01-25:
> > please could you confirm or not this?
> 
> According to the LVFS, `4f3fac7c-981b-58cc-bb58-8a3b90f31c9c` is a  SteelSeries Aerox 3 Wireless Mouse Mouse Update

And if my understanding is right, fwupd suggests to install the firmware id 40205 which is for the Aerox 3 Wireless USB Receiver USB Receiver Update. And this is an issue. But I cannot understand how that is possible.

yukijoou, 2024-01-31:
hi, i was able to update my device using [gnome-firmware](https://gitlab.gnome.org/World/gnome-firmware)'s fwupd frontend!

[image]

it allowed me to pick which device to update and to which version (either the dongle or the actual mouse), so i guess that circumvented the issue. i think @gportay's theory was correct.

maybe having a way to pick which device to update in fwupd would be a good temporary fix? though, honestly, i have no idea what gnome is doing on their end to make this work…

i can still downgrade for testing if needs be i think, i'm not happy with the solution, but the problem is technically solved.

yukijoou, 2024-01-31:
i will also note that when the mouse is plugged in via usb, and the dongle unplugged, gnome-firmware seems to show updates for both the dongle and the mouse as available for the mouse itself… so yeah, seems like something's going wrong when pulling data for which firmware updates are available for the device

[image]

and, when the mouse is connected through the dongle, the dongle has its own firmware updates

[image]

and the mouse has its own updates

[image]

re: what i said about the dongle not working for me btw, after the firmware update, it does work, *sometimes*… it seems quite flakey, and has a hard time reconnecting after a disconnect. so my hardware may be faulty, though i don't *think* this should matter much to the issue…?

i don't know anyone else with the same mouse who could test this out, maybe it's worth asking on some forum/reddit/discord thing about that to get other people's feedback

gportay, 2024-05-20:
TL;DR; I can reproduce the issue with the latest fwupd on Arch Linux; I confirm the issue is raised if upgrading the mouse via the USB cable; the upgrading works correctly if the device it connected to the USB Adapter.

	$ fwupdmgr --version
	compile   com.hughsie.libxmlb           0.3.19
	compile   org.freedesktop.Passim        0.1.7
	compile   com.hughsie.libjcat           0.2.1
	compile   org.freedesktop.fwupd         1.9.18
	runtime   org.freedesktop.fwupd-efi     1.6
	compile   org.freedesktop.gusb          0.4.8
	runtime   com.hughsie.libxmlb           0.3.19
	runtime   org.freedesktop.Passim        0.1.8
	runtime   org.freedesktop.gusb          0.4.8
	runtime   com.hughsie.libjcat           0.2.1
	runtime   org.kernel                    6.8.7-arch1-2
	runtime   org.freedesktop.fwupd         1.9.18

I have bought a second hand Aerox 3 Wireless; the device I have received come with the latest firmwares, so I have downgraded all of them using my archived firmware [repository](https://github.com/gportay/steelseries-firmware).

    $ git clone git@github.com:gportay/steelseries-firmware.git
    $ cd steelseries-firmware/
    $ make world
    $ cd aerox-3-wireless/mouse/1.9.3/
    $ fwupdmgr local-install SteelSeries_Aerox_3_Wireless_Mouse_1.9.3.cab --allow-older
    $ cd ../../dongle/1.3.1/
    $ fwupdmgr local-install SteelSeries_Aerox_3_Wireless_Dongle_1.3.1.cab --allow-older

    $ fwupdmgr get-devices
    (...)
    ├─Aerox 3 Wireless USB Receiver:
    │ │   Device ID:          54dda0fbfd5ac0dee5efc0ec4baefc9190d666ac
	│ │   Current version:    1.3.1
	│ │   Vendor:             SteelSeries (USB:0x1038)
	│ │   Install Duration:   13 seconds
	│ │   Update State:       Success
	│ │   GUIDs:              df0cb7f0-be88-532f-87f9-65662da376b9
	│ │                       291b0582-8d08-5714-88db-986911c744d9 ← USB\VID_1038&PID_1838
	│ │   Device Flags:       • Updatable
	│ │                       • Supported on remote server
	│ │                       • Cryptographic hash verification is available
	│ │ 
	│ └─Aerox 3 Wireless Mouse via USB Receiver:
	│       Device ID:        d068fe744d853a3ae2d51f78d034c1aa121ee21a
	│       Current version:  1.9.3
	│       Vendor:           SteelSeries (USB:0x1038)
	│       Install Duration: 38 seconds
	│       Battery:          80% (threshold 20%)
	│       GUID:             541b2713-d367-5240-a443-bed07f09ffbf ← STEELSERIES\VID_1038&PID_1838&PROTOCOL_FIZZ_TUNNEL
	│       Device Flags:     • Updatable
	│                         • Supported on remote server
	│                         • Cryptographic hash verification is available

    $ fwupdmgr get-updates	
    (...)
	├─Aerox 3 Wireless Mouse via USB Receiver:
	│ │   Device ID:          d068fe744d853a3ae2d51f78d034c1aa121ee21a
	│ │   Current version:    1.9.3
	│ │   Vendor:             SteelSeries (USB:0x1038)
	│ │   Install Duration:   38 seconds
	│ │   Battery:            80% (threshold 20%)
	│ │   GUID:               541b2713-d367-5240-a443-bed07f09ffbf ← STEELSERIES\VID_1038&PID_1838&PROTOCOL_FIZZ_TUNNEL
	│ │   Device Flags:       • Updatable
	│ │                       • Supported on remote server
	│ │                       • Cryptographic hash verification is available
	│ │ 
	│ ├─Aerox 3 Wireless Mouse Mouse Update:
	│ │     New version:      1.11.4
	│ │     Remote ID:        lvfs
	│ │     Release ID:       40204
	│ │     Summary:          Firmware for the SteelSeries Aerox 3 Wireless Gaming Mouse
	│ │     License:          Proprietary
[... 255 lines cut ...]
	║ • Fix issue with modifiers being processed after other keys on windows       ║
	║ • Fix BLE wake from sleep issue                                              ║
	║ • Incorporated improvements from latest SteelSeries libraries                ║
	║ • Updated lib-steel to address pixart jitter bug, smoothing bug, and 0       ║
	║ sample debounce bug                                                          ║
	║ • Faster wake from sleep (2s to 200ms) with slightly increased power         ║
	║ consumption                                                                  ║
	║ • Improved connection process to dongle                                      ║
	║ • BLE connection improvements - improved pairing and Chromebook support      ║
	║ • More consistent battery reporting                                          ║
	║ • Decreased timing to prevent unintentionally entering bootloader mode       ║
	║                                                                              ║
	║ Aerox 3 Wireless Mouse and all connected devices may not be usable while     ║
	║ updating.                                                                    ║
	╚══════════════════════════════════════════════════════════════════════════════╝
	Perform operation? [Y|n]: y
	Authenticating…          [***************************************]==== AUTHENTICATING FOR org.freedesktop.fwupd.downgrade-hotplug-trusted ====
	Authentication is required to downgrade the firmware on a removable device
	Authenticating as: gportay
	Password: 
	==== AUTHENTICATION COMPLETE ====
	Waiting…                 [***************************************] Less than one minute remaining…
	Successfully installed firmware

I still suspect an issue in the firmware proposed by fwupd either the quirk file or the cabinet file.

gportay, 2024-05-20:
Well, the quirk and the cabinet files seem corrects. I am suspecting something missing in the plugin between the parent and the child.

It appears fwupd proposes the parent firmware to upgrade the child if I am correct.

gportay, 2024-05-24:
I found the issue!

It comes from a mismatch in the `CounterpartGuid` in the [quirk](https://github.com/fwupd/fwupd/blob/c96f3cbb59844c2db3a033c1189caf3931d8ef67/plugins/steelseries/steelseries.quirk#L50) file.

The first counterpart GUID `USB\VID_1038&PID_1839` is the USB adapter (bootloader) **not** the USB mouse  (i.e. the mouse connected directly via a USB C -> A cable).

The true counterpart GUID is `USB\VID_1038&PID_183B` for the USB mouse (bootloader).

It is a "typo": `9` -> `B`

---

A bit of history.

The commit 8d4d2f208798811640a2fb081cf963ac2754fa9d has introduced a regression in the [CounterpartGuid](https://github.com/fwupd/fwupd/commit/8d4d2f208798811640a2fb081cf963ac2754fa9d#diff-57dfd6e4effc618d4f6529143457cd467828a22f7d31c374bd3a9f1cd4242c47L44) by appending the counterpart GUID for the Bluetooth HIDRAW. It has modified the counterpart GUID at the same time (certainly a nonexistent USB product of [Quatographic AG](https://the-sz.com/products/usbid/index.php?v=0x103B).

Then, the commit c96f3cbb59844c2db3a033c1189caf3931d8ef67 intends to fix back the counterpart GUID by taking the (mismatched) couterpart GUID of the USB adapter (bootloader) instead of the USB mouse (bootloader).

Since then, `fwupd` proposes the firmware for the USB adapter if an USB mouse device is connected via an USB C -> A cable.

I guess that issue was silent as the firmware version of the USB adapter was lower than the one for the USB mouse.

The issue became visible since both firmware hold the same version number, causing `fwupd` to propose the USB adapter firmware as first choice.

Hopefully, the mechanism to detect a wrong firmware raises a size issue! 

So my bad, sorry for that mess.

closed 2024-05-24 by hughsie