fwupd/fwupd #6707: Hmm; modify-config is broken...  Look what a few runs are doing:

superm1, 2024-01-20:
Hmm; modify-config is broken...  Look what a few runs are doing:

```
[fwupd]
DisabledPlugins=test;test_ble
OnlyTrusted=true
AllowEmulation=false
DisabledPlugins=
OnlyTrusted=true
AllowEmulation=false
DisabledPlugins=test;test_ble
OnlyTrusted=true
AllowEmulation=false
DisabledPlugins=test_ble
OnlyTrusted=true
AllowEmulation=false
```

_Originally posted by @superm1 in https://github.com/fwupd/fwupd/pull/6706#discussion_r1460641059_

superm1, 2024-01-21:
Reding [glib comments](https://github.com/GNOME/glib/blob/main/glib/gkeyfile.c#L152C4-L154C21), this seems like expected behavior for what we're throwing at it.

> groups in key files may contain the same key multiple times; the last entry wins.
> * Key files may also contain multiple groups with the same name; they are merged
> * together. 

I _guess_ it's because we merge two fwupd.conf files.

superm1, 2024-01-21:
do you have both a mutable and immutable conf specified?  I believe it's specific to that circumstance.

hughsie, 2024-01-21:
In that case, does something like this make sense?

			/* load the main daemon config file */
	-       if (!fu_config_add_location(self, configdir, error))
	-               return FALSE;
			if (!fu_config_add_location(self, configdir_mut, error))
					return FALSE;
	+       if (priv->items->len == 0) {
	+               g_autofree gchar *configdir = fu_path_from_kind(FU_PATH_KIND_SYSCONFDIR_PKG);
	+               if (!fu_config_add_location(self, configdir, error))
	+                       return FALSE;
	+       }
			if (!fu_config_reload(self, error))

i.e. the mutable one completely replaces the immutable one if it exists.

superm1, 2024-01-21:
So if the daemon encounters both it only uses mutable? And similarly if it starts with immutable and gets changed it writes mutable and stops using immutable next time?

hughsie, 2024-01-21:
Nah, my suggestion is too heavy a hammer. I'll have a think and write some tests on Monday. I think we only want to write out the *changed* versions to the mutable file. We're doing a dumb thing now.

closed 2024-01-23 by hughsie