fwupd/fwupd #6715: "WARNING: UEFI capsule updates not available or enabled in firmware setup" with `DisabledPlugins=test;test_ble;invalid;bios` in config

mattalexx, 2024-01-22:
**Describe the bug**
Running `fwupdmgr refresh --force`, I get:

```
WARNING: UEFI capsule updates not available or enabled in firmware setup
See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
```

[I understand](https://github.com/fwupd/fwupd/issues/6703) that my machine will not support this feature, so I want to disable the warning. In the [referenced wiki page](https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported), it says:

> This warning can be ignored if UEFI firmware updates are not desired. It can be disabled by adding DisabledPlugins=test;test_ble;invalid;bios to /etc/fwupd/daemon.conf or recompiling the daemon without UEFI support.

So I tried that, with no luck:

```
% sudo cat /etc/fwupd/daemon.conf
cat: /etc/fwupd/daemon.conf: No such file or directory
% echo 'DisabledPlugins=test;test_ble;invalid;bios' | sudo tee /etc/fwupd/daemon.conf
DisabledPlugins=test;test_ble;invalid;bios
% sudo fwupdmgr refresh --force
WARNING: UEFI capsule updates not available or enabled in firmware setup
See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
Updating lvfs
Downloading…             [           |                           ]
Successfully downloaded new metadata: 1 local device supported
```

Thinking it weird that the file didn't exist I investigated, and found this in the log:

https://github.com/fwupd/fwupd/commit/7fa422744c42d5f6a207d43c094d02dece64baf6

I edited the wiki page to reflect the new path, then tried with the actual config file:

```
% sudo rm /etc/fwupd/daemon.conf
% sudo cat /etc/fwupd/fwupd.conf
[fwupd]
DisabledPlugins=test;test_ble
OnlyTrusted=true
AllowEmulation=false
% sudo vim /etc/fwupd/fwupd.conf
% sudo cat /etc/fwupd/fwupd.conf
[fwupd]
DisabledPlugins=test;test_ble;invalid;bios
OnlyTrusted=true
AllowEmulation=false
```
Trying again:
```
% sudo fwupdmgr refresh --force
WARNING: UEFI capsule updates not available or enabled in firmware setup
See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
Updating lvfs
Downloading…             [              /                        ]
Successfully downloaded new metadata: 1 local device supported
```

Didn't work.

**Steps to Reproduce**
Set `DisabledPlugins` to `test;test_ble;invalid;bios` in `/etc/fwupd/fwupd.conf`. Run `fwupdmgr refresh --force`.

**Expected behavior**

No warnings:

```
% sudo fwupdmgr refresh --force
Updating lvfs
Downloading…             [              /                        ]
[... 147 lines cut ...]
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│
└─intel-spi:
Device ID:          051f0b361682a47120288c6f017bcc63815d05e1
Summary:            Memory Technology Device
Vendor:             DMI:Dell Inc.
GUIDs:              9a048892-31d3-5e76-93e0-0d2310b57476 ← MTD\NAME_intel-spi
d07b697a-6aff-5e03-b846-0bfaf5d2d5f7 ← MTD\VENDOR_Dell-Inc.&NAME_intel-spi
4a5cf905-78cb-5c77-b63e-25764c949a21 ← MTD\VENDOR_Dell-Inc.&PRODUCT_OptiPlex-9020&NAME_intel-spi
Device Flags:       • Internal device
• Updatable
• Needs a reboot after installation
• Cryptographic hash verification is available
```

</details>

**Additional questions**

- Operating system and version:
- Have you tried rebooting?
- Is this a regression?

superm1, 2024-01-22:
Have you restarted the service or rebooted since making that change?

mattalexx, 2024-01-24:
Sorry, I reported I'd rebooted but I guess I hadn't, because the error is gone after rebooting now. False alarm.