fwupd/fwupd #6717: Additional systemd options

RZR7332, 2024-01-22:
Would it be possible/feasible/realistic to update the systemd service file for fwupd to include some hardening options? I have built a service file (based largely on https://github.com/juju4/ansible-harden-systemd/blob/main/templates/fwupd.service.d-override.conf.j2) to include some additional sandboxing and security toggles.

I am not suggesting that fwupd is insecure or badly coded, but an extra layer never hurts in my opinion. I am using the below configuration (some I have removed as it is specific to my systems):

Sourced from https://github.com/juju4/ansible-harden-systemd/blob/master/templates/fwupd.service.d-override.conf.j2

Exposure 7.2 -> 2.4
Not matching
https://github.com/fwupd/fwupd/blob/main/data/org.freedesktop.fwupd.service.in

[Unit]
Description=Firmware update daemon [OVERRIDE IN EFFECT]

[Service]
PrivateTmp=yes
ProtectHome=yes
ProtectSystem=strict
RestrictAddressFamilies=AF_NETLINK AF_UNIX
ProtectControlGroups=yes
ProtectKernelModules=yes
RestrictRealtime=yes
ReadWritePaths=-/boot/efi -/efi/EFI -/boot/EFI

ProcSubset=pid
ProtectProc=ptraceable

ProtectKernelTunables=yes
ProtectKernelLogs=yes
ProtectHostname=yes
ProtectClock=yes

UMask=077

CapabilityBoundingSet=~CAP_SYS_ADMIN
CapabilityBoundingSet=~CAP_AUDIT_CONTROL CAP_AUDIT_READ CAP_DAC_READ_SEARCH
CapabilityBoundingSet=~CAP_SYS_RAWIO
CapabilityBoundingSet=~CAP_SYS_PTRACE
CapabilityBoundingSet=~CAP_DAC_OVERRIDE CAP_DAC_READ_SEARCH CAP_FOWNER CAP_IPC_OWNER
CapabilityBoundingSet=~CAP_NET_ADMIN
CapabilityBoundingSet=~CAP_KILL
CapabilityBoundingSet=~CAP_MKNOD
CapabilityBoundingSet=~CAP_NET_BIND_SERVICE CAP_NET_BROADCAST
CapabilityBoundingSet=~CAP_SYS_NICE CAP_SYS_RESOURCE
CapabilityBoundingSet=~CAP_SYS_BOOT
CapabilityBoundingSet=~CAP_LINUX_IMMUTABLE
CapabilityBoundingSet=~CAP_SYS_CHROOT
CapabilityBoundingSet=~CAP_BLOCK_SUSPEND
CapabilityBoundingSet=~CAP_LEASE
CapabilityBoundingSet=~CAP_SYS_PACCT
CapabilityBoundingSet=~CAP_SYS_TTY_CONFIG
CapabilityBoundingSet=~CAP_SETUID CAP_SETGID CAP_SETPCAP
CapabilityBoundingSet=~CAP_CHOWN CAP_FSETID CAP_SETFCAP
CapabilityBoundingSet=~CAP_NET_RAW
CapabilityBoundingSet=~CAP_IPC_LOCK

NoNewPrivileges=yes
RestrictNamespaces=yes
LockPersonality=yes
MemoryDenyWriteExecute=yes
RestrictSUIDSGID=yes

SystemCallFilter=@system-service @resources @privileged
SystemCallFilter=~@debug @mount @reboot @swap @raw-io @clock @module @obsolete
SystemCallErrorNumber=EPERM
SystemCallArchitectures=native

I have been running this for some months on Ubuntu Server 22.04 LTS without any issues.

Just a suggestion, if this is considered it is ideally done upstream at the source rather than users creating a custom service file.

RZR7332, 2024-01-22:
Suspect I will need a bit of guidance here. It seems clear to me that I need to fork and modify - but which file to modify? Where are these settings stored? I can see what looks like partial config in fwupd/data/fwupd.service.in but not sure that is the correct file to work on.

hughsie, 2024-01-22:
> fwupd/data/fwupd.service.in

Yes, that's the correct file. There's a bit of complexity to deal with different systemd versions, but it's mostly boilerplate.

RZR7332, 2024-01-22:
Awesome, thank you. I will try to come up with something presentable tomorrow (learning the ropes here as to how this whole process works).

I will also go through the settings tomorrow against the manual and figure out what the minimum systemd version is to support all the settings. In most cases I think the setting is ignored if it is not valid, so even on an older version it *should* not break anything.

Appreciate the help for a noob.

superm1, 2024-01-22:
What kind of devices were you updating? Some of those capabilities removal are things fwupd uses for some kinds of devices.

RZR7332, 2024-01-22:
@superm1 nothing fancy, perhaps why I have not seen any issues. Do you know which capabilities are required? I can update the pull and my config accordingly in order to not waste anyone's time.

superm1, 2024-01-22:
IIUC rawio is probably going to break ssd updates and disk access, sys-admin is going to break any devfs access.  Immutable is going to break UEFI updates.

Dropping the mount call might affect UEFI updates, depends on how the calls for udisks2 work.  Maybe it only breaks unmounting the disk?

A good idea is to reference everything used in the fwupd snap interface: https://github.com/snapcore/snapd/blob/master/interfaces/builtin/fwupd.go

You can see all the capabilities explicitly given to the snap.  Any of those you're tearing away will break _something_.

RZR7332, 2024-01-22:
This is most helpful - thank you! I will spend some time tomorrow and during the course of the week reworking the configuration to align with that link before proceeding any further.

Really appreciate the input and guidance, thank you.

RZR7332, 2024-01-23:
Making progress on a revised service file (some of the options I am a little uncertain about, but this is why we test). For reference, on Ubuntu 22.04, the below are defaults:

[Service]
Type=dbus
TimeoutSec=180
RuntimeDirectory=motd.d
RuntimeDirectoryPreserve=yes
BusName=org.freedesktop.fwupd
ExecStart=/usr/libexec/fwupd/fwupd
PrivateTmp=yes
ProtectHome=yes
ProtectSystem=full
SystemCallFilter=~@mount
ProtectControlGroups=yes
ProtectKernelModules=yes
RestrictRealtime=yes
ReadWritePaths=-/boot/efi -/efi/EFI -/boot/EFI -/boot/grub
ConfigurationDirectory=fwupd
StateDirectory=fwupd
CacheDirectory=fwupd
RestrictAddressFamilies=AF_NETLINK AF_UNIX AF_INET AF_INET6

From this, I conclude excluding the mount system call is not problematic.

RZR7332, 2024-01-29:
I am still working on this, have not had much time of late to dedicate.

@hughsie @superm1 with regards to the proposed changes for Fedora 40 (https://discussion.fedoraproject.org/t/f40-change-proposal-systemd-security-hardening-system-wide/96423/1) - is anyone from the project involved? I see fwupd is listed as one of the potentially targeted services.

hughsie, 2024-01-29:
> is anyone from the project involved

Me!? I'm a Red Hat employee.

RZR7332, 2024-01-29:
Apologies 😄 guess that answers that question! Not going to wait for the Fedora work but will try to keep tabs on whatever they may be doing.

superm1, 2024-01-29:
I don't think it's a good idea to let Fedora and upstream deviate in this area. If some default hardening is good enough for Fedora it should be good enough for upstream. If it's not it's masking a bug.

RZR7332, 2024-01-29:
> I don't think it's a good idea to let Fedora and upstream deviate in this area. If some default hardening is good enough for Fedora it should be good enough for upstream. If it's not it's masking a bug.

Agreed - hence why I asked the question in the first place.

RZR7332, 2024-03-13:
Apologies for radio silence and any lack of movement on this, I have not had much time to spare as of late. I see some of the settings I was going to suggest have been merged (https://github.com/fwupd/fwupd/pull/6860) - great news!

Are you guys open to further options/suggestions? I do still feel there is some room for improvement, especially around CapabilityBoundingSet and SystemCallFilter settings. I would like to take a cautious approach and introduce a few at a time to see if anything breaks, rather than just throwing everything into the file in one go and hoping for the best.

Either way, thank you for the merge (and to Rahul), this is progress 😄

superm1, 2024-03-13:
> Apologies for radio silence and any lack of movement on this, I have not had much time to spare as of late. I see some of the settings I was going to suggest have been merged [...]

Certainly open to more, but yes please let's take them one at a time and they're not candidates for the stable branch immediately.

RZR7332, 2024-03-13:
Understood - I read through the pull request and subsequent conversation - will only be backported to 1.9.x eventually after 2.0 has been released and out for a while. Totally understand.

I will try to get something submitted by end of week/weekend - will first target ProcSubset (which I am uncertain about in this case), as well as removing a few of the SystemCallFilters which are unlikely to be needed.

Thanks!

closed 2024-11-11 by hughsie

RZR7332, 2024-11-12:
> I think we did a lot of this now.

Thank you for your time and effort, much appreciated 😄