fwupd/fwupd #6721: Passim listens on 0.0.0.0:27500 and [::]:27500 until reboot

DonRichie, 2024-01-22:
**Describe the bug**
After executing fwupdmgr, the following ports are opened by the caching software "passim", which has been implicitly started by fwupdmgr:

```
LISTEN 0      10                0.0.0.0:27500      0.0.0.0:*
LISTEN 0      10                   [::]:27500         [::]:*
```

These ports are open to everyone in the network until the system has been rebooted.

**Steps to Reproduce**
Execute "fwupdmgr"

**Expected behavior**
In my opinion the software passim should be configured by fwupdmgr to only listen on localhost.
It should further be turned off again after usage. Making its usage optional, would also be an appreciated feature.

```
**fwupd version information**
fwupd 1.9.11-2
passim 0.1.5-1
```

```
fwupdmgr --version

~/sees> fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.15
compile   org.freedesktop.Passim        0.1.5
compile   com.hughsie.libjcat           0.2.0
compile   org.freedesktop.fwupd         1.9.11
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.Passim        0.1.5
runtime   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.0
runtime   org.kernel                    6.7.0-arch3-1
runtime   org.freedesktop.fwupd         1.9.11
```

**Additional information**
```
Installed from Arch Linux extra repository.
OS: Linux 6.7.0-arch3-1 #1 SMP PREEMPT_DYNAMIC Sat, 13 Jan 2024 14:37:14 +0000 x86_64 GNU/Linux
```

hughsie, 2024-01-22:
Err, isn't the point? passimd has to send the local file out to all machines on the local network.

EdJoPaTo, 2024-01-22:
Personal point of view: There is some dependency I didn't install or configure manually which is related to my hardware and its available openly on the network.
Even if this service is not sharing information related to my specific hardware or firmware state it is a red flag to me at first glance just by the relations of this software. My first reaction to this was to remove `passim` (and therefore `fwupd`) from my systems until I know why it is required for `fwupd` to share stuff on the network.

`fwupd` should be about updating firmware on my machine, not sharing stuff on the network. If that's an optional feature – fine. But it should be enabled actively then. Even when I install `passim` manually for some other reason to share stuff on my network `fwupd` shouldn't use it to share hardware related stuff just because it's there without any active configuration on my behalf.

`passim` isn't currently mentioned in the repo docs files except SECURITY.md. The [release notes of 1.9.5](https://github.com/fwupd/fwupd/releases/tag/1.9.5) mention passim as an optional feature. I assume it's a bug on the Arch Linux Package to install passim as a required dependency and therefore exposing it on the network?
(But even then, don't just use something available to the network just because it's there without the user / admin enabling such a feature)

superm1, 2024-01-22:
The policy for it is configured by the P2PPolicy configuration option in `fwupd.conf`.  I'd expect if you change that to `none` it won't run anymore.

It's described in the man page here: https://fwupd.github.io/libfwupdplugin/fwupd.conf.html

hughsie, 2024-01-22:
Google finds this: https://blogs.gnome.org/hughsie/2023/07/28/introducing-passim/

soyersoyer, 2024-04-23:
P2PPolicy=none does not disable passimd

closed 2024-04-24 by hughsie

hughsie, 2024-04-24:
> fwupd shouldn't just use a program when its installed

Why not? Passim was created with fwupd as it's intended use case.

> It should be an active decision to enable such a feature.

Why?

> Either passim is required then it should be a hard dependency, or it should be optional.

It's not strictly required, it's just recommended.

EdJoPaTo, 2024-04-24:
tldr: Arch Linux has passim as a hard dependency for fwupd. [wont fix](https://bugs.archlinux.org/task/79614)

Personal solution: don't keep fwupd installed, only install it once in a while to check for updates, then remove it completely from the system again.

hughsie, 2024-04-24:
> only install it once in a while to check for updates

That's literally your choice dude.

jason-oliveira, 2024-05-11:
This explains how my computer was infected with LogoFAIL. Thank you @DonRichie for this bug I can reference in later suits against my assailant.

hughsie, 2024-05-11:
@jason-oliveira ha, that made me laugh, thanks :)

stigok, 2025-12-02:
> The policy for it is configured by the P2PPolicy configuration option in `fwupd.conf`. I'd expect if you change that to `none` it won't run anymore.
> 
> It's described in the man page here: https://fwupd.github.io/libfwupdplugin/fwupd.conf.html

A working solution to disable this behaviour is adding the following line to your `/etc/fwupd/fwupd.conf`:
```
P2pPolicy=nothing
```
A process restart is required for this to take effect (`sudo kill passimd`).