fwupd/fwupd #6725: Mediatek scaler chip device is missing with snap fwupd 1.9.11

CragW, 2024-01-23:
**Describe the bug**
Mediatek scaler chip device is missing with snap fwupd 1.9.11

**Steps to Reproduce**
$ sudo fwupd.fwupdtool get-devices --plugins mediatek_scaler

**Expected behavior**
The device is shown in the device list, for example:
```
$ fwupdtool get-devices --plugins mediatek_scaler
Loading…                 [*********************                  ]
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
Dell Inc. OptiPlex AIO Plus 7420
│
└─Display Controller:
      Device ID:          7002ec1e191e016c9a498fd8450bd012598dba20
      Current version:    2.1.7
      Vendor:             Mediatek (PCI:0x1028)
      GUID:               f54c71fc-4404-5af9-87dd-6886fbd6af3b ← DISPLAY\VID_1028&PID_0C88&HWVER_2.1.2.1
      Device Flags:       • Internal device
                          • Updatable
                          • Cryptographic hash verification is available
                          • Unsigned Payload
```

**fwupd version information**
$ snap list fwupd
Name   Version               Rev   Tracking    Publisher       Notes
fwupd  1.9.11-85-g4e1992d78  5632  1.9.x/edge  richard-hughes  -

```shell
fwupdmgr --version
$ fwupd.fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.12
compile   com.hughsie.libxmlb           0.3.6
compile   io.snapcraft.fwupd            5632
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.9.12
runtime   org.kernel                    6.5.0-14-generic
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
snap

<details>

<summary>**fwupd device information**</summary>

```shell
08:35:30.054 FuDebug              verbose to debug (on console 0)
08:35:30.054 FuEngine             starting fwupd 1.9.12...
08:35:30.054 FuMain               locked /run/lock/snap.fwupd/fwupdtool
08:35:30.054 GLib                 Failed to set thread scheduler attributes: Operation not permitted
08:35:30.056 FuMain               failed to stop daemon: failed to find snap.fwupd.fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit snap.fwupd.fwupd.service not loaded.
Loading…: 0%
08:35:30.056 FuConfig             config /snap/fwupd/5632/etc/fwupd/fwupd.conf is immutable
08:35:30.056 FuConfig             skipping mode check for /snap/fwupd/5632/etc/fwupd/fwupd.conf as not writable
08:35:30.056 FuConfig             trying to load config values from /snap/fwupd/5632/etc/fwupd/fwupd.conf
08:35:30.056 FuCommon             mapped file /snap/fwupd/5632/etc/fwupd/fwupd.conf of size 0x77
08:35:30.056 FuConfig             trying to load config values from /var/snap/fwupd/common/var/etc/fwupd/fwupd.conf
08:35:30.056 FuCommon             mapped file /var/snap/fwupd/common/var/etc/fwupd/fwupd.conf of size 0x77
08:35:30.056 FuConfig             ::configuration loaded
08:35:30.056 FuRemoteList         loading /snap/fwupd/5632/share/fwupd/metainfo
08:35:30.056 XbSilo               attempting to load /var/snap/fwupd/common/var/cache/fwupd/metainfo.xmlb
08:35:30.056 XbSilo               file: d05b1b86-a270-2dd4-66ac-bcb1c2741512, current:d05b1b86-a270-2dd4-66ac-bcb1c2741512, cached: (null)
08:35:30.057 XbSilo               loading silo with file contents
[... 878 lines cut ...]
FuUdevBackend:
  Name:                 udev
  Enabled:              true
  DoneSetup:            true
  CanInvalidate:        false
FuBluezBackend:
  Name:                 bluez
  Enabled:              true
  DoneSetup:            true
  CanInvalidate:        false
FuMediatekScalerPlugin:
  Name:                 mediatek_scaler
  Flags:                ready

Idle…: 0%
08:35:33.166 FuEngine             resetting update motd timeout
[31m[1mWARNING[0m: This package has not been validated, it may not work properly.
No detected devices
08:35:33.173 FuMain               No detected devices

08:35:33.174 FuPluginTest         destroy

```

</details>

hughsie, 2024-01-23:
I assume it's this?

> 08:35:33.143 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-2/drm_dp_aux1: failed to open /dev/i2c-13: Operation not permitted

superm1, 2024-01-23:
Isn't that what @valentindavid has in the PR for snapd fixed?

Try that artifact snapd.

CragW, 2024-01-23:
Verify fixed with the artifact snapd.
```
$ snap list snapd
Name   Version                    Rev  Tracking       Publisher  Notes
snapd  2.61.1+git1a18cdf.1a18cdf  x1   latest/stable  -          snapd
```

The device is seen.
```
$ sudo fwupd.fwupdtool get-devices --plugins mediatek_scaler
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
Dell Inc. OptiPlex AIO Plus 7420
│
└─Display Controller:
      Device ID:          7002ec1e191e016c9a498fd8450bd012598dba20
      Current version:    2.1.7
      Vendor:             Mediatek (PCI:0x1028)
      GUID:               f54c71fc-4404-5af9-87dd-6886fbd6af3b ← DISPLAY\VID_1028&PID_0C88&HWVER_2.1.2.1
      Device Flags:       • Internal device
                          • Updatable
                          • Cryptographic hash verification is available
                          • Unsigned Payload

```

closed 2024-01-23 by superm1