fwupd/fwupd #6727: lvfs firmware.xml.xz -vs- firmware.xml.gz

pjwelsh, 2024-01-23:
**Describe the bug**
The file firmware.xml.gz (https://cdn.fwupd.org/downloads/firmware.xml.gz) is not equal to firmware.xml.xz (https://cdn.fwupd.org/downloads/firmware.xml.xz). firmware.xml.gz seems to be from last year.

**Steps to Reproduce**
Steps to reproduce the behavior.
fwupdmgr --assume-yes enable-remote lvfs
(time goes by and fwupd has a change to /etc/fwupd/remotes.d/lvfs.conf and RPM creates /etc/fwupd/remotes.d/lvfs.conf.rpmnew  - rpmnew file uses firmware.xml.xz but fwupd continues to use firmware.xml.gz)

**Expected behavior**
Keep firmware.xml.xz and firmware.xml.gz? *.spec better define lvfs.conf to better handle possibly critical changes (overwrite)?

**fwupd version information**
fwupd-1.8.16-1.el9.alma.1.x86_64
fwupdmgr --version
compile   org.freedesktop.fwupd         1.8.16
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.6
runtime   org.freedesktop.fwupd-efi     1.8.16
compile   org.freedesktop.gusb          0.3.8
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.3.8
runtime   org.kernel                    5.14.0-362.13.1.el9_3.x86_64
runtime   org.freedesktop.fwupd         1.8.16

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
AlmaLinux 9.x RPM based system

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

**Additional questions**

- Operating system and version: Currently AlmaLinux 9.3
- Have you tried rebooting? Yes
- Is this a regression?

hughsie, 2024-01-23:
If you can work out the patch to the .spec file I'll gladly apply it upstream, which means it also trickles into Fedora, centos and RHEL.

pjwelsh, 2024-01-23:
It appears the .xml.gz is *not* current 2024 data. Is that by design? Or why are they different?

hughsie, 2024-01-23:
the .xz content has a suffix -- which makes it CDN friendly. This is why we download the .xml.xz.jcat file first, then open that to find the "true" filename, then download the actual metadata.

hughsie, 2024-01-23:
Which also makes me wonder -- what fwupd did you have originally -- and what are you updating to? We test RHEL pretty hard, but we're doing no testing on Alma at all.

pjwelsh, 2024-01-23:
Based on the age of the OS installs from a couple of systems checked, this
looks to have most likely been an Almalinux 9.2 original install and
subsequent dnf update to Almalinux 9.3. Nothing interesting as best I can
tell.

Historically, I have a /etc/yum.repos.d/*.rpmnew "fixer" script to handle
similar'ish issues. That script is very utilitarian since it effectively
looks for enabled/disabled changes via dnf related tools/options to decide
what to do...
However, from a perspective, removing the "config" tag from the fwupd.spec
could help ensure changes are mandatory and not discretionary. I already
have a check/enable lvfs repo function associated with my fwupd script so
having it "disabled" by default does not hurt me.

-- 

*Disclaimer*

The information contained in this communication from the 
sender is confidential. It is intended solely for use by the recipient and 
others authorized to receive it. If you are not the recipient, you are 
hereby notified that any disclosure, copying, distribution or taking action 
in relation of the contents of this information is strictly prohibited and 
may be unlawful.

This email has been scanned for viruses and malware, and 
may have been automatically archived. Cassens

hughsie, 2024-02-04:
> I have a /etc/yum.repos.d/*.rpmnew "fixer" script

@superm1 maybe we should save the remote.conf into /var like we now do fwupd.conf too?

superm1, 2024-02-04:
Sure

sathieu, 2025-03-08:
Are there specifications about the repo format somewhere?

NB: I plan to write a script to partially sync the repo for air-gapped usage.

hughsie, 2025-03-08:
> Are there specifications about the repo format somewhere?

It's supposed to be a "standard" PULP-style repo -- i.e. a directory of files with a PULP_MANIFEST index.

> NB: I plan to write a script to partially sync the repo for air-gapped usage.

You can use foreman already (that's what most Red Hat customers are using) -- there's also https://gitlab.com/fwupd/lvfs-website/-/blob/master/contrib/sync-pulp.py?ref_type=heads that might be helpful.

hughsie, 2025-03-08:
> for full mirror sync, and not partial mirror

How do you define partial? e.g. just specific appstream ID components?

sathieu, 2025-03-09:
@hughsie I would only keep files matching hardware ids returned by `fwupdtool hwids`  and device GUIDS from `fwupdtool get-devices` of all machines. `firmware.xml.xz` would be synced as-is. Searching for ids in `<provides><firmware type="flashed">`

`sync-pulp.py` as similar logic, but on tags, and only *Firmware for the Hughski ColorHug2 Colorimeter* has tags.

hughsie, 2025-03-09:
> sync-pulp.py as similar logic, but on tags

Would you like to contribute functionality to `sync-pulp.py` to add that? The `fwupdmgr get-devices --json` command outputs a json file, and maybe we could pass that to `sync-pulp.py` to only download GUIDs in that JSON dump. I think that's useful for other people too!

sathieu, 2025-03-15:
> Would you like to contribute functionality to `sync-pulp.py` to add that? The `fwupdmgr get-devices --json` command outputs a json file, and maybe we could pass that to `sync-pulp.py` to only [...]

Here it is: https://gitlab.com/fwupd/lvfs-website/-/merge_requests/1455

I was not able to create the GUID file from the JSON, but the doc should help.

closed 2025-05-03 by hughsie