fwupd/fwupd #6730: Failed to open cab file error when local install is executed on Snap version of fwupd

jkolako-solidigm, 2024-01-25:
Failed to open cab file error when local install is executed on Snap version of  fwupd 1.9.12
[image]

No issue with apt-get version 1.7.9:
[image]

**Steps to Reproduce**
1. Install snap version of fwupd
2. Execute local installation with cab file

**Expected behavior**
Cab file is opened and FW update process isperformed

**fwupd version information**
compile   org.freedesktop.fwupd         1.9.12
compile   com.hughsie.libxmlb           0.3.6
compile   io.snapcraft.fwupd            5643
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.9.12
runtime   org.kernel                    6.2.0-26-generic

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

**Additional questions**

- Operating system and version: Ubuntu 22.04.3 LTS
- Have you tried rebooting? Yes
- Is this a regression?

hughsie, 2024-01-25:
Does it work without sudo?

jkolako-solidigm, 2024-01-25:
without sudo "invalid handle" error is returned

superm1, 2024-01-25:
OK, I'm quite confused. I could reproduce this before, https://github.com/fwupd/fwupd/pull/6654 fixed it for me, but now I can reproduce it again too.

@valentindavid can you advise at all?

@jkolako-solidigm for now; here are the workarounds:

1. Copy the CAB file into this directory: ` /var/snap/fwupd/common/share/fwupd/remotes.d/vendor/firmware/`.  This will use the `vendor-directory` remote.  With this remote it automatically gets extracted and metadata built for it.  You can then run `fwupdmgr update`.
2. Copy the CAB into `/root` and install using `sudo fwupdmgr install /root/foo.cab`.
3. Use `fwupdtool` ie `sudo fwupdtool install foo.cab`.

valentindavid, 2024-01-25:
Are there apparmor errors in the journal? What are the connections to the snap (`snap connections --all`)?

superm1, 2024-01-25:
Here's the full sequence of info (reproduced using edge channel):

```
$ sudo snap install fwupd --channel edge
cp /snap/fwupd/current/share/installed-tests/fwupd/fakedevice124.cab .
sudo fwupd.fwupdtool enable-test-devices
sudo fwupd.fwupdtool get-devices --plugins test
sudo dmesg --clear
fwupd (edge) 1.9.10-293-g591533e41 from Richard Hughes installed
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
Successfully enabled test devices
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
AMD Birman-PHX
│
└─Integrated Webcam™:
      Device ID:          08d460be0f1f9f128413f816022a6439e0078018
      Summary:            Fake webcam
      Current version:    1.2.2
      Minimum Version:    1.2.0
      Bootloader Version: 0.1.2
      Vendor:             ACME Corp. (USB:0x046D)
      GUID:               b585990a-003e-5270-89d5-3705a17f9a43
      Device Flags:       • Updatable
                          • System requires external power source
                          • Supported on remote server
                          • Cryptographic hash verification is available
                          • Unsigned Payload
      Device Requests:    • Message

$ fwupdmgr install ./fakedevice124.cab
WARNING: This package has not been validated, it may not work properly.
invalid handle
```

```
$ sudo dmesg
[  101.280021] kauditd_printk_skb: 9 callbacks suppressed
[  101.280025] audit: type=1400 audit(1706197663.105:272): apparmor="DENIED" operation="open" class="file" profile="snap.fwupd.fwupdmgr" name="/etc/init.d/" pid=4797 comm="bash" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[  106.211381] audit: type=1400 audit(1706197668.037:273): apparmor="DENIED" operation="open" class="file" profile="snap.fwupd.fwupdmgr" name="/etc/gnutls/config" pid=4862 comm="gio-querymodule" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[  106.226683] audit: type=1400 audit(1706197668.053:274): apparmor="DENIED" operation="open" class="file" profile="snap.fwupd.fwupdmgr" name="/etc/gnutls/config" pid=4832 comm="fwupdmgr" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[  106.231939] audit: type=1326 audit(1706197668.057:275): auid=1000 uid=1000 gid=1000 ses=2 subj=snap.fwupd.fwupdmgr pid=4865 comm="pkttyagent" exe="/snap/fwupd/5646/usr/bin/pkttyagent" sig=0 arch=c000003e syscall=314 compat=0 ip=0x77fa24dd3a7d code=0x50000
[  106.236035] audit: type=1107 audit(1706197668.061:276): pid=888 uid=101 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_signal"  bus="system" path="/org/freedesktop/PolicyKit1/Authority" interface="org.freedesktop.PolicyKit1.Authority" member="Changed" name=":1.12" mask="receive" pid=4865 label="snap.fwupd.fwupdmgr" peer_pid=902 peer_label="unconfined"
                exe="/usr/bin/dbus-daemon" sauid=101 hostname=? addr=? terminal=?'
[  106.266659] audit: type=1326 audit(1706197668.093:277): auid=1000 uid=1000 gid=1000 ses=2 subj=snap.fwupd.fwupdmgr pid=4832 comm="fwupdmgr" exe="/snap/fwupd/5646/bin/fwupdmgr" sig=0 arch=c000003e syscall=314 compat=0 ip=0x78fae7c44a7d code=0x50000
[  106.401542] audit: type=1400 audit(1706197668.225:278): apparmor="DENIED" operation="open" class="file" profile="snap.fwupd.fwupd" name="/etc/gnutls/config" pid=4887 comm="fwupd" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
[  106.428639] audit: type=1326 audit(1706197668.253:279): auid=4294967295 uid=0 gid=0 ses=4294967295 subj=snap.fwupd.fwupd pid=4887 comm="fwupd" exe="/snap/fwupd/5646/libexec/fwupd/fwupd" sig=0 arch=c000003e syscall=314 compat=0 ip=0x7ee17e3aba7d code=0x50000
[  108.645272] audit: type=1400 audit(1706197670.469:280): apparmor="DENIED" operation="file_receive" class="file" profile="snap.fwupd.fwupd" name="/home/test/fakedevice124.cab" pid=4887 comm="gdbus" requested_mask="r" denied_mask="r" fsuid=0 ouid=1000
```

$ snap connections --all
```
Interface               Plug                                             Slot                                                 Notes
alsa                    firefox:alsa                                     -                                                    -
appstream-metadata      snap-store:appstream-metadata                    :appstream-metadata                                  -
audio-playback          firefox:audio-playback                           :audio-playback                                      -
audio-playback          steam:audio-playback                             :audio-playback                                      -
audio-record            firefox:audio-record                             :audio-record                                        -
audio-record            steam:audio-record                               -                                                    -
avahi-observe           firefox:avahi-observe                            :avahi-observe                                       -
bluez                   fwupd:bluez                                      :bluez                                               -
bluez                   steam:bluez                                      -                                                    -
browser-support         firefox:browser-sandbox                          :browser-support                                     -
[... cut ...]
x11                     multipass:x11                                    :x11                                                 -
x11                     snap-store:x11                                   :x11                                                 -
x11                     snapd-desktop-integration:x11                    :x11                                                 -
x11                     steam:x11                                        :x11                                                 -
```

ifishtoo, 2024-04-14:
my system is a dell optiplex 3060. A variation of the these instructions can be found here https://www.dell.com/support/kbdoc/en-us/000131486/update-the-dell-bios-in-a-linux-or-ubuntu-environment. 

 Steps that worked for me:
1) download cab file
2) Right click file and choose expand here
3)lsblk make note of dev/? that is listed for /boot/efi
4)sudo  mount /dev/? /media/my-user
5)sudo cp .cab-file /media/my-user/EFI/BOOT/filename #note:autocomplete does not work for this directory
6) reboot F12 and choose bios flash update
7) navigate to firmware file
8)follow instructions on screen
9) remount EFI/BOOT and rm the firmware file.
10) umount /dev/?

hughsie, 2024-11-11:
@superm1 you can still see this?

closed 2025-05-04 by hughsie