fwupd/fwupd #6732: Fail to update HP USB-C Dock G4 with the CAB file from LVFS: signatures were required

d4s, 2024-01-25:
**Describe the bug**
Device local update has been refused with the CAB file: https://fwupd.org/downloads/04a53b0567c68a8500e0ce2a5d3642e794ed8ad79c7c055df893c3a0c1627c19-Thunderbolt_Dock_G4_V1.4.20.0.cab 

Fail with the result:
```
firmware signature missing or not trusted; set OnlyTrusted=false in /etc/fwupd/fwupd.conf ONLY if you are a firmware developer,
```

And in logs I see:
```
2024-01-25T06:42:28.528511Z INFO fwupd[5948]: FuCabinet            failed to verify dmc.metainfo.xml: signatures were required, but none supplied
2024-01-25T06:42:28.528511Z INFO fwupd[5948]: FuCabinet            processing file: dmc.metainfo.xml
2024-01-25T06:42:28.528661Z INFO fwupd[5948]: FuCabinet            processing release: 1.4.20.0
2024-01-25T06:42:28.548009Z INFO fwupd[5948]: FuCabinet            failed to verify payload Thunderbolt_Dock_G4_V1.4.20.0.bin: signatures were required,  but none supplied
```

According to logs the CAB file is correctly downloaded from cdn.fwupd.org with the size 3120777:
```
01/24 22:42:27.066 DEBUG|    connectionpool:0433|  "GET /downloads/04a53b0567c68a8500e0ce2a5d3642e794ed8ad79c7c055df893c3a0c1627c19-Thunderbolt_Dock_G4_V1.4.20.0.cab?ts=1699349465 HTTP/1.1" 200 3120777
```

**Steps to Reproduce**
Steps to reproduce the behavior.
```
CACHE_DIRECTORY='/var/cache/fwupd' fwupdmgr local-install --json --allow-older --allow-reinstall https://fwupd.org/downloads/04a53b0567c68a8500e0ce2a5d3642e794ed8ad79c7c055df893c3a0c1627c19-Thunderbolt_Dock_G4_V1.4.20.0.cab 6e4824c963088596f7e4cfd12f7e0304db418baf
```

**Expected behavior**
Should try to update after download.

**fwupd version information**
Please provide the version of the daemon and client:
N/A, 1.9.10 as far as I remember. ChromeOS build: brya-release/R122-15752.0.0

hughsie, 2024-01-25:
@d4s what libjcat version is that please? I see:

    fwupdmgr get-details 04a53b0567c68a8500e0ce2a5d3642e794ed8ad79c7c055df893c3a0c1627c19-Thunderbolt_Dock_G4_V1.4.20.0.cab
    ....
        Release Flags:    • Trusted payload
                          • Trusted metadata
    ....

and the archive looks okay to me. Does chromeos have jcat-tool available?

d4s, 2024-01-25:
@hughsie have no this info in logs. Hopefully I have the target ChromeBook. Will try to install and check tomorrow.

hughsie, 2024-01-26:
If you do, this is what I'm trying to verify:

	$ gcab -x 04a53b0567c68a8500e0ce2a5d3642e794ed8ad79c7c055df893c3a0c1627c19-Thunderbolt_Dock_G4_V1.4.20.0.cab
	$ jcat-tool verify firmware.jcat --public-keys /etc/pki/fwupd-metadata/ --public-keys /etc/pki/fwupd
	Thunderbolt_Dock_G4_V1.4.20.0.bin:
	    PASSED sha1: OK
	    PASSED sha256: OK
	    PASSED gpg: 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
	    PASSED pkcs7: O=Linux Vendor Firmware Project,CN=LVFS CA
	dmc.metainfo.xml:
	    PASSED sha1: OK
	    PASSED sha256: OK
	    PASSED gpg: 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
	    PASSED pkcs7: O=Linux Vendor Firmware Project,CN=LVFS CA

d4s, 2024-01-26:
Heh... have no access to exact ChromeOS build version (mine is 15747.0.0), but you are right:

```
redrix-rev3 /tmp # gcab -x 04a53b0567c68a8500e0ce2a5d3642e794ed8ad79c7c055df893c3a0c1627c19-Thunderbolt_Dock_G4_V1.4.20.0.cab
redrix-rev3 /tmp # echo $?
0
redrix-rev3 /tmp # jcat-tool verify firmware.jcat --public-keys /etc/pki/fwupd-metadata/ --public-keys /etc/pki/fwupd
Thunderbolt_Dock_G4_V1.4.20.0.bin:
    PASSED sha1: OK
    PASSED sha256: OK
    FAILED gpg: Jcat engine kind 'gpg' not supported
    PASSED pkcs7: O=Linux Vendor Firmware Project,CN=LVFS CA
    FAILED: Validation failed
dmc.metainfo.xml:
    PASSED sha1: OK
    PASSED sha256: OK
    FAILED gpg: Jcat engine kind 'gpg' not supported
    PASSED pkcs7: O=Linux Vendor Firmware Project,CN=LVFS CA
    FAILED: Validation failed
Validation failed
```

hughsie, 2024-01-26:
> signatures were required,  but none supplied

Hmm, according to https://github.com/hughsie/libjcat/blob/main/libjcat/jcat-context.c you should have seen other debugging, e.g. `signature failure: %s", error_local->message` -- if you set `G_MESSAGES_DEBUG=all` do you see any of that?

d4s, 2024-01-26:
Checked additionally with ChromeOS 123-15758.0.0

```
# fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.10
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.8
compile   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.gusb          0.4.5
runtime   org.kernel                    5.15.147-21525-gade18b8e6c7f
runtime   org.freedesktop.fwupd         1.9.10
```

@hughsie should I set `G_MESSAGES_DEBUG=all` for `jcat-tool` like below?
```
# export G_MESSAGES_DEBUG=all
redrix-rev3 /tmp # jcat-tool verify firmware.jcat --public-keys /etc/pki/fwupd-metadata/ --public-keys /etc/pki/fwupd
(jcat-tool:9080): GLib-GIO-DEBUG: 18:43:53.942: _g_io_module_get_default: Found default implementation local (GLocalVfs) for ?gio-vfs?
Thunderbolt_Dock_G4_V1.4.20.0.bin:
(jcat-tool:9080): Jcat-DEBUG: 18:43:53.944: reading ./Thunderbolt_Dock_G4_V1.4.20.0.bin with 3113282 bytes
    PASSED sha1: OK
    PASSED sha256: OK
    FAILED gpg: Jcat engine kind 'gpg' not supported
(jcat-tool:9080): Jcat-DEBUG: 18:43:53.957: got 1 PKCS7 signatures
    FAILED pkcs7: failed to verify data for O=Linux Vendor Firmware Project,CN=LVFS CA: Public key signature verification has failed. [-89]
    FAILED: Validation failed
dmc.metainfo.xml:
(jcat-tool:9080): Jcat-DEBUG: 18:43:53.959: reading ./dmc.metainfo.xml with 2102 bytes
    PASSED sha1: OK
    PASSED sha256: OK
    FAILED gpg: Jcat engine kind 'gpg' not supported
(jcat-tool:9080): Jcat-DEBUG: 18:43:53.959: got 1 PKCS7 signatures
    FAILED pkcs7: failed to verify data for O=Linux Vendor Firmware Project,CN=LVFS CA: Public key signature verification has failed. [-89]
    FAILED: Validation failed
Validation failed
```

hm... looks like there are no keys for LVFS pre-installed:
```
# file /etc/pki/
/etc/pki/: cannot open `/etc/pki/' (No such file or directory)
```

superm1, 2024-01-26:
Are you building a subjproject'd libjcat?  This might be the same problem we just hit on main with subprojected builds that was failing CI.

superm1, 2024-01-26:
IOW it's caused by https://github.com/fwupd/fwupd/commit/db755512117b7ee0ca4170c9c2ee4c53332d1349.
Perhaps we should just backport https://github.com/fwupd/fwupd/commit/c51df0280b1aa2eafea9a9dba44d6ab0034b44e2 to 1_9_X too.

d4s, 2024-01-26:
@superm1 I see `libjcat` is a separate package in ChromeOS:
https://chromium.googlesource.com/chromiumos/overlays/chromiumos-overlay/+/refs/heads/main/dev-libs/libjcat/

superm1, 2024-01-26:
> @superm1 I see `libjcat` is a separate package in ChromeOS:
> https://chromium.googlesource.com/chromiumos/overlays/chromiumos-overlay/+/refs/heads/main/dev-libs/libjcat/
> 

Did the pkgconfig not get used with fwupd then?

d4s, 2024-01-26:
hm... thats odd.
pkgconfig is used both for `fwupd` and `libjcat` but `gpgme` is not available in build tree for some reason.

superm1, 2024-01-26:
> hm... thats odd.
> pkgconfig is used both for `fwupd` and `libjcat` but `gpgme` is not available in build tree for some reason.

And what version of libjcat?

d4s, 2024-01-26:
> And what version of libjcat?

I put some output above, atm it is v1.8.0

superm1, 2024-01-26:
Yup; this makes sense then. It's the same bug as the subproject bug.   You don't have this commit:
https://github.com/hughsie/libjcat/commit/b62808f56a0c2f60fa62e32163bbd04ebea78eff

So you fall back to the "default" value that it's in fwupd.  This happens to be 0, and you end up with that behavior.

The correct solution here is to upgrade your libjcat to 0.1.9 or later.

hughsie, 2024-01-26:
That seems plausible; but _how did it ever work_ before?

superm1, 2024-01-26:
> That seems plausible; but _how did it ever work_ before?

That patch that changed behavior is in 1.9.10. that's when stuff starts to explode.

superm1, 2024-01-26:
I say chrome should update libjcat and we should backport c51df02 to 1_9_X _both_.

hughsie, 2024-01-26:
Ahh! And this is only triggering because the libjcat version in chromeos is so old that it never supported `supported_pkcs7`. Indeed. Cherry-picked and pushed.

hughsie, 2024-01-26:
@d4s can you either convince google to backport 08a4a8653a2bcea2cb4c1139b8bce2525d45b37a into fwupd or to update the libjcat version please.

d4s, 2024-01-26:
> > That seems plausible; but _how did it ever work_ before?
> 
> That patch that changed behavior is in 1.9.10. that's when stuff starts to explode.

hm... that's interesting. Please check [my comment](https://github.com/fwupd/fwupd/issues/6732#issuecomment-1911938540) -- `fwupd` version is 1.9.9.

superm1, 2024-01-27:
Well in 1.9.9 you're missing gpg. So there are two problems.

hughsie, 2024-01-27:
> Well in 1.9.9 you're missing gpg

IIRC, ChromeOS only ever used PKCS#7.

d4s, 2024-02-19:
After updating to libjcat 0.1.14 the issue disappeared.

closed 2024-02-19 by d4s