fwupd/fwupd #6734: redfish: About fwupd 1.9.12 version

vincent-tang1, 2024-01-26:
**Describe the question**
following this release comparing changes log, 
```
https://github.com/fwupd/fwupd/compare/1.9.12...main
```
it included below commit about  'redfish: Do not request the Advantech BMC to reboot'
```
https://github.com/fwupd/fwupd/commit/70632221dee8f9510f5897b2214755acbeed441b
```

but in fact, i downloaded this release code and check it
```
https://github.com/fwupd/fwupd/releases/tag/1.9.12
```

as result, seems like this commit  have not being merged into this release.

not sure if this is your expectation? thanks

hughsie, 2024-01-26:
Ohh, I didn't know this commit needed backporting. For future, pull requests with the "needs-backport" label are backported manually. I've one this now, and pushed to 1_9_X --- i'll be in the next 1.9.x release -- sorry.

closed 2024-01-26 by hughsie

vincent-tang1, 2024-01-26:
it's ok, thanks your help :)