fwupd/fwupd #6740: redfish : continously update firmware twice in some conditions

vincent-tang1, 2024-01-26:
**Precondition**

Our server have two BMC flash, one is for normal usage, the other is for backup purpose, see below command
```
root@FWA-6172-efi:/var# ipmitool hpm check

...

--------------------------------------------------------------------------
|ID  | Name        |                     Versions                        |
|    |             |     Active      |     Backup      |      Deferred   |
--------------------------------------------------------------------------
|   0| BL          |   0.30 00000000 | ---.-- -------- | ---.-- -------- |
|   1| BMC         |   0.99 00000000 |   0.99 00000000 | ---.-- -------- |
|   2| BMCONF      |   1.02 00000000 | ---.-- -------- | ---.-- -------- |
|*  3| FPGA        |   0.12 00000000 | ---.-- -------- | ---.-- -------- |
|*  4| BIOS        |   1.00 00000000 |   1.00 00000000 | ---.-- -------- |
|*  5| NVRAM       |   3.00 00000000 | ---.-- -------- | ---.-- -------- |
--------------------------------------------------------------------------
(*) Component requires Payload Cold Reset
```

Here is our firmware page on LVFS website
https://fwupd.org/lvfs/devices/com.advantech.fwa6172.10598c6b.bmc.firmware

Here is our linux bmc device from fwupd
```
root@FWA-6172-efi:/var# fwupdmgr get-devices
WARNING: This package has not been validated, it may not work properly.
Advantech FWA-6172
│
...
├─FWA6172 Platform Management:
│     Device ID:          62c1cd95692c5225826cf8568a460427ea3b1827
│     Summary:            Redfish host bus adapter
│     Current version:    0.99.0000
│     Vendor:             Advantech (REDFISH:ADVANTECH)
│     GUIDs:              10598c6b-6f24-518d-9065-bb948baa4907 ← REDFISH\VENDOR_Advantech&SOFTWAREID_894feb65-2ca8-4378-a5d3-41c78ff5b59f-BMC
│                         69ec5e09-0466-5679-83f5-0dc75e3ff4d5 ← REDFISH\VENDOR_Advantech&ID_BMC
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│
...
```

And fwupd use snap to install with --channel-edge
```
root@FWA-6172-efi:/var# snap list
Name               Version                  Rev    Tracking         Publisher       Notes
bare               1.0                      5      latest/stable    canonical✓      base
core22             20231123                 1033   latest/stable    canonical✓      base
firefox            122.0-2                  3687   latest/stable    mozilla✓        -
fwupd              1.9.10-319-gb24c81245    5651   latest/edge      richard-hughes  -
gnome-42-2204      0+git.ff35a85            141    latest/stable    canonical✓      -
gnome-firmware     45.0-16-g1e19cec         8      latest/stable    superm1         -
gtk-common-themes  0.1-81-g442e511          1535   latest/stable    canonical✓      -
lxd                5.19-8635f82             26200  latest/stable/…  canonical✓      -
snap-store         41.3-71-g709398e         959    latest/stable    canonical✓      -
snapd              2.61.1+git1663.ge016f71  20995  latest/edge      canonical✓      snapd
```

**Describe the question**

My question is, when execute "fwupdmgr update" command, it will  continously update firmware twice
```
root@FWA-6172-efi:/var# fwupdmgr update
WARNING: This package has not been validated, it may not work properly.
Devices with no available firmware updates:
 • BIOS
 • BL
 • BMCONF
 • FPGA
 • NVRAM
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade FWA6172 Platform Management from 0.99.0000 to 1.01.0000?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This unstable release fixes the following issues:                            ║
║                                                                              ║
║ • Fix bug for test 1                                                         ║
║ • Improve performance for test 1                                             ║
║                                                                              ║
║ Some new functionality has also been added:                                  ║
║                                                                              ║
║ • Add support for test 1                                                     ║
║                                                                              ║
║ FWA6172 Platform Management and all connected devices may not be usable      ║
║ while updating.                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y
Restarting device…       [*******************                    ]

==> when first updated successfully, wait for BMC reboot and login, after that will continously update again

Restarting device…       [************************************** ] Less than one minute remaining…

==> when ​secondary updated successfully , finally system show below message and exit it

Successfully installed firmware

```
As result, we can see both BMC flashes have been updated (0.99 -> 1.01)
```
root@FWA-6172-efi:/var# ipmitool hpm check

...

--------------------------------------------------------------------------
|ID  | Name        |                     Versions                        |
|    |             |     Active      |     Backup      |      Deferred   |
--------------------------------------------------------------------------
|   0| BL          |   0.30 00000000 | ---.-- -------- | ---.-- -------- |
|   1| BMC         |   1.01 00000000 |   1.01 00000000 | ---.-- -------- |
|   2| BMCONF      |   1.02 00000000 | ---.-- -------- | ---.-- -------- |
|*  3| FPGA        |   0.12 00000000 | ---.-- -------- | ---.-- -------- |
|*  4| BIOS        |   1.00 00000000 |   1.00 00000000 | ---.-- -------- |
|*  5| NVRAM       |   3.00 00000000 | ---.-- -------- | ---.-- -------- |
--------------------------------------------------------------------------
(*) Component requires Payload Cold Reset

```

**Q1 : Not sure if this is your exepectation, or this is a bug ?**
**Q2 : Do you know if fwupd have such a mechinism based on some condition ?**

Not only "fwupdmgr update" command, but also belwo command still lead to update twice situation
```
fwupdmgr reinstall
fwupdmgr downgrade
```

However..............................
If use following cases, above phenomenon will not happen , means it only update once.
1.  run 'snap restart fwupd' and then first execute "fwupdmgr update", it only update fw once. (But after execute the commmand for second time, above phenomenon still happen)
2. if use locally update "fwupdmgr install xxxxx.cab", it will always update once
3. if use "fwupdtool update", it will always update once

hughsie, 2024-01-26:
> ==> when first updated successfully, wait for BMC reboot and login, after that will continously update again

Does the BMC backup export via redfish as a separate fwupd device? If you can grab the `fwupd -vv` logs when the double update is happening that would help a lot.

vincent-tang1, 2024-01-26:
> Does the BMC backup export via redfish as a separate fwupd device? 

i think this should no export from my side, because redfish.quirk in source code is no this kind flag in Advantech GUID. And in firmware metada file, i have no adding this device flag, you can download CAB file as below link to check it.
https://fwupd.org/lvfs/devices/com.advantech.fwa6172.10598c6b.bmc.firmware

> If you can grab the fwupd -vv logs when the double update is happening that would help a lot.

you means this command "fwupdmgr update -vv"?

vincent-tang1, 2024-01-26:
> Does the BMC backup export via redfish as a separate fwupd device? 

i think no export BMC backup via redfish, you can see below two URI
```
GET /redfish/v1/UpdateService/FirmwareInventory
{
	"@odata.type":"#SoftwareInventoryCollection.SoftwareInventoryCollection",
	"@odata.context":"/redfish/v1/$metadata#SoftwareInventoryCollection.SoftwareInventoryCollection",
	"@odata.id":"/redfish/v1/UpdateService/FirmwareInventory",
	"@odata.etag":"5875ba4f0b3711de37e9",
	"Name":"Firmware Collection",
	"Members@odata.count":6,
	"Members":[
		{
			"@odata.id":"/redfish/v1/UpdateService/FirmwareInventory/BL"
		},
		{
			"@odata.id":"/redfish/v1/UpdateService/FirmwareInventory/BMC"
		},
		{
			"@odata.id":"/redfish/v1/UpdateService/FirmwareInventory/BMCONF"
		},
		{
			"@odata.id":"/redfish/v1/UpdateService/FirmwareInventory/FPGA"
		},
		{
			"@odata.id":"/redfish/v1/UpdateService/FirmwareInventory/BIOS"
		},
		{
			"@odata.id":"/redfish/v1/UpdateService/FirmwareInventory/NVRAM"
		}
	]
}

GET /redfish/v1/UpdateService/FirmwareInventory/BMC
{
	"@odata.type":"#SoftwareInventory.v1_3_0.SoftwareInventory",
	"@odata.context":"/redfish/v1/$metadata#SoftwareInventory.SoftwareInventory",
	"@odata.id":"/redfish/v1/UpdateService/FirmwareInventory/BMC",
	"@odata.etag":"b8555e2165cc1ae8ab94",
	"Id":"BMC",
	"Name":" BMC",
	"SoftwareId":"894feb65-2ca8-4378-a5d3-41c78ff5b59f-BMC",
	"Updateable":true,
	"Manufacturer":"Advantech",
	"Version":"1.01.0000"
}
```

and fwupd get-device as
```
root@FWA-6172-efi:/var/snap# fwupdmgr get-devices
WARNING: This package has not been validated, it may not work properly.
Advantech FWA-6172
│
├─BIOS:
│     Device ID:          562313e34c756a05a2e878861377765582bbf971
│     Summary:            Redfish host bus adapter
│     Current version:    1.00.0000
│     Vendor:             Advantech (REDFISH:ADVANTECH)
│     GUIDs:              467a7e47-2db8-5400-9fd1-8915df8e01d4 ← REDFISH\VENDOR_Advantech&SOFTWAREID_894feb65-2ca8-4378-a5d3-41c78ff5b59f-BIOS
│                         84591cf7-9cee-5641-99d5-11af7ab46ecd ← REDFISH\VENDOR_Advantech&ID_BIOS
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│
├─BL:
│     Device ID:          92b9cdf108e011d817c6a97824eab088f9dd1194
│     Summary:            Redfish host bus adapter
│     Current version:    0.30.0000
│     Vendor:             Advantech (REDFISH:ADVANTECH)
│     GUIDs:              208e148d-02da-52df-8458-857871ff624e ← REDFISH\VENDOR_Advantech&SOFTWAREID_894feb65-2ca8-4378-a5d3-41c78ff5b59f-BL
│                         ac7965d0-0781-5aa1-951d-4a263bb7a68b ← REDFISH\VENDOR_Advantech&ID_BL
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│
├─BMCONF:
│     Device ID:          234a9bdfc3540c1797d9693fe5389f3486b5942c
│     Summary:            Redfish host bus adapter
│     Current version:    1.02.0000
│     Vendor:             Advantech (REDFISH:ADVANTECH)
│     GUIDs:              2326661c-41db-5760-9443-5b6dec414802 ← REDFISH\VENDOR_Advantech&SOFTWAREID_894feb65-2ca8-4378-a5d3-41c78ff5b59f-BMCONF
│                         a92c90ab-448f-53e9-b0dc-ff6741f13b0a ← REDFISH\VENDOR_Advantech&ID_BMCONF
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│
├─FPGA:
│     Device ID:          c15823b7bd2b3abb8ec782bb77c941359a49fd1f
│     Summary:            Redfish host bus adapter
│     Current version:    0.12.0000
│     Vendor:             Advantech (REDFISH:ADVANTECH)
│     GUIDs:              70b760f5-078b-5281-84d5-a6b7531a4965 ← REDFISH\VENDOR_Advantech&SOFTWAREID_894feb65-2ca8-4378-a5d3-41c78ff5b59f-FPGA
│                         d53fb01b-be1c-572e-8e88-1711a52083b1 ← REDFISH\VENDOR_Advantech&ID_FPGA
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│
├─FWA6172 Platform Management:
│     Device ID:          62c1cd95692c5225826cf8568a460427ea3b1827
│     Summary:            Redfish host bus adapter
│     Current version:    1.01.0000
│     Vendor:             Advantech (REDFISH:ADVANTECH)
│     Update State:       Success
│     GUIDs:              10598c6b-6f24-518d-9065-bb948baa4907 ← REDFISH\VENDOR_Advantech&SOFTWAREID_894feb65-2ca8-4378-a5d3-41c78ff5b59f-BMC
│                         69ec5e09-0466-5679-83f5-0dc75e3ff4d5 ← REDFISH\VENDOR_Advantech&ID_BMC
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│
├─NVRAM:
│     Device ID:          e62d2a3a9c875839f7b57843190f695bca7f222f
│     Summary:            Redfish host bus adapter
│     Current version:    3.00.0000
│     Vendor:             Advantech (REDFISH:ADVANTECH)
│     GUIDs:              30a77683-3cfe-5fe8-b2ed-b89f0ba2a16d ← REDFISH\VENDOR_Advantech&SOFTWAREID_894feb65-2ca8-4378-a5d3-41c78ff5b59f-NVRAM
│                         afe3785d-abe4-546c-8889-1d3fbe16e6c1 ← REDFISH\VENDOR_Advantech&ID_NVRAM
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│
────────────────────────────────────────────────

```

vincent-tang1, 2024-01-26:
> If you can grab the fwupd -vv logs when the double update is happening that would help a lot.

could you guide me how to setup "fwupd -vv" and  grab this log ? thanks

superm1, 2024-01-26:
For the snap, the easy way is to modify `/etc/systemd/system/snap.fwupd.fwupd.service` and add `--verbose --verbose` to the `ExecStart` line.
Then run `systemctl daemon-reload`
Then `systemctl restart snap.fwupd.fwupd`
Then run the tool.

Your logs will show up in `journalctl -u snap.fwupd.fwupd.service` then.

Another way is to modify `/var/snap/fwupd/common/var/etc/fwupd/fwupd.conf` to add `VerboseDomains='*'` and then `systemctl restart snap.fwupd.fwupd`.

vincent-tang1, 2024-01-29:
> If you can grab the fwupd -vv logs when the double update is happening that would help a lot.

Run fwupmgr command **for first time** after restart fwupd. As result, only update firmware once, see log
```
# systemctl restart snap.fwupd.fwupd
# fwupdmgr reinstall
```
[0129-log1.log](https://github.com/fwupd/fwupd/files/14082272/0129-log1.log)

But, run fwupmgr command **for second times** (or later more times). As result, update firmware twice, see log
 ```
# fwupdmgr reinstall
```
[0129-log2.log](https://github.com/fwupd/fwupd/files/14082285/0129-log2.log)

whatever "fwupdmgr reinstall" or "fwupdmgr update" or "fwupdmgr downgrad", they have same above result. thanks

hughsie, 2024-01-29:
> But, run fwupmgr command for second times (or later more times). As result, update firmware twice, see log

Comparing those, I see in the first log:

	FuIdle               status now busy
	FuEngine             composite update 1: 62c1cd95692c5225826cf8568a460427ea3b1827 (/redfish/v1/UpdateService/FirmwareInventory/BMC: 0)
	FuEngine             install phase now composite-prepare

and in the second:

	FuIdle               status now busy
	FuEngine             composite update 1: 62c1cd95692c5225826cf8568a460427ea3b1827 (/redfish/v1/UpdateService/FirmwareInventory/BMC: 0)
	FuEngine             composite update 2: 62c1cd95692c5225826cf8568a460427ea3b1827 (/redfish/v1/UpdateService/FirmwareInventory/BMC: 0)
	FuEngine             install phase now composite-prepare

so installing the firmware twice is actually working as expected! The question is why we're detecting two composite updates. Either one or of these must be happening:

 * We're detecting more than one component from the archive, or detecting more than one metainfo file in the archive (I don't think it's this)
 * We're calling `fu_daemon_install_with_helper()` multiple times somehow. @vincent-tang1 can you see if we call `fu_daemon_install_with_helper()` more than once by putting in a `g_warning()` somewhere?
 * `fu_engine_get_devices_by_composite_id()` is returning more than one device for some reason
 * Multiple devices are matching in the `devices_possible` for loop (`10598c6b-6f24-518d-9065-bb948baa4907` only seems to be found on the BMC device)
 * There are multiple devices added with the same device ID -- which I didn't think the daemon allowed. @vincent-tang1 can you grab us the *entire* log of `sudo fwupdtool --plugins redfish get-devices -vv` (using whatever path snap gives you) so we can test this theory.
 
Thanks for reporting and helping us to debug this.

vincent-tang1, 2024-01-29:
> There are multiple devices added with the same device ID -- which I didn't think the daemon allowed. @vincent-tang1 can you grab us the entire log of sudo fwupdtool --plugins redfish get-devices -vv (using whatever path snap gives you) so we can test this theory.

please refer below log
[0129-log3.log](https://github.com/fwupd/fwupd/files/14085550/0129-log3.log)

> We're calling fu_daemon_install_with_helper() multiple times somehow. @vincent-tang1 can you see if we call fu_daemon_install_with_helper() more than once by putting in a g_warning() somewhere?

Or could you give me snap package you build (so you can add any log you want to) ? 
I also can easily apply it using `snap install <your_package>` and grap the logs for you

superm1, 2024-01-29:
> Or could you give me snap package you build (so you can add any log you want to) ?

@hughsie if you want to do it this way a draft PR with whatever debug would spit out a snap artifact now that can be used like this.

(It also spits out RPM/DEB for people to use too)

vincent-tang1, 2024-01-30:
> please refer below log [0129-log3.log](https://github.com/fwupd/fwupd/files/14085550/0129-log3.log)

not sure if this log is helpful for you ?

> Or could you give me snap package you build (so you can add any log you want to) ? I also can easily apply it using `snap install <your_package>` and grap the logs for you

Not sure if you would be willing to build a snap artifact containing the required logs and then provide me to grab ? ^^

hughsie, 2024-01-30:
Okay, I might have a fix, testing much appreciated ^^

vincent-tang1, 2024-01-31:
> [Only look for active devices in the daemon](https://github.com/fwupd/fwupd/commit/6b48abf851f1ed4846c6c4554be43b2ae81f7c8d) …

After applied your patch and follow [Building and Debugging](https://github.com/fwupd/fwupd/blob/main/docs/building.md) method in README to operate. As result, it does work and fine, seems like this issue is fixed.

I also want to test it in snap side, could you help to build it for me to verify because i do not know how to build this sanp version ? thanks!

hughsie, 2024-01-31:
@vincent-tang1 I think https://github.com/fwupd/fwupd/actions/runs/7715126801/job/21029209937?pr=6749 might have the CLI, e.g. grab `https://productionresultssa7.blob.core.windows.net/actions-results/3031cf38-b134-4c0a-8c14-394a1bd52f09/workflow-job-run-1f602c08-6376-5fae-6c91-b105014dca3a/artifacts/1c27324f013705cbf0c49f3d3bb103c7069ed5988889d426c10cfc25684f0a31.zip` and then `sudo snap install --dangerous fwupd_1.9.10-330-gd83e2aefa_amd64.snap`

closed 2024-01-31 by hughsie

vincent-tang1, 2024-01-31:
thanks for your efforts

seems like i have no permission to access this file
```
~# wget https://productionresultssa7.blob.core.windows.net/actions-results/3031cf38-b134-4c0a-8c14-394a1bd52f09/workflow-job-run-1f602c08-6376-5fae-6c91-b105014dca3a/artifacts/1c27324f013705cbf0c49f3d3bb103c7069ed5988889d426c10cfc25684f0a31.zip
--2024-01-31 14:11:29--  https://productionresultssa7.blob.core.windows.net/actions-results/3031cf38-b134-4c0a-8c14-394a1bd52f09/workflow-job-run-1f602c08-6376-5fae-6c91-b105014dca3a/artifacts/1c27324f013705cbf0c49f3d3bb103c7069ed5988889d426c10cfc25684f0a31.zip
Resolving productionresultssa7.blob.core.windows.net (productionresultssa7.blob.core.windows.net)... 52.239.172.36, 20.150.82.228, 20.150.88.228
Connecting to productionresultssa7.blob.core.windows.net (productionresultssa7.blob.core.windows.net)|52.239.172.36|:443... connected.
HTTP request sent, awaiting response... 409 Public access is not permitted on this storage account.
2024-01-31 14:11:30 ERROR 409: Public access is not permitted on this storage account..
```

However after i use `snap install fwupd --channel=edge` command, it can work and as result as fxied it, thank you ^^