fwupd/fwupd #6742: Update 8BITDO SF30 gamepad - failed to wait for detach replug

flaphoschi, 2024-01-28:
**Describe the question**
Hello!

I'm trying to update an *8BITDO* [SF30](https://download.8bitdo.com/Manual/Controller/N30+F30/N30+F30_Manual_V4.pdf?20220513) from firmware *2.67* to *2.70* with fwupd. It instructs me to unplug the controller and reboot it with pressing <kbd>L+R+Start</kbd> for 3 seconds. I'm using the short original calbe deliverd with the gamepad. If fails:

```
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade SFC30 Joystick from 2.67 to 2.70?                                    ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Support remapping the buttons by the RemapTools on Gamepad Mode.             ║
║                                                                              ║
║ SFC30 Joystick and all connected devices may not be usable while updating.   ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Action Required: Unplug the controller, hold down L+R+START for 3 seconds until both LEDs are flashing then reconnect the controller.
Waiting…                 [                                       ]
failed to wait for detach replug: device 983c3cffc6fd36d32b00b62928d30721eaeb93db did not come back
```

Steps to reproduce
1. Plugin gamepad # blue led blinks
2. `fwupdmgr update`
3. Unplug gamepad
4. Press <kbd>L+R+Start</kbd> for 3 seconds # no reboot, gamepad turns instead off
5. Press <kbd>L+R+Start</kbd> for 3 seconds # green and blue led on, both blinking
6. Operation fails

**fwupd version information**
```
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.12
compile   com.hughsie.libxmlb           0.3.15
compile   org.freedesktop.Passim        0.1.5
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
runtime   org.freedesktop.Passim        0.1.5
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    6.7.1-arch1-1
runtime   org.freedesktop.fwupd         1.9.12
```
**daemon seems to be automatically launched**
```
$ systemctl status fwupd.service 
● fwupd.service - Firmware update daemon
     Loaded: loaded (/usr/lib/systemd/system/fwupd.service; static)
     Active: active (running) since Sun 2024-01-28 12:21:41 CET; 1h 34min ago
       Docs: https://fwupd.org/
   Main PID: 2036 (fwupd)
      Tasks: 6 (limit: 36978)
     Memory: 36.5M (peak: 39.0M)
        CPU: 17.222s
     CGroup: /system.slice/fwupd.service
             └─2036 /usr/lib/fwupd/fwupd
```

**gamepad details**
``` ├─SFC30 Joystick:
│     Device ID:          983c3cffc6fd36d32b00b62928d30721eaeb93db
│     Summary:            A redesigned classic game controller
│     Current version:    2.67
│     Vendor:             8BitDo (USB:0x1235)
│     GUIDs:              043c7673-bd45-5e2e-b8b9-c22df2dfec2a
│                         b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e
│                         f94d3231-f6e1-5ef3-a4a0-dc819d74ae54 ← USB\VID_1235&PID_AB21
│     Device Flags:       • Updatable
│                         • Supported on remote server
│                         • Unsigned Payload
│     Device Requests:    • Message (custom)
```

What I'm doing wrong?
I stumpled across #565 but that was fixed a long time ago.

Thank you

hughsie, 2024-01-29:
> hold down L+R+START for 3 seconds until both LEDs are flashing

Did you get the LEDs to flash?

flaphoschi, 2024-01-29:
Yes. But only after doing it twice i.e. step number 5.

flaphoschi, 2024-02-02:
@hughsie 
Maybe I've a hint. Or my understanding of USB is wrong.

When I plug in the gamepad and call `lsusb`:
```
Bus 006 Device 021: ID 1235:ab21 Focusrite-Novation SFC30 Joystick
```
We see here the `idVendor` and `idProduct`. We don't see "8BitDo" because the `idVendor` is assigned to *Focusrite-Novation* which is followed by *SFC30 Joystick*.

Which is weird because *8BitDo* has an own `idVendor` value and also an `idProduct` assigned to `ab21`:

File `/usr/share/hwdata/usb.ids`
```
2dc8  8BitDo
        5006  M30 Bluetooth gamepad
        5750  Bootloader
        6000  SF30 Pro gamepad
        6001  SN30/SF30 Pro gamepad
        ab11  F30 gamepad
        ab12  N30 gamepad
        ab20  SN30/SF30 gamepad
        ab21  SF30 gamepad
```
But `lsusb` is showing "SFC30 Joystick" not "SF30 gamepad". Therefore I guess `lsusb` uses another source for the device name?

```
Bus 006 Device 022: ID 1235:ab21 Focusrite-Novation SFC30 Joystick
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1235 Focusrite-Novation
  idProduct          0xab21 SFC30 Joystick
  bcdDevice            0.01
  iManufacturer           1 SFC30             
  iProduct                2 SFC30 Joystick
  iSerial                 0 
  bNumConfigurations      1
```

Yep. It seems to use the value stored at `iProduct` which seems to be an index into data provided by the device itself. A plain `lsusb` lists therefore `bus:device idVendor_from_device:idProduct_from_device idVendor_from_file_usb_ids  iProduct_from_device`. Right?

And now the actual interesting part. Remember that I've to press <kbd>L+R+Start</kbd> to get into the update mode.

The SFC30/FC30 now identifies itself as:
```
Bus 006 Device 023: ID 0483:5750 STMicroelectronics LED badge -- mini LED display -- 11x44
```
🤪

This happens with two gamepads. I'm now not surprised that the update fails.
I will ask 8BitDo to take look itself.

For reference I will attach pictures of a gamepad because 8BitDo seems to use "Something30" in confusing ways:
[image]
[image]

flaphoschi, 2024-02-02:
https://github.com/fwupd/8bitdo-firmware/commit/b1a9079545592609caa3935689411d8f86548403

This is the only hint I could found in the repository:
```
   <!-- USB\VID_1235&PID_AB11 (old firmware) -->
```

hughsie, 2024-02-02:
Are you hitting this perhaps? https://github.com/fwupd/fwupd/blob/main/plugins/ebitdo/ebitdo-legacy.quirk

flaphoschi, 2024-02-03:
Thank you! It worked :slightly_smiling_face: 
In lack of knowledge I copied the file `ebitdo-legacy.quirk` to `/usr/share/quirks.d` and executed `# systemctl restart fwupd.service`[1]. The following update with `fwupdmgr update` went flawlessly and the gamepad is now on firmware `2.70` (previously `2.67`).

Follow up question
The command `fwupd get-updates` doesn't offer me a further upgrade to the `4.xx` release of the firmware. Is that intentional?

[1] Not sure if necessary. Maybe I didn't unplugged, pressed <kbd>L+R+Start</kbd> (twice...) and replugged fast enough.

hughsie, 2024-02-03:
> The command fwupd get-updates doesn't offer me a further upgrade to the 4.xx release of the firmware

Have you tried enabling the lvfs-testing remote?

flaphoschi, 2024-02-04:
That's it. Thank you! 

**Documentation**
If your 8BitDo *SFC30* gamepad is using the *legacy bootloader* it will present the wrong `idVendor` when in update mode.

1. Save the [ebitdo-legacy.quirk](https://github.com/fwupd/fwupd/blob/main/plugins/ebitdo/ebitdo-legacy.quirk) file in `/usr/share/quirks.d`
2. `systemctl restart fwupd.service` # not sure if needed
3. execute command to update i.e. `fwupdmgr {get-devices, refresh, get-updates, update}`
4. When requested to press <kbd>L+R+Start</kbd> you need it do to twice with the SFC30

If you want to upgrade to the newest firmware `4.20` you need to enable `lvfs-testing`:
Edit file `/etc/fwupd/remotes.d/lvfs-testing.conf` and set the key `Enabled` to `true`. Then execute all steps from above.

**Important**
Remove the quirk file after the update to avoid confusion with other devices using the same value for `idVendor`. It seems to be a common mistake by various vendors. And if you enabled *lvfs-testing* revert it back to the default i.e. `false`. You likely want to avoid using firmware from testing on productive systems.

closed 2024-02-04 by flaphoschi