fwupd/fwupd #6750: PS175 and RTD242 is not being detected after v1.9.8

rishabh-chromium, 2024-01-30:
**Describe the bug**
PS175 and RTD242 is not being probed correctly or not detected after 1.9.8 when I run `fwupdmgr get-devices`

**Steps to Reproduce**
* check out `v1.9.9`
* deploy on dut
* Run `fwupdmgr get-devices`
* <verify `RTD2142` and `PS175` isn't visible>

**Expected behavior**
Both devices should be detected

**fwupd version information**
1.9.8

```shell
fwupdmgr --version
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

compile   org.freedesktop.fwupd         1.9.8                                                                                                                                                                      
compile   com.hughsie.libxmlb           0.3.11                                                                                                                                                                     
compile   com.hughsie.libjcat           0.1.8                                                                                                                                                                      
compile   org.freedesktop.gusb          0.4.5                                                                                                                                                                      
runtime   org.freedesktop.gusb          0.4.5                                                                                                                                                                      
runtime   org.kernel                    4.19.306-15851-gd32c1abf5b8c                                                                                                                                               
runtime   org.freedesktop.fwupd         1.9.8    

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

```
HP Noibat
?
??Unknown Device:
?     Device ID:          b5881e53bf55f4e454a88a354e1f9432e5780820
?     GUID:               e6b305ba-40e8-5bcd-8283-db8acb711fcc ? DRM\VEN_FL_&DEV_2701
?   
??DA4032:
?     Device ID:          eafa35acc9a754da25475e870cca0b4460daa4ed
?     Current version:    0x3535303430393139
?     Vendor:             EMMC:0x000045
?     Serial Number:      0xe5cc6019
?     GUIDs:              cd0cd165-b9b4-521b-bebf-49a7d263a2c5 ? EMMC\NAME_DA4032
?                         d5927774-7e99-5ffe-ad3d-5946a78ce9d1 ? EMMC\MAN_0045&OEM_0100&NAME_DA4032
?                         93cf76ac-f6e6-58fa-bc94-afb8f595d0c4 ? EMMC\MAN_0045&NAME_DA4032&REV_0x3535303430393139
?                         c657049b-c758-5403-ac91-c3502f43e047 ? EMMC\MAN_0045&OEM_0100&NAME_DA4032&REV_0x3535303430393139
?                         5500994c-1b73-52b4-acf4-bb574b6f3029 ? EMMC\0069&0256&DA4032
?     Device Flags:       ? Internal device
?                         ? Updatable
?                         ? Supported on remote server
?   
??Display controller:
      Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
      Current version:    02
      Vendor:             PCI:0x8086
      GUIDs:              62f450c5-c503-5539-bd71-ebf0968533ed ? PCI\VEN_8086&DEV_9B21
                          f57faad5-ffa4-5358-934e-7f2351ead82e ? PCI\VEN_8086&DEV_9B21&SUBSYS_80862212
      Device Flags:       ? Internal device
                          ? Cryptographic hash verification is available
```

I also ran `sudo fwupdtool --verbose --no-timestamp --plugins parade-lspcon get-devices` in v1.9.7 and v1.9.8 and here are the outputs

[... 109 lines cut ...]
  CanInvalidate:        false                                                                            
FuUdevBackend:                                      
  Name:                 udev                                                                             
  Enabled:              true                                                                             
  DoneSetup:            true         
 CanInvalidate:        false                                                                            
FuBluezBackend:                                     
  Name:                 bluez                                                                            
  Enabled:              true                                                                             
  DoneSetup:            true                                                                             
  CanInvalidate:        false                                                                            
FuParadeLspconPlugin:                               
  Name:                 parade_lspcon                                                                    
  Flags:                ready                                                                            

FuEngine             resetting update motd timeout        
```

</details>

**Additional questions**

- Operating system and version: ChromeOS
- Have you tried rebooting?
- Is this a regression?

superm1, 2024-01-30:
To make sure it's not a bug already fixed in 1.9.10 or later, can you reproduce on an up to date 1_9_X branch please, and provide `--verbose --verbose` output?

superm1, 2024-01-30:
Or alternatively, can you bisect 1.9.7 to 1.9.8 to find what introduced the breakage?

rishabh-chromium, 2024-01-30:
I was at top of tree which shoes up as 2.0.0. I still got the same result as 1.9.8 with almost identical `verbose` message.
```
FuDebug              verbose to info (on console 1)                                                                                                                                                       [15/2650]
FuEngine             starting fwupd 2.0.0?                                                                                                                                                                         
Loading?                 [ -                                     ]FuRemoteList         loading remote from /etc/fwupd/remotes.d/fwupd-peripherals.conf                                                             
FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf                                                                                                                                    
FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf                                                                                                                                            
FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf                                                                                                                                
FuRemoteList         enabled remotes: fwupd-peripherals[1000], vendor-directory[1000], lvfs[0]                                                                                                                     
Loading?                 [                                       ]FuEngine             client certificate exists and working                                                                                       
Loading?                 [*                                      ]FuQuirks             loading quirks from /usr/share/fwupd/quirks.d                                                                               
FuQuirks             loading quirks from /var/lib/fwupd/quirks.d                                                                                                                                                   
Loading?                 [*                                      ]FuContext            SMBIOS Manufacturer=HP                                                                                                      
FuContext            SMBIOS EnclosureKind=3                                                                                                                                                                        
FuContext            SMBIOS Family=Google_Hatch                                                                                                                                                                    
FuContext            SMBIOS ProductName=Noibat                                                                                                                                                                     
FuContext            SMBIOS ProductSku=sku83886083                                                                                                                                                                 
FuContext            SMBIOS BiosVendor=coreboot                                                                                                                                                                    
FuContext            SMBIOS BiosVersion=Google_Noibat.13324.71.0                                                                                                                                                   
FuContext            SMBIOS BiosMajorRelease=00                                                          
FuContext            SMBIOS BiosMinorRelease=00                                                          
FuContext            SMBIOS FirmwareMajorRelease=00                                                      
FuContext            SMBIOS FirmwareMinorRelease=00                                                      
FuContext            SMBIOS BaseboardManufacturer=HP                                                     
FuContext            SMBIOS BaseboardProduct=Noibat                                                      
FuContext            failed to load fdt: cannot find /sys/firmware/fdt or override /var/lib/fwupd/system.dtb
FuContext            added udev subsystem watch of firmware-attributes                                                                                                                                             
Loading?                 [**                                     ]FuEngine             loading metadata for remote 'vendor-directory'
FuEngine             ignoring: /usr/share/fwupd/remotes.d/vendor/firmware/README.md                                                                                                                                
FuEngine             using /usr/share/fwupd/remotes.d/vendor/firmware/ee85c8b40e538cd41ecbdd0d10ef7ef6f48c12eddffe1c934c988ef8f77f375e-MCEM131.cab as metadata source
FuEngine             ignoring: Error opening directory ?/var/lib/fwupd/local.d?: No such file or directory                                                                                                         
[... cut ...]
  Flags:                ready                                                                            
                                                                                                         
FuEngine             resetting update motd timeout                                                       
                                                    
No detected devices                                                                                      
FuMain               No detected devices        
```

superm1, 2024-01-30:
As a guess is this caused by c068d08c5 perhaps?  Can you build c068d08c5~1 and c068d08c5 and confirm?

hughsie, 2024-01-31:
@rishabh-chromium hmm, i'll dig out hardware now -- I can't reproduce when using a dummy i2c name on my Lenovo Laptop. Can you get the `sudo fwupdtool -vv --no-timestamp --plugins parade-lspcon get-devices` log for v1.9.7 please -- the `-vv` is the same as `--verbose --verbose` -- and that will be huge so please attach it rather than pasting inline. Thanks.

hughsie, 2024-01-31:
So, the conclusion I've come to is that the old code never worked properly! I did the correct thing in:

	-if (!fu_device_build_instance_id_quirk(device, error, "I2C", "NAME", "FAMILY", NULL))
	+if (!fu_device_build_instance_id_full(device, FU_DEVICE_INSTANCE_FLAG_QUIRKS, error, "I2C", "NAME", "FAMILY", NULL))

i.e. we were using the `I2C\NAME_1AF80175:00&FAMILY_Google_Hatch` just for quirks (and not to match to firmware). The only GUID that was visible (and able to match to firmware) was `I2C\NAME_1AF80175:00` -- but that's a *generic* ID that's no supposed to be allowed to match firmware. But! That's exactly what https://fwupd.org/lvfs/firmware/8270 does -- it matches the generic ID -- but thankfully it was never pushed to public else it would have matched any i2c device (on any platform) with that generic name.

So what should we do? First, we need to apply https://github.com/fwupd/fwupd/pull/6752 and backport that to the stable branch. Then I think we need to modify the existing firmware to use  `I2C\NAME_1AF80175:00&FAMILY_Google_Hatch` rather than `I2C\NAME_1AF80175:00` -- it's only had two downloads and so I think it's more of a proof-of-concept than actual used firmware. I can do that after the PR is approved and merged.

But, _interestingly_ -- the PR that added `FU_DEVICE_INTERNAL_FLAG_NO_GENERIC_GUIDS` was added to find **PRECISELY** this kind of issue -- e.g. where Kevin was worried that we might accidentally match the wrong custom firmware for one vendor against a generic device. So although I'm sorry there was a regression -- I'm super glad we found the use of a generic GUID [almost] in the wild.

closed 2024-01-31 by hughsie

rishabh-chromium, 2024-01-31:
Stupid question, I don't fully understand why we needed to make that change to `parade_lspcon` and `realtek_mst` plugins. Couldn't we just change the firmware to use the correct GUID format?
Also, I don't have access to that firmware link. I am assuming we still need to change the firmware to use the non generic GUID?

hughsie, 2024-01-31:
>  I don't fully understand why we needed to make that change to parade_lspcon and realtek_mst plugins

Essentially, this changes a "_only use for quirk matching_ GUID into a allow _quirk, plus also allow matching against firmware_ GUID as the old GUID was common to all platforms, and [correctly] being hidden. The issue before was that there were no defined matching GUIDs for the device, and the daemon rightfully got upset.

> I am assuming we still need to change the firmware to use the non generic GUID?

I've added it just now. Do you want to be in the Google LVFS group too?

rishabh-chromium, 2024-01-31:
Okay that makes sense. Yes, I would love to be on the group. Do we need to change something on our side? Firmware version bump?

hughsie, 2024-01-31:
> Yes, I would love to be on the group

I've given you ACLs.

> Firmware version bump?

Nah, it's all automatic.

rishabh-chromium, 2024-01-31:
> I've given you ACLs.
Thank you

We probably need to change the RTD2142 firmware as well?

hughsie, 2024-01-31:
Good catch; https://fwupd.org/lvfs/firmware/8024 updated too.