fwupd/fwupd #6755: Failure to update system firmware of HP EliteBook 840 G9

stouffer, 2024-02-01:
**Describe the bug**
I have Ubuntu 22.04 installed on an HP EliteBook 840 G9. The system informed me that a firmware update was available for my laptop from version 1.02 to 1.09. When I ran `fwupdmgr get-updates` it also listed 1.081. However, when I tried to install the update to 1.09 using fwupd 1.7.9 (available from apt) I only ever saw a "fwupd-efi version 1.4" message and then the system immediately rebooted back into the OS. When I would run `fwupdmgr get-results` it would show that the update failed.

From the HP website, I was able to find many intermediate versions of the firmware (1.03, 1.06, 1.07, and 1.08), so I instead tried their suggested path. This included copying the .bin files to \boot\efi\EFI, using an external USB drive, etc. None of these ended up working. (I'm aware that is *not* the fault of fwupd, but I wanted to add this detail since it feels relevant.)

After a lot of failed attempts, I realized that the version of fwupd available as a snap is much more recent (1.9.2) so I purged the version installed by apt and installed the snap. The snap version also offered 1.09, but in addition listed 1.04, 1.05, 1.06, 1.07, 1.081, and 1.09. When I tried to upgrade to 1.09 using fwupd 1.9.2 I still obtained the same failure as before.

It then occurred to me that I might have success if I were able to start at the earliest version of the firmware (1.04) and update through the sequence. To even try this path, I needed to use `fwupdmgr block-firmware CHECKSUM` for all recent checksums and then unblock these one at a time using `fwupdmgr unblock-firmware CHECKSUM`. Lo and behold, if I left 1.04 as the only unblocked version, `fwupdmgr upgrade` succeeded. So I then went one-by-one and eventually made it to a successful install of the latest version of the firmware.

I was never able to determine the reason behind the original bug except that perhaps the BIOS was unable internally to go from 1.03 to 1.09. It's also possible that fwupd 1.7.9 could have succeeded once I had made it to 1.081. However, the need to go through the step-by-step process made me think that it would be extremely nice if it was more straightforward for a user to directly do

```shell
fwupdmgr upgrade GUID CHECKSUM
```

rather than having to muck around with `block-firmware` and `unblock-firmware`. So I suppose this is both a bug report and a feature request.

As I've listed most versions above, I've left the details below empty. However, I'm happy to go back and add those later if it could prove useful!

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

hughsie, 2024-02-01:
I've asked HP for their input. I'll let you know what I hear. It should be possible to upgrade directly to the latest, but doing the updates one at a time is something we support in fwupd, but it has to be "opted-in".

WMP, 2024-09-12:
I have this same, i need to download manually this file: https://fwupd.org/lvfs/devices/com.hp.laptop.system.U71.firmware version Version 0x1040000 and install it by `fwupdmgr install 1e0649ecdea92e9e2da48eb5db8f4c9c68911b11b1f137bc40357f4b8cb66d44-U71_01040000.cab`. I then installed the same firmware Version 0x1040100, although I don't know if this was needed. Then after I ran fwupdmgr update and the latest firmware Version 0x10c0100 was installed

closed 2024-11-11 by hughsie