fwupd/fwupd #6757: Config handling is both fragile and too complicated

hughsie, 2024-02-01:
I've got an _idea_. The basic idea:

Just have two (intially empty, or missing) directories in `/etc/fwupd/conf.d/` and `/var/lib/fwupd/conf.d`

The man file generator reads from the local `data/99-defaults.conf`

The distro installs `/etc/fwupd/conf.d/99-defaults.conf` and gets sorted last.

fwupd reads using FuConfig from `/var/lib/fwupd/conf.d`, then `/etc/fwupd/conf.d/` in reverse alpha order -- every duplicate key=value is overwritten.

When saving values from the daemon we write to `/etc/fwupd/conf.d/00-custom.conf` if writeable, or `/var/lib/fwupd/conf.d/custom.conf` otherwise.

If `/etc/fwupd/fwupd.conf` exists then it's **moved** into `/etc/fwupd/conf.d/00-custom.conf`
If `/var/lib/fwupd/fwupd.conf` exists then it's **moved** into `/var/lib/fwupd/conf.d/00-custom.conf`

If `/etc/fwupd/redfish.conf` exists then it's migrated (i.e. stripping default values) into `/etc/fwupd/conf.d/50-migrated-redfish.conf`
If `/etc/fwupd/thunderbolt.conf` exists then it's migrated into `/etc/fwupd/conf.d/50-migrated-thunderbolt.conf`

Am I missing something on why that wouldn't both work better than what we have now (allowing per-distro presets to be dropped in) and also waaaaay less complicated?

hughsie, 2024-02-01:
@superm1

superm1, 2024-02-01:
> fwupd reads using FuConfig from /var/lib/fwupd/conf.d, then /etc/fwupd/conf.d/ in reverse alpha order -- every duplicate key=value is overwritten.

I think this is backwards in your idea.  You should read /etc/ first (under the presumption it's immutable) followed by `/var/etc`.  Otherwise you can never override something hardcoded in (for example) SilverBlue or snap.

>write to /etc/fwupd/conf.d/00-custom.conf if writeable, or /var/lib/fwupd/conf.d/custom.conf otherwise

I think another thing that would help is to stop the daemon touching `/etc`.  Leave that for the sysadmin.  Let the daemon only touch `/var/etc`.

> Am I missing something on why that wouldn't both work better than what we have now (allowing per-distro presets to be dropped in)

I'm still not convinced we want per-distro presets.  Thinking about that `IdleTimeout` case I think it's a really good general argument for headless systems that only sparingly interact with the fwupd daemon when the motd says to do so to not have it running a lot.  We don't need to have a whole special custom configuration for that.  There's the heuristics we discussed in that other bug, which I think go a LONG way.

>  and also waaaaay less complicated?

I don't really think so - now as an end user I need to manually parse (*exaggeration*) 15 configuration files and figure out which key is in which and if I update `00-foo.conf` but `50-bar.conf` has the same value it doesn't work because it's higher priority?  It seems crazy to me.

I think we have it right with KISS.  In that vain - here's another crazy idea to simplify things.  What if `/etc/fwupd/fwupd.conf` is just a symlink to `/var/etc/fwupd/fwupd.conf` by default?

superm1, 2024-02-01:
> and also waaaaay less complicated?

Another thought - we dropped all conf options by default in main! Do we really _need_ it that complicated?

hughsie, 2024-02-02:
> I think another thing that would help is to stop the daemon touching /etc

I think that's a great idea. https://github.com/fwupd/fwupd/pull/6759

closed 2024-02-04 by hughsie