fwupd/fwupd #6782: fwupdate not deleting cap file

Batwam, 2024-02-07:
**Describe the bug**
fwupdate doesn't appear to delete cap file after update. My boot/efi partition is unfortunately very small and I am having to clean things up before being able to upgrade as there is otherwise not enough space on it. While this is a separate issue, when I looked at what files I could remove, I noticed that a fwupd cap file was still there.
[image]

**Steps to Reproduce**
Update Dell Optiplex firmware

**Expected behavior**
leftover cap file should be deleted if no longer required to save space

**fwupd version information**
Please provide the version of the daemon and client.

```shell
compile   org.freedesktop.fwupd         1.9.5
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.5
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.4.5
runtime   org.freedesktop.fwupd         1.9.5
runtime   org.kernel                    6.5.0-15-generic
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): apt

<details>

<summary>**fwupd device information**</summary>

Output of the fwupd devices recognized in your system:
```shell
Dell Inc. OptiPlex 7060
│
├─Unknown Device:
│     Device ID:          4a4907dbb1b96c6a1177dfd1b95eb41c781d1265
│     GUID:               1199a818-4c52-5137-b536-d59e2e2cada9 ← GPIO\ID_INT3450:00
│   
├─CoffeeLake-S GT2 [UHD Graphics 630]:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              36d6f0eb-c9e4-50ce-a5ae-88e528c3180b ← PCI\VEN_8086&DEV_3E92
│                         821c08db-2e4b-5fd6-bf34-49ee34e2a341 ← PCI\VEN_8086&DEV_3E92&SUBSYS_1028085A
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Core™ i5-8500T CPU @ 2.10GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f4
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         809a0b93-8a12-5338-a571-ad5583acf896 ← CPUID\PRO_0&FAM_06&MOD_9E
│                         72ec2ff3-49ff-5ec2-bdbb-525badd47543 ← CPUID\PRO_0&FAM_06&MOD_9E&STP_A
│     Device Flags:       • Internal device
│   
├─KXG60ZNV256G NVMe TOSHIBA 256GB:
│     Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Summary:            NVM Express solid state drive
│     Current version:    10604107
│     Vendor:             Toshiba Corporation (NVME:0x1179)
│     Serial Number:      595A28PSK00N
│     GUIDs:              8a412b95-5c07-5ce4-b63a-f9cbf254dd5d ← STORAGE-DELL-107443
│                         344c2804-4948-e811-842f-0ed5f89f718b
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
[... 105 lines cut ...]
      Current version:    RQR12.10_B0032
      Bootloader Version: BOT01.02_B0015
      Vendor:             HIDRAW:0x046D|USB:0x046D
      Install Duration:   30 seconds
      GUIDs:              9d131a0c-a606-580f-8eda-80587250b8d6
                          279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
      Device Flags:       • Updatable
                          • Supported on remote server
                          • Unsigned Payload
    
────────────────────────────────────────────────
Devices that have been updated successfully:
 • KXG60ZNV256G NVMe TOSHIBA 256GB (10604106 → 10604107)
 • UEFI dbx (217 → 371)
 • System Firmware (1.27.0 → 1.28.0)

```

</details>

**Additional questions**

- Operating system and version: Ubuntu 23.10
- Have you tried rebooting? yes
- Is this a regression? possibly since the files are meant to be deleted based on #431

hughsie, 2024-02-07:
This was fixed in 1.9.7 with b4111a9b70b78100e831c891daa203564b5739bd -- can you try that or newer please.

superm1, 2024-02-07:
please re-open if it still happens on newer fwupd
/close

closed 2024-02-07 by superm1

Batwam, 2024-02-07:
Sounds good. Next distro release in Apr should be 1.9.12 so I’ll wait for that and report back if the issue persists

Batwam, 2024-07-05:
I can confirm that this is now solved