fwupd/fwupd #6802: fwupd daemon doesnt poke offline/rescan ports

krankun, 2024-02-14:
**Describe the bug**
at boot the daemon should port pokes offline and rescan for each host router in 
/sys/bus/thunderbolt/devices/0-0/usb4_port1/offline
/sys/bus/thunderbolt/devices/0-0/usb4_port1/rescan 

/sys/bus/thunderbolt/devices/1-0/usb4_port1/offline
/sys/bus/thunderbolt/devices/1-0/usb4_port1/rescan 

**Steps to Reproduce**
boot the dut and see if retimer objects appear and disappear 
**Expected behavior**
retimer is enumerated
**fwupd version information**

rex-rev2 ~ # fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.10
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.1.14
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.1.14
runtime   org.freedesktop.gusb          0.4.5
runtime   org.kernel                    6.1.77-11622-g1d93a98343c1
runtime   org.freedesktop.fwupd         1.9.10

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

**Additional questions**

- Operating system and version:
- Have you tried rebooting?
- Is this a regression?
 Yes

hughsie, 2024-02-14:
Got the `-vv` daemon log please?

krankun, 2024-02-14:
[fwupd_full_log.txt](https://github.com/fwupd/fwupd/files/14288678/fwupd_full_log.txt)
daemon with fwupd -vv full log

hughsie, 2024-02-14:
I don't see that 2nd link -- it's 0 bytes, no?

krankun, 2024-02-14:
[fwupd_full_log.txt](https://github.com/fwupd/fwupd/files/14288706/fwupd_full_log.txt)

hughsie, 2024-02-14:
I see:

	21:26:54.593 FuDevice             using 65115f27e304a5dd08de4aba78d2d8627ebc542a for 1ab68780-21a7-dac2-ffff-ffffffffffff
	21:26:54.593 FuPluginThunderbolt  missing nvm_version attribute
	21:26:54.593 FuPluginThunderbolt  Could not find non-volatile memory location
	21:26:54.593 FuEngine             thunderbolt ignoring: updates are distributed as part of the platform

Is that expected?

krankun, 2024-02-14:
missing nvm_version attribute from /sys/bus/thunderbolt/devices/0-0/ is expected

hughsie, 2024-02-14:
> missing nvm_version attribute from /sys/bus/thunderbolt/devices/0-0/ is expected

Okay, makes sense. I did expect to see a thunderbolt _controller_ tho. What's going on there?

hughsie, 2024-11-11:
I think we've got this now; it at least works on the tbt4 system I have here. We still need emulation data of course.

closed 2024-11-11 by hughsie