fwupd/fwupd #6806: X1 yoga gen 4: Unable to update. Getting error: Failed to run update on reboot: expected 0.1.19 and got 0.1.18

shaked8634, 2024-02-15:
**Describe the bug**
When trying to update Embedded Controller or System Firmware, firmware seems to be downloaded and saved in ..,/fw/
Then after reboot to _fwupdx64.efi_, it starts the EFI, then system reboots right away (using OPAL disk encryption, no LUKS).
Firmware is not updated and I can see in `fwupd get-history` message for the 2 firmware binaries like this:

> failed to run update on reboot: expected 0.1.48 and got 0.1.40

0.1.48 (and 0.1.19) is the one I want to flush.

**Steps to Reproduce**
Execute `fwupdmgr upgrade` and reboot

**Expected behavior**
That firmware should be updated

**fwupd version information**
```shell
fwupdmgr --version                                                                                                                                           [0]
compile   com.hughsie.libxmlb           0.3.15
compile   org.freedesktop.Passim        0.1.5
compile   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.fwupd         1.9.13
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.Passim        0.1.5
runtime   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    6.7.4-arch1-1
runtime   org.freedesktop.fwupd         1.9.13
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Installed using pacman

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices                                                                                                                      [0]
LENOVO 20SA000GUS
│
├─Unknown Device:
│     Device ID:          59a11bc1795b6b83caa7eface68d0ad302cf5f86
│     GUID:               0422c218-5b8a-504d-8fba-43f8a29f4312 ← GPIO\ID_INT34BB:00
│   
├─B140QAN02.2:
│     Device ID:          aec1a869eb0df71b7cea6b3ac71d39b830faf164
│     GUID:               9eca33b6-1116-559b-a61a-23dd0b45f4f3 ← DRM\VEN_AUO&DEV_2236
│     Device Flags:       • Internal device
│   
├─CometLake-U GT2 [UHD Graphics]:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    02
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              5589d313-88e5-54d3-8f4c-b2ee3882d4bc ← PCI\VEN_8086&DEV_9B41
│                         33d24107-5e36-5432-867b-627ee477f9da ← PCI\VEN_8086&DEV_9B41&SUBSYS_17AA2292
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Core™ i7-10510U CPU @ 1.80GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f8
│     Vendor:             Intel
│     GUIDs:              561403e8-143a-5071-ab09-bf5e4c146983 ← CPUID\PRO_0&FAM_06&MOD_8E
│                         d5092916-1868-5182-8c70-12f83dab6372 ← CPUID\PRO_0&FAM_06&MOD_8E&STP_C
│     Device Flags:       • Internal device
│   
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
[... 278 lines cut ...]
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── initramfs-linux-lts-fallback.img
├── initramfs-linux-lts.img
├── intel-ucode.img
├── vmlinuz-linux
└── vmlinuz-linux-lts

7 directories, 18 files
```

**Additional questions**

- Operating system and version:
Arch Linux kernel: 6.7.4
- Have you tried rebooting?
Yes
- Is this a regression?
Yes, I was able to install firmwares more than a year ago
- Are you using an NVMe disk?
Yes
- Is secure boot enabled?
No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?
No

hughsie, 2024-02-15:
> and saved in ..,/fw/

What's the absolute path please? That looks weird. Do you also see the "fwupd-efi 1.4" text on reboot?

shaked8634, 2024-02-16:
Full path: /efi/EFI/arch/fw/

```bash
cat /etc/os-release
NAME="Arch Linux"
PRETTY_NAME="Arch Linux"
ID="arch"
BUILD_ID=rolling
ANSI_COLOR="38;2;23;147;209"
HOME_URL="https://archlinux.org/"
DOCUMENTATION_URL="https://wiki.archlinux.org/"
SUPPORT_URL="https://bbs.archlinux.org/"
BUG_REPORT_URL="https://gitlab.archlinux.org/groups/archlinux/-/issues"
PRIVACY_POLICY_URL="https://terms.archlinux.org/docs/privacy-policy/"
LOGO=archlinux-logo
```

Regarding the message following boot the fw firmware, I see the line with: the message "fwupd-efi 1.4" and then the next message is resetting

hughsie, 2024-02-16:
> I see the line with: the message "fwupd-efi 1.4" and then the next message is resetting

It looks like fwupd is doing everything it needs to do; could you try turning off secure boot in the BIOS? I'm wondering if this is something arch-linux specific.

shaked8634, 2024-02-16:
Secure boot and boot order lock are disabled.

shaked8634, 2024-02-16:
OK, I set `EnableEfiDebugging=true` and received the following output:
[image]

hughsie, 2024-02-16:
That looks entirely okay to me. @mrhpearson is there any way to debug this further? We seem to be setting up the capsule, it just fails to run.

mrhpearson, 2024-02-16:
From the efibootmgr output - I'm not seeing BootNext to Linux-Firmware-Updater. Isn't that needed to run?

@shaked8634 - anything displayed on the screen (even briefly) during the reboot? What happens if you go to the F12 boot menu and select the firmware-updater to run?

Mark