fwupd/fwupd #6807: failed to build result for 362301da643102b9f38477387e2193e57abaa590 (similar to issue #6101 I believe)

tanj9, 2024-02-16:
**Describe the bug**
I have recently upgraded my linux distro (from Zorin 16 Core  - based on Ubuntu 20 - to Zorin 17 Core - based on Ubuntu 22). Following that upgrade, I now have the following error message:
```
Voici les détails des erreurs du gestionnaire de paquets :
failed to build result for 362301da643102b9f38477387e2193e57abaa590
```

**Steps to Reproduce**
It happens automatically each time I start my computer.

**Expected behavior**
No error message.

**fwupd version information**
Please provide the version of the daemon and client.
not sure about this but I have a line `systemd-fwupd───4*[{fwupd}]` when I run `pstree` in my terminal.
```shell
fwupdmgr --version
```
output of fwupdmgr --version:
```
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.5.0-17-generic
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10
```
Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
I think  it came with the installation of Zorin 17's associated Gnome software. Otherwise in general I tend to use apt to install packages / softwares.

<details>

**<summary>fwupd device information</summary>**

Please provide the output of the fwupd devices recognized in your system.
```shell
fwupdmgr get-devices --show-all-devices
```
```
➜  fwupd fwupdmgr get-devices --show-all-devices
HP Pavilion Laptop 15-eh0xxx
│
├─HP Wide Vision HD Camera:
│     Device ID:          1f914e619b232dc4da68ec3389b98b20163003a2
│     Current version:    0.5
│     Vendor:             Quanta (USB:0x0408)
│     GUIDs:              1f13475a-7c1a-5f08-960c-1c22028d670b
│                         42b80f85-b452-532f-9928-cf600db7c7b2
│     Device Flags:       • Updatable
│   
├─KBG40ZNV1T02 KIOXIA:
│     Device ID:          3743975ad7f64f8d6575a9ae49fb3a8856fe186f
│     Summary:            NVM Express solid state drive
│     Current version:    HP00AE00
│     Vendor:             KIOXIA Corporation (NVME:0x1E0F)
│     GUIDs:              64c62d8c-7734-5f3f-8731-57e46cd314fd
│                         1eb5a892-ab5f-5404-9c09-667eba636664
│                         13747a71-316d-5f1e-a28c-3f8505a362fb
│                         ed8b4112-c3b7-5a8c-ac78-555ab3126306
│                         ace4dd7d-a902-5996-a0f2-9bf3fd6bc5f9
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│   
├─Ryzen 7 4700U with Radeon Graphics:
[... 161 lines cut ...]
          "KernelName" : "Linux",
          "RuntimeVersion(org.kernel)" : "5.15.0-46-generic",
          "CompileVersion(org.freedesktop.gusb)" : "0.3.4",
          "RuntimeVersion(com.dell.libsmbios)" : "2.4",
          "DistroId" : "zorin",
          "CpuArchitecture" : "x86_64",
          "BootTime" : "1663657851",
          "HostVendor" : "HP",
          "EfivarNvramUsed" : "33253",
          "HostProduct" : "HP Pavilion Laptop 15-eh0xxx",
          "SecureBoot" : "Enabled",
          "RuntimeVersion(org.freedesktop.fwupd)" : "1.7.5",
          "KernelVersion" : "5.15.0-46-generic"
        }
      ]
    }
  ]
}
```
</details>

And I attach my \var\lib\fwupd\pending.db file:
[pending.zip](https://github.com/fwupd/fwupd/files/14309158/pending.zip)

A final word: I would like to thank you in advance for helping me finding a solution to this issue. I know it is similar to a previous issue (#6101) that have been closed for a long time and hope I am not bothering by creating this issue.

tanj9, 2024-02-16:
A last comment: I have a dual boot windows / linux Zorin (but I have the intention to get rid of it in the coming months).

superm1, 2024-02-16:
I believe you need to ask your distro to update to fwupd 1.9.4 or later to pick up that fix.

closed 2024-02-16 by superm1

tanj9, 2024-02-23:
Thanks @superm1. Well noted.