fwupd/fwupd #6808: fwupd and Xen causes CPU hang

hackerfactor, 2024-02-16:
**Describe the bug**
I have a detailed description of the problem at: https://www.hackerfactor.com/blog/index.php?/archives/1021-The-Jitter-Bug.html

The basic summary: Running Xen on Ubuntu 18.04 - 22.04, occasionally one of the guest VMs (domu) would experience a CPU hang. If the VM has multiple CPUs, then one would be pinned at 100% usage but doing nothing and the others would be working fine. kernel.log shows a failure:
```
kernel: [2333839.516291] RIP: e030:zap_pte_range.isra.0+0x168/0x860
kernel: [2333839.516298] Code: 00 10 00 00 e8 c9 f6 ff ff 49 89 c0 48 85 c0 74 0b 48 83 7d 88 00 0f 85 0a 06 00 00 41 f6 47 20 01 0f 84 d7 02 00 00 4c 8b 23 <48> c7 03 00 00 00 00 4d 39 6f 10 4c 89 e8 49 0f 46 47 10 4d 39 77
kernel: [2333839.516316] RSP: e02b:ffffc9004114ba60 EFLAGS: 00010202
kernel: [2333839.516322] RAX: ffffea000192fe40 RBX: ffff88807854e760 RCX: 0000000000000125
kernel: [2333839.516330] RDX: 0000000000000000 RSI: 00007f2410aec000 RDI: 00000000135f9125
kernel: [2333839.516339] RBP: ffffc9004114bb10 R08: ffffea000192fe40 R09: 0000000000000000
kernel: [2333839.516347] R10: 0000000000000001 R11: 000000000000073f R12: 00000000135f9125
kernel: [2333839.516355] R13: 00007f2410aec000 R14: 00007f2410aed000 R15: ffffc9004114bc48
kernel: [2333839.516371] FS: 00007f2410bf1580(0000) GS:ffff88807d500000(0000) knlGS:0000000000000000
kernel: [2333839.516380] CS: e030 DS: 0000 ES: 0000 CR0: 0000000080050033
kernel: [2333839.516387] CR2: 00007f2410ae1151 CR3: 0000000002a0a000 CR4: 0000000000040660
kernel: [2333839.516398] Fixing recursive fault but reboot is needed!
```
After a few hours (or days), another VM CPU would experience a failure. If all of the CPUs on the VM failed, then the last one never even has the chance to log to kernel.log.

On the hypervisor (dom0), there are no reports of the problem. dom0 can always destroy the VM and restart it.

The problem is inconsistent and irregular. I finally tracked it down to fwupd running on dom0. On Ubuntu, this process runs twice a day with a random offset of 12 hours. This is why the crashes seem to happen at random times. The CPU could hang up to 10 minutes after fwupd runs.

Stopping fwupd seems to have resolved the problem.

**Steps to Reproduce**

1. Install Ubuntu 20.04 LTS server (18.04 or 22.04 also work; I haven't tested any of the non-LTS version, but I suspect that they work too).
2. Install xen.
3. Create a couple of VMs. Use 2 or more CPUs per VM.  Leave 1 CPU for dom0.
4. Give the CPUs something to do -- busywork that keeps the load at least 50% on all CPUs.
5. Wait 60 days. On my servers (Dell R320 and R340 servers), the problem is irregular. It appears to be a race condition, but at least one CPU would experience the problem within 30 days.  A search for the reboot message on Google shows other people with similar messages and they report failures as often as daily. I think the frequency is a combination of fwupd, xen, OS, and hardware.

**Expected behavior**
The CPU should not hang.

**fwupd version information**
On Ubuntu 20.04, it's fwupd 1.7.9-1~20.04.3  (the version from the Ubuntu software repositories). It was installed using apt during the basic server installation (part of the default installation).

**Additional questions**
- Operating system and version: Ubuntu 20.04 LTS, but also seen on 18.04 LTS and 22.04 LTS.
- Have you tried rebooting? Yes. In fact, the only resolution is to restart the VM. Dom0 never experiences this problem, but rebooting dom0 doesn't make the hangs go away.
- Is this a regression? No.

The problem appears to be a deadlock race condition. When dom0 runs fwupd, it can lock a CPU, causing one or more domu to hang and fail. My solution to this problem is to disable fwupd:
```
sudo systemctl stop fwupd fwupd-refresh fwupd-refresh.timer
sudo systemctl disable fwupd fwupd-refresh fwupd-refresh.timer
sudo systemctl mask fwupd fwupd-refresh fwupd-refresh.timer
```

hackerfactor, 2024-02-16:
Hi superm1,

I have 4 servers, but they are all in use right now. (I usually have 3 servers running in production and one spare for testing, but the spare is running a long-term indexing of content right now. The last time I did this indexing, it too about 2 months, and I'm 15 days into it.) It will be a few months before I can spin up a new dom0 for testing.

Thanks for the tip about `fwupdmgr refresh`.

hughsie, 2024-02-16:
Also, I'd love to know if we can bisect down to what plugin is causing it. e.g. `man fwupd.conf` -- e.g. setting `DisabledPlugins=msr` or `DisabledPlugins=uefi_capsule;redfish` (and then restarting the fwupd service)

closed 2024-11-11 by hughsie

morgyang, 2026-10-14:
We hit what looks like the same class of problem on KVM rather than
Xen, and found a mechanism that may explain this report too.

Each fwupd daemon start calls `statfs()` on efivarfs, which the kernel
serves with an EFI `QueryVariableInfo()` call. On our hosts (Dell R6715,
AMD EPYC, SEV-SNP guests) each call raises an SMI on every CPU. Rarely,
the CPU running a guest vCPU does not resume servicing interrupts after
the SMI, and the guest sees a vCPU stuck at 100% until it is released.
We reproduced it with fwupd and also with a bare
`stat -f /sys/firmware/efi/efivars` loop.

Disabling every UEFI plugin stopped the calls and the stalls, and so did
masking the units as described above. Details and measurements are in
#11198.