fwupd/fwupd #6809: Goodix firmware update fails, but actually succeeds

kiram9, 2024-02-16:
We are testing a goodix firmware update through LVFS. And the update will run, and fail with the following error, but the device will be successfully updated.
For our 13in notebook there is a mix of pass/fail:
https://fwupd.org/lvfs/firmware/52793/analytics/reports
For our 16in notebook: 
https://fwupd.org/lvfs/reports/585491/share/1e8c8470-a49c-571a-82fd-19c9fa32b8c3

The error message we get is:
failed to get device before update reload: failed to wait for device: device 6dc14cd3f5a7e031a01b561fa9619f84c5fb8b07 did not come back

However the device will be successfully updated and usable after this failure.

**Steps to Reproduce**
You can update a FPR on a framework 13 or 16 laptop to version 332 on our embargo channel: 

**Expected behavior**
Firmware update should be successful. 

**fwupd version information**
Please provide the version of the daemon and client.

```
compile   com.hughsie.libxmlb           0.3.15
compile   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.fwupd         1.9.13
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.13
runtime   org.kernel                    6.7.3-200.fc39.x86_64

```

Installed version via DNF on fedora 39. 

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
Framework Laptop 16 (AMD Ryzen 7040 Series)
│
├─AMD Ryzen 7 7840HS w/ Radeon 780M Graphics:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x0a704103
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              2a1df9de-fe02-59c8-8af1-32c8bb011f94 ← CPUID\PRO_0&FAM_19&MOD_74
│ │                       5a4bca32-6547-584f-bb36-920d9adede25 ← CPUID\PRO_0&FAM_19&MOD_74&STP_1
│ │   Device Flags:       • Internal device
│ │ 
│ ├─Graphics Processing Unit (GPU):
│ │ │   Device ID:        749147322f8a287de327bb33c3896221f97c59eb
│ │ │   Summary:          AMD AMD_PHOENIX_GENERIC
│ │ │   Current version:  022.012.000.024.000001
│ │ │   Vendor:           Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│ │ │   GUID:             6c23af4f-d6cd-5ffc-a502-0b85e472e7cb ← AMD\113-PHXGENERIC-001
│ │ │   Device Flags:     • Internal device
│ │ │ 
│ │ └─NE160QDM-NZ6:
│ │       Device ID:      2a0fcb19df98a33fb1bd9f2108d4146d0b4288c3
│ │       GUID:           62710139-0372-57e5-a59e-f83260745941 ← DRM\VEN_BOE&DEV_0BC9
│ │       Device Flags:   • Internal device
│ │     
│ ├─Secure Processor:
│ │     Device ID:        c54ab0237d7a8db8c717b68e0be78e4374a2a079
│ │     Current version:  00.2d.00.74
│ │     Bootloader Version:00.2d.00.74
│ │     Vendor:           Advanced Micro Devices, Inc. (PCI:0x1022)
│ │     GUIDs:            9eb6a793-7f97-5fb5-b49c-139d2ce3ee46 ← PCI\VEN_1022&DEV_15C7
│ │                       dd0885cd-4388-51ad-9899-40f15897ec4e ← PCI\VEN_1022&DEV_15C7&SUBSYS_F1110005
[... 155 lines cut ...]
- Is this a regression? No

I did a little digging, and it looks like this device has a flag set to wait for reconnect, but maybe it takes longer than the delay time. I think this FPR has dual bank firmware update, and I think subsequent updates may be faster if i try to reflash, so this may only happen sometimes, as our 13in logs show a mix of failures and successes.

Looking at the goodix driver I think the reboot delay time is 5 seconds which may need to be increased slightly.
  fu_goodixmoc_device_init(FuGoodixMocDevice *self)
          **fu_device_set_remove_delay(FU_DEVICE(self), 5000);**

Looking at kernel output: 
Original Update:
2024-02-15T09:11:56.614600-0800 fedora kernel: usb 1-4.1: USB disconnect, device number 7
2024-02-15T09:12:01.142580-0800 fedora kernel: usb 1-4.1: new full-speed USB device number 10 using xhci_hcd
2024-02-15T09:12:01.254563-0800 fedora kernel: usb 1-4.1: New USB device found, idVendor=27c6, idProduct=609c, bcdDevice= 1.00
**This time is right at 5 seconds for the device to disconnect and reconnect**

Test run while reinstalling the firmware, which resulted in a pass for the following two runs:
Run 1:
[88357.468239] usb 1-4.1: USB disconnect, device number 10
[88361.738582] usb 1-4.1: new full-speed USB device number 12 using xhci_hcd
[88361.846480] usb 1-4.1: New USB device found, idVendor=27c6, idProduct=609c, bcdDevice= 1.00

Run 2:
[88707.164612] usb 1-4.1: USB disconnect, device number 12
[88711.439028] usb 1-4.1: new full-speed USB device number 13 using xhci_hcd
[88711.546655] usb 1-4.1: New USB device found, idVendor=27c6, idProduct=609c, bcdDevice= 1.00

superm1, 2024-02-16:
So maybe we should back https://github.com/fwupd/fwupd/commit/582e92d6e379396c525d06f4adda0dc66ea20bb5 out and swap it with a longer delay you think?

hughsie, 2024-02-16:
Longer delay seems fine to me,!

superm1, 2024-02-16:
We also seem to be missing a dual bank flag from the above description.

kiram9, 2024-02-16:
@superm1 oh interesting, i did not know about this change. The device should come back without a reboot after firmware update. Did this change make it into any released versions yet? I think i have it because i see the reboot message after updating on fedora.

Based on my testing, the FPR is usable after updating, and i can enroll fingerprints without a reboot. It is a usb device, so a system reboot should not be required.

superm1, 2024-02-16:
> Did this change make it into any released versions yet?

Yes. It's made it into 1.9.13.  This is the PR for it:
https://github.com/fwupd/fwupd/pull/6569

> Based on my testing, the FPR is usable after updating, and i can enroll fingerprints without a reboot. It is a usb device, so a system reboot should not be required.

Right; that's my expectation too, but it was not behaving properly is why that quirk went in.  Can you please test the new PR?  There are artifacts on the PR you can fetch to test:
https://github.com/fwupd/fwupd/actions/runs/7934780806?pr=6810

kiram9, 2024-02-16:
Ok I tried downgrading and upgrading with your test build 5 times. and all passed! 

[test_downgrade_upgrade.zip](https://github.com/fwupd/fwupd/files/14315717/test_downgrade_upgrade.zip)

```
fwupdtool install --allow-older --allow-reinstall 330_add_lotus_guid.cab 6dc14cd3f5a7e031a01b561fa9619f84c5fb8b07

fwupdtool install --allow-older --allow-reinstall a66c3bb595697e8434661f03a92d57c86a045dfa30f72a79120799122442383b-goodix-moc-609c-v01000332.cab 6dc14cd3f5a7e031a01b561fa9619f84c5fb8b07```

superm1, 2024-02-16:
OK, that's great!   Too bad this didn't make 1.9.13, but we can add it for 1.9.14.

closed 2025-05-04 by hughsie