fwupd/fwupd #6823: fwupd latest version build on older Linux distro.

KaiChuan-Hsieh, 2024-02-20:
**Describe the question**
@superm1 

Hello,

I am evaluating to do version bump for Ubuntu 22.04 to the version as Ubuntu 24.04 to support customer new platform.

I find it will build break because of the meson dependency, https://github.com/fwupd/fwupd/commit/db0fbc61313e5249f8a0488d751e4134061260c0.

Do I have to update all packages required by the new fwupd version, or I can just fallback the meson version requirement.
May I have your suggestion how can I do it properly?

Thanks a lot for you professional feedback always.

**fwupd version information**
fwupd 1.7.9

```shell
fwupdmgr --version
```
ubuntu@ubuntu:~$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.kernel                    6.5.0-1013-oem
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10

superm1, 2024-02-20:
Do you need main branch?  I don't think so.  Build the 1_9_X branch.  It should build fine and is what is currently stable.  We'll be tagging a 1.9.14 release soon for an upgrade regression we recently caught.

superm1, 2024-02-20:
Oh but I suppose https://github.com/fwupd/fwupd/commit/6ab21cbfe6475dc65ce2cfc1f607486ec33ddc04 was the backported version.  I think as Ubuntu builds fwupd-efi separately it's safe to revert for the Ubuntu Jammy packaging.

KaiChuan-Hsieh, 2024-02-20:
Got it, thanks for your suggestion.

KaiChuan-Hsieh, 2024-02-22:
@superm1 

Hello,

Do you know when will 1.9.14 be released?
It is because 24.04 code freeze date is Feb. 29, I'd like to know if it can include the version you mentioned here.

Thanks,

superm1, 2024-02-22:
I think we'll make it before that cutoff.

KaiChuan-Hsieh, 2024-02-22:
@superm1 

Thanks for replying it promptly.

closed 2024-02-22 by KaiChuan-Hsieh

superm1, 2024-03-04:
> @superm1
> 
> Thanks for replying it promptly.

JFYI - 1.9.14 is in Noble in case you missed it.