fwupd/fwupd #6826: Update Error: Partition kind required

ShahramMohammed, 2024-02-21:
I tried to reinstall Intel ME on my computer but it terminates the process with the "partition kind required" message.

I'm not using udisks, but have modified `/etc/fwupd/fwupd.conf` to:
```
[fwupd]
EspLocation=/boot
```

Output of fwupdmgr --version:
```
compile   org.freedesktop.fwupd         1.9.13
compile   com.hughsie.libxmlb           0.3.15
compile   org.freedesktop.Passim        0.1.5
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.4
runtime   org.freedesktop.Passim        0.1.5
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    6.7.5-arch1-1
runtime   org.freedesktop.fwupd         1.9.13
```
fwupd was installed with Pacman.

Operating system: Arch Linux
Have you tried rebooting? Yes
Is this a regression? I had updated Intel ME before without having udisks installed or modifying the contents fwupd.conf. The update was somehow applied successfully (or at least reported as so) but it did also report that ESP partition was not detected or configured.

fwupdmgr get-devices --show-all-devices
```
LENOVO 20L8S6QF00 
│ 
├─B140HAN02.4: 
│     Device ID:          aec1a869eb0df71b7cea6b3ac71d39b830faf164 
│     GUID:               95d87dc5-7582-544f-b636-eaf9b2d4011c ← DRM\VEN_AUO&DEV_243D 
│     Device Flags:       • Internal device 
│ 
├─Core™ i5-8350U CPU @ 1.70GHz: 
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027 
│     Current version:    0x000000f4 
│     Vendor:             Intel 
│     GUIDs:              561403e8-143a-5071-ab09-bf5e4c146983 ← CPUID\PRO_0&FAM_06&MOD_8E 
│                         54bdc429-0179-5019-9cb6-8a343324c97c ← CPUID\PRO_0&FAM_06&MOD_8E&STP_A 
│     Device Flags:       • Internal device 
│ 
├─Embedded Controller: 
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60 
│     Summary:            UEFI ESRT device 
│     Current version:    0.1.23 
│     Minimum Version:    0.1.23 
│     Vendor:             Lenovo (DMI:LENOVO) 
│     Update State:       Success 
│     GUID:               4d254d6e-cd67-477b-97d5-bc3048af45c4 
│     Device Flags:       • Internal device 
│                         • Updatable 
│                         • System requires external power source 
│                         • Supported on remote server 
│                         • Needs a reboot after installation 
│                         • Device is usable for the duration of the update 
│     Device Requests:    • Message 
│ 
├─Intel Management Engine: 
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd 
│     Summary:            UEFI ESRT device 
│     Current version:    184.94.4494 
│     Minimum Version:    0.0.1 
│     Vendor:             Lenovo (DMI:LENOVO) 
│     Update State:       Failed 
│     Update Error:       partition kind required 
│     Last modified:      2024-02-21 06:33 
[... 105 lines cut ...]
│     Vendor:             DMI:LENOVO 
│     Update State:       Success 
│     GUID:               b7913cf4-ba8b-6972-af81-eff6ba94f5c6 
│     Device Flags:       • Internal device 
│                         • Updatable 
│                         • System requires external power source 
│                         • Needs a reboot after installation 
│                         • Device is usable for the duration of the update 
│     Device Requests:    • Message 
│ 
├─UEFI Platform Key: 
│     Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a 
│     Summary:            Lenovo Ltd. PK CA 2012 
│     Vendor:             Lenovo Ltd. 
│     GUID:               71599d14-9b31-5270-b3bd-74c494585820 ← UEFI\CRT_9AEF2123F4DE7C19AFABD909BB2C8CAC4411E07E 
│ 
└─UHD Graphics 620: 
      Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a 
      Current version:    07 
      Vendor:             Intel Corporation (PCI:0x8086) 
      GUIDs:              8b72a10c-1279-5f8e-a28a-34fb11a58240 ← PCI\VEN_8086&DEV_5917 
                          880ab73e-f6d1-51aa-91e3-e8499b3d5220 ← PCI\VEN_8086&DEV_5917&SUBSYS_17AA2258 
      Device Flags:       • Internal device 
                          • Cryptographic hash verification is available 
```

hughsie, 2024-02-21:
The problem is, we need the partition type and partition UUID to actually write the `DEVICE_PATH` that the EFI loader understands. I'm not against people writing the code to make fwupd work without UDisks, but it's not code that I'm personally going to write -- given than udisks already exists. Is there a reason you can't use udisks?

@superm1 I'm kinda thinking we want to harder depend on udisks for uefi-capsule updates.

superm1, 2024-02-21:
> The problem is, we need the partition type and partition UUID to actually write the DEVICE_PATH that the EFI loader understands. I'm not against people writing the code to make fwupd work without [...]

Yeah so we're at an inroads.  We either need to:
1. Add support into `/etc/fwupd.conf` for other information provided by udisks (type and UUID).  Deal with the breakage when this is populated incorrectly.
2. Add support for shelling out to some other tool to try to get that information from a given path.  Someone needs to do it.
3. Drop support for non-udisks for uefi-capsule.

This hardcoded path stuff has been NOTHING BUT PAIN for a long time.  I'm leaning on we tear out code that we have making udisks optional is the way to go.

> Any further instructions on what to do would be appreciated.

Once you've installed udisks, drop your hardcoded path from `fwupd.conf` and restart the daemon.  It should just work ™.

superm1, 2024-02-21:
I looked at the code, but I don't think we can actually drop the `EspLocation` key.  The reason is we still need it even for the udisks case if there are multiple ESPs in the system but udisks is picking the wrong one.  So yes I think making udisks a dependency for uefi-capsule sounds like the better solution.

superm1, 2024-02-21:
https://github.com/fwupd/fwupd/pull/6831

ShahramMohammed, 2024-02-22:
> Once you've installed udisks, drop your hardcoded path from fwupd.conf and restart the daemon. It should just work ™.

This does not work as I mentioned in my previous message, but just to be sure I tried it again. Still doesn't work.

Here the steps that I followed:
1. pacman -Syu udisks2
2. remove the `EspLocation=/boot` line from fwupd.conf which leaves the file like this:
```shell
[fwupd]
# use `man 5 fwupd.conf` for documentation
```
3.  systemctl restart fwupd.service
4. fwupdmgr refresh
```shell
WARNING: UEFI ESP partition not detected or configured
See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found for more information.
Metadata is up to date; use --force to refresh again.
```

I tried rebooting and reinstalling fwupd as well and it didn't work.

superm1, 2024-02-22:
Can you please share 

```
sudo fwupdtool esp-list --verbose
```

ShahramMohammed, 2024-02-22:
```shell
$ sudo fwupdtool esp-list --verbose 
04:20:43.469 FuDebug              verbose to info (on console 1) 
04:20:43.469 FuEngine             starting fwupd 1.9.13… 
Loading…                 [                                       ]04:20:43.476 FuEngine             client certificate now exists: nothing to do 
Loading…                 [*                                      ]04:20:43.483 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d 
04:20:43.483 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d 
Loading…                 [*                                      ]04:20:43.485 FuContext            SMBIOS Manufacturer=LENOVO 
04:20:43.485 FuContext            SMBIOS EnclosureKind=a 
04:20:43.485 FuContext            SMBIOS Family=ThinkPad T480s 
04:20:43.485 FuContext            SMBIOS ProductName=20L8S6QF00 
04:20:43.485 FuContext            SMBIOS ProductSku=LENOVO_MT_20L8_BU_Think_FM_ThinkPad T480s 
04:20:43.485 FuContext            SMBIOS BiosVendor=LENOVO 
04:20:43.485 FuContext            SMBIOS BiosVersion=N22ET79W (1.56 ) 
04:20:43.485 FuContext            SMBIOS BiosMajorRelease=01 
04:20:43.485 FuContext            SMBIOS BiosMinorRelease=38 
04:20:43.485 FuContext            SMBIOS FirmwareMajorRelease=01 
04:20:43.485 FuContext            SMBIOS FirmwareMinorRelease=17 
04:20:43.485 FuContext            SMBIOS BaseboardManufacturer=LENOVO 
04:20:43.485 FuContext            SMBIOS BaseboardProduct=20L8S6QF00 
04:20:43.485 FuContext            failed to load fdt: cannot find /sys/firmware/fdt or override /var/lib/fwupd/system.dtb 
04:20:43.485 FuContext            added udev subsystem watch of firmware-attributes 
04:20:43.554 FuBiosSettings       loaded 80 BIOS settings 
Loading…                 [**                                     ]04:20:43.554 FuEngine             ignoring: Error opening directory “/var/lib/fwupd/local.d”: No such file or directory 
04:20:43.554 FuEngine             ignoring: Error opening directory “/usr/share/fwupd/local.d”: No such file or directory 
04:20:43.555 FuEngine             2441 components now in silo 
Loading…                 [***                                    ]04:20:43.562 FuContext            added udev subsystem watch of platform 
04:20:43.562 FuContext            added udev subsystem watch of block for plugin android_boot 
04:20:43.562 FuContext            added udev subsystem watch of pci 
04:20:43.562 FuContext            added udev subsystem watch of i2c 
04:20:43.562 FuContext            added udev subsystem watch of i2c-dev 
04:20:43.562 FuContext            added udev subsystem watch of hidraw 
04:20:43.562 FuContext            added udev subsystem watch of gpio for plugin gpio 
04:20:43.562 FuContext            added udev subsystem watch of mei 
04:20:43.562 FuContext            added udev subsystem watch of iommu for plugin iommu 
04:20:43.562 FuContext            added udev subsystem watch of drm 
04:20:43.563 FuContext            added udev subsystem watch of drm_dp_aux_dev for plugin kinetic_dp 
04:20:43.563 FuContext            added udev subsystem watch of video4linux 
04:20:43.563 FuContext            added udev subsystem watch of msr for plugin msr 
04:20:43.563 FuContext            added udev subsystem watch of mtd for plugin mtd 
04:20:43.563 FuContext            added udev subsystem watch of nvme for plugin nvme 
Loading…                 [***                                    ]04:20:43.564 FuContext            added udev subsystem watch of serio 
04:20:43.564 FuContext            added udev subsystem watch of thunderbolt 
04:20:43.564 FuContext            added udev subsystem watch of tpm for plugin tpm 
04:20:43.564 FuEngine             plugins disabled: test, test_ble 
Loading…                 [***                                    ]04:20:43.565 FuContext            battery threshold now 25 
Loading…                 [************************************** ]04:20:43.565 FuEngine             FuUsbBackend: 
  Name:                 usb 
  Enabled:              true 
  DoneSetup:            false 
  CanInvalidate:        false 
FuUdevBackend: 
  Name:                 udev 
  Enabled:              true 
  DoneSetup:            false 
  CanInvalidate:        false 
FuBluezBackend: 
  Name:                 bluez 
  Enabled:              true 
  DoneSetup:            false 
  CanInvalidate:        false 
FuAcpiDmarPlugin: 
  Name:                 acpi_dmar 
FuAcpiFacpPlugin: 
  Name:                 acpi_facp 
FuAcpiIvrsPlugin: 
  Name:                 acpi_ivrs 
FuAcpiPhatPlugin: 
  Name:                 acpi_phat 
FuAlgoltekUsbPlugin: 
[... 320 lines cut ...]
  Name:                 msr 
  Order:                1 
FuThunderboltPlugin: 
  Name:                 thunderbolt 
  Order:                1 
FuUefiCapsulePlugin: 
  Name:                 uefi_capsule 
  Flags:                measure-system-integrity 
  Order:                1 
  FuUefiBackendLinux: 
    Name:               uefi 
    Enabled:            true 
    DoneSetup:          false 
    CanInvalidate:      false 
  DeviceGType:          FuUefiNvramDevice 
  BgrtSupported:        false 
FuDellPlugin: 
  Name:                 dell 
  Order:                2 
FuUefiRecoveryPlugin: 
  Name:                 uefi_recovery 
  Flags:                require-hwid 
  Order:                2 
04:20:43.575 FuEngine             resetting update motd timeout 
No ESP or BDP found

superm1, 2024-02-22:
Is udisks not working at all? Does udisksctl work?

ShahramMohammed, 2024-02-22:
udisksctl does seem to be installed and working. I just couldn't get it to mount /boot manually because it is already mounted.

<details>

<summary> udisksctl help </summary>

```shell

 udisksctl help
Usage:
  udisksctl COMMAND

Commands:
  help            Shows this information
  info            Shows information about an object
  dump            Shows information about all objects
  status          Shows high-level status
  monitor         Monitor changes to objects
  mount           Mount a filesystem
  unmount         Unmount a filesystem
  unlock          Unlock an encrypted device
  lock            Lock an encrypted device
  loop-setup      Set-up a loop device
  loop-delete     Delete a loop device
  power-off       Safely power off a drive
  smart-simulate  Set SMART data for a drive

Use "udisksctl COMMAND --help" to get help on each command.
```

</detials>

ShahramMohammed, 2024-02-22:
I don't know if it is relevant but I do have an encrypted drive. Although the boot partition itself is not encrypted.

superm1, 2024-02-22:
Can you share output for `udisksctl dump`

ShahramMohammed, 2024-02-22:
<details>

<summary> udisksctl dump </summary>

```shell
[1m[34m/org/freedesktop/UDisks2/Manager:[0m
  [1m[35morg.freedesktop.UDisks2.Manager:[0m
    [37mDefaultEncryptionType:[0m      luks2
    [37mSupportedEncryptionTypes:[0m   luks1
                                luks2
    [37mSupportedFilesystems:[0m       ext2
                                ext3
                                ext4
                                xfs
                                vfat
                                ntfs
                                f2fs
                                nilfs2
                                exfat
                                btrfs
                                udf
                                swap
    [37mVersion:[0m                    2.10.1
  [1m[35morg.freedesktop.UDisks2.Manager.NVMe:[0m
    [37mHostID:[0m             
    [37mHostNQN:[0m            nqn.2014-08.org.nvmexpress:uuid:c9fe1b4c-2a83-11b2-a85c-c0bf00a7b629

[1m[34m/org/freedesktop/UDisks2/block_devices/dm_2d0:[0m
  [1m[35morg.freedesktop.UDisks2.Block:[0m
    [37mConfiguration:[0m              [('fstab', {'fsname': <b'UUID=bc88eb7a-beef-416a-ae9f-eb99cad745ea'>, 'dir': <b'/'>, 'type': <b'btrfs'>, 'opts': <b'rw,relatime,ssd,space_cache=v2,subvolid=5,subvol=/'>, 'freq': <0>, 'passno': <0>})]
    [37mCryptoBackingDevice:[0m        '/org/freedesktop/UDisks2/block_devices/sda2'
    [37mDevice:[0m                     /dev/dm-0
    [37mDeviceNumber:[0m               65024
    [37mDrive:[0m                      '/'
    [37mHintAuto:[0m                   false
    [37mHintIconName:[0m               
    [37mHintIgnore:[0m                 false
    [37mHintName:[0m                   
    [37mHintPartitionable:[0m          false
    [37mHintSymbolicIconName:[0m       
    [37mHintSystem:[0m                 true
    [37mId:[0m                         by-id-dm-name-crypt
    [37mIdLabel:[0m                    
    [37mIdType:[0m                     btrfs
    [37mIdUUID:[0m                     bc88eb7a-beef-416a-ae9f-eb99cad745ea
    [37mIdUsage:[0m                    filesystem
    [37mIdVersion:[0m                  
    [37mMDRaid:[0m                     '/'
    [37mMDRaidMember:[0m               '/'
    [37mPreferredDevice:[0m            /dev/mapper/crypt
    [37mReadOnly:[0m                   false
    [37mSize:[0m                       255042347008
    [37mSymlinks:[0m                   /dev/disk/by-id/dm-name-crypt
                                /dev/disk/by-id/dm-uuid-CRYPT-LUKS2-f471e66f0fce425fb63a979ee30d7fca-crypt
                                /dev/disk/by-uuid/bc88eb7a-beef-416a-ae9f-eb99cad745ea
                                /dev/mapper/crypt
    [37mUserspaceMountOptions:[0m      
  [1m[35morg.freedesktop.UDisks2.Filesystem:[0m
    [37mMountPoints:[0m        /
    [37mSize:[0m               255042347008

[1m[34m/org/freedesktop/UDisks2/block_devices/sda:[0m
  [1m[35morg.freedesktop.UDisks2.Block:[0m
    [37mConfiguration:[0m              []
    [37mCryptoBackingDevice:[0m        '/'
    [37mDevice:[0m                     /dev/sda
    [37mDeviceNumber:[0m               2048
    [37mDrive:[0m                      '/org/freedesktop/UDisks2/drives/TOSHIBA_KSG60ZMV256G_M_2e2_2280_256GB_197A14Q6K8EN'
    [37mHintAuto:[0m                   false
    [37mHintIconName:[0m               
[... 223 lines cut ...]
    [37mApmEnabled:[0m                                 false
    [37mApmSupported:[0m                               false
    [37mPmEnabled:[0m                                  true
    [37mPmSupported:[0m                                true
    [37mReadLookaheadEnabled:[0m                       true
    [37mReadLookaheadSupported:[0m                     true
    [37mSecurityEnhancedEraseUnitMinutes:[0m           2
    [37mSecurityEraseUnitMinutes:[0m                   2
    [37mSecurityFrozen:[0m                             true
    [37mSmartEnabled:[0m                               true
    [37mSmartFailing:[0m                               false
    [37mSmartNumAttributesFailedInThePast:[0m          0
    [37mSmartNumAttributesFailing:[0m                  0
    [37mSmartNumBadSectors:[0m                         0
    [37mSmartPowerOnSeconds:[0m                        20606400
    [37mSmartSelftestPercentRemaining:[0m              0
    [37mSmartSelftestStatus:[0m                        success
    [37mSmartSupported:[0m                             true
    [37mSmartTemperature:[0m                           301.15000000000003
    [37mSmartUpdated:[0m                               1708577577
    [37mWriteCacheEnabled:[0m                          true
    [37mWriteCacheSupported:[0m                        true
```

</details>

hughsie, 2024-02-22:
Isn't that the same as https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Invalid-ESP-Partition -- i.e. `/dev/sda1` looks like a FAT32 partition and isn't set up to be an ESP. I don't know what the Arch installer is doing, but it's probably wrong.

ShahramMohammed, 2024-02-22:
> Isn't that the same as https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Invalid-ESP-Partition -- i.e. `/dev/sda1` looks like a FAT32 partition and isn't set up to be an ESP. I don't know what the Arch installer is doing, but it's probably wrong.

Thank you! This fixed the problem.
Although it broke grub for some reason. I had to chroot into the system and run grub-mkconfig then reinstall it.

closed 2024-02-28 by superm1