fwupd/fwupd #6842: fwupd crash in ubuntu 22.04.4 (xb-lzma-decompressor.c:132:xb_lzma_decomp ressor_convert: code should not be reached)

johnrgregg3, 2024-02-25:
**Describe the bug**
Pretty often Ubuntu gives a pop-up to report a system problem. "View details" reveals a fwupd core dump.

**Steps to Reproduce**
Steps to reproduce the behavior.
It seems to happen all by itself pretty often, but any manual invocation of fwupdmgr will cause the crash.

**Expected behavior**
A clear and concise description of what you expected to happen.
No crash/core dump

**fwupd version information**
Please provide the version of the daemon and client.
$ dpkg -l fwupd
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version                                  Architecture Description
+++-==============-========================================-============-==========================>
ii  fwupd          1.9.5-1pop0~1695754564~22.04~44c2221~dev amd64        Firmware update daemon

```shell
fwupdmgr --version
```
$ fwupdmgr --version
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
not sure - it came with Ubuntu
<details>
The details from the system log make this look a lot like 5554: https://github.com/fwupd/fwupd/issues/5554

Here is a bunch of relevant stuff from /var/log/messages:

Feb 25 16:20:51 riley dbus-daemon[1116]: [system] Activating via systemd: service name='org.freedesk
top.fwupd' unit='fwupd.service' requested by ':1.219' (uid=1000 pid=11849 comm="fwupdmgr --version "
)
Feb 25 16:20:51 riley systemd[1]: Starting Firmware update daemon...
Feb 25 16:20:51 riley fwupd[11858]: **
Feb 25 16:20:51 riley fwupd[11858]: ERROR:../src/xb-lzma-decompressor.c:132:xb_lzma_decompressor_con
vert: code should not be reached
Feb 25 16:20:51 riley fwupd[11858]: Bail out! ERROR:../src/xb-lzma-decompressor.c:132:xb_lzma_decomp
ressor_convert: code should not be reached
Feb 25 16:20:51 riley systemd[1]: fwupd.service: Main process exited, code=dumped, status=6/ABRT
Feb 25 16:20:51 riley systemd[1]: fwupd.service: Failed with result 'core-dump'.
Feb 25 16:20:51 riley systemd[1]: Failed to start Firmware update daemon.

In 5554, there was some problem with /var/lib/fwupd/metadata/lvfs/metadata.xml.xz . I successfully did a manual unxz of (a copy of) my metadata.xml.xz however, so I don't think the compressed file itself is corrupted.

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```
again, fwupdmgr fails/times out with a crash.
</details>

**Additional questions**

- Operating system and version:
- ubuntu 22.04.4, 6.6.10-76060610-generic #202401051437~1704728131~22.04~24d69e2~dev-Ubuntu SMP PREEMPT_DY x86_64 x86_64 x86_64 GNU/Linux

- Have you tried rebooting?
- yes
- Is this a regression?
- Possibly. It looks a lot like 5554.

superm1, 2024-02-25:
What's your libxmlb version? It sounds like you're probably missing the fix.

johnrgregg3, 2024-02-25:
I'm not sure how to check, but here are a bunch of lines returned from dpkg -l:

ii  libxml-libxml-perl                            2.0207+dfsg+really+2.0134-1                       
                    amd64        Perl interface to the libxml2 library
ii  libxml-namespacesupport-perl                  1.12-1.1                                          
                    all          Perl module for supporting simple generic namespaces
ii  libxml-parser-perl:amd64                      2.46-3build1                                      
                    amd64        Perl module for parsing XML files
ii  libxml-sax-base-perl                          1.09-1.1                                          
                    all          base class for SAX drivers and filters
ii  libxml-sax-expat-perl                         0.51-1                                            
                    all          Perl module for a SAX2 driver for Expat (XML::Parser)
ii  libxml-sax-perl                               1.02+dfsg-3                                       
                    all          Perl module for using and building Perl SAX2 XML processors
ii  libxml-simple-perl                            2.25-1                                            
                    all          Perl module for reading and writing XML
ii  libxml-twig-perl                              1:3.52-1                                          
                    all          Perl module for processing huge XML documents in tree mode
ii  libxml-xpathengine-perl                       0.14-1                                            
                    all          re-usable XPath engine for DOM-like trees
ii  libxml2:amd64                                 2.9.13+dfsg-1ubuntu0.3                            
                    amd64        GNOME XML library
ii  libxmlb2:amd64                                0.3.10-2pop1~1686237323~22.04~f779d98~dev         
                    amd64        Binary XML library
ii  libxmlsec1:amd64                              1.2.33-1build2                                                        amd64        XML security library
ii  libxmlsec1-nss:amd64                          1.2.33-1build2                                                        amd64        Nss engine for the XML security library

If any of this is wrong, would that indicate a problem in Ubuntu itself? There might be a higher-level package management bug here. Is the relevant package the libxmlb2 one (0.3.10-2pop)?

superm1, 2024-02-26:
The fix is in libxmlb 0.3.11: https://github.com/hughsie/libxmlb/commit/f5921320454898138af90c6a154000e7bf49b621

I suspect PopOS didn't backport it. You should contact them.

FYI: This isn't the package from Ubuntu.

If you can still reproduce on libxmlb 0.3.11 (or newer) please let us know and reopen the issue.

closed 2024-02-26 by superm1

bcook128, 2024-03-08:
I ran into the same issue (System76 machine running Ubuntu 22.04.4).

Seems like it's an issue with the metadata files:

/var/lib/fwupd/metadata/lvfs/metadata.xml.xz
/var/lib/fwupd/metadata/lvfs/metadata.xml.xz.jcat

Deleting them both and running fwupdmgr refresh --force fixed it for me.

superm1, 2024-03-08:
> I ran into the same issue (System76 machine running Ubuntu 22.04.4).
> 
> Seems like it's an issue with the metadata files:
> 
> /var/lib/fwupd/metadata/lvfs/metadata.xml.xz
> /var/lib/fwupd/metadata/lvfs/metadata.xml.xz.jcat
> 
> Deleting them both and running fwupdmgr refresh --force fixed it for me.

You should raise an issue with System76. I already provided the fix above.