fwupd/fwupd #6849: test plugin doesn't update to 1.2.4

rishabh-chromium, 2024-02-27:
**Describe the bug**
I think after #6722 the test device is unable to upgrade to v1.2.4. I added `TestDevices=true` in fwupd.conf and verified that test plugin is enabled. I ran `fwupdtool enable-test-devices` anyway to be safe (didn't think it changed anything since the plugins were enabled anyway and `get-devices` and `get-updates` was working as expected). I then ran `fwupdmgr get-devices` and `fwupdmgr get-updates` and worked as expected. I then did `fwupdmgr install ${DEVICE_ID} 1.2.4` which seemed to work normally until the last line `device version not updated on success, 1.2.4 != 1.2.2`.

Also I ran `fwupdtool disable-test-devices` and noticed the `test` and `test_ble` plugins were still enabled when I ran `fwupdmgr get-plugins | grep test`

**Steps to Reproduce**
`fwupdmgr install ${TEST_DEVICE_ID} 1.2.4`

**Expected behavior**
Should upgrade normally

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
```

```
compile   org.freedesktop.fwupd         1.9.13
compile   com.hughsie.libxmlb           0.3.11
compile   com.hughsie.libjcat           0.2.0
compile   org.freedesktop.gusb          0.4.5
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.5
runtime   org.kernel                    5.15.148-21852-gd7ab448d89bd
runtime   org.freedesktop.fwupd         1.9.13
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

source

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

**Additional questions**

- Operating system and version: chromeos
- Have you tried rebooting? Yes
- Is this a regression?

superm1, 2024-02-27:
We actually test this flow in CI, for example here is a  recent Debian job that tested it: https://github.com/fwupd/fwupd/actions/runs/8058739433/job/22012041772

Can you show the logs for your failure?  Is it only on 1_9_X you see it?

rishabh-chromium, 2024-02-27:
Yes, it is updating correctly on 1.9.10. 
Here is the output of `fwupdmgr install ${DEVICE_ID} 1.2.4`
```
jinlon-rev5 ~ # fwupdmgr install 08d460be0f1f9f128413f816022a6439e0078018 1.2.4 -v
(fwupdmgr:6266): GLib-DEBUG: 10:34:38.446: setenv()/putenv() are not thread-safe and should not be used after threads are created
(fwupdmgr:6266): GLib-GIO-DEBUG: 10:34:38.447: _g_io_module_get_default: Found default implementation dummy (GDummyProxyResolver) for ?gio-proxy-resolver?
(fwupdmgr:6266): GLib-GIO-DEBUG: 10:34:38.447: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:6266): Fwupd-DEBUG: 10:34:38.449: Emitting ::status-changed() [idle]
(fwupdmgr:6266): FuMain-DEBUG: 10:34:38.453: ignoring extra input 1.2.4
????????????????????????????????????????????????????????????????????????????????
? Upgrade Integrated Webcam? from 1.2.2 to 1.2.4?                              ?
????????????????????????????????????????????????????????????????????????????????
? Fixes another bug with the flux capacitor to prevent time going backwards.   ?
?                                                                              ?
? Test firmware with Signed Report tested by Google on ChromeosTrusted flag    ?
? should show                                                                  ?
?                                                                              ?
? Integrated Webcam? and all connected devices may not be usable while         ?
? updating.                                                                    ?
????????????????????????????????????????????????????????????????????????????????
Perform operation? [Y|n]: 
(fwupdmgr:6266): FuMain-DEBUG: 10:34:39.600: ignoring 08d460be0f1f9f128413f816022a6439e0078018: no matching releases for device
Fwupd-INFO: 10:34:39.607: downloading https://storage.googleapis.com/chromeos-localmirror/lvfs/test/3fab34cfa1ef97238fb24c5e40a979bc544bb2b0967b863e43e7d58e0d9a923f-fakedevice124.cab
(fwupdmgr:6266): Fwupd-DEBUG: 10:34:39.607: Emitting ::status-changed() [downloading]
Fwupd-INFO: 10:34:39.648: download progress: 73%                 ]
Downloading?             [****************************           ]Fwupd-INFO: 10:34:39.649: download progress: 100%
(fwupdmgr:6266): Fwupd-DEBUG: 10:34:39.649: Emitting ::status-changed() [idle]
Fwupd-INFO: 10:34:39.649: status-code was 200
(fwupdmgr:6266): Fwupd-DEBUG: 10:34:39.651: Emitting ::status-changed() [decompressing]
Decompressing?           [***************************************](fwupdmgr:6266): Fwupd-DEBUG: 10:34:39.655: Emitting ::status-changed() [waiting-for-auth]
Authenticating?          [ -                                     ](fwupdmgr:6266): Fwupd-DEBUG: 10:34:39.725: Emitting ::status-changed() [device-busy]
Waiting?                 [ -                                     ](fwupdmgr:6266): Fwupd-DEBUG: 10:34:39.725: Emitting ::device-changed(08d460be0f1f9f128413f816022a6439e0078018)
(fwupdmgr:6266): Fwupd-DEBUG: 10:34:39.725: Emitting ::device-changed(08d460be0f1f9f128413f816022a6439e0078018)
(fwupdmgr:6266): Fwupd-DEBUG: 10:34:39.732: Emitting ::device-changed(08d460be0f1f9f128413f816022a6439e0078018)
(fwupdmgr:6266): Fwupd-DEBUG: 10:34:39.733: Emitting ::device-changed(08d460be0f1f9f128413f816022a6439e0078018)
(fwupdmgr:6266): Fwupd-DEBUG: 10:34:39.733: Emitting ::status-changed() [device-write]
Writing?                 [                                       ](fwupdmgr:6266): Fwupd-DEBUG: 10:34:39.734: Emitting ::status-changed() [device-restart]
Restarting device?       [                                       ](fwupdmgr:6266): Fwupd-DEBUG: 10:34:39.734: Emitting ::status-changed() [device-write]
Writing?                 [                                       ](fwupdmgr:6266): Fwupd-DEBUG: 10:34:39.734: Emitting ::status-changed() [decompressing]
Decompressing?           [                                       ](fwupdmgr:6266): Fwupd-DEBUG: 10:34:39.736: Emitting ::status-changed() [device-write]
Writing?                 [                                       ](fwupdmgr:6266): Fwupd-DEBUG: 10:34:39.737: Emitting ::status-changed() [device-verify]
Verifying?               [                                       ](fwupdmgr:6266): Fwupd-DEBUG: 10:34:39.738: Emitting ::status-changed() [device-restart]
Restarting device?       [*************************************  ](fwupdmgr:6266): Fwupd-DEBUG: 10:34:39.739: Emitting ::status-changed() [device-busy]
Waiting?                 [************************************** ](fwupdmgr:6266): Fwupd-DEBUG: 10:34:39.747: Emitting ::device-changed(08d460be0f1f9f128413f816022a6439e0078018)
(fwupdmgr:6266): Fwupd-DEBUG: 10:34:39.755: Emitting ::device-changed(08d460be0f1f9f128413f816022a6439e0078018)
(fwupdmgr:6266): Fwupd-DEBUG: 10:34:39.756: Emitting ::device-changed(08d460be0f1f9f128413f816022a6439e0078018)
(fwupdmgr:6266): Fwupd-DEBUG: 10:34:39.756: Emitting ::device-changed(08d460be0f1f9f128413f816022a6439e0078018)
(fwupdmgr:6266): Fwupd-DEBUG: 10:34:39.756: Emitting ::device-changed(08d460be0f1f9f128413f816022a6439e0078018)
Waiting?                 [***************************************](fwupdmgr:6266): Fwupd-DEBUG: 10:34:39.757: Emitting ::device-changed(08d460be0f1f9f128413f816022a6439e0078018)
(fwupdmgr:6266): Fwupd-DEBUG: 10:34:39.765: Emitting ::device-changed(08d460be0f1f9f128413f816022a6439e0078018)
(fwupdmgr:6266): Fwupd-DEBUG: 10:34:39.765: Emitting ::status-changed() [idle]

[... cut ...]
(fwupdmgr:8387): Fwupd-DEBUG: 16:35:39.375: Emitting ::device-changed(08d460be0f1f9f128413f816022a6439e0078018)
(fwupdmgr:8387): Fwupd-DEBUG: 16:35:39.392: Emitting ::device-changed(08d460be0f1f9f128413f816022a6439e0078018)
(fwupdmgr:8387): Fwupd-DEBUG: 16:35:39.408: Emitting ::device-changed(08d460be0f1f9f128413f816022a6439e0078018)
(fwupdmgr:8387): Fwupd-DEBUG: 16:35:39.459: Emitting ::changed()
(fwupdmgr:8387): Fwupd-DEBUG: 16:35:39.488: Emitting ::status-changed() [unknown]
(fwupdmgr:8387): Fwupd-DEBUG: 16:35:39.488: Emitting ::status-changed() [idle]
```

rishabh-chromium, 2024-02-27:
Verified, 2.0.0 also fails with the same error

superm1, 2024-02-27:
This works for me too:
```
$ fwupdmgr install 08d460be0f1f9f128413f816022a6439e0078018
WARNING: This package has not been validated, it may not work properly.
0.      Cancel
1.      1.2.4
2.      1.2.3
Choose release [0-2]: 1
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Integrated Webcam™ from 1.2.2 to 1.2.4?                              ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Fixes another bug with the flux capacitor to prevent time going backwards.   ║
║                                                                              ║
║ Integrated Webcam™ and all connected devices may not be usable while         ║
║ updating.                                                                    ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Waiting…                 [***************************************]
Successfully installed firmware
```

superm1, 2024-02-27:
I think we need to see your _daemon log_ when this failure happens for you.

rishabh-chromium, 2024-02-27:
What command show I use to get daemon logs?

superm1, 2024-02-27:
Add `-v` to the systemd unit that launches the daemon (or however you launch it)

rishabh-chromium, 2024-02-27:
I noticed that the `fwupd-test.conf` file path has changed, but for me it is still in `/etc/fwupd/remotes.d/fwupd-test.conf`. Is it supposed to be in `/usr/share/fwupd/remotes.d/` instead?

rishabh-chromium, 2024-02-27:
This is the last log message i see in `fwupd.log`. The logs seem identical between 1.9.10 and 1.9.13

```
2024-02-27T22:12:35.571694Z INFO fwupd[10987]: FuDeviceList         no replug or re-enumerate required
2024-02-27T22:12:35.589256Z INFO fwupd[10987]: FuEngine             install phase now cleanup
2024-02-27T22:12:35.589256Z INFO fwupd[10987]: FuDeviceList         no replug or re-enumerate required
2024-02-27T22:12:35.590199Z INFO fwupd[10987]: FuEngine             cleanup -> FuDevice:
2024-02-27T22:12:35.590199Z INFO fwupd[10987]:  DeviceId:             08d460be0f1f9f128413f816022a6439e0078018
2024-02-27T22:12:35.590199Z INFO fwupd[10987]:  Name:                 Integrated Webcam(TM)
2024-02-27T22:12:35.590199Z INFO fwupd[10987]:  Status:               device-busy
2024-02-27T22:12:35.590199Z INFO fwupd[10987]:  Percentage:           99
2024-02-27T22:12:35.590199Z INFO fwupd[10987]:  Guid:                 b585990a-003e-5270-89d5-3705a17f9a43
2024-02-27T22:12:35.590199Z INFO fwupd[10987]:  Summary:              Fake webcam
2024-02-27T22:12:35.590199Z INFO fwupd[10987]:  Plugin:               test
2024-02-27T22:12:35.590199Z INFO fwupd[10987]:  Protocol:             com.acme.test
2024-02-27T22:12:35.590199Z INFO fwupd[10987]:  Flags:                updatable|require-ac|supported|registered|can-verify|can-verify-image|unsigned-payload
2024-02-27T22:12:35.590199Z INFO fwupd[10987]:  RequestFlags:         allow-generic-message
2024-02-27T22:12:35.590199Z INFO fwupd[10987]:  Vendor:               ACME Corp.
2024-02-27T22:12:35.590199Z INFO fwupd[10987]:  VendorId:             USB:0x046D
2024-02-27T22:12:35.590199Z INFO fwupd[10987]:  Version:              1.2.2
2024-02-27T22:12:35.590199Z INFO fwupd[10987]:  VersionLowest:        1.2.0
2024-02-27T22:12:35.590199Z INFO fwupd[10987]:  VersionBootloader:    0.1.2
2024-02-27T22:12:35.590199Z INFO fwupd[10987]:  VersionFormat:        triplet
2024-02-27T22:12:35.590199Z INFO fwupd[10987]:  Icon:                 preferences-desktop-keyboard
2024-02-27T22:12:35.590199Z INFO fwupd[10987]:  Created:              2024-02-27
2024-02-27T22:12:35.590199Z INFO fwupd[10987]:  Modified:             2024-02-27
2024-02-27T22:12:35.590199Z INFO fwupd[10987]:  UpdateState:          success
2024-02-27T22:12:35.590199Z INFO fwupd[10987]:  AcquiesceDelay:       50
2024-02-27T22:12:35.590199Z INFO fwupd[10987]:  Order:                0
2024-02-27T22:12:35.590199Z INFO fwupd[10987]:  nr-update:            1
2024-02-27T22:12:35.590199Z INFO fwupd[10987]:  BestDevice:           /dev/urandom
2024-02-27T22:12:35.590199Z INFO fwupd[10987]:  InternalFlags:        update-pending
2024-02-27T22:12:35.590328Z INFO fwupd[10987]: FuDeviceList         no replug or re-enumerate required
2024-02-27T22:12:35.590809Z INFO fwupd[10987]: FuEngine             Updating Integrated Webcam(TM) took 0.033912 seconds
2024-02-27T22:12:35.599231Z INFO fwupd[10987]: FuDeviceList         no replug or re-enumerate required
```

I don't see any error message in the file since the install started

rishabh-chromium, 2024-02-27:
Here is the whole daemon log:

```
2024-02-27T22:12:21.680044Z INFO fwupd[10987]: FuDeviceList         no replug or re-enumerate required
2024-02-27T22:12:21.681934Z INFO fwupd[10987]: FuCabinet            verified metadata fakedevice124.metainfo.xml: 2
2024-02-27T22:12:21.681934Z INFO fwupd[10987]: FuCabinet            processing file: fakedevice124.metainfo.xml
2024-02-27T22:12:21.682153Z INFO fwupd[10987]: FuCabinet            processing release: 1.2.4
2024-02-27T22:12:21.683083Z INFO fwupd[10987]: FuCabinet            verified payload fakedevice124.bin: 2
2024-02-27T22:12:21.683970Z INFO fwupd[10987]: FuEngine             composite update 1: 08d460be0f1f9f128413f816022a6439e0078018 1.2.2->1.2.4 (n/a: 0)
2024-02-27T22:12:21.683970Z INFO fwupd[10987]: FuEngine             install phase now composite-prepare
2024-02-27T22:12:21.683970Z INFO fwupd[10987]: FuDeviceList         no replug or re-enumerate required
2024-02-27T22:12:21.683970Z INFO fwupd[10987]: FuEngine             failed to find CPU device
2024-02-27T22:12:21.720667Z INFO fwupd[10987]: FuEngine             install phase now prepare
2024-02-27T22:12:21.720667Z INFO fwupd[10987]: FuDeviceList         no replug or re-enumerate required
2024-02-27T22:12:21.730190Z INFO fwupd[10987]: FuEngine             prepare -> FuDevice:
2024-02-27T22:12:21.730190Z INFO fwupd[10987]:  DeviceId:             08d460be0f1f9f128413f816022a6439e0078018
2024-02-27T22:12:21.730190Z INFO fwupd[10987]:  Name:                 Integrated Webcam(TM)
2024-02-27T22:12:21.730190Z INFO fwupd[10987]:  Status:               device-busy
2024-02-27T22:12:21.730190Z INFO fwupd[10987]:  Guid:                 b585990a-003e-5270-89d5-3705a17f9a43
2024-02-27T22:12:21.730190Z INFO fwupd[10987]:  Summary:              Fake webcam
2024-02-27T22:12:21.730190Z INFO fwupd[10987]:  Plugin:               test
2024-02-27T22:12:21.730190Z INFO fwupd[10987]:  Protocol:             com.acme.test
2024-02-27T22:12:21.730190Z INFO fwupd[10987]:  Flags:                require-ac|supported|registered|can-verify|can-verify-image|updatable-hidden|unsigned-payload
2024-02-27T22:12:21.730190Z INFO fwupd[10987]:  Problems:             update-in-progress
2024-02-27T22:12:21.730190Z INFO fwupd[10987]:  RequestFlags:         allow-generic-message
2024-02-27T22:12:21.730190Z INFO fwupd[10987]:  Vendor:               ACME Corp.
2024-02-27T22:12:21.730190Z INFO fwupd[10987]:  VendorId:             USB:0x046D
2024-02-27T22:12:21.730190Z INFO fwupd[10987]:  Version:              1.2.2
2024-02-27T22:12:21.730190Z INFO fwupd[10987]:  VersionLowest:        1.2.0
2024-02-27T22:12:21.730190Z INFO fwupd[10987]:  VersionBootloader:    0.1.2
2024-02-27T22:12:21.730190Z INFO fwupd[10987]:  VersionFormat:        triplet
2024-02-27T22:12:21.730190Z INFO fwupd[10987]:  Icon:                 preferences-desktop-keyboard
2024-02-27T22:12:21.730190Z INFO fwupd[10987]:  Created:              2024-02-27
2024-02-27T22:12:21.730190Z INFO fwupd[10987]:  Modified:             2024-02-27
2024-02-27T22:12:21.730190Z INFO fwupd[10987]:  UpdateError:          An update is in progress
2024-02-27T22:12:21.730190Z INFO fwupd[10987]:  AcquiesceDelay:       50
2024-02-27T22:12:21.730190Z INFO fwupd[10987]:  Order:                0
2024-02-27T22:12:21.730190Z INFO fwupd[10987]:  BestDevice:           /dev/urandom
2024-02-27T22:12:21.730190Z INFO fwupd[10987]:  InternalFlags:        update-pending
2024-02-27T22:12:21.730190Z INFO fwupd[10987]:  Inhibit:              [update-in-progress] An update is in progress
2024-02-27T22:12:21.730515Z INFO fwupd[10987]: FuDeviceList         no replug or re-enumerate required
2024-02-27T22:12:21.730850Z INFO fwupd[10987]: FuDeviceList         no replug or re-enumerate required
2024-02-27T22:12:21.731008Z INFO fwupd[10987]: FuEngine             install phase now detach
2024-02-27T22:12:21.731008Z INFO fwupd[10987]: FuDeviceList         no replug or re-enumerate required
2024-02-27T22:12:21.731257Z INFO fwupd[10987]: FuEngine             detach -> FuDevice:
2024-02-27T22:12:21.731257Z INFO fwupd[10987]:  DeviceId:             08d460be0f1f9f128413f816022a6439e0078018
2024-02-27T22:12:21.731257Z INFO fwupd[10987]:  Name:                 Integrated Webcam(TM)
2024-02-27T22:12:21.731257Z INFO fwupd[10987]:  Status:               device-restart
2024-02-27T22:12:21.731257Z INFO fwupd[10987]:  Percentage:           1
2024-02-27T22:12:21.731257Z INFO fwupd[10987]:  Guid:                 b585990a-003e-5270-89d5-3705a17f9a43
2024-02-27T22:12:21.731257Z INFO fwupd[10987]:  Summary:              Fake webcam
2024-02-27T22:12:21.731257Z INFO fwupd[10987]:  Plugin:               test
2024-02-27T22:12:21.731257Z INFO fwupd[10987]:  Protocol:             com.acme.test
2024-02-27T22:12:21.731257Z INFO fwupd[10987]:  Flags:                require-ac|supported|registered|can-verify|can-verify-image|updatable-hidden|unsigned-payload
2024-02-27T22:12:21.731257Z INFO fwupd[10987]:  Problems:             update-in-progress
2024-02-27T22:12:21.731257Z INFO fwupd[10987]:  RequestFlags:         allow-generic-message
2024-02-27T22:12:21.731257Z INFO fwupd[10987]:  Vendor:               ACME Corp.
2024-02-27T22:12:21.731257Z INFO fwupd[10987]:  VendorId:             USB:0x046D
2024-02-27T22:12:21.731257Z INFO fwupd[10987]:  Version:              1.2.2
2024-02-27T22:12:21.731257Z INFO fwupd[10987]:  VersionLowest:        1.2.0
2024-02-27T22:12:21.731257Z INFO fwupd[10987]:  VersionBootloader:    0.1.2
[... cut ...]
2024-02-27T22:12:35.590199Z INFO fwupd[10987]:  nr-update:            1
2024-02-27T22:12:35.590199Z INFO fwupd[10987]:  BestDevice:           /dev/urandom
2024-02-27T22:12:35.590199Z INFO fwupd[10987]:  InternalFlags:        update-pending
2024-02-27T22:12:35.590328Z INFO fwupd[10987]: FuDeviceList         no replug or re-enumerate required
2024-02-27T22:12:35.590809Z INFO fwupd[10987]: FuEngine             Updating Integrated Webcam(TM) took 0.033912 seconds
2024-02-27T22:12:35.599231Z INFO fwupd[10987]: FuDeviceList         no replug or re-enumerate required
```

superm1, 2024-02-28:
Can you share whole fwupd.conf they matches your behavior?

rishabh-chromium, 2024-02-28:
Stripped everything earlier:
```
kohaku-rev6 ~ # cat /etc/fwupd/fwupd.conf 
[fwupd]
TestDevices=true
```

Here is the 1.9.10 fwupd.conf
```
dragonair-rev6 ~ # cat /etc/fwupd/fwupd.conf 
# use `man 5 fwupd.conf` for documentation
[fwupd]
DisabledPlugins=
OnlyTrusted=true
AllowEmulation=false
TrustedUids=1000;20134;20130
TrustedReports=DistroId=chromeos&VendorId=16;DistroId=chromeos&VendorId=1923;DistroId=chromeos&VendorId=$OEM
HostBkc=chromium
IdleTimeout=0

```

superm1, 2024-02-28:
> I noticed that the `fwupd-test.conf` file path has changed, but for me it is still in `/etc/fwupd/remotes.d/fwupd-test.conf`. Is it supposed to be in `/usr/share/fwupd/remotes.d/` instead?

yes it should be moved, but I don't _think_ that's your issue.  Is it possible that you have an "old" version of the CAB?  I seem to remember that we updated this recently on LVFS.  Maybe you have a stale version.

rishabh-chromium, 2024-02-28:
Yes, that was the problem. I updated the cab file to the newer cab file available and it worked. It's weird that it failed in only 1.9.13 and not any of the other versions?

superm1, 2024-02-28:
There was a nuanced bug that was caught that the test plugin didn't really update the version that was fixed around when that test mode code was added.  So it makes sense to me.  Glad we got down to the root cause.

closed 2024-02-28 by superm1