fwupd/fwupd #6852: BIOS firmware updated pass on embrago and local install command but failed when putting in Testing remote.

allenyo1, 2024-02-27:
**Describe the bug**
After turning state into testing state. execute fwupdmgr refresh, fwupdmgr update
terminal showed No updatable devices.

[image]

But the cab file using by fwupdmgr install .cab, or
copy xxx-embargo.conf on etc/fwupd/remotes.d/
fwupdmgr refresh & update was passed.

https://fwupd.org/lvfs/firmware/83453

**Steps to Reproduce**
Reinstall Ubuntu 22.04, run below commands
sudo apt install fwupd
fwupdmgr refresh
fwupdmgr update

**Expected behavior**
like embargo steps as below
[image]

**fwupd version information**

```shell
fwupdmgr --version
```
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.5.0-17-generic
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): apt

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

[get-device.txt](https://github.com/fwupd/fwupd/files/14417403/get-device.txt)

**System UEFI configuration**
Please provide the output of the following commands:

```shell
efibootmgr -v
```
[boot.txt](https://github.com/fwupd/fwupd/files/14417450/boot.txt)

```shell
efivar -l | grep fw
```

```shell
tree /boot
```

**Additional questions**

- Operating system and version: ubuntu 22.04
- Have you tried rebooting? yes
- Is this a regression?
- Are you using an NVMe disk? yes
- Is secure boot enabled? no
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?

hughsie, 2024-02-27:
For firmware in testing, you need to enable lvfs-testing using `fwupdmgr enable-remote lvfs-testing` -- does that help? Also not that the test and stable metadata only get updated every 4 hours.

allenyo1, 2024-02-27:
Hi Richard,
    Nice to meet you. Thanks for your rapid reply:) 
Because current in embargo state, I'll try to enable lvfs-testing after 4 hours.

Thank you very much!!

hughsie, 2024-02-27:
> I'll try to enable lvfs-testing after 4 hours.

Sometimes it's a lot quicker, 4h is worst-case-scenario. It depends on how lucky you're feeling. :)

allenyo1, 2024-03-05:
Hi Richard,
    sorry for reply lately. it worked after running enable-remote lvfs-testing.

But currently we study fwupdmgr get-bios-settings and set-bios-settings commands.
Could you kind to tip how to enable or trigger "PolicyKit Authentication" on Ubuntu 22.04?
https://github.com/fwupd/fwupd/blob/main/docs/bios-settings.md  >> sorry, I can't get understanding on this webpage description.
so that I can't use get/set-bios-settings command to r/w BIOS setup value & setup password.

Many appreciate!!

allenyo1, 2024-03-07:
> You need a kernel driver that supports the firmware attributes class API for this feature to work.

after key-in: sudo modprobe firmware_attributes_class
                    sudo fwupdmgr get-bios-setting
still reply This system doesn;t support firmware settings.

[image]

was command wrong? or need something installation ?   Thank you.

superm1, 2024-03-07:
Sorry I was unclear. You need to _develop_ such a driver and submit it to the kernel.

This is the API the driver needs to support:

https://www.kernel.org/doc/Documentation/ABI/testing/sysfs-class-firmware-attributes