fwupd/fwupd #6853: I failed to install firmware updates for 4 years because available firmware updates are only proposed when connected to AC power

mcatanzaro, 2024-02-27:
**Describe the bug**
I have an X1 Carbon 6th generation laptop, which has lots of firmware updates available on LVFS. Today when running `hostnamectl` to set my hostname I noticed that the laptop's firmware was last updated in 2018. I've had this laptop for 4 years now, so that means I've never installed any firmware updates in that time. I had just assumed that the laptop was too old to receive firmware updates via LVFS, and didn't realize otherwise until checking the list of supported devices earlier today.

After talking with Richard we discovered the problem is I've simply never checked for updates when plugged into AC power. When not plugged in, fwupdmgr and graphical app centers like GNOME Software and Discover just don't show the updates at all. So the secret is just to plug in my charger, and then firmware updates become available. I would argue that my experience indicates we we cannot expect users to ever do this.

**Steps to Reproduce**
1. Unplug AC charger, if connected
2. Check for updates using fwupdmgr or GNOME Software

**Expected behavior**
All available updates should be presented to the user. It's fine to not allow installing them without connecting to AC power

**fwupd version information**
Please provide the version of the daemon and client.

```shell
$ fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.13
compile   com.hughsie.libxmlb           0.3.15
compile   org.freedesktop.Passim        0.1.5
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.Passim        0.1.5
runtime   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd         1.9.13
runtime   org.kernel                    6.8.0-0.rc5.41.fc40.x86_64
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): This is Fedora's fwupd-1.9.13-1.fc40

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
LENOVO 20KGS23S00
│
├─B140HAN03.1:
│     Device ID:          aec1a869eb0df71b7cea6b3ac71d39b830faf164
│     GUID:               ee2712cb-3447-5820-8011-bc436248574b ← DRM\VEN_AUO&DEV_313D
│     Device Flags:       • Internal device
│  
├─Core™ i7-8650U CPU @ 1.90GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f4
│     Vendor:             Intel
│     GUIDs:              561403e8-143a-5071-ab09-bf5e4c146983 ← CPUID\PRO_0&FAM_06&MOD_8E
│                         54bdc429-0179-5019-9cb6-8a343324c97c ← CPUID\PRO_0&FAM_06&MOD_8E&STP_A
│     Device Flags:       • Internal device
│  
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    0.1.22
│     Minimum Version:    0.1.22
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
[... 117 lines cut ...]
      Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
      Current version:    07
      Vendor:             Intel Corporation (PCI:0x8086)
      GUIDs:              8b72a10c-1279-5f8e-a28a-34fb11a58240 ← PCI\VEN_8086&DEV_5917
                          0284ca6a-a235-5433-b190-1de21c753494 ← PCI\VEN_8086&DEV_5917&SUBSYS_17AA225C
      Device Flags:       • Internal device
                          • Cryptographic hash verification is available
   
────────────────────────────────────────────────
Devices that have been updated successfully:
 • UEFI dbx (77 → 371)
 • Embedded Controller (0.1.10 → 0.1.22)
 • Intel Management Engine (184.60.3561 → 184.94.4494)
 • System Firmware (0.1.34 → 0.1.62)
```

Please note I took the above get-devices output _after_ plugging into AC power and updating my firmware earlier today.

</details>

**Additional questions**

- Operating system and version: Fedora 40
- Have you tried rebooting? N/A
- Is this a regression? No

superm1, 2024-02-27:
Isn't this a bug for GNOME software?  We already show in fwupdmgr output messages about this.

Conan-Kudo, 2024-02-27:
It's more serious than that. All software centers that talk to fwupd don't get any data when not plugged in. This also affects Plasma Discover and every other software center talking to fwupd.

superm1, 2024-02-27:
I don't disagree it's a pervasive problem but again - it's a bug for every GUI not for the daemowe handle this in the daemon and matching command line tool.

mcatanzaro, 2024-02-27:
> Isn't this a bug for GNOME software?

Not sure, but Discover is affected too, as is fwupdmgr.

> We already show in fwupdmgr output messages about this.

No, I checked for updates using `fwupdmgr update` earlier today and am quite certain there was no such message.

But there is also `fwupdmgr get-updates` which does show a warning "Device requires AC power to be connected". That's how I figured out what was wrong. This is not the command users use to apply updates, though.

superm1, 2024-02-27:
I just tried on a Framework 16 running Fedora 39 with fwupd 1.9.13.  This is what I'm talking about:

[image]

mcatanzaro, 2024-02-27:
> I just tried on a Framework 16 running Fedora 39 with fwupd 1.9.13. This is what I'm talking about:

That looks like the output of `fwupdmgr get-updates`, but again, this is not the command that users run to install updates. If you just run `fwupdmgr update` it looks like there are no updates available.

superm1, 2024-02-27:
Ahh I see what you mean.

@hughsie what do you think on this?

closed 2024-02-28 by hughsie

joakim-tjernlund, 2024-03-12:
I the battery is say >=80% full, why refuse update at all?
Just looks odd when you have 100% battery but update is refused anyway.

stevenengler, 2024-06-10:
Has this fix (#6856) made it to a stable version yet? This doesn't seem to work for me on Fedora 40, which has version 1.9.20. Specifically with AC disconnected I'm told there are no updates:

```text
$ fwupdmgr update
Devices with no available firmware updates:
 • PC611 NVMe SK hynix 512GB
 • TPM
 • Thunderbolt host controller
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • UEFI dbx
```

and with AC connected I get:

```text
$ fwupdmgr update
Devices with no available firmware updates:
 • PC611 NVMe SK hynix 512GB
 • TPM
 • Thunderbolt host controller
 • UEFI Device Firmware
 • UEFI Device Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 1.9.1 to 1.28.0?                                ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • This release contains security updates as disclosed in the Dell            ║
║ Security Advisory.                                                           ║
║                                                                              ║
║ XPS 15 9500 must remain plugged into a power source for the duration of the  ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]:
```

mcatanzaro, 2024-06-11:
> Has this fix (https://github.com/fwupd/fwupd/pull/6856) made it to a stable version yet?

Nope. It will be in 1.10.

I think it's not urgent because nearly all users will install firmware updates via GNOME Software or Plasma Discover. [GNOME Software is already fixed.](https://gitlab.gnome.org/GNOME/gnome-software/-/issues/2465) We might need a KDE bug report still.

stevenengler, 2024-06-11:
> Nope. It will be in 1.10.

Okay thanks! I saw that it was merged to main, but didn't realize there were separate version branches (ex: `1_9_X`).