fwupd/fwupd #6861: Test device firmware version doesn't change after successful update

rishabh-chromium, 2024-02-29:
**Describe the bug**
I updated the test plugin from 1.2.2 to 1.2.4 but `fwupdmgr get-devices` still says 1.2.2. I narrowed down the pr to #6340

**Steps to Reproduce**
`fwupdmgr get-devices --json` says 1.2.2
`fwupdmgr install {DEVICE_ID} 1.2.4`
Verify successful update
`fwupdmgr get-devices --json` still says 1.2.2

**Expected behavior**
Should change version number

rishabh-chromium, 2024-02-29:
Found this in the daemon logs:
```
2024-02-29T02:43:59.276937Z INFO fwupd[5764]: FuRelease            invalid release version 0x4e7fdf9522e2800: value 353530343039313920 was above maximum 4294967295
```

rishabh-chromium, 2024-02-29:
I verified and this error shows even in `1.9.10`
Verified on 2 different devices

hughsie, 2024-02-29:
Weird. What fwupd version please? I'm trying to reproduce here.

hughsie, 2024-02-29:
Ohh hang on; I think we were supposed to backport the fix from https://github.com/fwupd/fwupd/issues/6836 and we got carried away with the reset config stuff. How about we do something like https://github.com/fwupd/fwupd/pull/6864 and put that in the next 1.9.x for ChromeOS? @rishabh-chromium if you test with the commit from hughsie/19x-test-device does it all magically work properly?

rishabh-chromium, 2024-02-29:
Weirdly it works as expected in 1.9.13... Not sure how?

hughsie, 2024-02-29:
> Weirdly it works as expected in 1.9.13... Not sure how?

Well, we could bisect from 1.9.10 to 1.9.13 -- I can't think of anything specific it will be. The most likely was 186987cb31a8638366058721c9a874f56cb01f53 but that was already fixed in 1.9.8

superm1, 2024-02-29:
I just tested this on Ubuntu 24.04 which now has fwupd 1.9.14.
This was my test flow:

```
sudo apt install fwupd-tests
sudo fwupdtool enable-test-devices
sudo systemctl restart fwupd
fwupdmgr update
```

I don't see any problems with it.